bluez/bluez #660: Apple Magic Mouse 2 can't successfully initiate connection

jim-minter, 2023-11-22:
Hi,

I'm running Ubuntu 22.04, kernel 6.2.12, bluez 5.64-0ubuntu1.  My Apple Magic Mouse 2 works when my laptop initiates a connection to it, but can't initiate a connection the other way around.  This is annoying because (for example) after resume, instead of being able to click the mouse to have it reconnect, I currently have to manually switch the mouse off and on, then run `bluetoothctl connect A8:91:3D:3D:80:D3`.  In comparison, my Apple Magic Keyboard reconnects as expected in the same circumstances.

When the mouse tries and fails to initiate a connection, using `btmon` I see it sends `Link Key Request`, and the laptop replies `Link Key Request Negative Reply`.  Shortly afterwards the mouse disconnects.  Note that the file `/var/lib/bluetooth/8C:17:59:71:49:EE/A8:91:3D:3D:80:D3/info` does not contain a `[LinkKey]` stanza.

In comparison, when the laptop successfully initiates the connection, using `btmon` I see the same `Link Key Request` / `Link Key Request Negative Reply` handshake, but then some sort of dynamic link key appears to be successfully negotiated (see `Simple Pairing Complete`, `Link Key Notification`, `Auth Complete` logs).

Is there a bug whereby this dynamic link key negotiation isn't attempted when the mouse initiates a connection?

`bluetoothctl info A8:91:3D:3D:80:D3` shows that the mouse is paired and trusted.

<details>
<summary>bluetoothctl info A8:91:3D:3D:80:D3 output</summary>

```
# bluetoothctl info A8:91:3D:3D:80:D3
Device A8:91:3D:3D:80:D3 (public)
	Name: Jim Minter’s Magic Mouse
	Alias: Jim Minter’s Magic Mouse
	Class: 0x00000580
	Icon: input-mouse
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	WakeAllowed: yes
	LegacyPairing: no
	UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	Modalias: bluetooth:v004Cp0269d0192
```

</details>

<details>
<summary>btmon of failed mouse-initiated connection attempt</summary>

```
# btmon
Bluetooth monitor ver 5.64
= Note: Linux version 6.2.12-060212-generic (x86_64)
= Note: Bluetooth subsystem version 2.22
= New Index: 8C:17:59:71:49:EE (Primary,USB,hci0)
= Open Index: 8C:17:59:71:49:EE
= Index Info: 8C:17:59:71:49:EE (Intel Corp.)
@ MGMT Open: bluetoothd (privileged) version 1.22
> HCI Event: Connect Request (0x04) plen 10
        Address: A8:91:3D:3D:80:D3 (Apple, Inc.)
        Class: 0x000580
          Major class: Peripheral (mouse, joystick, keyboards)
          Minor class: 0x20
        Link type: ACL (0x01)
< HCI Command: Accept Connection Request (0x01|0x0009) plen 7
        Address: A8:91:3D:3D:80:D3 (Apple, Inc.)
        Role: Central (0x00)
> HCI Event: Command Status (0x0f) plen 4
      Accept Connection Request (0x01|0x0009) ncmd 1
        Status: Success (0x00)
> HCI Event: Role Change (0x12) plen 8
        Status: Success (0x00)
        Address: A8:91:3D:3D:80:D3 (Apple, Inc.)
        Role: Central (0x00)
> HCI Event: Vendor (0xff) plen 4
        26 00 01 01
> HCI Event: Connect Complete (0x03) plen 11
        Status: Success (0x00)
        Handle: 256
        Address: A8:91:3D:3D:80:D3 (Apple, Inc.)
        Link type: ACL (0x01)
[... 1337 lines cut ...]
< ACL Data TX: Handle 256 flags 0x00 dlen 12
      L2CAP: Disconnection Request (0x06) ident 9 len 4
        Destination CID: 66
        Source CID: 66
< HCI Command: Exit Sniff Mode (0x02|0x0004) plen 2
        Handle: 256
> HCI Event: Command Status (0x0f) plen 4
      Exit Sniff Mode (0x02|0x0004) ncmd 1
        Status: Success (0x00)
> HCI Event: Number of Completed Packets (0x13) plen 5
        Num handles: 1
        Handle: 256
        Count: 1
> HCI Event: Mode Change (0x14) plen 6
        Status: Success (0x00)
        Handle: 256
        Mode: Active (0x00)
        Interval: 0.000 msec (0x0000)
> ACL Data RX: Handle 256 flags 0x02 dlen 12
      L2CAP: Disconnection Response (0x07) ident 9 len 4
        Destination CID: 66
        Source CID: 66
```

</details>

Vudentz, 2023-11-22:
I suspect this has to do with:

https://github.com/bluez/bluez/commit/9741e47aebcdcff343dd2ca7dadc3ec41b66744f

If possible try to reproduce with it.

jim-minter, 2023-11-23:
Awesome, I built from 5.70 and from master (https://github.com/bluez/bluez/commit/58e6ef54e672798e2621cae87356c66de14d011f), and in both cases the mouse now successfully initiates a connection itself.

(In case it helps anyone else, I had a teething problem where initially the mouse didn't work at all after upgrading bluetoothd.  I saw the error `Rejected connection from !bonded device` in the bluetoothd logs.  Running `bluetoothctl info` showed `Paired: yes, Trusted: yes, Connected: yes` but `Bonded: no`.  Re-pairing using blueman-manager didn't help, I'm guessing because my blueman-manager version is too old to understand bonding.  After re-pairing using `bluetoothctl pair`, things worked fine).

For posterity:

<details>
<summary>btmon log of a successful mouse-initiated connection attempt</summary>

```
# btmon
Bluetooth monitor ver 5.70
= Note: Linux version 6.2.12-060212-generic (x86_64)
= Note: Bluetooth subsystem version 2.22
= New Index: 8C:17:59:71:49:EE (Primary,USB,hci0)
= Open Index: 8C:17:59:71:49:EE
= Index Info: 8C:17:59:71:49:EE (Intel Corp.)
@ MGMT Open: bluetoothd (privileged) version 1.22
> HCI Event: Connect Request (0x04) plen 10
        Address: A8:91:3D:3D:80:D3 (Apple, Inc.)
        Class: 0x000580
          Major class: Peripheral (mouse, joystick, keyboards)
          Minor class: 0x20
        Link type: ACL (0x01)
< HCI Command: Accept Connection Request (0x01|0x0009) plen 7
        Address: A8:91:3D:3D:80:D3 (Apple, Inc.)
        Role: Central (0x00)
> HCI Event: Command Status (0x0f) plen 4
      Accept Connection Request (0x01|0x0009) ncmd 1
        Status: Success (0x00)
> HCI Event: Role Change (0x12) plen 8
        Status: Success (0x00)
        Address: A8:91:3D:3D:80:D3 (Apple, Inc.)
        Role: Central (0x00)
> HCI Event: Vendor (0xff) plen 4
      Intel PTT Switch Notification (0x26)
        Handle: 256
        Packet type table: Enhanced data rate (0x01)
> HCI Event: Connect Complete (0x03) plen 11
        Status: Success (0x00)
        Handle: 256
        Address: A8:91:3D:3D:80:D3 (Apple, Inc.)
        Link type: ACL (0x01)
        Encryption: Disabled (0x00)
< HCI Command: Read Remote Supported Features (0x01|0x001b) plen 2
        Handle: 256 Address: A8:91:3D:3D:80:D3 (Apple, Inc.)
> HCI Event: Command Status (0x0f) plen 4
      Read Remote Supported Features (0x01|0x001b) ncmd 1
        Status: Success (0x00)
> HCI Event: Read Remote Supported Features (0x0b) plen 11
        Status: Success (0x00)
        Handle: 256 Address: A8:91:3D:3D:80:D3 (Apple, Inc.)
        Features: 0xbc 0x06 0x86 0x7e 0x08 0x04 0x79 0x87
          Encryption
          Slot offset
          Timing accuracy
          Role switch
          Sniff mode
          Power control requests
          Channel quality driven data rate (CQDDR)
          Paging parameter negotiation
          Power control
          Broadcast Encryption
          Enhanced Data Rate ACL 2 Mbps mode
          Enhanced Data Rate ACL 3 Mbps mode
          Enhanced inquiry scan
          Interlaced inquiry scan
          Interlaced page scan
          RSSI with inquiry results
          AFH capable peripheral
[... 392 lines cut ...]
> HCI Event: Command Complete (0x0e) plen 7
      Read RSSI (0x05|0x0005) ncmd 1
        Status: Success (0x00)
        Handle: 256 Address: A8:91:3D:3D:80:D3 (Apple, Inc.)
        RSSI: 0 dBm (0x00)
< HCI Command: Read Link Quality (0x05|0x0003) plen 2
        Handle: 256 Address: A8:91:3D:3D:80:D3 (Apple, Inc.)
> HCI Event: Command Complete (0x0e) plen 7
      Read Link Quality (0x05|0x0003) ncmd 1
        Status: Success (0x00)
        Handle: 256 Address: A8:91:3D:3D:80:D3 (Apple, Inc.)
        Link quality: 0xff
< HCI Command: Read Transmit Power Level (0x03|0x002d) plen 3
        Handle: 256 Address: A8:91:3D:3D:80:D3 (Apple, Inc.)
        Type: Current Transmit Power Level (0x00)
> HCI Event: Command Complete (0x0e) plen 7
      Read Transmit Power Level (0x03|0x002d) ncmd 1
        Status: Success (0x00)
        Handle: 256 Address: A8:91:3D:3D:80:D3 (Apple, Inc.)
        TX power: 12 dbm (0x0c)
```

</details>

Thanks for your help @Vudentz !

closed 2023-11-23 by jim-minter