bluez/bluez #662: Various bluetoothd crashes with Basic Audio Profile

tlvince, 2023-11-23:
I'm working through an issue with Bluetooth LE Audio with a new headset (Creative Zen Hybrid Pro). With help from @pv, we believe BAP's triggering a bug in the adapter's firmware (MediaTek MT7922) - in short, the headset connection establishes, but the adapter hits a bug and restarts. See [No sink created for Creative Zen Hybrid Pro LE Audio LC3 device](https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/3666#note_2179246) and [MediaTek MT7922 controller crashes after LE Setup Isochronous Data Path](https://lore.kernel.org/linux-bluetooth/38cb99f2b63dc55763e9e2c8ae4d4cb14afc6770.camel@tlvince.com/#r) for details.

Debugging this has also triggered a few crashes in BlueZ. Here're some logs for a few different scenarios:

1. [BlueZ crashed from boot](https://github.com/bluez/bluez/files/13453004/bluez-from-boot-headset-off.txt) - maybe a cache issue - went away after clearing `/var/lib/bluetooth/*`
2. [After connecting the headset and starting pipewire](https://github.com/bluez/bluez/files/13453008/bluez-device-connected-started-pipewire.txt)

Some related logs if they're of use:

- Not a crash, but example of the linked issue with the adapter cycling [from bluez](https://github.com/bluez/bluez/files/13453012/bluez-adapter-cycling.txt) and [pipewire](https://github.com/bluez/bluez/files/13453015/pipewire-adapter-cycling.txt)
- [dmesg.log](https://github.com/bluez/bluez/files/13453021/dmesg.log)
- [btsnoop-log.txt](https://github.com/bluez/bluez/files/13453025/btsnoop-log.txt) showing the adapter coredump

Versions:

- bluez: [58e6ef54e672798e2621cae87356c66de14d011f](https://github.com/tlvince/nixos-config/blob/d88813de461df2c4301b7e81bfffe97c5bf2f3f0/overlays/bluez.nix#L29-L34) with [these flags](https://github.com/tlvince/nixos-config/blob/d88813de461df2c4301b7e81bfffe97c5bf2f3f0/overlays/bluez.nix#L68-L88) (including `--enable-asan`)
- Linux 6.6.1
- pipewire [1030d4ccbb3ac43c50a01547345916cac3a4706f](https://gitlab.freedesktop.org/pipewire/pipewire/-/commits/1030d4ccbb3ac43c50a01547345916cac3a4706f)

```
❯ cat /etc/bluetooth/main.conf 
[General]
ControllerMode=le
Experimental=true
KernelExperimental=6fbaf188-05e0-496a-9885-d6ddfdb4e03e

[Policy]
AutoEnable=true
```

Vudentz, 2024-01-04:
@tlvince seems to be working correctly with the patch above, but pipewire also needs to adopt the same logic for calculating the resulting SDU in case it wants to multiplex l+r over the same CIS.

@pv what is the plan for pipewire to adopt ChannelLocation and multiplexing in general?

pv, 2024-01-05:
Pipewire already supports multiplexing channels in same CIS.

I have this device and looked at it a while ago, and it seems like it has firmware bugs that cause problems:
https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/3666#note_2183197

Vudentz, 2024-01-05:
> Pipewire already supports multiplexing channels in same CIS.
> 
> I have this device and looked at it a while ago, and it seems like it has firmware bugs that cause problems: [gitlab.freedesktop.org/pipewire/pipewire/-/issues/3666#note_2183197](https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/3666#note_2183197)
> 
> Also, the device has `Supported_Max_Codec_Frames_Per_SDU == 1` in its PAC, even though it can multiplex two channels in a CIS. The BAP spec is unclear on whether this value is frames per channel or not, but IIUC it should be `2`. Because of this, PipeWire currently configures a mono stream. In principle we could (and maybe should) instead ignore this value and just use the channel count bitmasks.

I see, yeah Id go with ChannelLocation bitmask and also try setting it when configuring, not sure if pw does that already. The way PAC records has its capabilities is not very clear about the audio configuration it supports, so it is a little bit of a hit or miss right now, in theory the device could have multiple records of LC3 to try to narrow down its capabilities. Btw, is the Supported_Max_Codec_Frames_Per_SDU = 1 perhaps only for the source? Need to record the HCI trace to see unde what ASE does that comes from.

pv, 2024-01-05:
> I see, yeah Id go with ChannelLocation bitmask and also try setting it when configuring, not sure if pw does that already. The way PAC records has its capabilities is not very clear about the audio [...]

We pick the channels based on PACS Locations, and set the channel allocation in the LTV config. The new `ChannelAllocation` is not supported yet in Pipewire, but it's only pending on me finding time to work on it.

> Btw, is the Supported_Max_Codec_Frames_Per_SDU = 1 perhaps only for the source? Need to record the HCI trace to see unde what ASE does that comes from.

IIRC, it's the Sink PAC that has Supported_Max_Codec_Frames_Per_SDU=1 and Supported_Audio_Channel_Counts=0x3, to get it to work I needed to skip our Supported_Max_Codec_Frames_Per_SDU value check.

closed 2024-01-05 by BluezTestBot

pv, 2024-01-06:
The latest Creative Zen Hybrid Pro firmware update (1.2.231215.1500, 06-01-2024) seems to break connecting in LE mode altogether for me (also with bluetoothctl not pipewire), the device disconnects while ASE configuration is in progress.

runderwo, 2024-12-24:
@pv Is your configuration still broken?  I am setting up these headphones and encountering consistent disconnection during the initial handshake, and I wonder if it's the same issue.

runderwo, 2024-12-24:
Adding dump.btsnoop log from pipewire+AX210 which is broken and btsnoop_hci.log from Android where the headphones work.  The Android log captures an initial non-LE pairing, and then an upgrade to LE by toggling the slider on the connection properties.
[dump.btsnoop.log.gz](https://github.com/user-attachments/files/18235463/dump.btsnoop.log.gz)
[btsnoop_hci.log.gz](https://github.com/user-attachments/files/18235461/btsnoop_hci.log.gz)

Also including pipewire-bluez.log from wireplumber with debug options, which has some weird corruption in it which makes me wonder about wireplumber itself.
[pipewire-bluez.log.gz](https://github.com/user-attachments/files/18235462/pipewire-bluez.log.gz)