bluez/bluez #666: [Q][5.55-3] Paired and trusted speaker not detected by pulseaudio on powerup - endpoint is not registered, but should it be?

MaxMotovilov, 2023-11-28:
- OS: Raspberry Pi OS (bullseye), kernel 6.1.21+ (on P0W 1.1)
- bluez 5.55-3
- pulseaudio 14.2-2
- Device: Bose Solo II Soundbar (Vendor UUID 00001000-d102-11e1-9b23-00025b00a5a5)

**Problem:** when I power on the speaker and turn on the BT, the *Connected* property goes to *yes*. The pulseaudio BT module does not, however, discover and load the device module and create the sink. It does so only after an explicit `bluetoothctl connect` and then everything works fine until the speaker is turned off again.

**Question:** should the endpoint registration happen when a previously paired and trusted device comes online? Pulseaudio is listening for endpoint registrations (`org.bluez.MediaEndpoint1`); based on my observations with `dbus-monitor` only `org.bluez.MediaPlayer1` is registered when speaker is turned on. Lacking domain knowledge, I cannot tell whether this is an issue with the BT stack or a use case not covered by pulseaudio BT modules.

**More details**

When speaker is powered off:

```
$ bluetoothctl info AC:BF:71:AC:BC:A1
Device AC:BF:71:AC:BC:A1 (public)
	Name: Bose Solo
	Alias: Bose Solo
	Class: 0x00240428
	Icon: audio-card
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Vendor specific           (00001000-d102-11e1-9b23-00025b00a5a5)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
```

When turning on speaker and its BT:

```
$ dbus-monitor --system "type='signal',sender='org.bluez'"
...
signal time=1701145275.598554 sender=:1.8 -> destination=(null destination) serial=579 path=/org/bluez/hci0/dev_AC_BF_71_AC_BC_A1; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
   string "org.bluez.Device1"
   array [
      dict entry(
         string "Connected"
         variant             boolean true
      )
   ]
   array [
   ]
signal time=1701145275.664281 sender=:1.8 -> destination=(null destination) serial=580 path=/; interface=org.freedesktop.DBus.ObjectManager; member=InterfacesAdded
   object path "/org/bluez/hci0/dev_AC_BF_71_AC_BC_A1/player0"
   array [
      dict entry(
         string "org.freedesktop.DBus.Introspectable"
         array [
         ]
      )
      dict entry(
         string "org.bluez.MediaPlayer1"
         array [
            dict entry(
               string "Position"
               variant                   uint32 0
            )
            dict entry(
               string "Device"
               variant                   object path "/org/bluez/hci0/dev_AC_BF_71_AC_BC_A1"
            )
         ]
      )
      dict entry(
         string "org.freedesktop.DBus.Properties"
         array [
         ]
      )
   ]
signal time=1701145275.687078 sender=:1.8 -> destination=(null destination) serial=581 path=/org/bluez/hci0/dev_AC_BF_71_AC_BC_A1; interface=org.freedesktop.DBus.Properties; member=PropertiesChanged
   string "org.bluez.MediaControl1"
   array [
      dict entry(
         string "Player"
         variant             object path "/org/bluez/hci0/dev_AC_BF_71_AC_BC_A1/player0"
      )
      dict entry(
         string "Connected"
         variant             boolean true
      )
   ]
   array [
      dict entry(
   ]
```

Vudentz, 2023-11-28:
@MaxMotovilov 5.55 is very old, any chance to test with something more recent?

MaxMotovilov, 2023-11-28:
> @MaxMotovilov 5.55 is very old, any chance to test with something more recent?

I'd have to build from source - either *for* or *on* the P0W. My familiarity with that thing is all of one weekend :laughing:  so I thought I'd ask before starting on that path. Is it generally expected that  automatic reconnection to a known device also registers an endpoint - or not? Because if the answer is "no", the issue is with pulseaudio and I can attempt to fix it by installing a separate dbus listener that would fork off `bluetoothctl connect` when it sees device coming up.

For that matter, I can try and repro this on my regular Linux desktop with recent bluez. Probably what I ought to do next anyway.

Vudentz, 2023-11-28:
> > @MaxMotovilov 5.55 is very old, any chance to test with something more recent?
> 
> I'd have to build from source - either _for_ or _on_ the P0W. My familiarity with that thing is all of one weekend 😆 so I thought I'd ask before starting on that path. Is it generally expected that automatic reconnection to a known device also registers an endpoint - or not? Because if the answer is "no", the issue is with pulseaudio and I can attempt to fix it by installing a separate dbus listener that would fork off `bluetoothctl connect` when it sees device coming up.

No, it doesn't reconnect when you register a new endpoint, it just attempts to setup a stream if one is not configured by that requires the device to be connected already.

I recall we have some issue about reconnections when rebooting for example, we probably should have a reconnection policy if we detect that a device was disconnected due to daemon restart, system reboot, etc, so when it restart it would reconnect them automatically but we also need to wait until the system registers its endpoints in order to do that properly.

> For that matter, I can try and repro this on my regular Linux desktop with recent bluez. Probably what I ought to do next anyway.

You might have a better luck with pipewire, since it does seem to reconnect automatically all audio devices once it restarts, or bluetoothd restarts, which is quite annoying if you have many audio devices but for regular users that sort of does the job until we have the policy Ive just described above.

MaxMotovilov, 2023-11-29:
@Vudentz same behavior with 5.64 on Mint 21.

MaxMotovilov, 2023-11-29:
After digging a little in the docs here, I think what happens is that only the remote control target profile (MediaPlayer) is automatically connected when the soundbar powers up. The audio sink profile doesn't. I am guessing it is the device's quirk and not the software stack's - would love to be corrected by knowledgeable people. The cure seems to require a DBus listening service after all, along the lines of https://github.com/nE0sIghT/bluetooth-buggy-reconnect but with somewhat different logic.

closed 2025-08-04 by github-actions[bot]