bluez/bluez #672: L2CAP/COS/CED/BI-02-C PTS test fails

fdanis-oss, 2023-12-08:
`L2CAP/COS/CED/BI-02-C` test send a malformed L2CAP signaling packet with 2 commands in it (a connection request and an unknown command) and expect to get a connection response packet and a command reject packet. The second is not sent by BlueZ.

I attached:
- BlueZ kernel dynamic debug log: [L2CAP_COS_CED_BI-02-C.log](https://github.com/bluez/bluez/files/13612282/L2CAP_COS_CED_BI-02-C.log)
- btmon record: [L2CAP_COS_CED_BI-02-C.btsnoop.zip](https://github.com/bluez/bluez/files/13612310/L2CAP_COS_CED_BI-02-C.btsnoop.zip)
- PTS log: [L2CAP_COS_CED_BI_02_C_2023_12_07_08_56_26.xml.zip](https://github.com/bluez/bluez/files/13612315/L2CAP_COS_CED_BI_02_C_2023_12_07_08_56_26.xml.zip)

fdanis-oss, 2023-12-13:
Patch merged upstream

closed 2023-12-13 by fdanis-oss

secit, 2023-12-14:
@fdanis-oss may I ask how your test setup was?
Looks like I have the same issue, but the patch didn't work for me.

fdanis-oss, 2023-12-14:
@secit with `TSPX_psm` set to `1011` I run `l2test -d -P 4113` before starting the test