bluez/bluez #683: test failure in test-vcp

andyrtr, 2023-12-14:
bluez 5.71 fails the test here (Arch Linux): 
```
[1;**
ERROR:src/shared/tester.c:949:test_io_recv: assertion failed (len == iov->iov_len): (5 == 7)
39mVOCS/SR/SPE/BI-01-C[0m - [0;35mteardown complete[0m
[1;39mVOCS/SR/SPE/BI-01-C[0m - [0;30mdone[0m

[1;39mVOCS/SR/CP/BV-01-C[0m - [0;30minit[0m
[1;39mVOCS/SR/CP/BV-01-C[0m - [0;34msetup[0m
[1;39mVOCS/SR/CP/BV-01-C[0m - [0;34msetup complete[0m
[1;39mVOCS/SR/CP/BV-01-C[0m - [0;30mrun[0m
gatt_notify_cb: Failed to send notification
Bail out! ERROR:src/shared/tester.c:949:test_io_recv: assertion failed (len == iov->iov_len): (5 == 7)
FAIL unit/test-vcp (exit status: 134)
```

[test-suite.log](https://github.com/bluez/bluez/files/13676801/test-suite.log)

Vudentz, 2023-12-15:
@andyrtr what arch are you running? There have been some fixes for Big Endian but perhaps we missed something.

Antiz96, 2023-12-15:
> @andyrtr what arch are you running? There have been some fixes for Big Endian but perhaps we missed something.

Allow me to answer (Andy and I are both the maintainers of [the Arch Linux bluez package](https://archlinux.org/packages/extra/x86_64/bluez/)).
This test failure in test-vcp is happening on Arch Linux running on the x86_64 architecture. The build and test instructions are [there](https://gitlab.archlinux.org/archlinux/packaging/packages/bluez/-/blob/main/PKGBUILD?ref_type=heads).

jhatler, 2024-01-06:
@andyrtr @Antiz96 Does arch build the bluez package with link time optimizations enabled? I ran into this issue on my Gentoo system and found that removing the "-flto" option from my CFLAGS produced a build which passed all the tests.

Antiz96, 2024-01-09:
> @andyrtr @Antiz96 Does arch build the bluez package with link time optimizations enabled? I ran into this issue on my Gentoo system and found that removing the "-flto" option from my CFLAGS produced a build which passed all the tests.

@jhatler Nice catch, I confirm that building without lto makes all the tests pass!

macmpi, 2024-04-15:
[test-vcp fails ](https://gitlab.alpinelinux.org/macmpi/aports/-/jobs/1345179)on latest **bluez 5.74** on Alpine (musl) x86 and x86_64 platforms.
Test passes on other platforms (armhf, armv7, ...)

Vudentz, 2024-04-15:
@maheshtalewad looks like you were the one responsible to write test-vcp, can you please check what is going on here?

macmpi, 2024-04-16:
can look [there](https://github.com/bluez/bluez/issues/726#issuecomment-1900632058) for segfault on Alpine (musl is often less permissive than libc on memory allocation: maybe easier to track potential memory allocation issues there).

xry111, 2025-05-20:
I no longer see it with gcc-15.1.0.  But maybe this is just some blind luck.

xry111, 2025-05-20:
> I no longer see it with gcc-15.1.0. But maybe this is just some blind luck.

Indeed blind luck.  It does not happen with 5.79 but then happens with 5.82.

macmpi, 2025-09-20:
FWIW: fixed with 5.84