bluez/bluez #686: Bluez 5.71 causes initial auto-connect to fail with [Failed to load link keys for hci0] - git bisect done

Antiz96, 2023-12-16:
Hello,

Bluez 5.71 introduced a regression where the initial auto-connect for Bluetooth devices doesn't work anymore.

Indeed, at system startup, starting a Bluetooth device (headset in my case) will cause it to go in "connected/disconnected" state loop, requiring to initialize a manual connection first (with sometimes multiple attempts needed) before getting it connected and working as intended.

`systemctl status bluetooth` reports the following error:

```
 $ systemctl status bluetooth
```
> ● bluetooth.service - Bluetooth service
>    Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset: disabled)
>     Active: active (running) since Fri 2023-12-15 11:03:18 CET; 25s ago
>       Docs: man:bluetoothd(8)
>   Main PID: 592 (bluetoothd)
>     Status: "Running"
>      Tasks: 1 (limit: 38404)
>     Memory: 3.1M (peak: 3.6M)
>        CPU: 33ms
>     CGroup: /system.slice/bluetooth.service
>             └─592 /usr/lib/bluetooth/bluetoothd
>
> déc. 15 11:03:18 Arch-Desktop bluetoothd[592]: Battery Provider Manager created
> **déc. 15 11:03:18 Arch-Desktop bluetoothd[592]: Failed to load link keys for hci0: Invalid Parameters (0x0d)**
> déc. 15 11:03:38 Arch-Desktop bluetoothd[592]: Endpoint registered: sender=:1.28 path=/MediaEndpoint/A2DPSink/sbc
> déc. 15 11:03:38 Arch-Desktop bluetoothd[592]: Endpoint registered: sender=:1.28 path=/MediaEndpoint/A2DPSource/sbc
> déc. 15 11:03:38 Arch-Desktop bluetoothd[592]: Endpoint registered: sender=:1.28 path=/MediaEndpoint/A2DPSink/sbc_xq_453
> déc. 15 11:03:38 Arch-Desktop bluetoothd[592]: Endpoint registered: sender=:1.28 path=/MediaEndpoint/A2DPSource/sbc_xq_453
> déc. 15 11:03:38 Arch-Desktop bluetoothd[592]: Endpoint registered: sender=:1.28 path=/MediaEndpoint/A2DPSink/sbc_xq_512
> déc. 15 11:03:38 Arch-Desktop bluetoothd[592]: Endpoint registered: sender=:1.28 path=/MediaEndpoint/A2DPSource/sbc_xq_512
> déc. 15 11:03:38 Arch-Desktop bluetoothd[592]: Endpoint registered: sender=:1.28 path=/MediaEndpoint/A2DPSink/sbc_xq_552
> déc. 15 11:03:38 Arch-Desktop bluetoothd[592]: Endpoint registered: sender=:1.28 path=/MediaEndpoint/A2DPSource/sbc_xq_552

I bisected the bug with `git bisect` and it turns out that the faulty commit is [d5536e0cd431e22be9a1132be6d4af2445590633](https://github.com/bluez/bluez/commit/d5536e0cd431e22be9a1132be6d4af2445590633).
I confirm that reverting it fixes the issue.

Additional informations:
- OS: Arch Linux (x86_64)
- Kernel: [Linux 6.6.7.arch1](https://archlinux.org/packages/core/x86_64/linux/)
- Bluetooth device: Sony WH-100XM3 (although the bug was reproduced with other devices)
- Bluetooth adapter: Asus BT500 (although the bug was reproduced with other adapter)
- Build instructions for the `bluez` Arch Linux package: https://gitlab.archlinux.org/archlinux/packaging/packages/bluez/-/blob/main/PKGBUILD?ref_type=heads
- Arch Linux downstream ticket: https://gitlab.archlinux.org/archlinux/packaging/packages/bluez/-/issues/4

If you need any more tests or information, feel free to ask :)

crunk1, 2023-12-16:
I also got `Failed to load link keys for hci0: Invalid Parameters (0x0d)`

Arch Linux (x86_64)
Linux 6.6.7.arch1
Adapter: [Mpow BH519A Bluetooth 5.1 USB Adapter for PC](https://www.xmpow.com/products/mpow-bh519a-bluetooth-5-1-usb-adapter-for-pc)

Downgrading to 5.70 resolved the issue.

XKQuestion, 2023-12-16:
Hello,
The kernel needs to apply the following patch:
[Bluetooth: MGMT/SMP: Fix address type when using SMP over BREDR/LE - kernel/git/bluetooth/bluetooth-next.git - Bluetooth kernel development tree](https://git.kernel.org/pub/scm/linux/kernel/git/bluetooth/bluetooth-next.git/commit/?id=df24d38bc7acd546eebd5567ea2d41f8eb2d6afa)

heftig, 2023-12-16:
No, I think bluez needs to fix this. Why was a release made depending on a kernel change that's not even in mainline? It's entirely unreasonable.

XKQuestion, 2023-12-16:
Could you try the following patch?
```diff
diff --git a/src/adapter.c b/src/adapter.c
index ee70b00d2..1ab09f0f1 100644
--- a/src/adapter.c
+++ b/src/adapter.c
@@ -4347,7 +4347,7 @@ static void load_link_keys(struct btd_adapter *adapter, GSList *keys,
                struct link_key_info *info = l->data;
 
                bacpy(&key->addr.bdaddr, &info->bdaddr);
-               key->addr.type = info->bdaddr_type;
+               key->addr.type = BDADDR_BREDR;
                key->type = info->type;
                memcpy(key->val, info->key, 16);
                key->pin_len = info->pin_len;
```

bjowi, 2023-12-17:
I appllied the BDADDR_BREDR patch, and it helped inn my case. I'm assuming I had the same problem as the original poster, the error message was certainly the same.

Antiz96, 2023-12-17:
I confirm the change included in https://github.com/tedd-an/bluez-upstream-test/pull/484 fixes the issue.

closed 2023-12-20 by BluezTestBot

zeroepoch, 2023-12-21:
@Vudentz, IMO (as an user) this issue is enough of a regression to warrant a new release. I can't reconnect my BT audio devices at all until applying the fix mentioned in the PR and then repeating the initial pairing. I would guess many people are going to have a broken BT setup soon once they update to Bluez 5.71. I created RedHat bug https://bugzilla.redhat.com/show_bug.cgi?id=2255481 for this issue because Fedora just picked up the change and it caused me some frustration trying to join a meeting a few days ago. I finally had time to investigate the problem and located this issue and confirmed things are good again after applying the patch.

pallaswept, 2024-01-03:
Came by to report this and found existing cases about it already so his is just a +1. I gather from the above that the patch to resolve the issue is in place, but there's no release yet, so distributions won't be rolling it out yet. OpenSUSE Tumbleweed rolled out 5.71 today, and after a reboot none of my BT devices would connect. After some experimentation I found that un-pairing and re-pairing them appeared to work, but after a reboot, same failure. Same errors as mentioned above.

> IMO (as an user) this issue is enough of a regression to warrant a new release. I would guess many people are going to have a broken BT setup soon once they update to Bluez 5.71

This dude saw the future coming. Can we please have a new release @Vudentz? I've been working on packaging today so I've been doing updates all day long in close succession, so I will certainly be one of the first of many Tumbleweed users to find their BT broken over the coming few days as people do an update and get this broken release. I'll report this downstream to them as well, perhaps they'll patch their 5.71 in the meantime, but a new release would go a long way to helping distros sort this out.

Vudentz, 2024-01-03:
> Came by to report this and found existing cases about it already so his is just a +1. I gather from the above that the patch to resolve the issue is in place, but there's no release yet, so [...]

There are some other regression I want to be fixed before 5.72.

pallaswept, 2024-01-03:
Understood, and thanks for the speedy feedback. [I've reported this to OpenSUSE downstream](https://bugzilla.opensuse.org/show_bug.cgi?id=1218522) so hopefully they'll apply the patch to 5.7.1 in the meantime.

pallaswept, 2024-01-03:
FYI OpenSUSE have patched 5.71 with [your code](https://github.com/bluez/bluez/commit/7ad5669402c9acff8e4cc808edc12a41df36654e) and it's been pushed to release already. Should roll out very soon. They sure were quick!

zeroepoch, 2024-01-04:
FYI, Fedora 38/39 also has updated packages now pushed to stable containing this patch.