bluez/bluez #687: Unable to pair vivo TWS Air2 earphones (LE Audio)

testuser7, 2023-12-16:
I have problem with pairing vivo TWS Air2 earphones as a LE Audio device.
I'm using Intel AX210 card and BlueZ 5.70 configured according to this manual: https://gitlab.freedesktop.org/pipewire/pipewire/-/wikis/LE-Audio-+-LC3-support
When `ControllerMode = le` is set, the device is not even displayed during scan.

Services discovered using bettercap:
```
┌──────────────┬──────────────────────────────────────┬─────────────────────┬─────────────────────────────┐
│   Handles    │      Service > Characteristics       │     Properties      │            Data             │
├──────────────┼──────────────────────────────────────┼─────────────────────┼─────────────────────────────┤
│ 0001 -> 0007 │ Generic Access (1800)                │                     │                             │
│ 0003         │     Device Name (2a00)               │ READ                │ vivo TWS Air200             │
│ 0005         │     Appearance (2a01)                │ READ                │ Unknown                     │
│ 0007         │     2aa6                             │ READ                │ 00                          │
│              │                                      │                     │                             │
│ 000c -> 0015 │ Generic Attribute (1801)             │                     │                             │
│ 000e         │     Service Changed (2a05)           │ INDICATE            │                             │
│ 0011         │     2b29                             │ READ, WRITE         │ 01                          │
│ 0013         │     2b2a                             │ READ                │ ´Zµp´DÚ[^îðº1e¸0dM          │
│ 0015         │     2b3a                             │ READ                │ 01                          │
│              │                                      │                     │                             │
│ 0016 -> 0016 │ 12994b7e6d4742158c9eaae9a1095ba3     │                     │                             │
│              │                                      │                     │                             │
│ 0017 -> 001f │ 00001100d10211e19b2300025b00a5a5     │                     │                             │
│ 0019         │     00001101d10211e19b2300025b00a5a5 │ WRITE               │                             │
│ 001b         │     00001102d10211e19b2300025b00a5a5 │ READ, NOTIFY        │ 0000                        │
│ 001e         │     00001103d10211e19b2300025b00a5a5 │ READ, WRITE, NOTIFY │ 0000                        │
│              │                                      │                     │                             │
│ 0020 -> 0023 │ 0837                                 │                     │                             │
│ 0022         │     00000838d10211e19b2300025b00a5a5 │ READ, WRITE, NOTIFY │ request not supported       │
│              │                                      │                     │                             │
│ 0024 -> 0030 │ 184e                                 │                     │                             │
│ 0026         │     2bc6                             │ WRITE, NOTIFY       │                             │
│ 0029         │     2bc4                             │ READ, NOTIFY        │ insufficient authentication │
│ 002c         │     2bc4                             │ READ, NOTIFY        │ insufficient authentication │
│ 002f         │     2bc5                             │ READ, NOTIFY        │ insufficient authentication │
│              │                                      │                     │                             │
│ 0031 -> 0043 │ 1850                                 │                     │                             │
│ 0033         │     2bcd                             │ READ, NOTIFY        │ insufficient authentication │
│ 0036         │     2bce                             │ READ, NOTIFY        │ insufficient authentication │
│ 0039         │     2bca                             │ READ, WRITE, NOTIFY │ insufficient authentication │
│ 003c         │     2bc9                             │ READ, NOTIFY        │ insufficient authentication │
│ 003f         │     2bcc                             │ READ, WRITE, NOTIFY │ insufficient authentication │
│ 0042         │     2bcb                             │ READ, NOTIFY        │ insufficient authentication │
│              │                                      │                     │                             │
│ 0044 -> 0047 │ 184d                                 │                     │                             │
│ 0046         │     2bc3                             │ READ, WRITE, NOTIFY │ insufficient authentication │
│              │                                      │                     │                             │
│ 0048 -> 004a │ 1855                                 │                     │                             │
│ 004a         │     8fc9                             │ READ                │ insufficient authentication │
│              │                                      │                     │                             │
│ 004b -> 0056 │ 1846                                 │                     │                             │
│ 004d         │     2b84                             │ READ, NOTIFY        │ insufficient authentication │
│ 0050         │     2b85                             │ READ, NOTIFY        │ insufficient authentication │
│ 0053         │     2b86                             │ READ, WRITE, NOTIFY │ insufficient authentication │
│ 0056         │     2b87                             │ READ                │ insufficient authentication │
│              │                                      │                     │                             │
│ 0057 -> 0058 │ 1853                                 │                     │                             │
│              │                                      │                     │                             │
└──────────────┴──────────────────────────────────────┴─────────────────────┴─────────────────────────────┘
```

testuser7, 2023-12-18:
After pairing the device, I have the following information:
```
[vivo TWS Air2]# info
Device xx:xx:xx:xx:xx:xx (public)
	Name: vivo TWS Air2
	Alias: vivo TWS Air2
	Class: 0x00248404 (2393092)
	Icon: audio-headset
	Paired: yes
	Bonded: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Vendor specific           (00000837-d102-11e1-9b23-00025b00a5a5)
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (8486eb04-d102-11e1-9b23-00025b00a5a5)
	UUID: Vendor specific           (99999999-9999-9999-9999-999999999999)
	Modalias: bluetooth:v02B0p0000d001F
	Battery Percentage: 0x64 (100)
```
Controller info:
```
[bluetooth]# show
Controller xx:xx:xx:xx:xx:xx (public)
	Name: xxx
	Alias: xxx
	Class: 0x007c010c (8126732)
	Powered: yes
	PowerState: on
	Discoverable: no
	DiscoverableTimeout: 0x000000b4 (180)
	Pairable: yes
	UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
	UUID: Audio Stream Control      (0000184e-0000-1000-8000-00805f9b34fb)
	UUID: Volume Offset Control     (00001845-0000-1000-8000-00805f9b34fb)
	UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
	UUID: Volume Control            (00001844-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00005005-0000-1000-8000-0002ee000001)
	UUID: Microphone Control        (0000184d-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
	UUID: Published Audio Capabil.. (00001850-0000-1000-8000-00805f9b34fb)
	UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
	UUID: Broadcast Audio Scan      (0000184f-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v1D6Bp0246d0546
	Discovering: no
	Roles: central
	Roles: peripheral
	Roles: central-peripheral
	ExperimentalFeatures: BlueZ Experimental Simu.. (671b10b5-42c0-4696-9227-eb28d1b049d6)
	ExperimentalFeatures: BlueZ Experimental Blue.. (330859bc-7506-492d-9370-9a6f0614037f)
	ExperimentalFeatures: BlueZ Experimental ISO... (6fbaf188-05e0-496a-9885-d6ddfdb4e03e)
	ExperimentalFeatures: BlueZ Experimental LL p.. (15c0a148-c273-11ea-b3de-0242ac130004)
	ExperimentalFeatures: BlueZ Offload Codecs      (a6695ace-ee7f-4fb9-881a-5fac66c629af)
Advertising Features:
	ActiveInstances: 0x00 (0)
	SupportedInstances: 0x0c (12)
	SupportedIncludes: tx-power
	SupportedIncludes: appearance
	SupportedIncludes: local-name
	SupportedSecondaryChannels: 1M
	SupportedSecondaryChannels: 2M
	SupportedSecondaryChannels: Coded
	SupportedCapabilities Key: MinTxPower
	SupportedCapabilities Value: 0xffffffde (-34)
	SupportedCapabilities Key: MaxTxPower
	SupportedCapabilities Value: 0x0007 (7)
	SupportedCapabilities Key: MaxAdvLen
	SupportedCapabilities Value: 0x1f (31)
	SupportedCapabilities Key: MaxScnRspLen
	SupportedCapabilities Value: 0x1f (31)
	SupportedFeatures: CanSetTxPower
	SupportedFeatures: HardwareOffload
Advertisement Monitor Features:
	SupportedMonitorTypes: or_patterns
	SupportedFeatures: controller-patterns
```

Vudentz, 2023-12-18:
@testuser7 are you sure it is advertising? From the looks of it seems to be doing available over Classic given that it has found A2DP UUID, are you sure you are in LE mode only?

testuser7, 2023-12-18:
> @testuser7 are you sure it is advertising? From the looks of it seems to be doing available over Classic given that it has found A2DP UUID, are you sure you are in LE mode only?

Here is `le only` mode controller info:
```
[bluetooth]# show
Controller xx:xx:xx:xx:xx:xx (public)
	Name: xxx
	Alias: xxx
	Class: 0x00000000 (0)
	Powered: yes
	PowerState: on
	Discoverable: no
	DiscoverableTimeout: 0x000000b4 (180)
	Pairable: yes
	UUID: Published Audio Capabil.. (00001850-0000-1000-8000-00805f9b34fb)
	UUID: Audio Stream Control      (0000184e-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Broadcast Audio Scan      (0000184f-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Volume Control            (00001844-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Microphone Control        (0000184d-0000-1000-8000-00805f9b34fb)
	UUID: Volume Offset Control     (00001845-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v1D6Bp0246d0546
	Discovering: no
	Roles: central
	Roles: peripheral
	ExperimentalFeatures: BlueZ Experimental ISO... (6fbaf188-05e0-496a-9885-d6ddfdb4e03e)
Advertising Features:
	ActiveInstances: 0x00 (0)
	SupportedInstances: 0x0c (12)
	SupportedIncludes: tx-power
	SupportedIncludes: appearance
	SupportedIncludes: local-name
	SupportedSecondaryChannels: 1M
	SupportedSecondaryChannels: 2M
	SupportedSecondaryChannels: Coded
	SupportedCapabilities Key: MinTxPower
	SupportedCapabilities Value: 0xffffffde (-34)
	SupportedCapabilities Key: MaxTxPower
	SupportedCapabilities Value: 0x0007 (7)
	SupportedCapabilities Key: MaxAdvLen
	SupportedCapabilities Value: 0x1f (31)
	SupportedCapabilities Key: MaxScnRspLen
	SupportedCapabilities Value: 0x1f (31)
	SupportedFeatures: CanSetTxPower
	SupportedFeatures: HardwareOffload
Advertisement Monitor Features:
	SupportedMonitorTypes: or_patterns
	SupportedFeatures: controller-patterns
```
But then the earphones are not discovered...

Vudentz, 2023-12-19:
> Here is `le only` mode controller info:
>
> ```
> [bluetooth]# show
> Controller xx:xx:xx:xx:xx:xx (public)
> 	Name: xxx
> 	Alias: xxx
> 	Class: 0x00000000 (0)
> 	Powered: yes
> 	PowerState: on
> 	Discoverable: no [...]

Perhaps you should contact tha manufacturer then, I wonder if is trying to do GATT over BR/EDR and then once the services are discovered then do connect LE, etc, that said afaik LE Advertisements are required by LE Audio, anyway we currently don't support GATT over BR/EDR as a client in BlueZ.

testuser7, 2023-12-20:
> Perhaps you should contact tha manufacturer then, I wonder if is trying to do GATT over BR/EDR and then once the services are discovered then do connect LE, etc, that said afaik LE Advertisements are required by LE Audio, anyway we currently don't support GATT over BR/EDR as a client in BlueZ.

Are there any headphones that work as an LE Audio device with BlueZ?

Vudentz, 2023-12-20:
> Are there any headphones that work as an LE Audio device with BlueZ?

Yes, I tested with Galaxy Buds 2 Pro and Creative Zen Hybrid Pro.

testuser7, 2023-12-20:
> Yes, I tested with Galaxy Buds 2 Pro and Creative Zen Hybrid Pro.

Can you show how services are discovered using bettercap or something similar?

Vudentz, 2023-12-20:
> > Yes, I tested with Galaxy Buds 2 Pro and Creative Zen Hybrid Pro.
> 
> Can you show how services are discovered using bettercap or something similar?

bluetoothd does automatically discover all the services using GATT Discover procedure, Im not really familiar with bettercap, we do have our own tools like btmon to decode the traffic if that is what you are after, bluetoothctl can also list the discovered GATT attributes but in case of PACS/ASCS they are normally claimed by the bluetoothd in order to avoid conflicts with upper layers.

Btw, I wouldn't use bettercap if that is doing read procedures, btmon can actually decode the traffic e.g. 6c81bae99bc593af75913951ed553842862e95a8

testuser7, 2023-12-20:
Using btmon I see these events:
```
> HCI Event: LE Meta Event (0x3e) plen 51                                                                                                                                                            #39 36.638545
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0013
            Props: 0x0013
              Connectable
              Scannable
              Use legacy advertising PDUs
            Data status: Complete
          Legacy PDU Type: ADV_IND (0x0013)
          Address type: Public (0x00)
          Address: xx:xx:xx:xx:xx:xx (vivo Mobile Communication Co., Ltd.)
          Primary PHY: LE 1M
          Secondary PHY: No packets
          SID: no ADI field (0xff)
          TX power: 127 dBm
          RSSI: -53 dBm (0xcb)
          Periodic advertising interval: 0.00 msec (0x0000)
          Direct address type: Public (0x00)
          Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
          Data length: 0x19
        02 01 08 15 ff 37 08 08 01 4c 99 92 12 df 79 0b  .....7...L....y.
        3e 64 64 00 65 3f 00 00 00                       >dd.e?...       
        Flags: 0x08
          Simultaneous LE and BR/EDR (Controller)
        Company: vivo Mobile Communication Co., Ltd. (2103)
          Data: 08014c999212df790b3e646400653f000000
> HCI Event: LE Meta Event (0x3e) plen 41                                                                                                                                                            #40 36.639541
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x001b
            Props: 0x001b
              Connectable
              Scannable
              Scan response
              Use legacy advertising PDUs
            Data status: Complete
          Legacy PDU Type: SCAN_RSP to an ADV_SCAN_IND (0x001b)
          Address type: Public (0x00)
          Address: xx:xx:xx:xx:xx:xx (vivo Mobile Communication Co., Ltd.)
          Primary PHY: LE 1M
          Secondary PHY: No packets
          SID: no ADI field (0xff)
          TX power: 127 dBm
          RSSI: -53 dBm (0xcb)
          Periodic advertising interval: 0.00 msec (0x0000)
          Direct address type: Public (0x00)
          Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
          Data length: 0x0f
        0e ff 37 08 08 00 00 00 00 00 00 00 00 00 00     ..7............ 
        Company: vivo Mobile Communication Co., Ltd. (2103)
          Data: 0800000000000000000000
@ MGMT Event: Device Found (0x0012) plen 54                                                                                                                                                     {0x0001} 36.639636
        LE Address: xx:xx:xx:xx:xx:xx (vivo Mobile Communication Co., Ltd.)
        RSSI: -53 dBm (0xcb)
        Flags: 0x00000000
        Data length: 40
        Flags: 0x08
          Simultaneous LE and BR/EDR (Controller)
        Company: vivo Mobile Communication Co., Ltd. (2103)
          Data: 08014c999212df790b3e646400653f000000
        Company: vivo Mobile Communication Co., Ltd. (2103)
          Data: 0800000000000000000000
```

Vudentz, 2023-12-20:
> Using btmon I see these events:
>
> ```
>       LE Extended Advertising Report (0x0d)
>         Num reports: 1
>         Entry 0
>           Event type: 0x0013
>             Props: 0x0013 [...]

So it does looks like it indicating the support of both LE and BR/EDR in which case BR/EDR is currently the preferred bearer, we could change that but I think we would need to detect that it does indeed support LE Audio but from the looks of it doesn't appear to advertise its support so we would only discover after the GATT Discover procedure, anyway when you set as LE only mode it shall prefer LE so I wonder why you cannot discover it, perhaps there is a bug related to setting LE only mode and having Simultaneous LE and BR/EDR it would still prefer BR/EDR even though that is not enabled.

testuser7, 2023-12-20:
> So it does looks like it indicating the support of both LE and BR/EDR in which case BR/EDR is currently the preferred bearer, we could change that but I think we would need to detect that it does [...]

So changes to the source code are needed?

testuser7, 2024-01-09:
Is this condition causing a problem: https://github.com/bluez/bluez/blob/master/src/device.c#L4469 ?

testuser7, 2024-03-06:
@Vudentz Have any changes been made recently that would help correctly detect these headphones in LE mode?

I just checked on a phone with Android 13 that supports LE Audio and the headphones are also not detected as an LE device.

testuser7, 2024-11-19:
I was able to pair earphones today in `le` mode:
```
Device 4C:99:92:12:DF:79 (public)
	Name: vivo TWS Air2
	Alias: vivo TWS Air2
	Paired: yes
	Bonded: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Unknown                   (00000837-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00001100-d102-11e1-9b23-00025b00a5a5)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Coordinated Set Identif.. (00001846-0000-1000-8000-00805f9b34fb)
	UUID: Microphone Control        (0000184d-0000-1000-8000-00805f9b34fb)
	UUID: Audio Stream Control      (0000184e-0000-1000-8000-00805f9b34fb)
	UUID: Published Audio Capabil.. (00001850-0000-1000-8000-00805f9b34fb)
	UUID: Common Audio              (00001853-0000-1000-8000-00805f9b34fb)
	UUID: Telephony and Media Audio (00001855-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (12994b7e-6d47-4215-8c9e-aae9a1095ba3)
	Sets.Key: /org/bluez/hci0/set_19ffc6078275905b0e580388d093efa5
	Sets.Value.Rank: 0x01 (1)
```
But device is recognized as `unknown`:
[image]

testuser7, 2024-11-20:
@Vudentz is it possible to change the bluetooth class? It seems to be incorrect:
```
api.bluez5.class = "0x000000"
```

Vudentz, 2024-11-20:
> @Vudentz is it possible to change the bluetooth class? It seems to be incorrect:
> 
> ```
> api.bluez5.class = "0x000000"
> ```

No idea where this is coming from, there is an entry on main.conf but that is irrelevant here since LE doesn't use device class, it has been replaced with appearance, but that is only for the peripheral to set.

testuser7, 2024-11-20:
If the class is missing I'm not able to use device as an sound output.

After some changes I'm able to use it as an sound output:
```
Device xx:xx:xx:xx:xx:xx (public)
	Name: vivo TWS Air2
	Alias: vivo TWS Air2
	Class: 0x00248404 (2393092)
	Icon: audio-headset
	Paired: yes
	Bonded: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Unknown                   (00000837-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00001100-d102-11e1-9b23-00025b00a5a5)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Coordinated Set Identif.. (00001846-0000-1000-8000-00805f9b34fb)
	UUID: Microphone Control        (0000184d-0000-1000-8000-00805f9b34fb)
	UUID: Audio Stream Control      (0000184e-0000-1000-8000-00805f9b34fb)
	UUID: Published Audio Capabil.. (00001850-0000-1000-8000-00805f9b34fb)
	UUID: Common Audio              (00001853-0000-1000-8000-00805f9b34fb)
	UUID: Telephony and Media Audio (00001855-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (12994b7e-6d47-4215-8c9e-aae9a1095ba3)
	Sets.Key: /org/bluez/hci0/set_19ffc6078275905b0e580388d093efa5
	Sets.Value.Rank: 0x01 (1)
```
```
Device xx:xx:xx:xx:xx:xx (public)
	Name: vivo TWS Air2
	Alias: vivo TWS Air2
	Paired: yes
	Bonded: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Unknown                   (00000837-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00001100-d102-11e1-9b23-00025b00a5a5)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Coordinated Set Identif.. (00001846-0000-1000-8000-00805f9b34fb)
	UUID: Microphone Control        (0000184d-0000-1000-8000-00805f9b34fb)
	UUID: Audio Stream Control      (0000184e-0000-1000-8000-00805f9b34fb)
	UUID: Published Audio Capabil.. (00001850-0000-1000-8000-00805f9b34fb)
	UUID: Common Audio              (00001853-0000-1000-8000-00805f9b34fb)
	UUID: Telephony and Media Audio (00001855-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (12994b7e-6d47-4215-8c9e-aae9a1095ba3)
	Sets.Key: /org/bluez/hci0/set_19ffc6078275905b0e580388d093efa5
	Sets.Value.Rank: 0x02 (2)
```
But the sound in one earphone is distorted...

closed 2025-08-02 by github-actions[bot]