bluez/bluez #688: Unable to Pair Dualshock3 on bluez 5.71

mmbossoni, 2023-12-17:
Before bluez 5.71:
1 - Connect dualshock 3 through usb cable
2 - bluez would ask trust + authorize
3 - disconnect cable
4 - press PS button and you are device is connected

With bluez 5.71:
On Step 4, when pressing the PS button, bluez asks for a PIN even though device doesn't require one.

I've made sure PS3 plugin is installed.

Let me know which kind of debug info is needed

dmanlfc, 2023-12-20:
confirmed, same issue - `[Sony PLAYSTATION(R)3 Controller]# hci0 04:98:F3:76:A8:DD auth failed with status 0x05 (Authentication Failed)
[Sony PLAYSTATION(R)3 Controller]# Failed to pair: org.bluez.Error.AuthenticationFailed
[Sony PLAYSTATION(R)3 Controller]# hci0 04:98:F3:76:A8:DD type BR/EDR disconnected with reason 4
[CHG] Device 04:98:F3:76:A8:DD Connected: noluetooth
[bluetooth]# hci0 04:98:F3:76:A8:DD type BR/EDR connected eir_len 5
[CHG] Device 04:98:F3:76:A8:DD Connected: yestroller`

rolling back to 5.70 for now.

mmbossoni, 2023-12-20:
After further testing, this might be a dup of https://github.com/bluez/bluez/issues/673, since changing the ClassicBondedOnly config fixed it for me

kohane27, 2023-12-21:
Downgrading to 5.64-2 works for me.

legluondunet, 2023-12-25:
> Downgrading to 5.64-2 works for me.

You do not need to downgrade bluez packages, on Arch, created input.conf file in /etc/bluetooth folder, then added this lines:

```
[General]
ClassicBondedOnly=false
```

restart bluetooth:
sudo systemctl restart bluetooth

my Dualshock 3 works again without pin code.

xadips, 2024-01-03:
Yup the `ClassicBondedOnly=false` helps on multiple controllers since the default change

KaroLaunonen, 2024-02-21:
Great! Finally solution for Dualshock 3. It's been broken for me since 5.68 but this fixes it on 5.72. Thanks!

DavisNT, 2024-03-14:
Setting `ClassicBondedOnly=false` will also enable the CVE-2023-45866 security vulnerability.

For details see:
- https://ubuntu.com/security/CVE-2023-45866 
- https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/profiles/input?id=25a471a83e02e1effb15d5a488b3f0085eaeb675

manuelafm, 2025-04-30:
FYI:
It should now be possible to connect the DualShock3 with Bluetooth with limited exposure to [CVE-2023-45866](https://ubuntu.com/security/CVE-2023-45866) and without changing `ClassicBondedOnly=false`: https://github.com/bluez/bluez/issues/1165

closed 2025-09-13 by github-actions[bot]