bluez/bluez #689: [Regression] [5.69] Cannot pair or make WMR controllers stay connected anymore

Meister1593, 2023-12-18:
OS: Arch
Used `downgrade` to bisect problematic bluez version

Before 5.69:
I try to pair, it has loong wait on kde plasma dialog, fails but stays connected for short 15-30 seconds, then disconnects.
After restarting controllers, it automatically connects and seemingly works (althrough no `/dev/js` devices being produced, only `/dev/hidraw*` and they don't seem to be affected by udev rules)

On/After 5.69:
I try to pair, it almost instantly fails on kde plasma dialog and then stays connected for short 15-30 seconds, then disconnectes. Restarting controller connects it again, but for 5 seconds and it disconnects.
```
Agent registered
[bluetooth]# scan le
Discovery started
[CHG] Controller 00:1A:7D:DA:71:13 Discovering: yes
[NEW] Device D8:C4:97:05:3C:3D Motion controller - Left
[bluetooth]# pair D8:C4:97:05:3C:3D
Attempting to pair with D8:C4:97:05:3C:3D
[CHG] Device D8:C4:97:05:3C:3D Connected: yes
[CHG] Device D8:C4:97:05:3C:3D Bonded: yes
[CHG] Device D8:C4:97:05:3C:3D Modalias: usb:v045Ep065Bd0208
[CHG] Device D8:C4:97:05:3C:3D UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[CHG] Device D8:C4:97:05:3C:3D UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device D8:C4:97:05:3C:3D ServicesResolved: yes
[CHG] Device D8:C4:97:05:3C:3D Paired: yes
Pairing successful
[CHG] Device D8:C4:97:05:3C:3D ServicesResolved: no
[CHG] Device D8:C4:97:05:3C:3D Connected: no -- here it disconnects after pair
[CHG] Device D8:C4:97:05:3C:3D Class: 0x00000508 (1288)
[CHG] Device D8:C4:97:05:3C:3D Icon: input-gaming
[CHG] Device D8:C4:97:05:3C:3D Connected: yes -- connects after turning on
[CHG] Device D8:C4:97:05:3C:3D Connected: no -- and quickly disconnects
```

I wasn't sure what kind of logs needed, so if there is anything more needed, please let me know.

loco-choco, 2023-12-19:
I am having a similar issue on 5.70, but first it said that the pairing fail, and now if I turn off and on the controller, it automatically connects to the machine with the following dmesg log:
```bash
[ 6289.428823] hid-generic 0005:045E:065B.000B: unknown main item tag 0x0
[ 6289.428874] hid-generic 0005:045E:065B.000B: unknown main item tag 0x0
[ 6289.429435] input: Motion controller - Right as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/bluetooth/hci0/hci0:56/0005:045E:065B.000B/input/input33
[ 6289.429838] input: Motion controller - Right as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/bluetooth/hci0/hci0:56/0005:045E:065B.000B/input/input34
[ 6289.430155] hid-generic 0005:045E:065B.000B: input,hidraw4: BLUETOOTH HID v2.08 Device [Motion controller - Right] on 00:1a:7d:da:71:03
```

closed 2025-08-02 by github-actions[bot]

Wintch, 2026-10-04:
For anyone landing here on a recent BlueZ: on 5.82 (Debian 13, kernel 6.12) HP Reverb G2 controllers (045e:066a, classic BR/EDR HID, not BLE) pair and stay connected to a host adapter (TP-Link UB500 Plus, RTL8761B) with `pairable on`, then opening the HID control (PSM 0x11) and interrupt (PSM 0x13) L2CAP channels from the host, in that order, at security level MEDIUM (what Windows does), then `trust`. The hidraw nodes `0005:045E:066A` appear afterwards and Monado's `wmr_bt_controller` / the Oasis driver read them. The controllers only keep the HID link up while awake (a few seconds of stillness drops them to a 20 Hz idle rate, and they sleep later), so wake them before expecting the nodes.

Script and notes: https://github.com/Wintch/reverb-g2/blob/main/scripts/bt-controllers/bt-winorder.py and https://github.com/Wintch/reverb-g2/blob/main/docs/136-g2-controllers-system-bluetooth.md

I'm not claiming this explains the 5.69 regression, only that pairing works today with that order