bluez/bluez #690: BAP sometime not using the right address type to connect

wnmusic, 2023-12-20:
For a LE controller with no public address, (all 0),  the bt_io_connect function will not work!  This is because it always use LE_PUBLIC address as the source address. But in above case, the address type is actually LE_RANDOM. 

A simple Patch has also been included. 
[bap_addr.patch](https://github.com/bluez/bluez/files/13722726/bap_addr.patch)

Vudentz, 2023-12-20:
> For a LE controller with no public address, (all 0), the bt_io_connect function will not work! This is because it always use LE_PUBLIC address as the source address. But in above case, the address [...]

Oh are you talking about devices using privacy? If you can please send a proper patch to the mailing list so we get CI to test it.

wnmusic, 2024-01-07:
Yes, this for LE devices with no public address (all 0 ).   I wonder which mailing list shall I send to ? I am pretty new to the bluez dev community, or shall I do a pull request instead ? 

Thanks. 
Ning

Vudentz, 2025-05-22:
This has been fixed by 87cabb26793554ad9f67f0b23ecf19869b20dd17

closed 2025-05-22 by Vudentz