bluez/bluez #691: [5.70] Failure to connect to Jabra headset

Chaman-veteran, 2023-12-20:
Hello,

I'm on 6.6.7-arch1-1 and since update to 5.70 I struggle a lot to connect to my Jabra headset.
It's a Jabra Elite 85h with firmware 1.6.3.

When trying to connect using bluetoothctl, I'm faced with :
```
[bluetooth]# connect 70:BF:92:11:57:5A
Attempting to connect to 70:BF:92:11:57:5A
[CHG] Device 70:BF:92:11:57:5A Connected: yes
Failed to connect: org.bluez.Error.Failed br-connection-unknown
[CHG] Device 70:BF:92:11:57:5A Connected: no
[CHG] Device 70:BF:92:11:57:5A Connected: yes
```

On btmon, I can see some errors, such as
`= bluetoothd: src/profile.c:ext_connect() Hands-Free Voice gateway failed connect to 70:BF:92:11:57:5A: Connection refused (111)`
`= bluetoothd: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 70:BF:92:11:57:5A: Connection reset by peer (104)`
`= bluetoothd: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 70:BF:92:11:57:5A: Permission denied (13)`

My network controller is:  MEDIATEK Corp. MT7921 802.11ax PCI Express Wireless Network Adapter

If needed I can give BTSnoop log file. 

I didn't found something like this on the issues but I may be missing something, so is it a known issue?
And do you have any hints on how to fix this?

Thanks :)

Vudentz, 2023-12-20:
@Chaman-veteran it is probably a good idea to add the btmon logs, from the looks of it doesn't even connect so I wonder if there is some pairing related problem.

Chaman-veteran, 2023-12-20:
@Vudentz  Thanks for your quick reply, here's the log [bluetooth.log](https://github.com/bluez/bluez/files/13729323/bluetooth.log)

Chaman-veteran, 2023-12-20:
@Vudentz It was a pairing related problem and it's fixed (for me), thanks a lot!
I just had to forgot and redo the pairing between my headset and my computer.

Idk if it was something you were aware it could do while releasing the 5.71, which is why I don't close this issue it for now.

lnicola, 2023-12-21:
I had the same issue with my keyboard (Keychron K8), re-pairing fixed it. I noticed `Failed to load link keys for hci0: Invalid Parameters (0x0d)` in the logs, not sure if that's new. My Bluetooth mouse worked fine.

This is unrelated to the 5.70 (`ClassicBondedOnly`) change.

Chaman-veteran, 2024-03-09:
Closing it as it seemed to be an epiphenomenon, maybe due to a peculiar update or idk, and there's no motion on this issue nor ppl that seems to still have this problem

closed 2024-03-09 by Chaman-veteran