bluez/bluez #692: Impossible to use bluetoothctl in shell scripts.

johnoyler-quext, 2023-12-20:
Even `expect` is difficult to use, when I can't connect to any device until scanning is turned on, and that floods the screen with rssi settings. Even a proper filter that limits it to the device I'm attempting to communicate with sees rssi messages often enough that they interfere, and the connection doesn issue a proper message, but changes the prompt with the rssi message. Connections to BLE devices last all of 3 seconds, in which I have to issue upwards of 6 to 8 commands to accomplish anything. This last problem would be no big deal, except that it's not scriptable, so I can't arrange all of the commands and issue them at once.

Other bluetooth command line tools are deprecated, and this one is described as "command line tool" everywhere, which it is very much not. It's a console application. You might as well call Microsoft Word a command line application... it's possible to launch it from cmd.exe, after all. I cannot issue two commands as arguments to the executable, one to execute after the other. There are no pages anywhere on the internet that describe how to use these commands. The help menu is bizarre, look at this:

```
Usage:
	bluetoothctl [--options] [commands]
```

Does this mean I can issue more than one command? Apparently not. No big deal, check the man page (emptier than the help menu). This is not a tool made for a power user, anyone exploring unsupported bluetooth devices, someone who would wish to automate anything that could be accomplished with this tool (whatever that is), or any other class of user I can imagine.

Even should you manage to to get it to connect in a way that a subsequent single command works, much of the functionality relies on a previous command having been successfully run (select-attribute before read).

```
$ bluetoothctl -- gatt.select-attribute 52052c11-e701-4782-99f5-8ce88dbb1502
$ bluetoothctl -- gatt.read 52052c11-e701-4782-99f5-8ce88dbb1502
No attribute selected
```

Something as simple as allowing me to stage up several commands to be run in sequence would go a long way, either in the interactive or command-line version of this. Making it more expect-friendly would make it scriptable.  But this seems to be a mess, and if there is a more appropriate tool then bluetoothctl has so fully filled up Google that I can get no other results after days of trying.

Vudentz, 2023-12-21:
@johnoyler-quext well perhaps you can use -s/--init-script <file> and then have the commands in the file.

sonus89, 2024-06-15:
@Vudentz Why do we have a functionality like **gatt.select-attribute** if it is not possible to use it? 

Could you help us out with a working example?
I don't want to enter interactive mode just read a GATT attribute

sonus89, 2024-06-15:
@johnoyler-quext 

I figured out the best way to automate bluetoothctl
I found the solution on the Arch wiki page 

echo -e "command1\ncommand2\n" | bluetoothctl

charles-dyfis-net, 2024-11-09:
@pneumus, the OP already knows that exists -- `expect`, which the ticket references, is a cleaner / automated version that's able to do things like wait for output, do conditional logic, etc (f/e, only run command 2 if command 1 succeeds; or only run command 1 after we see a log message that the daemon is successfully connected).

But as the OP says, `expect` is a messy hack compared to a proper programmatic interface; just blindly writing to stdin, as the Arch wiki suggests, is even _more_ of a messy hack, and it's deeply unfortunate if there's nothing better suited to purpose available.

andrelaszlo, 2025-04-17:
Some ideas that would make scripting easier, from less to more work (guessing):

- Just disabling ansi output would make expect easier to use.
- Consistent prompt, to make it easier to know when to send commands. I've seen `[bluetooth]#` , which recently seems to have changed to `[bluetoothctl]>` - breaking my expect script. Today I'm getting `[MX Master 3]>` (my mouse) or `[Keychron K8 Pro]>` (my keyboard) - breaking my script again!
- Flag to only write output of commands to stdout, to make it more predictable.
- Built-in command to wait for something to change.
- Scripting mode with simpler input/output.
- IPC protocol for controlling bluetoothctl from a different process, and some high-level wrapper (Python?). Something like https://man.archlinux.org/man/sway-ipc.7.en perhaps.

Here's my script to re-pair my headset (It can only be paired with 2 devices at a time, so this is a hack I've found to quickly fix that issue if I've had it paired to something else):

<details>
<summary>Here's my hacky expect script, if it helps anyone:</summary>

```tcl
#!/usr/bin/expect

# Problematic, keeps changing format
# set prompt "\\\[bluetooth\\\]#"
# set prompt "\\\[bluetoothctl\\\]>"
set prompt "\\\[.*\\\]>"

set address "12:34:56:C0:FF:EE"
set timeout 60

proc remove_if_exists {address prompt} {
    expect *
    send "devices\r"

    # It prints the prompt again before printing the output
    expect -re $prompt

    expect {
        -re $address {
            send_user "Removing existing device\n"

            send "remove $address\r"

            expect -re "Device has been removed"
        }
        -re $prompt {
            send_user "No existing device found\n"
        }
    }
    expect -re $prompt
    expect *
}

proc pair_device {address} {
    expect *
    send "pair $address\r"

    expect {
        -re "Pairing successful" { return true }
        -re "Failed to pair" { return false }
    }
}

proc wait_for_device {address filter} {
    set timeout 1
    send_user "\nWaiting for device\n"
    expect *
    send "devices $filter\r"
    expect {
        timeout { wait_for_device $address $filter}
        -re "Device $address"
    }
}

if {![exp_debug]} {
    trap {
        send \x03
        send_user "\nInterrupted...\n"

        expect -re $prompt
        send "scan off\r"

        expect -re $prompt
        send "quit\r"
        exit
    } SIGINT
}

spawn bluetoothctl

expect -re $prompt
remove_if_exists $address $prompt

send "scan on\r"
wait_for_device $address ""

pair_device $address

expect *
send "trust $address\r"
expect -re "$address trust succeeded"

expect -re $prompt
expect *
send "connect $address\r"

wait_for_device $address "Connected"

expect -re $prompt
send "quit\r"
expect eof
```
</details>

mhantsch, 2025-05-22:
This is a valid issue, I don't think it should be closed just because no work has been done on it in the past 30 days...

Vudentz, 2025-05-22:
> Some ideas that would make scripting easier, from less to more work (guessing):
> 
>     * Just disabling ansi output would make expect easier to use.
> 
>     * Consistent prompt, to make it easier to know when to send commands. I've seen `[bluetooth]#` , which recently seems to have changed to `[bluetoothctl]>` - breaking my expect script. Today I'm getting `[MX Master 3]>` (my mouse) or `[Keychron K8 Pro]>` (my keyboard) - breaking my script again!
> 
>     * Flag to only write output of commands to stdout, to make it more predictable.
> 
>     * Built-in command to wait for something to change.
> 
>     * Scripting mode with simpler input/output.
> 
>     * IPC protocol for controlling bluetoothctl from a different process, and some high-level wrapper (Python?). Something like https://man.archlinux.org/man/sway-ipc.7.en perhaps.
> 
> 
> Here's my script to re-pair my headset (It can only be paired with 2 devices at a time, so this is a hack I've found to quickly fix that issue if I've had it paired to something else):
> Here's my hacky expect script, if it helps anyone:
> 
> #!/usr/bin/expect
> 
> # Problematic, keeps changing format
> # set prompt "\\\[bluetooth\\\]#"
> # set prompt "\\\[bluetoothctl\\\]>"
> set prompt "\\\[.*\\\]>"
> 
> set address "12:34:56:C0:FF:EE"
> set timeout 60
> 
> proc remove_if_exists {address prompt} {
>     expect *
>     send "devices\r"
> 
>     # It prints the prompt again before printing the output
>     expect -re $prompt
> 
>     expect {
>         -re $address {
>             send_user "Removing existing device\n"
> 
>             send "remove $address\r"
> 
>             expect -re "Device has been removed"
>         }
>         -re $prompt {
>             send_user "No existing device found\n"
>         }
>     }
>     expect -re $prompt
>     expect *
> }
> 
> proc pair_device {address} {
>     expect *
>     send "pair $address\r"
> 
>     expect {
>         -re "Pairing successful" { return true }
>         -re "Failed to pair" { return false }
>     }
> }
> 
> proc wait_for_device {address filter} {
>     set timeout 1
>     send_user "\nWaiting for device\n"
>     expect *
>     send "devices $filter\r"
>     expect {
>         timeout { wait_for_device $address $filter}
>         -re "Device $address"
>     }
> }
> 
> if {![exp_debug]} {
>     trap {
>         send \x03
>         send_user "\nInterrupted...\n"
> 
>         expect -re $prompt
>         send "scan off\r"
> 
>         expect -re $prompt
>         send "quit\r"
>         exit
>     } SIGINT
> }
> 
> spawn bluetoothctl
> 
> expect -re $prompt
> remove_if_exists $address $prompt
> 
> send "scan on\r"
> wait_for_device $address ""
> 
> pair_device $address
> 
> expect *
> send "trust $address\r"
> expect -re "$address trust succeeded"
> 
> expect -re $prompt
> expect *
> send "connect $address\r"
> 
> wait_for_device $address "Connected"
> 
> expect -re $prompt
> send "quit\r"
> expect eof

Not sure why you are using a python script when you can use native scritpting: bluetoothctl -s or bluetoothctl> script?

If it is just because there is no connect by address command perhaps we can make connect internally scan and connect, or alternatively use https://github.com/bluez/bluez/blob/master/doc/org.bluez.Adapter.rst#object-connectdevicedict-properties-experimental

closed 2025-05-22 by Vudentz