bluez/bluez #696: Cannot connect Sixaxis PS3 controller through Bluetooth

AdelKS, 2023-12-27:
Hello,

I have been trying to connect my Sixaxis (PS3) controller and I simply cannot get it to work with bluetooth: the controller works in USB, I get prompted to trust it and authorize it (KDE), but when I disconnect the USB connection, the controller communicates with my Bluetooth receiver but the connection fails.

Here are logs with `btmon` when I connect the controller with USB, the disconnect it for the bluetooth exchange to happen, until the controller turns off (after it fails to get a slot assigned, as it indicated its LED lights for slot 1,2,3 or 4), using `bluez-5.62` and `bluez-5.71`

[btmon-5.62.log.txt](https://github.com/bluez/bluez/files/13781860/btmon-5.62.log.txt)
[btmon-5.71.log.txt](https://github.com/bluez/bluez/files/13781861/btmon-5.71.log.txt)

I did set `ClassicBondedOnly=false` in `/etc/bluetooth/input.conf`.
I tried `bluez` `5.71` down to version `5.62`, kernel `6.6.8` and kernel `6.1.69`.

`dmesg` shows these relevant lines
```
kernel: usb 3-4: USB disconnect, device number 12
bluetoothd[612007]: sixaxis: compatible device connected: Sony PLAYSTATION(R)3 Controller (054C:0268 /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:09:00.3/usb3/3-4/3-4:1.0/0003:054C:0268.000F/hidraw/hidraw7
)
kernel: sony 0003:054C:0268.000F: input,hiddev100,hidraw7: USB HID v1.11 Joystick [Sony PLAYSTATION(R)3 Controller] on usb-0000:09:00.3-4/input0
kernel: input: Sony PLAYSTATION(R)3 Controller as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:09:00.3/usb3/3-4/3-4:1.0/0003:054C:0268.000F/input/input32
kernel: usb 3-4: Manufacturer: Sony
kernel: usb 3-4: Product: PLAYSTATION(R)3 Controller
kernel: usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
kernel: usb 3-4: New USB device found, idVendor=054c, idProduct=0268, bcdDevice= 1.00
kernel: usb 3-4: new full-speed USB device number 12 using xhci_hcd
```

legluondunet, 2023-12-30:
In  your /etc/bluetooth/input.conf file, you have to add
[General]
before 
ClassicBondedOnly=false

`[General]
ClassicBondedOnly=false`

AdelKS, 2023-12-30:
Hello, thank you for your feedback! As I mentioned in my first comment I did use `ClassicBondedOnly=false` in that file, unfortunately.

AdelKS, 2023-12-30:
Here's my file `/etc/bluetooth/input.conf`

```
# Configuration file for the input service

# This section contains options which are not specific to any
# particular interface
[General]

# Set idle timeout (in minutes) before the connection will
# be disconnect (defaults to 0 for no timeout)
#IdleTimeout=30

# Enable HID protocol handling in userspace input profile
# Defaults to false (HIDP handled in HIDP kernel module)
#UserspaceHID=true

# Limit HID connections to bonded devices
# The HID Profile does not specify that devices must be bonded, however some
# platforms may want to make sure that input connections only come from bonded
# device connections. Several older mice have been known for not supporting
# pairing/encryption.
# Defaults to true for security.
ClassicBondedOnly=false

# LE upgrade security
# Enables upgrades of security automatically if required.
# Defaults to true to maximize device compatibility.
#LEAutoSecurity=true
```

endrift, 2024-06-21:
Is there any way to carve out an exception to ClassicBondedOnly=true for the sixaxis controller? I don't want to re-expose users to CVE-2023-45866 but I'd like the DS3 to work too.

AdelKS, 2024-09-28:
It's back to working with bluez `5.78` and linux kernel `6.11.0`

closed 2024-09-28 by AdelKS

endrift, 2025-02-27:
This is not fixed for `ClassicBondedOnly=true` (the default), which means the only way to pair with the Sixaxis still is to expose yourself to the CVE.