bluez/bluez #697: Keyboard pair successful, connection fails with "permission denied"

dmfay, 2023-12-28:
I can't get a bluetooth keyboard to connect to my Linux machine. It pairs, but the connection almost always fails (I _have_ gotten it to type twice out of many attempts). The keyboard connects to my phone readily. The bluetooth service logs generally show these errors:

```
Dec 28 00:11:16 medusa bluetoothd[2279]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info
Dec 28 00:11:45 medusa bluetoothd[2279]: profiles/input/device.c:control_connect_cb() connect to 54:46:6B:0F:22:A7: Permission denied (13)
```

I have also seen but cannot reproduce a `Too many levels of symbolic links (40)` connection error and once got it into a state where the device was "Busy" and didn't talk.

Here's a bluetoothctl session:

```
[bluetooth]# hci0 new_settings: powered bondable ssp br/edr le secure-conn cis-central cis-peripheral
[bluetooth]# Agent registered
[bluetooth]# [CHG] Controller ma:ca:dd:re:ss Pairable: yes
[bluetooth]# show
Controller ma:ca:dd:re:ss (public)
        Manufacturer: 0x0002 (2)
        Version: 0x0c (12)
        Name: medusa
        Alias: medusa
        Class: 0x00000104 (260)
        Powered: yes
        Discoverable: no
        DiscoverableTimeout: 0x000000b4 (180)
        Pairable: yes
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0547
        Discovering: no
        Roles: central
        Roles: peripheral
Advertising Features:
        ActiveInstances: 0x00 (0)
        SupportedInstances: 0x0c (12)
        SupportedIncludes: tx-power
        SupportedIncludes: appearance
        SupportedIncludes: local-name
        SupportedSecondaryChannels: 1M
        SupportedSecondaryChannels: 2M
        SupportedSecondaryChannels: Coded
[bluetooth]# scan on
[bluetooth]# SetDiscoveryFilter success
[bluetooth]# hci0 type 7 discovering on
[bluetooth]# Discovery started
[bluetooth]# [CHG] Controller ma:ca:dd:re:ss Discovering: yes
[bluetooth]# [NEW] Device 54:46:6B:0F:22:A7 Bluetooth 3.0 Keyboard
[bluetooth]# pair 54:46:6B:0F:22:A7
Attempting to pair with 54:46:6B:0F:22:A7
[bluetooth]# hci0 device_flags_changed: 54:46:6B:0F:22:A7 (BR/EDR)
[bluetooth]#      supp: 0x00000001  curr: 0x00000000
[bluetooth]# hci0 type 7 discovering off
[bluetooth]# hci0 54:46:6B:0F:22:A7 type BR/EDR connected eir_len 29
[CHG] Device 54:46:6B:0F:22:A7 Connected: yes
[Bluetooth 3.0 Keyboard]# [agent] PIN code: 933126
[Bluetooth 3.0 Keyboard]# hci0 new_link_key 54:46:6B:0F:22:A7 type 0x00 pin_len 0 store_hint 1
[Bluetooth 3.0 Keyboard]# hci0 device_flags_changed: 54:46:6B:0F:22:A7 (BR/EDR)
[Bluetooth 3.0 Keyboard]#      supp: 0x00000001  curr: 0x00000001
[Bluetooth 3.0 Keyboard]# [CHG] Device 54:46:6B:0F:22:A7 Bonded: yes
[Bluetooth 3.0 Keyboard]# [CHG] Device 54:46:6B:0F:22:A7 WakeAllowed: yes
[Bluetooth 3.0 Keyboard]# [CHG] Device 54:46:6B:0F:22:A7 Modalias: usb:v04E8p7021d0001
[Bluetooth 3.0 Keyboard]# [CHG] Device 54:46:6B:0F:22:A7 ServicesResolved: yes
[Bluetooth 3.0 Keyboard]# [CHG] Device 54:46:6B:0F:22:A7 Paired: yes
[Bluetooth 3.0 Keyboard]# Pairing successful
[Bluetooth 3.0 Keyboard]# hci0 54:46:6B:0F:22:A7 type BR/EDR disconnected with reason 2
[Bluetooth 3.0 Keyboard]# [CHG] Device 54:46:6B:0F:22:A7 ServicesResolved: no
[... 836 lines cut ...]
< HCI Command: Disconnect (0x01|0x0006) plen 3                                                                         #92 [hci0] 38.964746
        Handle: 256 Address: 54:46:6B:0F:22:A7 (Shenzhen CZTIC Electronic Technology Co., Ltd)
        Reason: Remote User Terminated Connection (0x13)
> HCI Event: Command Status (0x0f) plen 4                                                                              #93 [hci0] 38.967635
      Disconnect (0x01|0x0006) ncmd 1
        Status: Success (0x00)
> HCI Event: Disconnect Complete (0x05) plen 4                                                                         #94 [hci0] 39.016636
        Status: Success (0x00)
        Handle: 256 Address: 54:46:6B:0F:22:A7 (Shenzhen CZTIC Electronic Technology Co., Ltd)
        Reason: Connection Terminated By Local Host (0x16)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                 {0x0002} [hci0] 39.016646
        BR/EDR Address: 54:46:6B:0F:22:A7 (Shenzhen CZTIC Electronic Technology Co., Ltd)
        Reason: Connection terminated by local host (0x02)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                 {0x0001} [hci0] 39.016646
        BR/EDR Address: 54:46:6B:0F:22:A7 (Shenzhen CZTIC Electronic Technology Co., Ltd)
        Reason: Connection terminated by local host (0x02)
< HCI Command: Write Scan Enable (0x03|0x001a) plen 1                                                                  #95 [hci0] 39.034774
        Scan enable: Page Scan (0x02)
> HCI Event: Command Complete (0x0e) plen 4                                                                            #96 [hci0] 39.037634
      Write Scan Enable (0x03|0x001a) ncmd 2
        Status: Success (0x00)

```

</details>

zeroepoch, 2023-12-31:
This might be related to #686. I was seeing similar permission denied issues until applying the patch which resolves that issue.

ben-krieger, 2024-12-28:
I had this issue connecting an Apple Magic Trackpad with Bluez 5.78. Upgrading to 5.79 had no effect. Then I used `bluetoothctl` to remove, pair, trust, and then connect the device at which point the trackpad functioned as an input device.

Previously, when I tried to connect with bluetoothctl, I got: `Failed to connect: org.bluez.Error.Failed br-connection-create-socket`. When I used blueman-applet, I got an I/O error message and `profiles/input/device.c:control_connect_cb() connect to ...: Permission denied (13)` in bluetoothd logs.

MrSorcus, 2025-06-16:
> I had this issue connecting an Apple Magic Trackpad with Bluez 5.78. Upgrading to 5.79 had no effect. Then I used `bluetoothctl` to remove, pair, trust, and then connect the device at which point [...]

I have the same problem with the touchpad on Arch Linux (6.14.10-arch1-1).
But successfully connects to Windows and Android devices.

closed 2025-09-22 by github-actions[bot]