bluez/bluez #698: Removing /var/lib/bluetooth does not clear devices

dmfay, 2023-12-28:
Trying to get a clean slate for testing #697, I ran into an apparently unrelated problem: the device is not removed even when /var/lib/bluetooth has been wiped and the service restarted. `remove` in bluetoothctl does not clear it either. I have to reboot the computer I'm trying to connect the keyboard to.

I've followed these steps:

- `systemctl stop bluetooth`
- `rm -rf /var/lib/bluetooth`
- `systemctl start bluetooth`

then enter `bluetoothctl` as follows:

```
[bluetooth]# hci0 new_settings: powered bondable ssp br/edr le secure-conn cis-central cis-peripheral
[bluetooth]# Agent registered
[bluetooth]# [CHG] Controller ma:ca:dd:re:ss Pairable: yes
[bluetooth]# devices
[bluetooth]# scan on
[bluetooth]# SetDiscoveryFilter success
[bluetooth]# hci0 type 7 discovering on
[bluetooth]# Discovery started
[bluetooth]# [CHG] Controller ma:ca:dd:re:ss Discovering: yes
[NEW] Device 54:46:6B:0F:22:A7 Bluetooth 3.0 Keyboard
[bluetooth]# pair 54:46:6B:0F:22:A7
Attempting to pair with 54:46:6B:0F:22:A7
[bluetooth]# hci0 device_flags_changed: 54:46:6B:0F:22:A7 (BR/EDR)
[bluetooth]#      supp: 0x00000001  curr: 0x00000000
[bluetooth]# hci0 type 7 discovering off
[bluetooth]# hci0 device_flags_changed: 54:46:6B:0F:22:A7 (BR/EDR)
[bluetooth]#      supp: 0x00000001  curr: 0x00000001
[bluetooth]# Failed to pair: org.bluez.Error.AlreadyExists
[bluetooth]# [CHG] Device 54:46:6B:0F:22:A7 WakeAllowed: yes
```

## vitals

uname: `Linux medusa 6.6.8-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 21 Dec 2023 19:01:01 +0000 x86_64 GNU/Linux`
bluetoothctl: 5.71

## btmon

<details><summary>logs</summary>

```
Bluetooth monitor ver 5.71
btmon[13858]: = Note: Linux version 6.6.8-arch1-1 (x86_64)                                                                                                                                                                                                                         0.266270
btmon[13858]: = Note: Bluetooth subsystem version 2.22                                                                                                                                                                                                                             0.266272
= New Index: ma:ca:dd:re:ss (Primary,USB,hci0)                                                                                                                                                                                                                           [hci0] 0.266273
= Open Index: ma:ca:dd:re:ss                                                                                                                                                                                                                                             [hci0] 0.266274
= Index Info: ma:ca:dd:re:ss (Intel Corp.)                                                                                                                                                                                                                               [hci0] 0.266274
bluetoothd[13768]: @ MGMT Open: bluetoothd (privileged) version 1.22                                                                                                                                                                                                      {0x0001} 0.266275
bluetoothctl[13910]: @ MGMT Open: bluetoothctl (privileged) version 1.22                                                                                                                                                                                                  {0x0002} 3.696121
bluetoothd[13768]: @ MGMT Command: Set Bondable (0x0009) plen 1                                                                                                                                                                                                    {0x0001} [hci0] 3.697001
        Bondable: Enabled (0x01)
@ MGMT Event: Command Complete (0x0001) plen 7                                                                                                                                                                                                                     {0x0001} [hci0] 3.697005
      Set Bondable (0x0009) plen 4
        Status: Success (0x00)
        Current settings: 0x000c0ad1
          Powered
          Bondable
          Secure Simple Pairing
          BR/EDR
          Low Energy
          Secure Connections
          CIS Central
          CIS Peripheral
@ MGMT Event: New Settings (0x0006) plen 4                                                                                                                                                                                                                         {0x0002} [hci0] 3.697006
        Current settings: 0x000c0ad1
          Powered
          Bondable
          Secure Simple Pairing
          BR/EDR
[... 208 lines cut ...]
      Remove Device (0x0034) plen 7
        Status: Success (0x00)
        BR/EDR Address: 54:46:6B:0F:22:A7 (Shenzhen CZTIC Electronic Technology Co., Ltd)
< HCI Command: Write Scan Enable (0x03|0x001a) plen 1                                                                                                                                                                                                                  #16 [hci0] 20.402406
        Scan enable: No Scans (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                                                                                            #17 [hci0] 20.405242
      Write Scan Enable (0x03|0x001a) ncmd 2
        Status: Success (0x00)
bluetoothctl[13910]: @ MGMT Close: bluetoothctl                                                                                                                                                                                                                          {0x0002} 21.959061
bluetoothd[13768]: @ MGMT Command: Set Bondable (0x0009) plen 1                                                                                                                                                                                                   {0x0001} [hci0] 21.960653
        Bondable: Disabled (0x00)
@ MGMT Event: Command Complete (0x0001) plen 7                                                                                                                                                                                                                    {0x0001} [hci0] 21.960659
      Set Bondable (0x0009) plen 4
        Status: Success (0x00)
        Current settings: 0x000c0ac1
          Powered
          Secure Simple Pairing
          BR/EDR
          Low Energy
          Secure Connections
          CIS Central
          CIS Peripheral
```

</details>

dmfay, 2023-12-28:
## with `remove`

After the stop-delete-start process, in a fresh `bluetoothctl` session:

```
[bluetooth]# Agent registered
[bluetooth]# [CHG] Controller ma:ca:dd:re:ss Pairable: yes
[bluetooth]# scan on
[bluetooth]# SetDiscoveryFilter success
[bluetooth]# hci0 type 7 discovering on
[bluetooth]# Discovery started
[bluetooth]# [CHG] Controller ma:ca:dd:re:ss Discovering: yes
[bluetooth]# [NEW] Device B8:D7:AF:CF:7C:91 G9Xm2
[NEW] Device 54:46:6B:0F:22:A7 Bluetooth 3.0 Keyboard
pair 54:46:6B:0F:22:A7
Attempting to pair with 54:46:6B:0F:22:A7
[bluetooth]# hci0 device_flags_changed: 54:46:6B:0F:22:A7 (BR/EDR)
[bluetooth]#      supp: 0x00000001  curr: 0x00000000
[bluetooth]# hci0 type 7 discovering off
[bluetooth]# hci0 device_flags_changed: 54:46:6B:0F:22:A7 (BR/EDR)
[bluetooth]#      supp: 0x00000001  curr: 0x00000001
[bluetooth]# Failed to pair: org.bluez.Error.AlreadyExists
[bluetooth]# [CHG] Device 54:46:6B:0F:22:A7 WakeAllowed: yes
[bluetooth]# remove 54:46:6B:0F:22:A7
[bluetooth]# [DEL] Device 54:46:6B:0F:22:A7 Bluetooth 3.0 Keyboard
[bluetooth]# Device has been removed
hci0 type 7 discovering on
[bluetooth]# devices
Device B8:D7:AF:CF:7C:91 G9Xm2
[NEW] Device 54:46:6B:0F:22:A7 Bluetooth 3.0 Keyboard
[bluetooth]# pair 54:46:6B:0F:22:A7
Attempting to pair with 54:46:6B:0F:22:A7
[bluetooth]# hci0 device_flags_changed: 54:46:6B:0F:22:A7 (BR/EDR)
[bluetooth]#      supp: 0x00000001  curr: 0x00000000
[bluetooth]# hci0 type 7 discovering off
[bluetooth]# hci0 device_flags_changed: 54:46:6B:0F:22:A7 (BR/EDR)
[bluetooth]#      supp: 0x00000001  curr: 0x00000001
[bluetooth]# Failed to pair: org.bluez.Error.AlreadyExists
[bluetooth]# [CHG] Device 54:46:6B:0F:22:A7 WakeAllowed: yes
```

### btmon

<details><summary>logs</summary>

```
Bluetooth monitor ver 5.71
btmon[18221]: = Note: Linux version 6.6.8-arch1-1 (x86_64)                                                                                                                                                                                                                         0.369043
btmon[18221]: = Note: Bluetooth subsystem version 2.22                                                                                                                                                                                                                             0.369044
= New Index: ma:ca:dd:re:ss (Primary,USB,hci0)                                                                                                                                                                                                                           [hci0] 0.369045
= Open Index: ma:ca:dd:re:ss                                                                                                                                                                                                                                             [hci0] 0.369045
= Index Info: ma:ca:dd:re:ss (Intel Corp.)                                                                                                                                                                                                                               [hci0] 0.369045
bluetoothd[18176]: @ MGMT Open: bluetoothd (privileged) version 1.22                                                                                                                                                                                                      {0x0001} 0.369046
bluetoothctl[18260]: @ MGMT Open: bluetoothctl (privileged) version 1.22                                                                                                                                                                                                  {0x0002} 3.217900
bluetoothd[18176]: @ MGMT Command: Set Bondable (0x0009) plen 1                                                                                                                                                                                                    {0x0001} [hci0] 3.218928
        Bondable: Enabled (0x01)
@ MGMT Event: Command Complete (0x0001) plen 7                                                                                                                                                                                                                     {0x0001} [hci0] 3.218931
      Set Bondable (0x0009) plen 4
        Status: Success (0x00)
        Current settings: 0x000c0ad1
          Powered
          Bondable
          Secure Simple Pairing
          BR/EDR
          Low Energy
          Secure Connections
          CIS Central
          CIS Peripheral
@ MGMT Event: New Settings (0x0006) plen 4                                                                                                                                                                                                                         {0x0002} [hci0] 3.218932
        Current settings: 0x000c0ad1
[... 597 lines cut ...]
      Remove Device (0x0034) plen 7
        Status: Success (0x00)
        BR/EDR Address: 54:46:6B:0F:22:A7 (Shenzhen CZTIC Electronic Technology Co., Ltd)
< HCI Command: Write Scan Enable (0x03|0x001a) plen 1                                                                                                                                                                                                                  #37 [hci0] 25.462675
        Scan enable: No Scans (0x00)
[... cut ...]
@ MGMT Event: Command Complete (0x0001) plen 7                                                                                                                                                                                                                    {0x0001} [hci0] 26.860765
      Set Bondable (0x0009) plen 4
        Status: Success (0x00)
        Current settings: 0x000c0ac1
          Powered
          Secure Simple Pairing
          BR/EDR
          Low Energy
          Secure Connections
          CIS Central
          CIS Peripheral
```

</details>

Vishal8523, 2025-02-27:
@Vudentz : Any inputs here ? Facing the same issue

Vudentz, 2025-02-27:
@Vishal8523 @dmfay seems like the kernel still has access to the keys for some reason:

```
@ MGMT Event: Command Complete (0x0001) plen 10                                                                                                                                                                                                                   {0x0001} [hci0] 25.462585
      Pair Device (0x0019) plen 7
        Status: Already Paired (0x13)
        BR/EDR Address: 54:46:6B:0F:22:A7 (Shenzhen CZTIC Electronic Technology Co., Ltd)
```

Perhaps you had removed the files from the filesystem without first removing from bluetoothd? Anyway you can try with bluetoothctl> mgmt unpair, that shall cleanup the keys from the kernel.

Perhaps we can attempt to unpair automatically when this situation is detected, but first check if the latest git still has similar problem.

Vishal8523, 2025-03-05:
@Vudentz : Any inputs here ?

closed 2025-08-02 by github-actions[bot]