bluez/bluez #700: btmgmt hangs if its stdin is redirected

sleirsgoevy, 2023-12-29:
The command `btmgmt public-addr XX:XX:XX:XX:XX:XX` (possibly other btmgmt subcommands too) attempts to register fd 0 with `epoll_ctl`. If epoll does not support that fd (for example, when redirecting from `/dev/null`), the syscall returns EPERM and the utility proceeds to wait on a partially-initialized epoll, which hangs.

This is probably a duplicate of #110, but with a more descriptive name and description.

hlovdal, 2024-04-22:
The same can be triggered by a simple `btmgmt info < /dev/null`:

```text
$ strace btmgmt info < /dev/null
execve("/usr/bin/btmgmt", ["btmgmt", "info"], 0x7fffdb0cca28 /* 96 vars */) = 0
brk(NULL)                               = 0x55787e8d8000
arch_prctl(0x3001 /* ARCH_??? */, 0x7fff497d3020) = -1 EINVAL (Invalid argument)
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=184207, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 184207, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f6d786ef000
close(3)                                = 0
openat(AT_FDCWD, "/lib64/libreadline.so.8", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0755, st_size=369736, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f6d786ed000
mmap(NULL, 367288, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f6d78693000
mmap(0x7f6d786aa000, 196608, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x17000) = 0x7f6d786aa000
mmap(0x7f6d786da000, 36864, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x47000) = 0x7f6d786da000
mmap(0x7f6d786e3000, 36864, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x50000) = 0x7f6d786e3000
mmap(0x7f6d786ec000, 2744, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f6d786ec000
close(3)                                = 0
openat(AT_FDCWD, "/lib64/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0 \203\2\0\0\0\0\0"..., 832) = 832
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
newfstatat(3, "", {st_mode=S_IFREG|0755, st_size=2420152, ...}, AT_EMPTY_PATH) = 0
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
mmap(NULL, 1973104, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f6d784b1000
mmap(0x7f6d784d7000, 1441792, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x26000) = 0x7f6d784d7000
mmap(0x7f6d78637000, 319488, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x186000) = 0x7f6d78637000
mmap(0x7f6d78685000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1d3000) = 0x7f6d78685000
mmap(0x7f6d7868b000, 31600, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f6d7868b000
close(3)                                = 0
openat(AT_FDCWD, "/lib64/libtinfo.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0755, st_size=215904, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 215328, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f6d7847c000
mmap(0x7f6d7848a000, 81920, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xe000) = 0x7f6d7848a000
mmap(0x7f6d7849e000, 57344, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x22000) = 0x7f6d7849e000
mmap(0x7f6d784ac000, 20480, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2f000) = 0x7f6d784ac000
close(3)                                = 0
mmap(NULL, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f6d78479000
arch_prctl(ARCH_SET_FS, 0x7f6d78479740) = 0
set_tid_address(0x7f6d78479a10)         = 1681756
set_robust_list(0x7f6d78479a20, 24)     = 0
rseq(0x7f6d7847a060, 0x20, 0, 0x53053053) = 0
mprotect(0x7f6d78685000, 16384, PROT_READ) = 0
mprotect(0x7f6d784ac000, 16384, PROT_READ) = 0
mprotect(0x7f6d786e3000, 12288, PROT_READ) = 0
mprotect(0x55787d74b000, 12288, PROT_READ) = 0
mprotect(0x7f6d7874e000, 8192, PROT_READ) = 0
prlimit64(0, RLIMIT_STACK, NULL, {rlim_cur=8192*1024, rlim_max=RLIM64_INFINITY}) = 0
munmap(0x7f6d786ef000, 184207)          = 0
getrandom("\x24\x6d\x07\xda\xeb\x3c\xcf\x3a", 8, GRND_NONBLOCK) = 8
brk(NULL)                               = 0x55787e8d8000
brk(0x55787e8f9000)                     = 0x55787e8f9000
epoll_create1(EPOLL_CLOEXEC)            = 3
socket(AF_BLUETOOTH, SOCK_RAW|SOCK_CLOEXEC|SOCK_NONBLOCK, BTPROTO_HCI) = 4
bind(4, {sa_family=AF_BLUETOOTH, hci_dev=htobs(65535), hci_channel=HCI_CHANNEL_CONTROL}, 6) = 0
epoll_ctl(3, EPOLL_CTL_ADD, 4, {events=0, data={u32=2123204400, u64=93976007640880}}) = 0
epoll_ctl(3, EPOLL_CTL_MOD, 4, {events=EPOLLIN, data={u32=2123204400, u64=93976007640880}}) = 0
getsockopt(4, SOL_BLUETOOTH, BT_SNDMTU, 0x55787e8d849a, [0]) = 0
setsockopt(4, SOL_BLUETOOTH, BT_SNDMTU, "\377\377", 2) = 0
epoll_ctl(3, EPOLL_CTL_ADD, 0, {events=0, data={u32=2123206736, u64=93976007643216}}) = -1 EPERM (Operation not permitted)
rt_sigprocmask(SIG_BLOCK, [INT USR2 TERM CHLD], NULL, 8) = 0
signalfd4(-1, [INT USR2 TERM CHLD], 8, SFD_CLOEXEC|SFD_NONBLOCK) = 5
epoll_ctl(3, EPOLL_CTL_ADD, 5, {events=0, data={u32=2123206736, u64=93976007643216}}) = 0
epoll_ctl(3, EPOLL_CTL_MOD, 5, {events=EPOLLIN, data={u32=2123206736, u64=93976007643216}}) = 0
epoll_wait(3,                         <-- Hangs forever...
```
[... 10 lines cut ...]
getsockopt(4, SOL_BLUETOOTH, BT_SNDMTU, 0x5645639aa49a, [0]) = 0
setsockopt(4, SOL_BLUETOOTH, BT_SNDMTU, "\377\377", 2) = 0
epoll_ctl(3, EPOLL_CTL_ADD, 0, {events=0, data={u32=1671082064, u64=94856023814224}}) = 0
epoll_ctl(3, EPOLL_CTL_MOD, 4, {events=EPOLLIN|EPOLLOUT, data={u32=1671079728, u64=94856023811888}}) = 0
rt_sigprocmask(SIG_BLOCK, [INT USR2 TERM CHLD], NULL, 8) = 0
[... cut ...]
epoll_ctl(3, EPOLL_CTL_MOD, 4, {events=EPOLLIN|EPOLLOUT, data={u32=1671079728, u64=94856023811888}}) = 0
epoll_wait(3, [{events=EPOLLOUT, data={u32=1671079728, u64=94856023811888}}], 10, -1) = 1
writev(4, [{iov_base="\4\0\0\0\0\0", iov_len=6}], 1) = 6
epoll_ctl(3, EPOLL_CTL_MOD, 4, {events=EPOLLIN, data={u32=1671079728, u64=94856023811888}}) = 0
epoll_wait(3, [{events=EPOLLIN, data={u32=1671079728, u64=94856023811888}}], 10, -1) = 1
...
```

where `epoll_ctl(3, EPOLL_CTL_ADD, 0,` does not fail, and there is an additional `epoll_ctl(3, EPOLL_CTL_MOD, 4, {events=EPOLLIN|EPOLLOUT, ` call.

tramseyer, 2024-06-28:
Ended up wrapping my "btmgmt info" call via script as suggested on https://stackoverflow.com/questions/66532619/redirecting-output-of-bluez-btmgmt-to-file-from-systemd-service to fake a tty and prevent it from hanging.
`script -qc "btmgmt info"`

closed 2025-08-02 by github-actions[bot]