bluez/bluez #702: PS4 controller: can only use it one time, then connect and immedaitly disconnect

legluondunet, 2024-01-01:
I have the same issue on 
- Manjaro OS: Manjaro 23.1.1 with  Kernel: x86_64 Linux 6.6.8-2-MANJARO
- Bluez 5.71-3

I can add PS4 controller with Blutooth, use it as long as I need. But if I disconnect it, then start PS4 controller again, it will connect then immediatly disconnect. I have to removed it and add it again like a new controller each time I want to use it.
Restart Bluetooth server workaround does not work for me.
I can reproduce this issue with command line on blutoothctl:

```

[CHG] Device 98:B6:E9:5E:CF:07 Connected: yes
[Wireless Controller]# [CHG] Device 70:09:71:9E:2B:E3 ManufacturerData.Key: 0x0075 (117)
[Wireless Controller]# [CHG] Device 70:09:71:9E:2B:E3 ManufacturerData.Value:
[Wireless Controller]#   42 04 01 20 6f 21 0d 00 22 01 41 01 01 00 01 00  B.. o!..".A.....
[Wireless Controller]#   00 00 00 00 00 00 00 05                          ........        
[Wireless Controller]# [CHG] Device 70:09:71:9E:2B:E3 ManufacturerData.Key: 0xff19 (65305)
[Wireless Controller]# [CHG] Device 70:09:71:9E:2B:E3 ManufacturerData.Value:
[Wireless Controller]#   00 75 00 09 01 00 00 00 06 01 00 00 00 00 00 00  .u..............
[Wireless Controller]#   00 00 00 00 00 00 00 00                          ........        
[CHG] Device 98:B6:E9:5E:CF:07 Connected: no
[bluetooth]# [CHG] Device 70:09:71:9E:2B:E3 ManufacturerData.Key: 0x0075 (117)

```

I try to downgrade bluez and bluez-utils packages to 5.68 versioon but it issue still here.

Thank you for your help.

Vudentz, 2024-01-02:
@legluondunet perhaps something with link-key not being found/valid for some reason, try collecting the HCI trace using btmon.

legluondunet, 2024-01-02:
"try collecting the HCI trace using btmon" sorry I don't know how to do that, could you explain me? Until now  I used Bluetoothctl to watch bluetooth log.

Vudentz, 2024-01-02:
@legluondunet https://manpages.org/btmon, you can record the bluetooth traffic by doing something like:

> sudo btmon -w <filename.log>

Then attempt to connect using bluetoothctl, etc.

legluondunet, 2024-01-03:
Hello, this is a trace, I just try to connect my ds4 and as usual, it connected and then immediately disconnected.

[btmon_PS4_controller.log.zip](https://github.com/bluez/bluez/files/13824279/btmon_PS4_controller.log.zip)

jacobwhall, 2025-04-10:
@legluondunet this sounds similar to an issue I'm experiencing. Does [this workaround](https://github.com/bluez/bluez/issues/488#issuecomment-2791684978) work for you by chance?

legluondunet, 2025-04-10:
hello @jacobwhall , I used the same workaround as you:
`sudo btmgmt ssp off`
and this command resolved my issue.
Do you know what ssp option is?