bluez/bluez #703: Huge latency with SONY WF-C500 but not with SONY WH-1000XM5

gabriele2000, 2024-01-02:
For reference, with debug logs and stuff like that, I'll just leave my [Pipewire's gitlab issue](https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/3591).

Long story short: it seems that my girlfriend's WF-C500 have a very high latency (they almost have half a second delay) and this causes discomfort if we're watching a movie together, for example.

A few days ago I bought the WH-1000XM5 (I wanted them since months) and I was VERY surprised (and pissed off at the same time) seeing that **using the LDAC codec** they have a low latency, like, I played games with them and I noticed that the audio is stellar AND there's such low latency that I believe that:
- something is wrong with pipewire (I'm starting to think that everthing is fine after months of trying to fix this issue)
- something is not reported in BlueZ concerning those earbuds

The weird thing is that the website "Rtings" reports both my GF's earbuds and my headphones as "not suitable for gaming because of high bluetooth latency".

I really don't know where to bang my head anymore.

Vudentz, 2024-01-02:
@gabriele2000 I wouldn't be surprised this is due to the fact there are true wireless type vs WH-1000XM5 which is a single device, but for the host stack we have no idea that there are 2 earbuds, at least no on classic, so perhaps there is something with the way the codec left and right frames are generate that adds a lot of latency when synchronizing left and right earbuds, this is a little hard to know exactly what is going on without the vendor telling us what is the problem, anyway try collecting the HCI trace using btmon so we can try to use --analyze to check if that is something with the transmission that takes too long.

gabriele2000, 2024-01-03:
> @gabriele2000 I wouldn't be surprised this is due to the fact there are true wireless type vs WH-1000XM5 which is a single device, but for the host stack we have no idea that there are 2 earbuds, at [...]

I'll collect the traces as soon as I can.

I've noticed that compiling the master version (I was lagging behind 2 weeks) most of the lag has disappeared.
My bluetooth version is 4.2, I'm waiting for the new wifi card for my laptop but it won't arrive, damn AliExpress...

By the way, I was thinking: the official page for the C500 says that they are two separate connections, so right and left.

My card could be overloaded but I don't know why now I have connected them PLUS MINE and there is not an issue, the latency is ok despite the fact that my card could be overloaded...
Did you guys make some compression adjustment or something like that?

gabriele2000, 2024-01-05:
> try collecting the HCI trace using btmon

[btmon-c500.zip](https://github.com/bluez/bluez/files/13843893/btmon-c500.zip)

Vudentz, 2024-01-16:
> monitor/btmon --analyze ~/Downloads/btmon-c500.txt 
```
Bluetooth monitor ver 5.71
Trace contains 2909 packets

Found BR/EDR controller with index 65535
  BD_ADDR 00:00:00:00:00:00
  0 commands
  0 events
  0 ACL packets
  0 SCO packets
  0 ISO packets
  0 vendor diagnostics
  2 system notes
  0 user logs
  1 control messages 
  0 unknown opcodes

Found BR/EDR controller with index 0
  BD_ADDR 34:41:5D:03:1D:1B (Intel Corp.)
  0 commands
  1451 events
  1452 ACL packets
  0 SCO packets
  0 ISO packets
  0 vendor diagnostics
  0 system notes
  0 user logs
  0 control messages 
  0 unknown opcodes
  Found unknown connection with handle 256
        Address: 00:00:00:00:00:00 (OUI 00-00-00)
        Connection setup missing
        RX packets: 1/1
        RX Latency: 0-0 msec (~0 msec)
        RX size: 6-6 octets (~6 octets)
        TX packets: 1451/1451
        TX Latency: 5-17 msec (~12 msec)
        TX size: 7-528 octets (~527 octets)
        TX speed: ~547 Kb/s

                                                                               
  1000 +-+-----------------------------------------------------------------+   
        +|        +                                                        |   
        +|        |                                             TX +-----+ |   
        +|    +   |                                                        |   
        +|    |   |                                                        |   
   100 +-|    |   |    +             +             +                       |   
        +|    |   |    |         +   |             |                       |   
        +|    |   |    |         |   |             |                       |   
        +|    |   |    |         |   |             |                       |   
        +|    |   |    |         |   |             |                       |   
         |    |   |    |         |   |             |    +                  |   
    10 +-|    |   |    |         |   |         +   |    |                  |   
        +|    |   |    |         |   |         |   |    |                  |   
        +|    |   |    |    +    |   |    +    |   |    |                  |   
        +|    |   |    |    |    |   |    |    |   |    |                  |   
        +|    |   |    |    |    |   |    |    |   |    |             +    |   
     1 +-|    |   |    |    |    |   |    |    |   |    |         +   |    |   
        +|    |   |    |    |    |   |    |    |   |    |         |   |    |   
        ++-----------------------------------------------------------------+   
         +        +         +        +         +        +         +        +   
         4        6         8        10        12       14        16       18  
                                    Latency (ms)                               
                                                                               
  Found TX L2CAP channel with CID 68
        TX packets: 1450/1450
        TX Latency: 5-17 msec (~6 msec)
        TX size: 528-528 octets (~528 octets)
        TX speed: ~548 Kb/s
[... 107 lines cut ...]
        +|  | |  |  |                                           TX +-----+ |   
        +|  | |  |  |                                                      |   
        +|  | |  |  | +                                                    |   
   100 +-|  | |  |  | |  +                                                 |   
        +|  | |  |  | |  | +                                               |   
        +|  | |  |  | |  | |  +    +                                       |   
        +|  | |  |  | |  | |  |  + |       +                               |   
        +|  | |  |  | |  | |  |  | |     + |       +                       |   
         |  | |  |  | |  | |  |  | |  +  | |     + |  +                    |   
    10 +-|  | |  |  | |  | |  |  | |  |  | |     | |  |                    |   
        +|  | |  |  | |  | |  |  | |  |  | |     | |  |  +                 |   
        +|  | |  |  | |  | |  |  | |  |  | |     | |  |  |                 |   
        +|  | |  |  | |  | |  |  | |  |  | |     | |  |  | +               |   
        +|  | |  |  | |  | |  |  | |  |  | |     | |  |  | |               |   
     1 +-|  | |  |  | |  | |  |  | |  |  | |  +  | |  |  | |    +  +       |   
        +|  | |  |  | |  | |  |  | |  |  | |  |  | |  |  | |    |  |       |   
        ++-----------------------------------------------------------------+   
         +            +            +             +            +            +   
         0            5            10            15           20           25  
                                    Latency (ms)
```

Worst case is actually better on c500, and this actually looks very tamed btw, Ive seem a lot worse latencies that just 20ms, but this is just part of the overall latency perhaps there is some added latency at the source endpoint (e.g. pipewire) or at the rendering side.

@pv does pw do anything with reported Delay property: https://github.com/bluez/bluez/blob/master/doc/org.bluez.MediaTransport.rst#uint16-delay-readwrite-optional

pv, 2024-01-16:
> Worst case is actually better on c500, and this actually looks very tamed btw, Ive seem a lot worse latencies that just 20ms, but this is just part of the overall latency perhaps there is some added latency at the source endpoint (e.g. pipewire) or at the rendering side.

Pipewire has only MAX(encoding delay, processing quantum) but this is of order some tens of ms. AFAICS the latency seen in this case is on the device side.

> @pv does pw do anything with reported Delay property: https://github.com/bluez/bluez/blob/master/doc/org.bluez.MediaTransport.rst#uint16-delay-readwrite-optional

The AVDTP Delay report value is passed on to applications as informative stream latency. E.g. video player apps and browsers make use of it to get A/V sync by delaying the video (and this does works nicely). Most other apps, e.g. music players, notification sounds etc. generally ignore the stream latency. In all cases there's no additional latency in audio output.

I've seen headset playback latencies 200ms+ also on another TWS device, but it had it also with Android. Due to delay reporting, this is generally only visible in games etc. interactive and not in video playback.

gabriele2000, 2024-01-16:
So you guys are telling me that sony advertises them as "low-latency" like mine, but they're not?
I refuse to believe it, they're bluetooth 5.0 TWS, I have bluetooth 4.2, I guess if the package gets unstuck my network card will shed some light on the issue.

pv, 2024-01-16:
You should try the device with a phone to compare: install a piano app that makes a sound when screen tapped, and press a button. For accurate values you can record with external mic and measure time difference in hitting the phone screen and sound output from earbud. As I mentioned earlier, looking at video A/V sync does not help due to the latency compensation in browsers and other video players.

gabriele2000, 2024-01-16:
> You should try the device with a phone to compare: install a piano app that makes a sound when screen tapped, and tap the screen. Compare with phone speaker. For accurate values you can record with external mic and measure time difference in hitting the phone screen and sound output from earbud. For apples to apples, check the codec the phone is using. As I mentioned earlier, looking at video A/V sync does not help due to the latency compensation in browsers and other video players.

Yeah, I remember using a game weeks ago... the latency is even worse than using the computer, but I guess it's a known issue, I remember seeing it as a known issue one day.

ValeZAA, 2025-03-28:
>LDAC codec they have a low latency,

No, they do not. Just like my WH-1000XM5 your WH-1000XM5 has 200 ms latency in A2DP LDAC. LMAO

gabriele2000, 2025-03-28:
BTW this issue was fixed, it was the adapter limitation since I had bluetooth 4.2.
I bought a bluetooth 5.3  / Wifi 6 card, fixed

closed 2025-03-28 by gabriele2000