bluez/bluez #705: Segfault on SONY MDR-1000X

matteopaolucci, 2024-01-04:
Hello,

Recently my headphones have started to disconnect after any kind of media playback (music, youtube...) stops. The problems have arisen with the last 15 versions. Before, the headphones worked perfectly. I would like to point out that the headphone still works fine on Android and IOS. 

System info:
- Distro: Fedora 39
- bluez version: 5.71-2.fc39
- Kernel version: 6.6.8-200.fc39.x86_64
- bluetooth adapter: RTL8821CE

Messages at startup:

- **journalctl -u bluetooth.service**

```
-- Boot bc9d61a371924b568bc1c37e68360218 --
Jan 04 11:20:27 fedora systemd[1]: Starting bluetooth.service - Bluetooth service...
Jan 04 11:20:27 fedora (uetoothd)[1602]: bluetooth.service: ConfigurationDirectory 'bluetooth' already exists but the mode is different. (File system: 555 ConfigurationDirectoryMode: 755)
Jan 04 11:20:27 fedora bluetoothd[1602]: Bluetooth daemon 5.71
Jan 04 11:20:27 fedora bluetoothd[1602]: Starting SDP server
Jan 04 11:20:27 fedora bluetoothd[1602]: src/plugin.c:plugin_init() System does not support csip plugin
Jan 04 11:20:27 fedora bluetoothd[1602]: profiles/audio/micp.c:micp_init() D-Bus experimental not enabled
Jan 04 11:20:27 fedora bluetoothd[1602]: src/plugin.c:plugin_init() System does not support micp plugin
Jan 04 11:20:27 fedora bluetoothd[1602]: src/plugin.c:plugin_init() System does not support vcp plugin
Jan 04 11:20:27 fedora bluetoothd[1602]: src/plugin.c:plugin_init() System does not support mcp plugin
Jan 04 11:20:27 fedora bluetoothd[1602]: src/plugin.c:plugin_init() System does not support bass plugin
Jan 04 11:20:27 fedora bluetoothd[1602]: src/plugin.c:plugin_init() System does not support bap plugin
Jan 04 11:20:27 fedora bluetoothd[1602]: Bluetooth management interface 1.22 initialized
Jan 04 11:20:27 fedora bluetoothd[1602]: src/adapter.c:reset_adv_monitors_complete() Failed to reset Adv Monitors: Failed (0x03)
Jan 04 11:20:27 fedora bluetoothd[1602]: Battery Provider Manager created
Jan 04 11:20:27 fedora bluetoothd[1602]: Failed to clear UUIDs: Failed (0x03)
Jan 04 11:20:27 fedora bluetoothd[1602]: Failed to add UUID: Failed (0x03)
Jan 04 11:20:27 fedora bluetoothd[1602]: Failed to add UUID: Failed (0x03)
Jan 04 11:20:27 fedora bluetoothd[1602]: Failed to add UUID: Failed (0x03)
Jan 04 11:20:27 fedora systemd[1]: Started bluetooth.service - Bluetooth service.
Jan 04 11:20:27 fedora bluetoothd[1602]: Failed to load link keys for hci0: Invalid Parameters (0x0d)
```

Messages after crash:

- **journalctl -u bluetooth.service**
 
```
Jan 04 11:35:50 fedora systemd[1]: bluetooth.service: Main process exited, code=dumped, status=11/SEGV
Jan 04 11:35:50 fedora systemd[1]: bluetooth.service: Failed with result 'core-dump'.
Jan 04 11:35:50 fedora systemd[1]: Starting bluetooth.service - Bluetooth service...
Jan 04 11:35:50 fedora (uetoothd)[6672]: bluetooth.service: ConfigurationDirectory 'bluetooth' already exists but the mode is different. (File system: 555 ConfigurationDirectoryMode: 755)
Jan 04 11:35:50 fedora bluetoothd[6672]: Bluetooth daemon 5.71
Jan 04 11:35:50 fedora systemd[1]: Started bluetooth.service - Bluetooth service.
Jan 04 11:35:50 fedora bluetoothd[6672]: Starting SDP server
Jan 04 11:35:50 fedora bluetoothd[6672]: src/plugin.c:plugin_init() System does not support csip plugin
Jan 04 11:35:50 fedora bluetoothd[6672]: profiles/audio/micp.c:micp_init() D-Bus experimental not enabled
Jan 04 11:35:50 fedora bluetoothd[6672]: src/plugin.c:plugin_init() System does not support micp plugin
Jan 04 11:35:50 fedora bluetoothd[6672]: src/plugin.c:plugin_init() System does not support vcp plugin
Jan 04 11:35:50 fedora bluetoothd[6672]: src/plugin.c:plugin_init() System does not support mcp plugin
Jan 04 11:35:50 fedora bluetoothd[6672]: src/plugin.c:plugin_init() System does not support bass plugin
Jan 04 11:35:50 fedora bluetoothd[6672]: src/plugin.c:plugin_init() System does not support bap plugin
Jan 04 11:35:50 fedora bluetoothd[6672]: Bluetooth management interface 1.22 initialized
Jan 04 11:35:50 fedora bluetoothd[6672]: Battery Provider Manager created
Jan 04 11:35:50 fedora bluetoothd[6672]: Failed to set privacy: Rejected (0x0b)
Jan 04 11:35:50 fedora bluetoothd[6672]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/ldac
Jan 04 11:35:50 fedora bluetoothd[6672]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSink/aptx_hd
Jan 04 11:35:50 fedora bluetoothd[6672]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/aptx_hd
Jan 04 11:35:50 fedora bluetoothd[6672]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSink/aptx
Jan 04 11:35:50 fedora bluetoothd[6672]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/aptx
Jan 04 11:35:50 fedora bluetoothd[6672]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSink/aac
Jan 04 11:35:50 fedora bluetoothd[6672]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/aac
Jan 04 11:35:50 fedora bluetoothd[6672]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSink/sbc
Jan 04 11:35:50 fedora bluetoothd[6672]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/sbc
Jan 04 11:35:50 fedora bluetoothd[6672]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSink/sbc_xq
Jan 04 11:35:50 fedora bluetoothd[6672]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/sbc_xq
Jan 04 11:35:50 fedora bluetoothd[6672]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/aptx_ll_1
Jan 04 11:35:50 fedora bluetoothd[6672]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/aptx_ll_0
Jan 04 11:35:50 fedora bluetoothd[6672]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
Jan 04 11:35:50 fedora bluetoothd[6672]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
Jan 04 11:35:50 fedora bluetoothd[6672]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/faststream
Jan 04 11:35:50 fedora bluetoothd[6672]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/faststream_duplex
Jan 04 11:35:50 fedora bluetoothd[6672]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSink/opus_05
Jan 04 11:35:50 fedora bluetoothd[6672]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/opus_05
Jan 04 11:35:50 fedora bluetoothd[6672]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSink/opus_05_duplex
Jan 04 11:35:50 fedora bluetoothd[6672]: Endpoint registered: sender=:1.87 path=/MediaEndpoint/A2DPSource/opus_05_duplex
Jan 04 11:35:50 fedora bluetoothd[6672]: Failed to load link keys for hci0: Invalid Parameters (0x0d)
```

- **dmesg**

```
[  935.693346] bluetoothd[1602]: segfault at 563ce8ac2cd3 ip 000056398a33ae96 sp 00007ffc00c23160 error 4 in bluetoothd[56398a319000+e0000] likely on CPU 2 (core 1, socket 0)
[  935.693374] Code: 4c 89 f7 e8 ac ad 0b 00 31 c0 e9 51 ff ff ff 0f 1f 44 00 00 f3 0f 1e fa 55 48 8b 7a 08 48 89 e5 41 56 41 55 41 54 4c 8b 22 53 <49> 8b 44 24 20 4d 8b b4 24 88 00 00 00 4c 8b 28 48 85 ff 74 19 c7
```

nickmansrob, 2024-01-07:
Related to https://github.com/bluez/bluez/issues/708

enorrmann, 2024-01-10:
same here running debian sid kernel 6.6.9-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.6.9-1 (2024-01-01) x86_64 GNU/Linux

travissuel, 2024-01-11:
I was experiencing this issue on Fedora 39 using a Sony SRS-XB33, and it was caused by #701. Fedora just pushed [bluez-5.71-3.fc39](https://bodhi.fedoraproject.org/updates/FEDORA-2024-c501bb0b43) which backports the patch, and I am no longer experiencing the segfault

enorrmann, 2024-01-11:
> I was experiencing this issue on Fedora 39 using a Sony SRS-XB33, and it was caused by #701. Fedora just pushed [bluez-5.71-3.fc39](https://bodhi.fedoraproject.org/updates/FEDORA-2024-c501bb0b43) which backports the patch, and I am no longer experiencing the segfault

I can confirm, installing bluez from source solved the issue on debian (waiting for the update)

matteopaolucci, 2024-01-11:
> I was experiencing this issue on Fedora 39 using a Sony SRS-XB33, and it was caused by #701. Fedora just pushed [bluez-5.71-3.fc39](https://bodhi.fedoraproject.org/updates/FEDORA-2024-c501bb0b43) which backports the patch, and I am no longer experiencing the segfault

I confirm version [bluez-5.71-3.fc39](https://bodhi.fedoraproject.org/updates/FEDORA-2024-c501bb0b43) of Fedora solve the issue

closed 2024-01-11 by matteopaolucci

phpony, 2024-01-15:
> I can confirm, installing bluez from source solved the issue on debian (waiting for the update)

Debian has a set of it's own 14 patches for Bluez so the better practice would be to [build and install deb from patched source package](https://github.com/bluez/bluez/issues/708#issuecomment-1892137902) instead of using standalone source build.