bluez/bluez #708: 5.71 Upgrade brought frequent bluetoothd segfaults to my Fedora 38

steevithak, 2024-01-06:
I'm running Fedora 38 on an Intel NUC13L3Kv7 (with Intel Core i7-1370P cpu). I have three bluetooth devices:
1. A Kensington Smart Mouse
2. Edifier R1700BT speakers
3. Sony WH-CH700N headphones

This system has been very stable with no bluetooth problems until Fedora pushed the bluez 5.71 update out on Dec 30. Since then, I'm getting multiple bluetooth crashes with kernel segfaults each day. The crashes seem to occur at random times. Sometimes things will work for an hour or several hours with no problems and then another crash. I haven't seen anything obvious that precedes the crash.

After a crash occurs all bluetooth devices are in a disconnected state. Because one is my mouse, recovery is a bit complicated. I have to physically connect a USB mouse and use that to access the GNOME Bluetooth control panel where I stop and then restart the bluetooth system. And then I have to select each of the BT devices and click "connect" to get them working again. After that, bluetooth will work again for a while, until the next crash. 

Fedora also pushed a bluez 5.71-2 update out on Jan 3 but it did not have any corrective effect on the segfaults. 

I'm not sure what additional information would be helpful in debugging this but if you have suggestions I'll try to provide it. Maybe it's worth noting I previously ran the same 3 BT devices on an Intel NUC8i7BEH for years with Fedora 30 to 37 with no bluetooth problems, so I'm really doubtful it's a problem with my hardware as little has changed.

Here is the journalctl -f output from a recent crash:

```
Jan 05 11:33:02 fedora audit[12591]: ANOM_ABEND auid=4294967295 uid=0 gid=0 ses=4294967295 subj=system_u:system_r:bluetooth_t:s0 pid=12591 comm="bluetoothd" exe="/usr/libexec/bluetooth/bluetoothd" sig=11 res=1
Jan 05 11:33:02 fedora kernel: bluetoothd[12591]: segfault at 55f1bbdb0b7c ip 000055f4e3174e36 sp 00007ffc96788450 error 4 in bluetoothd[55f4e3153000+e0000] likely on CPU 7 (core 12, socket 0)
Jan 05 11:33:02 fedora kernel: Code: 4c 89 f7 e8 ac ad 0b 00 31 c0 e9 51 ff ff ff 0f 1f 44 00 00 f3 0f 1e fa 55 48 8b 7a 08 48 89 e5 41 56 41 55 41 54 4c 8b 22 53 <49> 8b 44 24 20 4d 8b b4 24 88 00 00 00 4c 8b 28 48 85 ff 74 19 c7
Jan 05 11:33:02 fedora audit: BPF prog-id=114 op=LOAD
Jan 05 11:33:02 fedora audit: BPF prog-id=115 op=LOAD
Jan 05 11:33:02 fedora audit: BPF prog-id=116 op=LOAD
Jan 05 11:33:02 fedora systemd[1]: Started [systemd-coredump@2-14112-0.service](mailto:systemd-coredump@2-14112-0.service) - Process Core Dump (PID 14112/UID 0).
Jan 05 11:33:02 fedora audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-coredump@2-14112-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jan 05 11:33:02 fedora systemd-coredump[14113]: [🡕] Process 12591 (bluetoothd) of user 0 dumped core.
                                                Module libudev.so.1 from rpm systemd-253.14-1.fc38.x86_64
                                                Module sixaxis.so from rpm bluez-5.71-2.fc38.x86_64
                                                Module liblz4.so.1 from rpm lz4-1.9.4-2.fc38.x86_64
                                                Module libzstd.so.1 from rpm zstd-1.5.5-1.fc38.x86_64
                                                Module liblzma.so.5 from rpm xz-5.4.1-1.fc38.x86_64
                                                Module libcap.so.2 from rpm libcap-2.48-8.fc38.x86_64
                                                Module libsystemd.so.0 from rpm systemd-253.14-1.fc38.x86_64
                                                Module libpcre2-8.so.0 from rpm pcre2-10.42-1.fc38.1.x86_64
                                                Module libdbus-1.so.3 from rpm dbus-1.14.10-1.fc38.x86_64
                                                Module libglib-2.0.so.0 from rpm glib2-2.76.6-1.fc38.x86_64
                                                Module bluetoothd from rpm bluez-5.71-2.fc38.x86_64
                                                Stack trace of thread 12591:
                                                #0  0x000055f4e3174e36 a2dp_suspend_complete (bluetoothd + 0x43e36)
                                                #1  0x000055f4e315b04f finalize_suspend.lto_priv.0 (bluetoothd + 0x2a04f)
                                                #2  0x000055f4e315d270 suspend_cfm.lto_priv.0 (bluetoothd + 0x2c270)
                                                #3  0x000055f4e316b469 session_cb (bluetoothd + 0x3a469)
                                                #4  0x00007f79334ce4fc g_main_context_dispatch (libglib-2.0.so.0 + 0x5c4fc)
                                                #5  0x00007f793352c6b8 g_main_context_iterate.isra.0 (libglib-2.0.so.0 + 0xba6b8)
                                                #6  0x00007f79334cdaff g_main_loop_run (libglib-2.0.so.0 + 0x5baff)
                                                #7  0x000055f4e3157bed main (bluetoothd + 0x26bed)
                                                #8  0x00007f7933266b8a __libc_start_call_main (libc.so.6 + 0x27b8a)
                                                #9  0x00007f7933266c4b __libc_start_main@@GLIBC_2.34 (libc.so.6 + 0x27c4b)
                                                #10 0x000055f4e3158d35 _start (bluetoothd + 0x27d35)
                                                ELF object binary architecture: AMD x86-64
Jan 05 11:33:02 fedora systemd[1]: [systemd-coredump@2-14112-0.service](mailto:systemd-coredump@2-14112-0.service): Deactivated successfully.
Jan 05 11:33:02 fedora audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-coredump@2-14112-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jan 05 11:33:02 fedora audit: BPF prog-id=116 op=UNLOAD
Jan 05 11:33:02 fedora audit: BPF prog-id=115 op=UNLOAD
Jan 05 11:33:02 fedora audit: BPF prog-id=114 op=UNLOAD
Jan 05 11:33:02 fedora systemd[1]: bluetooth.service: Main process exited, code=dumped, status=11/SEGV
Jan 05 11:33:02 fedora systemd[1]: bluetooth.service: Failed with result 'core-dump'.
Jan 05 11:33:02 fedora audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=bluetooth comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
Jan 05 11:33:02 fedora systemd[1]: bluetooth.service: Consumed 2.460s CPU time.
Jan 05 11:33:02 fedora systemd[1]: Starting bluetooth.service - Bluetooth service...
Jan 05 11:33:02 fedora bluetoothd[14149]: Bluetooth daemon 5.71
Jan 05 11:33:02 fedora systemd[1]: Started bluetooth.service - Bluetooth service.
Jan 05 11:33:02 fedora audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=bluetooth comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jan 05 11:33:02 fedora bluetoothd[14149]: Starting SDP server
Jan 05 11:33:02 fedora bluetoothd[14149]: src/plugin.c:plugin_init() System does not support csip plugin
Jan 05 11:33:02 fedora bluetoothd[14149]: profiles/audio/micp.c:micp_init() D-Bus experimental not enabled
Jan 05 11:33:02 fedora bluetoothd[14149]: src/plugin.c:plugin_init() System does not support micp plugin
Jan 05 11:33:02 fedora bluetoothd[14149]: src/plugin.c:plugin_init() System does not support vcp plugin
Jan 05 11:33:02 fedora bluetoothd[14149]: src/plugin.c:plugin_init() System does not support mcp plugin
Jan 05 11:33:02 fedora bluetoothd[14149]: src/plugin.c:plugin_init() System does not support bass plugin
Jan 05 11:33:02 fedora bluetoothd[14149]: src/plugin.c:plugin_init() System does not support bap plugin
[... 5 lines cut ...]
Jan 05 11:33:02 fedora audit: BPF prog-id=117 op=LOAD
Jan 05 11:33:02 fedora audit: BPF prog-id=118 op=LOAD
Jan 05 11:33:02 fedora audit: BPF prog-id=119 op=LOAD
Jan 05 11:33:02 fedora gsd-media-keys[2577]: Unable to get default source
Jan 05 11:33:02 fedora systemd[1]: Starting systemd-hostnamed.service - Hostname Service...
Jan 05 11:33:02 fedora bluetoothd[14149]: Endpoint registered: sender=:1.86 path=/MediaEndpoint/A2DPSource/ldac
Jan 05 11:33:02 fedora bluetoothd[14149]: Endpoint registered: sender=:1.86 path=/MediaEndpoint/A2DPSink/aac
Jan 05 11:33:02 fedora bluetoothd[14149]: Endpoint registered: sender=:1.86 path=/MediaEndpoint/A2DPSource/aac
[... cut ...]
Jan 05 11:33:02 fedora audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-hostnamed comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jan 05 11:33:02 fedora abrt-server[14164]: Deleting problem directory ccpp-2024-01-05-11:33:02.622954-12591 (dup of ccpp-2023-12-30-17:16:35.769038-4599)
Jan 05 11:33:03 fedora abrt-notification[14223]: [🡕] Process 4599 (bluetoothd) crashed in a2dp_suspend_complete()
```

nickmansrob, 2024-01-07:
I also have this exact same issue. I downgraded to 5.69 and this solved my issue for now.
I included an excerpt of the output of `journalctl -xeu bluetooth` in the attachment.
[dump-bluetooth.log](https://github.com/bluez/bluez/files/13854956/dump-bluetooth.log)

phpony, 2024-01-15:
[This patch](https://github.com/BluezTestBot/bluez/commit/83cfad1badee6aae77eb15177ccc917249ab9bb3.patch) fixes everything.

```
apt-get install build-essential fakeroot devscripts
mkdir -p /usr/src/debian/; cd /usr/src/debian
apt source bluez
apt build-dep bluez
wget https://github.com/BluezTestBot/bluez/commit/83cfad1badee6aae77eb15177ccc917249ab9bb3.patch
patch bluez-5.71/profiles/audio/transport.c 83cfad1badee6aae77eb15177ccc917249ab9bb3.patch
cd bluez-5.71/
debuild -b -uc -us
dpkg -i ../bluez_5.71-1_amd64.deb
```

No more segfaults.

closed 2025-08-01 by github-actions[bot]