bluez/bluez #712: Failed service discovery for Klipsch The Three Plus (BLE)

wleese, 2024-01-06:
Originally posted here https://github.com/hbldh/bleak/issues/1491

* bleak version:  0.21.1
* Python version: 3.11.2
* Operating System: Linux rpi400 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
* BlueZ version: bluetoothctl:  5.71  (self compiled)

### Description

Running the script https://github.com/hbldh/bleak/blob/master/examples/service_explorer.py modified just to list services, shows all service UUIDs when run on MacOS. However when run on Raspbian (rpi400), zero services are found.

### What I Did

Ran this command on MacOS and then on Raspbian:

MacOS
```
python3 service_explorer.py --name "Klipsch The Three Plus" -d

Here's the debug logs from starting bluetooth with debug logging and then running the script which simply scans and tries to discover services.

I have:
-  I added a 10 second delay before disconnecting, to separate those logs from those which are relevant.
- Combined the bluez logs with the script logging, to highlight the timing of events (looking at the time stamps, I'm not confident that console output buffering didn't throw things off a little bit)

```
wleese@rpi400:~/repos/bleak-test $ ./bin/python test_macos_works.py  --name "Klipsch The Three Plus"

2024-01-06 08:56:24,839 __main__ INFO: starting scan...
2024-01-06 08:56:25,098 __main__ INFO: connecting to device...

Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/adapter.c:set_discovery_filter() sender :1.54
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/adapter.c:parse_discovery_filter_dict() filtered discovery params: transport: 6 rssi: 32767 pathloss: 32767  duplicate data: false discoverable false pattern (null)
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/adapter.c:set_discovery_filter() successfully pre-set filter
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/adapter.c:start_discovery() sender :1.54
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/adapter.c:update_discovery_filter()
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/adapter.c:discovery_filter_to_mgmt_cp()
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/adapter.c:trigger_start_discovery()
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/adapter.c:cancel_passive_scanning()
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/adapter.c:start_discovery_timeout()
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/adapter.c:start_discovery_timeout() adapter->current_discovery_filter == 1
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/adapter.c:start_discovery_timeout() sending MGMT_OP_START_SERVICE_DISCOVERY 127, 6, 0
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/shared/mgmt.c:send_request() [0x0000] command 0x003a
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/shared/mgmt.c:can_read_data() [0x0000] command 0x003a complete: 0x00
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/adapter.c:start_discovery_complete() status 0x00
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x0013
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/adapter.c:discovering_callback() hci0 type 6 discovering 1 method 1
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x0012
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/adapter.c:device_found_callback() hci0 addr 47:6E:49:3C:9C:74, rssi -42 flags 0x0004 eir_len 27
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/device.c:device_create() dst 47:6E:49:3C:9C:74
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/device.c:device_new() address 47:6E:49:3C:9C:74
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/device.c:device_new() Creating device /org/bluez/hci0/dev_47_6E_49_3C_9C_74
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/device.c:device_set_legacy() legacy 0
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/device.c:device_set_rssi_with_delta() rssi -42
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/device.c:device_set_flags() flags 26
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x0012
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/adapter.c:device_found_callback() hci0 addr 67:F2:3D:3C:F2:DC, rssi -34 flags 0x0000 eir_len 31
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/device.c:device_create() dst 67:F2:3D:3C:F2:DC
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/device.c:device_new() address 67:F2:3D:3C:F2:DC
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/device.c:device_new() Creating device /org/bluez/hci0/dev_67_F2_3D_3C_F2_DC
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/device.c:device_set_legacy() legacy 0
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/device.c:device_set_rssi_with_delta() rssi -34
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/device.c:device_set_flags() flags 26
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x0012
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/adapter.c:device_found_callback() hci0 addr 54:B4:31:6B:F7:19, rssi -43 flags 0x0000 eir_len 31
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/device.c:device_create() dst 54:B4:31:6B:F7:19
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/device.c:device_new() address 54:B4:31:6B:F7:19
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/device.c:device_new() Creating device /org/bluez/hci0/dev_54_B4_31_6B_F7_19
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/device.c:device_set_legacy() legacy 0
Jan 06 08:56:24 rpi400 bluetoothd[2322]: src/device.c:device_set_rssi_with_delta() rssi -43
[... 126 lines cut ...]
Jan 06 08:56:39 rpi400 bluetoothd[2322]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr CB:4E:FD:F1:E2:8E type 2 status 0xe
Jan 06 08:56:39 rpi400 bluetoothd[2322]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
Jan 06 08:56:39 rpi400 bluetoothd[2322]: src/device.c:device_bonding_failed() status 14
Jan 06 08:56:39 rpi400 bluetoothd[2322]: src/adapter.c:resume_discovery()
Jan 06 08:56:39 rpi400 bluetoothd[2322]: src/shared/att.c:disconnect_cb() Channel 0x5597b42c30 disconnected: Software caused connection abort
Jan 06 08:56:39 rpi400 bluetoothd[2322]: src/device.c:att_disconnected_cb()
Jan 06 08:56:39 rpi400 bluetoothd[2322]: src/device.c:att_disconnected_cb() Software caused connection abort (103)
Jan 06 08:56:39 rpi400 bluetoothd[2322]: src/gatt-client.c:btd_gatt_client_disconnected() Device disconnected. Cleaning up.
Jan 06 08:56:39 rpi400 bluetoothd[2322]: src/device.c:att_disconnected_cb() Automatic connection disabled
Jan 06 08:56:39 rpi400 bluetoothd[2322]: src/gatt-database.c:btd_gatt_database_att_disconnected()
Jan 06 08:56:39 rpi400 bluetoothd[2322]: attrib/gattrib.c:g_attrib_unref() 0x5597b0e960: g_attrib_unref=0
```

### Logs

tcpdump dump from raspbian:
[failed_bleak_debian.pcap.zip](https://github.com/bluez/bluez/files/13851061/failed_bleak_debian.pcap.zip)

packetLogger capture from MacOS - this file is also readable by Wireshark, although it might need some filtering for readability. e.g. `_ws.col.protocol != "PKTLOG"`
[working_service_discovery_bleak_corebluetooth.zip](https://github.com/bluez/bluez/files/13851062/working_service_discovery_bleak_corebluetooth.zip)

I've also been unable to list service uuids with bluetoothctl on raspbian.

Next to MacOS, I also installed LineageOS/Android on the same Raspberry 400. The official application to connect to this device worked. Here's the btsnoop file I captured using Androids builtin bug report feature, also readable by Wireshark.
[btsnoop_hci.log.zip](https://github.com/bluez/bluez/files/13851057/btsnoop_hci.log.zip)

wleese, 2024-01-12:
Anything I can do to help move this forward?

wleese, 2024-01-21:
Same problem on Raspberry Pi 5.

dheerajluffy, 2024-04-21:
i am also facing the exact issue, been looking for a solution from days but cant find any.

closed 2025-08-01 by github-actions[bot]