bluez/bluez #715: BlueZ returns a mac address that is 00:00:00:00:00:00

Nilon123456789, 2024-01-09:
https://github.com/hbldh/bleak/issues/1490

# Here are the info on the original issue :

bleak version: 0.21.1
Python version: 3.9.2
Operating System: Raspbian GNU/Linux 11 (bullseye)
BlueZ version (bluetoothctl -v) in case of Linux: 5.66
Description
I'm trying to get the mac address from a BLE device. But it returns a mac address equal to -> 00:00:00:00:00:00

## What I Did
Of course I tried to restart the rasp, I also tried to see if there bluetooth drivers I needed to install, but didn't find any. I was unable to find any good thread for this type of problem.

## My code
Here is the code I use to get the mac addresses.

```python
import asyncio
from bleak import BleakScanner

async def discover_devices():
devices = await BleakScanner.discover(timeout=20)
for device in devices:
print(f"Device: {device.name}, Address: {device.address}")

asyncio.run(discover_devices())
```

Nilon123456789, 2024-01-10:
It the address from the advertisement that is detected as 00:00:00:00:00 even tho the advertisement address is not 00:00:00:00:00. Since I can detect it correctly with my Mac

Vudentz, 2024-01-10:
> It the address from the advertisement that is detected as 00:00:00:00:00 even tho the advertisement address is not 00:00:00:00:00. Since I can detect it correctly with my Mac

Do you have the btmon trace? Also are you sure it is not using and RPA that resolves to 00:00:00:00:00:00?

Nilon123456789, 2024-01-10:
> RPA that resolves to 00:00:00:00:00:00

I don't think so, since on my Mac I can clearly get the MAC address of my device broadcast. 

I will look to get the btmon trace

Nilon123456789, 2024-01-10:
Here is the dump.

[hcidump.log](https://github.com/bluez/bluez/files/13890955/hcidump.log)

Nilon123456789, 2024-01-10:
I don't know if it has something to do with it. But I can't seem to be able to enable the `LE 2M` as an entry. I tried enabling it using ` sudo hcitool -i hci0 cmd 08 31 03 07 07` but it only enabled the `LE 1M` and `LE Coded`. I know that my advertisement is using both `LE 1M` and `LE 2M` 

Here is the description of the extended advertisement of the device (nrf connect)
> Advertise with extended advertising. This options enables extended advertising in the advertising set. In extended advertising the advertising set will send a small header packet on the three primary advertising channels. This small header points to the advertising data packet that will be sent on one of the 37 secondary advertising channels. The advertiser will send primary advertising on LE 1M PHY, and secondary advertising on LE 2M PHY. Connections will be established on LE 2M PHY. Without this option the advertiser will send advertising data on the three primary advertising channels.

Output from `sudo btmon -T -i 0`
```
< HCI Command: LE Set Extended Scan Parameters (0x08|0x0041) plen 13                                                                                                     #3 [hci0] 108.259728
        Own address type: Random (0x01)
        Filter policy: Accept all advertisement (0x00)
        PHYs: 0x05
        Entry 0: LE 1M
          Type: Active (0x01)
          Interval: 11.250 msec (0x0012)
          Window: 11.250 msec (0x0012)
        Entry 1: LE Coded
          Type: Active (0x01)
          Interval: 11.250 msec (0x0012)
          Window: 11.250 msec (0x0012)
```

Vudentz, 2024-01-10:
```
> HCI Event: LE Meta Event (0x3e) plen 64                                                                                                                                                                                                                                 #8 [hci0] 70.208696
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0000
            Props: 0x0000
            Data status: Complete
          Address type: Reserved (0xff)
          Address: 00-00-00-00-00-00
          Primary PHY: LE 1M
          Secondary PHY: LE 2M
          SID: 0x00
          TX power: 127 dBm
          RSSI: -68 dBm (0xbc)
          Periodic advertising interval: 0.00 msec (0x0000)
          Direct address type: Public (0x00)
          Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
          Data length: 0x26
        17 16 ab cd 00 26 01 16 3d 02 25 1d 03 28 56 04  .....&..=.%..(V.
        16 5b 05 00 00 fe 02 5d 0d 09 4c 52 49 4d 61 20  .[.....]..LRIMa 
        74 65 73 74 20 31                                test 1          
        Service Data: Unknown (0xcdab)
          Data: 002601163d02251d03285604165b050000fe025d
        Name (complete): LRIMa test 1
```

The addres type is not even valid (0xff), no wonder it cannot be resolved, now I don't know what magic iOS does to resolve the address but is definately not compliant to the Bluetooth spec, perhaps it is using the Service Data to display the address, which is IMO a layer violation if we start to have to look into the actual data to figure out this sort of information.

Nilon123456789, 2024-01-12:
I did a wireshark capture using the [nrf sniffer tool ](https://www.nordicsemi.com/Products/Development-tools/nrf-sniffer-for-bluetooth-le/). You can clearly see the MAC address of the device (f0:ca:f0:ca:01:d5)
[capture.pcapng.zip](https://github.com/bluez/bluez/files/13924141/capture.pcapng.zip)

closed 2025-09-22 by github-actions[bot]