bluez/bluez #716: [5.66] pairing iOS devices barely works

Sylensky, 2024-01-09:
I have developed a application with a custom pairing agent following the AgentManager and Agent DBus API with following controller settings:
- set sspmode to disable
- registered agent with "KeyboardDisplay" capability

With that configuration i was able to register the agent and got the passkey prompt on my android device when trying to establish a connection to it.

Pairing and connecting with said android device works flawlessly.

Now when it comes to pairing a iOS device the passkey prompt visibility varies from 3s-15s and aborts the pairing process after that for some reason. Attaching to btmon and looking for odd behaviour only showed a difference in a role request in form of the iOS device wanting to change to peripheral role which is not permitted instead of center role.

What i also noticed is that iOS devices act differently when being already connected to the controller hitting the "Forget this Device" button and trying to pair with the controller again.
 In this case the device is only pairable when i completely reboot the system.

Are there any hints on what is going wrong with iOS devices or did someone else experience similiar behaviour?

Prashaanth06, 2025-04-15:
I am using bluez 5.65, even I am facing the same issue when I am "forget the device" and then connect back it doesn't happen.

Prashaanth06, 2025-04-16:
Has this issue been addressed after 5.65 bluez release? If yes, possible to point to that particular commit

closed 2025-08-01 by github-actions[bot]