bluez/bluez #717: Unable to respond to or initiate Connection Parameter Update Procedure as a Central

0xff07, 2024-01-10:
It seems that there's no way to respond to invalid L2CAP connection parameters as a Central responder, nor can it initiate Connection Parameter Update Procedure. The constraints can be set in the following files of debugfs:

1. `/sys/kernel/debug/bluetooth/hci0/conn_min_interval`
2. `/sys/kernel/debug/bluetooth/hci0/conn_max_interval`

But the comparison against those 2 values doesn's seem happen when Linux is a Central. In particular, in the `l2cap_conn_param_update_req()` of the `net/bluetooth/l2cap_core.c` [1].

I can see this functionality had been once in this commit of Linux kernel:

Bluetooth: validate BLE connection interval updates[2]
(c49a8682fc5d298d44e8d911f4fa14690ea9485e)

but was reverted in this commit due to regression concerns.

Revert "Bluetooth: validate BLE connection interval updates"
(68d19d7d995759b96169da5aac313363f92a9075) 

I'd like to understand if there's any other way to do this, or an further discussion regarding this? Thank you.

[1] https://elixir.bootlin.com/linux/latest/source/net/bluetooth/l2cap_core.c#L5616
[2] https://lore.kernel.org/linux-bluetooth/20190815104419.161177fa@kemnade.info/t/
[3] https://lore.kernel.org/lkml/20191003154253.184041960@linuxfoundation.org/

Vudentz, 2024-01-10:
@0xff07 You can initiate the intervals by setting it via main.conf:

https://github.com/bluez/bluez/blob/master/src/main.conf#L213

The problem is that if the remote end attempts to update we store its value then the value of main.conf are superceeded, perhaps you are after rejecting interval updates from the peripheral? This sort of interface is not available right now, usually the peripheral knows best what interval works for them as they might have some limitation in term of battery, etc, that said sometimes the central is connected to multiple devices and changing the interval may result in collisions at link-layer.

0xff07, 2024-01-12:
Thank you. Now I see that rejection may not work at least in kernel. I can also confirm that setting parameters in `main.conf` works, although the parameters seems to be set by in an "LE Extended Create Connection" HCI command, instead of using the L2CAP Connection Parameter Update Procedure:

```
< HCI Command: LE Extended Create Connection (0x08|0x0043) plen 26                                                                                                                       #8 [hci0] 4.136896
        Filter policy: Accept list is not used (0x00)
        Own address type: Public (0x00)
        Peer address type: Public (0x00)
        Peer address: MA:CA:DD:RE:SS (OUI C0-07-E8)
        Initiating PHYs: 0x01
        Entry 0: LE 1M
          Scan interval: 60.000 msec (0x0060)
          Scan window: 60.000 msec (0x0060)
          Min connection interval: 62.50 msec (0x0032)
          Max connection interval: 87.50 msec (0x0046)
          Connection latency: 1 (0x0001)
          Supervision timeout: 5000 msec (0x01f4)
          Min connection length: 0.000 msec (0x0000)
          Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                                #9 [hci0] 4.141766
      LE Extended Create Connection (0x08|0x0043) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 31                                                                                                                                               #10 [hci0] 4.455103
      LE Enhanced Connection Complete (0x0a)
        Status: Success (0x00)
        Handle: 10
        Role: Central (0x00)
        Peer address type: Public (0x00)
        Peer address:  MA:CA:DD:RE:SS  (OUI C0-07-E8)
        Local resolvable private address: 00:00:00:00:00:00 (Non-Resolvable)
        Peer resolvable private address: 00:00:00:00:00:00 (Non-Resolvable)
        Connection interval: 82.50 msec (0x0042)
        Connection latency: 1 (0x0001)
        Supervision timeout: 5000 msec (0x01f4)
        Central clock accuracy: 0x00
@ MGMT Event: Device Connected (0x000b) plen 34                                                                                                                                    {0x0002} [hci0] 4.455163
        LE Address:  MA:CA:DD:RE:SS  (OUI C0-07-E8)
        Flags: 0x00000008
          Connection Locally Initiated
        Data length: 21
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        16-bit Service UUIDs (partial): 1 entry
          Generic Access Profile (0x1800)
        Name (complete): PTS-GAP-2712
@ MGMT Event: Device Connected (0x000b) plen 34                                                                                                                                    {0x0001} [hci0] 4.455163
        LE Address: MA:CA:DD:RE:SS  (OUI C0-07-E8)
        Flags: 0x00000008
          Connection Locally Initiated
        Data length: 21
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        16-bit Service UUIDs (partial): 1 entry
          Generic Access Profile (0x1800)
        Name (complete): PTS-GAP-2712
```

I'd like to know further on:
1. would it be possible to initiate it using the L2CAP Connection Parameter Update Procedure?
2. Is it possible to initiate the Connection Parameter Update Procedure after a connection has been establish?

JamesLiang23, 2025-02-11:
> [@0xff07](https://github.com/0xff07) You can initiate the intervals by setting it via main.conf:
>
> https://github.com/bluez/bluez/blob/master/src/main.conf#L213
>
> The problem is that if the remote end [...]

@Vudentz I observed the behavior of storing the connection parameter and re-using it in the subsequent connection creations, in my case I would like not to store the value for subsequent connections, updating the parameter for the current connection is fine. In order to do this, whenever the peripheral request update to connection parameters, I manually edit the main.conf and restart the bluetooth service, then the stored value will be cleared somehow. This is really painful.

Could you please let me know where blueZ store the connection parameters (it will be great if you can point me to the source code of this logic) and possibly how I can remove the stored value using some sort of API or file system operations? Thank you.

Vudentz, 2025-02-11:
> [@Vudentz](https://github.com/Vudentz) I observed the behavior of storing the connection parameter and re-using it in the subsequent connection creations, in my case I would like not to store the [...]

Perhaps you are after something PPCP implementation is doing (d6fe19bccba048a68def9410dff6eb4aa82c6848), that ends up calling btd_device_set_conn_param -> btd_adapter_load_conn_param so you can overwrite the value used in the connection with that.

JamesLiang23, 2025-02-11:
> > > [@0xff07](https://github.com/0xff07) You can initiate the intervals by setting it via main.conf:
> > > https://github.com/bluez/bluez/blob/master/src/main.conf#L213
> > > The problem is that if the remote end attempts to update we store its value then the value of main.conf are superceeded, perhaps you are after rejecting interval updates from the peripheral? This sort of interface is not available right now, usually the peripheral knows best what interval works for them as they might have some limitation in term of battery, etc, that said sometimes the central is connected to multiple devices and changing the interval may result in collisions at link-layer.
> > 
> > 
> > [@Vudentz](https://github.com/Vudentz) I observed the behavior of storing the connection parameter and re-using it in the subsequent connection creations, in my case I would like not to store the value for subsequent connections, updating the parameter for the current connection is fine. In order to do this, whenever the peripheral request update to connection parameters, I manually edit the main.conf and restart the bluetooth service, then the stored value will be cleared somehow. This is really painful.
> > Could you please let me know where blueZ store the connection parameters (it will be great if you can point me to the source code of this logic) and possibly how I can remove the stored value using some sort of API or file system operations? Thank you.
> 
> Perhaps you are after something PPCP implementation is doing ([d6fe19b](https://github.com/bluez/bluez/commit/d6fe19bccba048a68def9410dff6eb4aa82c6848)), that ends up calling btd_device_set_conn_param -> btd_adapter_load_conn_param so you can overwrite the value used in the connection with that.

Thanks for your quick response. I may not express my question clearly, here is what I observed:

1. My Linux box tries to create a BLE connection to a peripheral with connection parameters of min and max connection intervals as 7.5ms and 200ms (this is the value I set in main.conf). And the connection is made successfully.

2. Immediately after the connection creation, the remote end sends a connection parameter update request to my Linux box and set the min and max connection intervals to 6 (7.5ms). The Linux box accepted the parameter.
 
3. From now on, every time the Linux box tries to create a connection to the same remote peripheral, it always use 7.5ms as the min and max connection interval in the connection parameters, the min and max conn interval value in the main.conf are superseded. Rebooting the Linux box does not reset the stored value, the only way to reset the stored value is to make changes to main.conf (e.g. change min conn interval to another value) and restart the bluetooth service.

I want to figure out where blueZ persists the previously negotiated connection parameters, is it stored in /var/lib/bluetooth/{hci_mac}/{device_mac} folder or it's stored somewhere else?

Vudentz, 2025-07-21:
@0xff07 @JamesLiang23 we maybe able to put these parameters into the recently introduced org.bluez.Bearer.LE1:

https://github.com/bluez/bluez/blob/master/doc/org.bluez.Bearer.LE.rst

closed 2025-09-22 by github-actions[bot]