bluez/bluez #719: Regression: Setting volume via Dbus might timeout (>= 5.67)

BenKaufmann, 2024-01-12:
There is a regression in `profiles/audio/transport.c:set_volume()`. 

In versions <= 5.66, this function always called `g_dbus_pending_property_success()` even if the volume to be set was equal to the current a2dp volume. 

However, starting with version 5.67, `set_volume()` no longer calls `g_dbus_pending_property_success()`  or `g_dbus_pending_property_error()` in that case resulting in eventual client timeouts.

Vudentz, 2024-01-12:
@BenKaufmann did you attempt to bisect what is the exact patch that introduced such regression?

BenKaufmann, 2024-01-12:
> @BenKaufmann did you attempt to bisect what is the exact patch that introduced such regression?

It's probably this commit: https://github.com/bluez/bluez/commit/1de41786c1a22fe0d4f55805f3006ce66e5c65a6

Vudentz, 2024-01-12:
> > @BenKaufmann did you attempt to bisect what is the exact patch that introduced such regression?
> 
> It's probably this commit: [1de4178](https://github.com/bluez/bluez/commit/1de41786c1a22fe0d4f55805f3006ce66e5c65a6)

Yeah, this has probably been fixed upstream by commit e4764af7622825a9d7dfd4ecc2066733b03f5dbf as it does end up always calling g_dbus_pending_property_success or g_dbus_pending_property_error:

https://github.com/bluez/bluez/blob/master/profiles/audio/transport.c#L930

closed 2025-07-31 by github-actions[bot]