bluez/bluez #722: Dealing with vendor specific HCI commands

jonas2515, 2024-01-14:
Apparently a lot of (probably all) Broadcom cards support the HCI_BRCM_SET_ACL_PRIORITY vendor specific command, and that command needs to be sent for any connection handle once an A2DP stream is active. This fixes periodic (every few minutes) stuttering of the audio stream.

Android appears to send this command to all broadcom devices (based on the manufacturer of the card), see https://github.com/LineageOS/android_system_bt/blob/lineage-19.1/stack/l2cap/l2c_utils.cc#L2246. The old blueZ stack of android was also using this (see https://android.googlesource.com/platform/external/bluetooth/bluez/+/9a8383ad9ee9516756abeeb5081de9651a6294ed), and same thing goes for MacOS.

I've confirmed this method works just fine with BlueZ on the BCM4377 in an M1 macbook and audio is a lot more reliable afterwards, here's a quick hack implementing that: https://github.com/jonas2515/bluez/commit/b7ca931d7f6ffb5f788bf80d0c8557f8607017dc

The fact that this vendor specific command needs to be sent on start and end of A2DP streams means we probably have to do it from BlueZ instead of the kernel, that's a bit unfortunate given that HCI commands are usually created within the kernel.

I'm wondering if this could be implemented as a plugin, that would need some kind of hook into a2dp profile code, but that seems a lot better than stuffing it into the a2dp code.

Vudentz, 2024-01-16:
Perhaps this can be archieved with some custom interface to the driver e.g. hdev->set_prio, BlueZ already has priority concept based on SO_PRIORITY.

Anyway, this is sort of a bad design from the manufactorer if it doesn't work well without it when there is a single connection in place, also I do wonder what happens if there are multiple A2DP connections/multi-point?

jonas2515, 2024-01-16:
> Perhaps this can be archieved with some custom interface to the driver e.g. hdev->set_prio, BlueZ already has priority concept based on SO_PRIORITY.

Yeah I was wondering if that's a good idea, too. But not sure, the whole use-case seems a bit too high level to me to do that on a driver level, especially if we also want to add that `HCI_Write_Automatic_Flush_Timeout` quirk (as recommended by Broadcom in [this document](https://github.com/AsahiLinux/docs/files/13221453/713af2f7430822a2f4fc35057fd5ba88eb17389782993159514e76093700ea4c.pdf).

I think I'd prefer if we had a plugin in BlueZ that sends the vendor specific HCI commands from userspace. Possibly we could also have a MGMT API for sending freeform vendor specific commands, where the kernel only enforces OGF = 63 and sends vendor specific events with code = 255 to userspace?

> Anyway, this is sort of a bad design from the manufactorer if it doesn't work well without it when there is a single connection in place, also I do wonder what happens if there are multiple A2DP connections/multi-point?

Yeah it's bad design no doubt.. It does work quite well though, even for multiple A2DP connections (where all streams get prioritized using the vendor specific API).

Vudentz, 2024-01-16:
> Yeah I was wondering if that's a good idea, too. But not sure, the whole use-case seems a bit too high level to me to do that on a driver level, especially if we also want to add that [...]

Well the driver is normally the one responsible for vendor specific commands and we don't really do HCI commands via userspace anymore since that could mess up the HCI state machine, specially now that we have hci_cmd_sync mechanism in place Id like to stick to that as much as possible.

> I think I'd prefer if we had a plugin in BlueZ that sends the vendor specific HCI commands from userspace. Possibly we could also have a MGMT API for sending freeform vendor specific commands, where [...]

Ok, well I guess broadcom didn't bother with BlueZ then, or do you work for them? Id always assumed that since they are submitting drivers and firmware they are testing it on Linux in general (distros) but it looks like they only bothered doing it for Android.

jonas2515, 2024-01-16:
> Well the driver is normally the one responsible for vendor specific commands and we don't really do HCI commands via userspace anymore since that could mess up the HCI state machine, specially now that we have hci_cmd_sync mechanism in place Id like to stick to that as much as possible.

Agreed, although if we added MGMT API for vendor specific stuff like this, we could insert those commands into the hci_sync machinery without problems I think. Would that be something you could live with?

Fwiw, I can also see this being useful for other optimizations such as parsing vendor specific link quality reports or for tinkering with other vendor specific APIs for eg. BT LE direction finding.

> Ok, well I guess broadcom didn't bother with BlueZ then, or do you work for them? Id always assumed that since they are submitting drivers and firmware they are testing it on Linux in general (distros) but it looks like they only bothered doing it for Android.

Heh, I don't work for them and I don't plan to! Yeah it seems like (as most vendors) they aren't actually interesting in a smooth experience on linux and only do the bare minimum of testing :/

The reason I found this is that I recently got an M1 Macbook and am looking into improving the bluetooth experience on the Apple ARM laptops (see also https://github.com/AsahiLinux/docs/issues/85). From my findings there, it looks like there's a ton of custom firmware functionality only added for Apple devices, and while most of those things aren't interesting to us, there might be a few things which will significantly improve performance on linux too.

jonas2515, 2024-04-09:
Okay, so finally getting back to this now that I have a bit better understanding of the whole bluetooth socket concept: What you're suggesting is that pipewire would set SO_PRIORITY to 6, which the kernel could then interpret as "this is a real-time stream", and therefore 1) call `hdev->set_high_prio_connection(connectionId, TRUE)` (or `hdev->set_connection_prio(6)`, and 2) set the automatic flush timeout to 200ms (or possibly that should just be a new option on the socket?).

SebastianForslund, 2024-08-07:
>Agreed, although if we added MGMT API for vendor specific stuff like this, we could insert those commands into the hci_sync machinery without problems I think. Would that be something you could live with?

This would be an excellent feature. Some Bluetooth modules use vendor specific commands for configuration. AFAIK, to use these commands today one has to resort to hcitool. Like Vudentz described, this leads to problems.

closed 2025-07-31 by github-actions[bot]