bluez/bluez #726: [5.72] test_aics_unit_testcases (test-vcp.c) fails on x86_64

macmpi, 2024-01-17:
Trying to build bluez 5.72 on Alpine (musl): it now fails on x86_64 (only)
```
FAIL: unit/test-vcp
============================================================================
Testsuite summary for bluez 5.72
============================================================================
# TOTAL: 31
# PASS:  30
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0
```
Dunno how to get more detailed info on actual failure.
Other arches  ARM (32 & 64) , x86, etc are fine.
`test_aics_unit_testcases` seems the offending part:[ disabling it](https://github.com/macmpi/bluez/commit/1d5a2ec8eb96e9186aeb2826927057ddc7a9d0a6) fixes build.

Vudentz, 2024-01-17:
@macmpi try running it with valgrind to see if that reports anything.

macmpi, 2024-01-19:
some more logs details on this:
```
==================================
   bluez 5.72: ./test-suite.log
==================================
# TOTAL: 31
# PASS:  30
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0
.. contents:: :depth: 2
FAIL: unit/test-vcp
===================
VOCS/SR/SGGIT/SER/BV-01-C - init
VOCS/SR/SGGIT/SER/BV-01-C - setup
VOCS/SR/SGGIT/SER/BV-01-C - setup complete
VOCS/SR/SGGIT/SER/BV-01-C - run
VOCS/SR/SGGIT/SER/BV-01-C - test passed
VOCS/SR/SGGIT/SER/BV-01-C - teardown
VOCS/SR/SGGIT/SER/BV-01-C - teardown complete
VOCS/SR/SGGIT/SER/BV-01-C - done
VOCS/SR/SGGIT/CHA/BV-01-C - init
VOCS/SR/SGGIT/CHA/BV-01-C - setup
VOCS/SR/SGGIT/CHA/BV-01-C - setup complete
VOCS/SR/SGGIT/CHA/BV-01-C - run
VOCS/SR/SGGIT/CHA/BV-01-C - test passed
VOCS/SR/SGGIT/CHA/BV-01-C - teardown
VOCS/SR/SGGIT/CHA/BV-01-C - teardown complete
VOCS/SR/SGGIT/CHA/BV-01-C - done
VOCS/SR/SGGIT/CHA/BV-02-C - init
VOCS/SR/SGGIT/CHA/BV-02-C - setup
VOCS/SR/SGGIT/CHA/BV-02-C - setup complete
VOCS/SR/SGGIT/CHA/BV-02-C - run
VOCS/SR/SGGIT/CHA/BV-02-C - test passed
VOCS/SR/SGGIT/CHA/BV-02-C - teardown
VOCS/SR/SGGIT/CHA/BV-02-C - teardown complete
VOCS/SR/SGGIT/CHA/BV-02-C - done
VOCS/SR/SGGIT/CHA/BV-03-C - init
VOCS/SR/SGGIT/CHA/BV-03-C - setup
VOCS/SR/SGGIT/CHA/BV-03-C - setup complete
VOCS/SR/SGGIT/CHA/BV-03-C - run
VOCS/SR/SGGIT/CHA/BV-03-C - test passed
VOCS/SR/SGGIT/CHA/BV-03-C - teardown
VOCS/SR/SGGIT/CHA/BV-03-C - teardown complete
VOCS/SR/SGGIT/CHA/BV-03-C - done
VOCS/SR/SGGIT/CHA/BV-04-C - init
VOCS/SR/SGGIT/CHA/BV-04-C - setup
VOCS/SR/SGGIT/CHA/BV-04-C - setup complete
VOCS/SR/SGGIT/CHA/BV-04-C - run
VOCS/SR/SGGIT/CHA/BV-04-C - test passed
VOCS/SR/SGGIT/CHA/BV-04-C - teardown
VOCS/SR/SGGIT/CHA/BV-04-C - teardown complete
VOCS/SR/SGGIT/CHA/BV-04-C - done
VOCS/SR/SGGIT/CP/BI-01-C - init
VOCS/SR/SGGIT/CP/BI-01-C - setup
VOCS/SR/SGGIT/CP/BI-01-C - setup complete
VOCS/SR/SGGIT/CP/BI-01-C - run
VOCS/SR/SGGIT/CP/BI-01-C - test passed
VOCS/SR/SGGIT/CP/BI-01-C - teardown
VOCS/SR/SGGIT/CP/BI-01-C - teardown complete
VOCS/SR/SGGIT/CP/BI-01-C - done
VOCS/SR/SGGIT/CP/BI-02-C - init
VOCS/SR/SGGIT/CP/BI-02-C - setup
VOCS/SR/SGGIT/CP/BI-02-C - setup complete
VOCS/SR/SGGIT/CP/BI-02-C - run
VOCS/SR/SGGIT/CP/BI-02-C - test passed
VOCS/SR/SGGIT/CP/BI-02-C - teardown
VOCS/SR/SGGIT/CP/BI-02-C - teardown complete
VOCS/SR/SGGIT/CP/BI-02-C - done
VOCS/SR/SGGIT/CP/BI-03-C - init
VOCS/SR/SGGIT/CP/BI-03-C - setup
VOCS/SR/SGGIT/CP/BI-03-C - setup complete
VOCS/SR/SGGIT/CP/BI-03-C - run
VOCS/SR/SGGIT/CP/BI-03-C - test passed
VOCS/SR/SGGIT/CP/BI-03-C - teardown
VOCS/SR/SGGIT/CP/BI-03-C - teardown complete
VOCS/SR/SGGIT/CP/BI-03-C - done
VOCS/SR/SPE/BI-01-C - init
VOCS/SR/SPE/BI-01-C - setup
VOCS/SR/SPE/BI-01-C - setup complete
VOCS/SR/SPE/BI-01-C - run
VOCS/SR/SPE/BI-01-C - test passed
VOCS/SR/SPE/BI-01-C - teardown
VOCS/SR/SPE/BI-01-C - teardown complete
VOCS/SR/SPE/BI-01-C - done
VOCS/SR/CP/BV-01-C - init
VOCS/SR/CP/BV-01-C - setup
VOCS/SR/CP/BV-01-C - setup complete
VOCS/SR/CP/BV-01-C - run
VOCS/SR/CP/BV-01-C - test passed
VOCS/SR/CP/BV-01-C - teardown
VOCS/SR/CP/BV-01-C - teardown complete
VOCS/SR/CP/BV-01-C - done
AICS/SR/SGGIT/CHA/BV-01-C - init
AICS/SR/SGGIT/CHA/BV-01-C - setup
AICS/SR/SGGIT/CHA/BV-01-C - setup complete
AICS/SR/SGGIT/CHA/BV-01-C - run
AICS/SR/SGGIT/CHA/BV-01-C - test passed
AICS/SR/SGGIT/CHA/BV-01-C - teardown
AICS/SR/SGGIT/CHA/BV-01-C - teardown complete
AICS/SR/SGGIT/CHA/BV-01-C - done
AICS/SR/SGGIT/CHA/BV-02-C - init
AICS/SR/SGGIT/CHA/BV-02-C - setup
AICS/SR/SGGIT/CHA/BV-02-C - setup complete
AICS/SR/SGGIT/CHA/BV-02-C - run
AICS/SR/SGGIT/CHA/BV-02-C - test passed
AICS/SR/SGGIT/CHA/BV-02-C - teardown
AICS/SR/SGGIT/CHA/BV-02-C - teardown complete
AICS/SR/SGGIT/CHA/BV-02-C - done
AICS/SR/SGGIT/CHA/BV-03-C - init
AICS/SR/SGGIT/CHA/BV-03-C - setup
AICS/SR/SGGIT/CHA/BV-03-C - setup complete
AICS/SR/SGGIT/CHA/BV-03-C - run
AICS/SR/SGGIT/CHA/BV-03-C - test passed
AICS/SR/SGGIT/CHA/BV-03-C - teardown
AICS/SR/SGGIT/CHA/BV-03-C - teardown complete
AICS/SR/SGGIT/CHA/BV-03-C - done
AICS/SR/SGGIT/CHA/BV-04-C - init
AICS/SR/SGGIT/CHA/BV-04-C - setup
AICS/SR/SGGIT/CHA/BV-04-C - setup complete
AICS/SR/SGGIT/CHA/BV-04-C - run
AICS/SR/SGGIT/CHA/BV-04-C - test passed
AICS/SR/SGGIT/CHA/BV-04-C - teardown
AICS/SR/SGGIT/CHA/BV-04-C - teardown complete
AICS/SR/SGGIT/CHA/BV-04-C - done
AICS/SR/SGGIT/CHA/BV-05-C - init
AICS/SR/SGGIT/CHA/BV-05-C - setup
AICS/SR/SGGIT/CHA/BV-05-C - setup complete
AICS/SR/SGGIT/CHA/BV-05-C - run
AICS/SR/SGGIT/CHA/BV-05-C - test passed
AICS/SR/SGGIT/CHA/BV-05-C - teardown
AICS/SR/SGGIT/CHA/BV-05-C - teardown complete
AICS/SR/SGGIT/CHA/BV-05-C - done
AICS/SR/SGGIT/CHA/BV-06-C - init
AICS/SR/SGGIT/CHA/BV-06-C - setup
AICS/SR/SGGIT/CHA/BV-06-C - setup complete
AICS/SR/SGGIT/CHA/BV-06-C - run
AICS/SR/SGGIT/CHA/BV-06-C - test passed
AICS/SR/SGGIT/CHA/BV-06-C - teardown
AICS/SR/SGGIT/CHA/BV-06-C - teardown complete
AICS/SR/SGGIT/CHA/BV-06-C - done
AICS/SR/SGGIT/CP/BI-01-C - init
AICS/SR/SGGIT/CP/BI-01-C - setup
AICS/SR/SGGIT/CP/BI-01-C - setup complete
AICS/SR/SGGIT/CP/BI-01-C - run
AICS/SR/SGGIT/CP/BI-01-C - test passed
AICS/SR/SGGIT/CP/BI-01-C - teardown
AICS/SR/SGGIT/CP/BI-01-C - teardown complete
AICS/SR/SGGIT/CP/BI-01-C - done
AICS/SR/SGGIT/CP/BI-02-C - init
AICS/SR/SGGIT/CP/BI-02-C - setup
AICS/SR/SGGIT/CP/BI-02-C - setup complete
AICS/SR/SGGIT/CP/BI-02-C - run
AICS/SR/SGGIT/CP/BI-02-C - test passed
AICS/SR/SGGIT/CP/BI-02-C - teardown
Segmentation fault (core dumped)
FAIL unit/test-vcp (exit status: 139)
```
produced vcp test and codedump are below
[vcptest_cdmp.tar.gz](https://github.com/bluez/bluez/files/13991064/vcptest_cdmp.tar.gz)

xry111, 2024-01-20:
Maybe related to #683.

macmpi, 2024-01-23:
Thank for the ref @xry111 : tried without  "-flto" option in CFLAGS, but no more success unfortunately.

macmpi, 2024-03-07:
seems fixed in 5.73.

macmpi, 2024-04-16:
is back from 5.74

macmpi, 2024-04-17:
@pv saw you just nailed-down some memory allocation issues: would you see anything similar in running `test_aics_unit_testcases` (ref earlier [segfault](https://github.com/bluez/bluez/issues/726#issuecomment-1900632058))?
(latest post-5.75 patches do not fix it yet).

pv, 2024-04-17:
If you get segfaults, it's simplest to compile with ASAN enabled, as if it crashes then the backtraces will usually tell you immediately what's going on.

macmpi, 2024-04-17:
Thanks for the tip. 
I'm afraid I don't know how to get ASAN to work within [Alpine CI builds ](https://gitlab.alpinelinux.org/alpine/aports/-/blob/master/main/bluez/APKBUILD) which I rely upon to do my builds & tests...
So I'll leave it here and just disable that offending test, until someone with more skills eventually fix it for good.

pv, 2024-04-17:
Why not just `./bootstrap && ./configure --enable-asan && make check`? no need to involve some package system?

omnivagant, 2024-04-24:
```
gatt_notify_cb: Failed to send notification
Bail out! ERROR:src/shared/tester.c:949:test_io_recv: assertion failed (len == iov->iov_len): (5 == 6)
Aborted
FAIL unit/test-vcp (exit status: 134)
```
this is also failing with bluez 5.73 on our 3.20 package builders (where we're rebuilding world for the upcoming stable release)

macmpi, 2024-05-17:
@Vudentz : This test is still failing on `5.76` on `x86` and `x86_64`: with this in log
```
ERROR:src/shared/tester.c:981:test_io_recv: assertion failed (len == iov->iov_len): (5 == 6)
```

macmpi, 2024-07-09:
same on 5.77, despite all the memory fixes.
```
AICS/SR/SGGIT/CP/BI-01-C - test passed
AICS/SR/SGGIT/CP/BI-01-C - teardown
AICS/SR/SGGIT/CP/BI-01-C - teardown complete
AICS/SR/SGGIT/CP/BI-01-C - done
AICS/SR/SGGIT/CP/BI-02-C - init
AICS/SR/SGGIT/CP/BI-02-C - setup
AICS/SR/SGGIT/CP/BI-02-C - setup complete
AICS/SR/SGGIT/CP/BI-02-C - run
AICS/SR/SGGIT/CP/BI-02-C - test passed
AICS/SR/SGGIT/CP/BI-02-C - teardown
**
ERROR:src/shared/tester.c:981:test_io_recv: assertion failed (len == iov->iov_len): (5 == 6)
AICS/SR/SGGIT/CP/BI-02-C - teardown complete
AICS/SR/SGGIT/CP/BI-02-C - done
AICS/SR/CP/BV-01-C - init
AICS/SR/CP/BV-01-C - setup
AICS/SR/CP/BV-01-C - setup complete
AICS/SR/CP/BV-01-C - run
AICS/SR/CP/BV-01-C - test passed
AICS/SR/CP/BV-01-C - teardown
AICS/SR/CP/BV-01-C - teardown complete
AICS/SR/CP/BV-01-C - done
AICS/SR/SPE/BI-01-C - init
AICS/SR/SPE/BI-01-C - setup
AICS/SR/SPE/BI-01-C - setup complete
AICS/SR/SPE/BI-01-C - run
gatt_notify_cb: Failed to send notification
Bail out! ERROR:src/shared/tester.c:981:test_io_recv: assertion failed (len == iov->iov_len): (5 == 6)
Aborted (core dumped)
FAIL unit/test-vcp (exit status: 134)
```

gudnimg, 2025-03-23:
I am seeing the same problem in BlueZ 5.80. The issue is confirmed to appears on `qemuarm64` and `qemux86-64`. BlueZ 5.79 behaved fine. I don't see anything obvious when comparing 5.79 and 5.80 releases.

[image]

gudnimg, 2025-03-23:
I'm building with libc right now to confirm the issue is not just with musl. I will also try later to run the tests with valgrind and report back.

gudnimg, 2025-03-23:
Hopefully I'm running this correctly with valgrind: `valgrind --leak-check=full --show-leak-kinds=all --track-origins=yes --verbose --log-file=valgrind-bluez-log.txt ptest-runner bluez5`

It seems to have found one error. Not sure if this is useful though:

```
root@qemux86-64:~# cat valgrind-bluez-log.txt 
==356== Memcheck, a memory error detector
==356== Copyright (C) 2002-2024, and GNU GPL'd, by Julian Seward et al.
==356== Using Valgrind-3.24.0-fcdaa47426-20241101 and LibVEX; rerun with -h for copyright info
==356== Command: ptest-runner bluez5
==356== Parent PID: 352
==356== 
--356-- 
--356-- Valgrind options:
--356--    --leak-check=full
--356--    --show-leak-kinds=all
--356--    --track-origins=yes
--356--    --verbose
--356--    --log-file=valgrind-bluez-log.txt
--356-- Contents of /proc/version:
--356--   Linux version 6.12.19-yocto-standard (oe-user@oe-host) (x86_64-oe-linux-gcc (GCC) 14.2.0, GNU ld (GNU Binutils) 2.44) #1 SMP PREEMPT_DYNAMIC Wed Mar 19 23:04:08 UTC 2025
--356-- 
--356-- Arch and hwcaps: AMD64, LittleEndian, amd64-cx16-rdtscp-sse3-ssse3-avx-f16c-rdrand
--356-- Page sizes: currently 4096, max supported 4096
--356-- Valgrind library directory: /usr/libexec/valgrind
--356-- Reading syms from /usr/bin/ptest-runner
--356-- Reading syms from /lib/ld-linux-x86-64.so.2
--356--   Considering /lib/ld-linux-x86-64.so.2 ..
--356--   .. CRC mismatch (computed f88c9772 wanted 4e32298a)
--356--   Considering /lib/.debug/ld-linux-x86-64.so.2 ..
--356--   .. CRC is valid
--356-- Reading syms from /usr/libexec/valgrind/memcheck-amd64-linux
--356--    object doesn't have a dynamic symbol table
--356-- Scheduler: using generic scheduler lock implementation.
--356-- Reading suppressions file: /usr/libexec/valgrind/default.supp
==356== embedded gdbserver: reading from /tmp/vgdb-pipe-from-vgdb-to-356-by-root-on-???
==356== embedded gdbserver: writing to   /tmp/vgdb-pipe-to-vgdb-from-356-by-root-on-???
==356== embedded gdbserver: shared mem   /tmp/vgdb-pipe-shared-mem-vgdb-356-by-root-on-???
==356== 
==356== TO CONTROL THIS PROCESS USING vgdb (which you probably
==356== don't want to do, unless you know exactly what you're doing,
==356== or are doing some strange experiment):
==356==   /usr/libexec/valgrind/../../bin/vgdb --pid=356 ...command...
==356== 
==356== TO DEBUG THIS PROCESS USING GDB: start GDB like this
==356==   /path/to/gdb ptest-runner
==356== and then give GDB the following command
==356==   target remote | /usr/libexec/valgrind/../../bin/vgdb --pid=356
==356== --pid is optional if only one valgrind process is running
==356== 
--356--   Considering /usr/libexec/valgrind/memcheck-amd64-linux ..
--356--   .. CRC mismatch (computed b8ef1b21 wanted 898e0b3a)
--356-- REDIR: 0x40265a0 (ld-linux-x86-64.so.2:strlen) redirected to 0x580bb3c2 (???)
--356-- REDIR: 0x4024fa0 (ld-linux-x86-64.so.2:strcmp) redirected to 0x580bb3dc (???)
--356-- REDIR: 0x4024d70 (ld-linux-x86-64.so.2:index) redirected to 0x580bb3f7 (???)
--356-- Reading syms from /usr/libexec/valgrind/vgpreload_core-amd64-linux.so
--356-- Reading syms from /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so
==356== WARNING: new redirection conflicts with existing -- ignoring it
--356--     old: 0x04024fa0 (strcmp              ) R-> (0000.0) 0x580bb3dc ???
--356--     new: 0x04024fa0 (strcmp              ) R-> (2016.0) 0x0484f390 strcmp
==356== WARNING: new redirection conflicts with existing -- ignoring it
--356--     old: 0x040265a0 (strlen              ) R-> (0000.0) 0x580bb3c2 ???
--356--     new: 0x040265a0 (strlen              ) R-> (2007.0) 0x0484e0c0 strlen
--356-- REDIR: 0x40243d0 (ld-linux-x86-64.so.2:mempcpy) redirected to 0x48539a0 (mempcpy)
--356-- Reading syms from /lib/libc.so.6
--356--   Considering /lib/libc.so.6 ..
--356--   .. CRC mismatch (computed d2664d11 wanted 6d9069c9)
--356--   Considering /lib/.debug/libc.so.6 ..
--356--   .. CRC is valid
--356-- REDIR: 0x4026740 (ld-linux-x86-64.so.2:strncmp) redirected to 0x484eb80 (strncmp)
[... 58 lines cut ...]
--356-- REDIR: 0x4904bc0 (libc.so.6:__memcmp_sse2) redirected to 0x4851c60 (__memcmp_sse2)
==356== 
==356== HEAP SUMMARY:
==356==     in use at exit: 7 bytes in 1 blocks
==356==   total heap usage: 491 allocs, 490 frees, 59,753 bytes allocated
==356== 
==356== Searching for pointers to 1 not-freed blocks
==356== Checked 103,656 bytes
==356== 
==356== 7 bytes in 1 blocks are definitely lost in loss record 1 of 1
==356==    at 0x48447B2: malloc (vg_replace_malloc.c:446)
==356==    by 0x4900A6D: strdup (strdup.c:42)
==356==    by 0x402803: ??? (in /usr/bin/ptest-runner)
==356==    by 0x4883E1A: (below main) (libc_start_call_main.h:58)
==356== 
==356== LEAK SUMMARY:
==356==    definitely lost: 7 bytes in 1 blocks
==356==    indirectly lost: 0 bytes in 0 blocks
==356==      possibly lost: 0 bytes in 0 blocks
==356==    still reachable: 0 bytes in 0 blocks
==356==         suppressed: 0 bytes in 0 blocks
==356== 
==356== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)

```

thesamesam, 2025-03-24:
I wouldn't use `--verbose` with Valgrind, it's very noisy. The CRC issues in that output are to do with Valgrind reading debug information, not an issue in bluez.

gudnimg, 2025-03-24:
@thesamesam here is the output when I omit the --verbose option:

```
root@qemux86-64:~# cat valgrind-bluez-log.txt 
==305== Memcheck, a memory error detector
==305== Copyright (C) 2002-2024, and GNU GPL'd, by Julian Seward et al.
==305== Using Valgrind-3.24.0 and LibVEX; rerun with -h for copyright info
==305== Command: ptest-runner bluez5
==305== Parent PID: 298
==305== 
==305== 
==305== HEAP SUMMARY:
==305==     in use at exit: 7 bytes in 1 blocks
==305==   total heap usage: 491 allocs, 490 frees, 59,753 bytes allocated
==305== 
==305== 7 bytes in 1 blocks are definitely lost in loss record 1 of 1
==305==    at 0x48447B2: malloc (vg_replace_malloc.c:446)
==305==    by 0x4900A6D: strdup (strdup.c:42)
==305==    by 0x402803: ??? (in /usr/bin/ptest-runner)
==305==    by 0x4883E1A: (below main) (libc_start_call_main.h:58)
==305== 
==305== LEAK SUMMARY:
==305==    definitely lost: 7 bytes in 1 blocks
==305==    indirectly lost: 0 bytes in 0 blocks
==305==      possibly lost: 0 bytes in 0 blocks
==305==    still reachable: 0 bytes in 0 blocks
==305==         suppressed: 0 bytes in 0 blocks
==305== 
==305== For lists of detected and suppressed errors, rerun with: -s
==305== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
```

thesamesam, 2025-03-24:
Thanks! OK, just a memory leak (so not a problem here anyway). What is interesting is, it.. looks like the test didn't fail here either (there's no abort).

I'm wondering if it actually ran "properly" or not. Could you humour me and try adding `--trace-children=yes` to Valgrind's arguments?

gudnimg, 2025-03-24:
This option `--trace-children=yes` is causing my QEMU simulator to crash/reboot. Currently I'm running all the tests, not just test-vcp. 

I'll see if I can patch BlueZ locally on my end such that I only run test-vcp. Hopefully that will run without issues.

gudnimg, 2025-03-24:
I managed to run only `test-vcp`, and when I use `--trace-children=yes` the test passes. Without, the test fails. A big difference here though is using `--trace-children` significantly slows down the testing time. Smells like a timing problem somehow.

Valgrind output looks the same though.

```
root@qemux86-64:~# cat valgrind-bluez-log.txt 
==400== Memcheck, a memory error detector
==400== Copyright (C) 2002-2024, and GNU GPL'd, by Julian Seward et al.
==400== Using Valgrind-3.24.0 and LibVEX; rerun with -h for copyright info
==400== Command: ptest-runner bluez5
==400== Parent PID: 354
==400== 
==400== 
==400== HEAP SUMMARY:
==400==     in use at exit: 7 bytes in 1 blocks
==400==   total heap usage: 491 allocs, 490 frees, 59,753 bytes allocated
==400== 
==400== 7 bytes in 1 blocks are definitely lost in loss record 1 of 1
==400==    at 0x48447B2: malloc (vg_replace_malloc.c:446)
==400==    by 0x4900A6D: strdup (strdup.c:42)
==400==    by 0x402803: ??? (in /usr/bin/ptest-runner)
==400==    by 0x4883E1A: (below main) (libc_start_call_main.h:58)
==400== 
==400== LEAK SUMMARY:
==400==    definitely lost: 7 bytes in 1 blocks
==400==    indirectly lost: 0 bytes in 0 blocks
==400==      possibly lost: 0 bytes in 0 blocks
==400==    still reachable: 0 bytes in 0 blocks
==400==         suppressed: 0 bytes in 0 blocks
==400== 
==400== For lists of detected and suppressed errors, rerun with: -s
==400== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
```

```
Test Summary
------------
VOCS/SR/SGGIT/SER/BV-01-C                            Passed       1.879 seconds
VOCS/SR/SGGIT/CHA/BV-01-C                            Passed       0.322 seconds
VOCS/SR/SGGIT/CHA/BV-02-C                            Passed       0.290 seconds
VOCS/SR/SGGIT/CHA/BV-03-C                            Passed       0.255 seconds
VOCS/SR/SGGIT/CHA/BV-04-C                            Passed       0.258 seconds
VOCS/SR/SGGIT/CP/BI-01-C                             Passed       0.379 seconds
VOCS/SR/SGGIT/CP/BI-02-C                             Passed       0.287 seconds
VOCS/SR/SGGIT/CP/BI-03-C                             Passed       0.291 seconds
VOCS/SR/SPE/BI-01-C                                  Passed       0.315 seconds
VOCS/SR/CP/BV-01-C                                   Passed       5.032 seconds
AICS/SR/SGGIT/CHA/BV-01-C                            Passed       0.275 seconds
AICS/SR/SGGIT/CHA/BV-02-C                            Passed       0.272 seconds
AICS/SR/SGGIT/CHA/BV-03-C                            Passed       0.274 seconds
AICS/SR/SGGIT/CHA/BV-04-C                            Passed       0.272 seconds
AICS/SR/SGGIT/CHA/BV-05-C                            Passed       0.256 seconds
AICS/SR/SGGIT/CHA/BV-06-C                            Passed       0.259 seconds
AICS/SR/SGGIT/CP/BI-01-C                             Passed       0.403 seconds
AICS/SR/SGGIT/CP/BI-02-C                             Passed       0.310 seconds
AICS/SR/CP/BV-01-C                                   Passed       5.062 seconds
AICS/SR/SPE/BI-01-C                                  Passed       0.330 seconds
Total: 20, Passed: 20 (100.0%), Failed: 0, Not Run: 0
Overall execution time: 17.2 seconds
PASS: test-vcp
All 1 tests passed
DURATION: 42
END: /usr/lib/bluez5/ptest
2025-03-24T20:27
STOP: ptest-runner
TOTAL: 1 FAIL: 0
```

thesamesam, 2025-03-24:
I see the following:
* Valgrind uninitialised bytes notice and the test passes
* The test fails without Valgrind

```
$ valgrind -q --error-exitcode=1 unit/test-vcp

==741346== Syscall param socketcall.bind(my_addr.rc_bdaddr) points to uninitialised byte(s)
==741346==    at 0x4B70407: bind (syscall-template.S:120)
==741346==    by 0x131106: bt_log_open (log.c:118)
==741346==    by 0x131106: bt_log_open (log.c:95)
==741346==    by 0x1311A1: bt_log_sendmsg (log.c:59)
==741346==    by 0x131362: bt_log_vprintf (log.c:152)
==741346==    by 0x12F03B: tester_log (tester.c:166)
==741346==    by 0x12F1DD: next_test_case (tester.c:478)
==741346==    by 0x12F364: start_tester (tester.c:769)
==741346==    by 0x4918375: g_main_dispatch (gmain.c:3398)
==741346==    by 0x49193D7: UnknownInlinedFun (gmain.c:4249)
==741346==    by 0x49193D7: g_main_context_iterate_unlocked.isra.0 (gmain.c:4314)
==741346==    by 0x491AA3E: g_main_loop_run (gmain.c:4516)
==741346==    by 0x131454: mainloop_run (mainloop-glib.c:66)
==741346==    by 0x131834: mainloop_run_with_signal (mainloop-notify.c:189)
==741346==  Address 0x1ffefff1d6 is on thread 1's stack
==741346==  in frame #1, created by bt_log_open (log.c:96)
==741346==
VOCS/SR/SGGIT/SER/BV-01-C - init
VOCS/SR/SGGIT/SER/BV-01-C - setup
VOCS/SR/SGGIT/SER/BV-01-C - setup complete
VOCS/SR/SGGIT/SER/BV-01-C - run
VOCS/SR/SGGIT/SER/BV-01-C - test passed
VOCS/SR/SGGIT/SER/BV-01-C - teardown
VOCS/SR/SGGIT/SER/BV-01-C - teardown complete
VOCS/SR/SGGIT/SER/BV-01-C - done
[...]
AICS/SR/SPE/BI-01-C                                  Passed       0.006 seconds
Total: 20, Passed: 20 (100.0%), Failed: 0, Not Run: 0
Overall execution time: 0.417 seconds
```

and
```
$ unit/test-vcp

VOCS/SR/SGGIT/SER/BV-01-C - init
VOCS/SR/SGGIT/SER/BV-01-C - setup
VOCS/SR/SGGIT/SER/BV-01-C - setup complete
[...]

AICS/SR/CP/BV-01-C - setup complete
AICS/SR/CP/BV-01-C - run
gatt_notify_cb: Failed to send notification
**
ERROR:src/shared/tester.c:981:test_io_recv: assertion failed (len == iov->iov_len): (5 == 6)
Bail out! ERROR:src/shared/tester.c:981:test_io_recv: assertion failed (len == iov->iov_len): (5 == 6)
Aborted                    (core dumped) unit/test-vcp
```

so definitely seems timing related. I haven't looked at what the uninitialised warning is about.

gudnimg, 2025-03-24:
@thesamesam I get the exact same result as you when I use the same command. It seems as if when the test executes very fast, then it fails.

gudnimg, 2025-03-24:
From the `???` below I wonder if I'm missing some debug symbols. I suspect the trace looks identical to yours.
```
root@qemux86-64:/usr/lib/bluez5/ptest# valgrind -q --error-exitcode=1 unit/test-vcp

==356== Syscall param socketcall.bind(my_addr.rc_bdaddr) points to uninitialised byte(s)
==356==    at 0x4AB4EDB: bind (syscall-template.S:120)
==356==    by 0x131BA4: ??? (in /usr/lib/bluez5/ptest/unit/test-vcp)
==356==    by 0x131C3E: ??? (in /usr/lib/bluez5/ptest/unit/test-vcp)
==356==    by 0x131DD2: ??? (in /usr/lib/bluez5/ptest/unit/test-vcp)
==356==    by 0x12FDED: ??? (in /usr/lib/bluez5/ptest/unit/test-vcp)
==356==    by 0x12FF6D: ??? (in /usr/lib/bluez5/ptest/unit/test-vcp)
==356==    by 0x1300F4: ??? (in /usr/lib/bluez5/ptest/unit/test-vcp)
==356==    by 0x48B46D8: ??? (in /usr/lib/libglib-2.0.so.0.8400.0)
==356==    by 0x48B77C6: ??? (in /usr/lib/libglib-2.0.so.0.8400.0)
==356==    by 0x48B819E: g_main_loop_run (in /usr/lib/libglib-2.0.so.0.8400.0)
==356==    by 0x131EB4: ??? (in /usr/lib/bluez5/ptest/unit/test-vcp)
==356==    by 0x1321EE: ??? (in /usr/lib/bluez5/ptest/unit/test-vcp)
==356==  Address 0x1fff0006ce is on thread 1's stack
==356== 
```

gudnimg, 2025-03-29:
Nothing new to report from my side, but here is a summary of what we have found so far:

- Valgrind tool does not detect any problems related to this bug.
- The problem appears on both libc and musl systems.
- The problem is confirmed to reproduce on 64-bit ARM and 64-bit x86 QEMU simulator. Likely more platforms too.
- The bug can be reproduced depending on how you execute the test. For certain commands the test pass but for some it fails. This hints at a timing problems or a race condition.

|  Command | Does it reproduce the problem | Comment |
| -------- | ------- | ------- |
| `unit/test-vcp` | ❌  Yes    | Running the executable directly |
| `valgrind -q --error-exitcode=1 unit/test-vcp`  | ✅  No     |  |
| `valgrind --leak-check=full --show-leak-kinds=all --track-origins=yes --log-file=valgrind-bluez-log.txt ptest-runner bluez5` | ❌  Yes      | ptest-runner bluez5 runs the following script from Yocto Project [run-ptest script](https://github.com/openembedded/openembedded-core/blob/master/meta/recipes-connectivity/bluez5/bluez5/run-ptest) |
| `valgrind --leak-check=full --show-leak-kinds=all --track-origins=yes --trace-children=yes --log-file=valgrind-bluez-log.txt ptest-runner bluez5` | ❌  Yes      | Same as above except only added `--trace-children=yes` |

gudnimg, 2025-05-20:
> This issue is stale because it has been open 30 days with no activity. Remove stale label or comment or this will be closed in 5 days.

Not stale

gudnimg, 2025-06-04:
Hmm, I'm getting a segmentation fault on BlueZ 5.83. Haven't tested running with valgrind yet.

```
root@qemux86-64:/usr/lib/bluez5/ptest# unit/test-vcp

VOCS/SR/SGGIT/SER/BV-01-C - init
VOCS/SR/SGGIT/SER/BV-01-C - setup
VOCS/SR/SGGIT/SER/BV-01-C - setup complete
VOCS/SR/SGGIT/SER/BV-01-C - run
VOCS/SR/SGGIT/SER/BV-01-C - test passed
VOCS/SR/SGGIT/SER/BV-01-C - teardown
VOCS/SR/SGGIT/SER/BV-01-C - teardown complete
VOCS/SR/SGGIT/SER/BV-01-C - done

VOCS/SR/SGGIT/CHA/BV-01-C - init
VOCS/SR/SGGIT/CHA/BV-01-C - setup
VOCS/SR/SGGIT/CHA/BV-01-C - setup complete
VOCS/SR/SGGIT/CHA/BV-01-C - run
VOCS/SR/SGGIT/CHA/BV-01-C - test passed
VOCS/SR/SGGIT/CHA/BV-01-C - teardown
VOCS/SR/SGGIT/CHA/BV-01-C - teardown complete
VOCS/SR/SGGIT/CHA/BV-01-C - done

VOCS/SR/SGGIT/CHA/BV-02-C - init
VOCS/SR/SGGIT/CHA/BV-02-C - setup
VOCS/SR/SGGIT/CHA/BV-02-C - setup complete
VOCS/SR/SGGIT/CHA/BV-02-C - run
VOCS/SR/SGGIT/CHA/BV-02-C - test passed
VOCS/SR/SGGIT/CHA/BV-02-C - teardown
VOCS/SR/SGGIT/CHA/BV-02-C - teardown complete
VOCS/SR/SGGIT/CHA/BV-02-C - done

VOCS/SR/SGGIT/CHA/BV-03-C - init
VOCS/SR/SGGIT/CHA/BV-03-C - setup
VOCS/SR/SGGIT/CHA/BV-03-C - setup complete
VOCS/SR/SGGIT/CHA/BV-03-C - run
VOCS/SR/SGGIT/CHA/BV-03-C - test passed
VOCS/SR/SGGIT/CHA/BV-03-C - teardown
VOCS/SR/SGGIT/CHA/BV-03-C - teardown complete
VOCS/SR/SGGIT/CHA/BV-03-C - done

VOCS/SR/SGGIT/CHA/BV-04-C - init
VOCS/SR/SGGIT/CHA/BV-04-C - setup
VOCS/SR/SGGIT/CHA/BV-04-C - setup complete
VOCS/SR/SGGIT/CHA/BV-04-C - run
VOCS/SR/SGGIT/CHA/BV-04-C - test passed
VOCS/SR/SGGIT/CHA/BV-04-C - teardown
VOCS/SR/SGGIT/CHA/BV-04-C - teardown complete
VOCS/SR/SGGIT/CHA/BV-04-C - done

VOCS/SR/SGGIT/CP/BI-01-C - init
VOCS/SR/SGGIT/CP/BI-01-C - setup
VOCS/SR/SGGIT/CP/BI-01-C - setup complete
VOCS/SR/SGGIT/CP/BI-01-C - run
VOCS/SR/SGGIT/CP/BI-01-C - test passed
VOCS/SR/SGGIT/CP/BI-01-C - teardown
VOCS/SR/SGGIT/CP/BI-01-C - teardown complete
VOCS/SR/SGGIT/CP/BI-01-C - done

VOCS/SR/SGGIT/CP/BI-02-C - init
VOCS/SR/SGGIT/CP/BI-02-C - setup
VOCS/SR/SGGIT/CP/BI-02-C - setup complete
VOCS/SR/SGGIT/CP/BI-02-C - run
VOCS/SR/SGGIT/CP/BI-02-C - test passed
VOCS/SR/SGGIT/CP/BI-02-C - teardown
VOCS/SR/SGGIT/CP/BI-02-C - teardown complete
VOCS/SR/SGGIT/CP/BI-02-C - done

VOCS/SR/SGGIT/CP/BI-03-C - init
VOCS/SR/SGGIT/CP/BI-03-C - setup
VOCS/SR/SGGIT/CP/BI-03-C - setup complete
VOCS/SR/SGGIT/CP/BI-03-C - run
VOCS/SR/SGGIT/CP/BI-03-C - test passed
VOCS/SR/SGGIT/CP/BI-03-C - teardown
VOCS/SR/SGGIT/CP/BI-03-C - teardown complete
VOCS/SR/SGGIT/CP/BI-03-C - done

VOCS/SR/SPE/BI-01-C - init
VOCS/SR/SPE/BI-01-C - setup
VOCS/SR/SPE/BI-01-C - setup complete
VOCS/SR/SPE/BI-01-C - run
VOCS/SR/SPE/BI-01-C - test passed
VOCS/SR/SPE/BI-01-C - teardown
VOCS/SR/SPE/BI-01-C - teardown complete
VOCS/SR/SPE/BI-01-C - done

VOCS/SR/CP/BV-01-C - init
VOCS/SR/CP/BV-01-C - setup
VOCS/SR/CP/BV-01-C - setup complete
VOCS/SR/CP/BV-01-C - run
VOCS/SR/CP/BV-01-C - test passed
VOCS/SR/CP/BV-01-C - teardown
VOCS/SR/CP/BV-01-C - teardown complete
VOCS/SR/CP/BV-01-C - done

AICS/SR/SGGIT/CHA/BV-01-C - init
AICS/SR/SGGIT/CHA/BV-01-C - setup
AICS/SR/SGGIT/CHA/BV-01-C - setup complete
AICS/SR/SGGIT/CHA/BV-01-C - run
AICS/SR/SGGIT/CHA/BV-01-C - test passed
AICS/SR/SGGIT/CHA/BV-01-C - teardown
AICS/SR/SGGIT/CHA/BV-01-C - teardown complete
AICS/SR/SGGIT/CHA/BV-01-C - done

AICS/SR/SGGIT/CHA/BV-02-C - init
AICS/SR/SGGIT/CHA/BV-02-C - setup
AICS/SR/SGGIT/CHA/BV-02-C - setup complete
AICS/SR/SGGIT/CHA/BV-02-C - run
AICS/SR/SGGIT/CHA/BV-02-C - test passed
AICS/SR/SGGIT/CHA/BV-02-C - teardown
AICS/SR/SGGIT/CHA/BV-02-C - teardown complete
AICS/SR/SGGIT/CHA/BV-02-C - done

AICS/SR/SGGIT/CHA/BV-03-C - init
AICS/SR/SGGIT/CHA/BV-03-C - setup
AICS/SR/SGGIT/CHA/BV-03-C - setup complete
AICS/SR/SGGIT/CHA/BV-03-C - run
AICS/SR/SGGIT/CHA/BV-03-C - test passed
AICS/SR/SGGIT/CHA/BV-03-C - teardown
AICS/SR/SGGIT/CHA/BV-03-C - teardown complete
AICS/SR/SGGIT/CHA/BV-03-C - done

AICS/SR/SGGIT/CHA/BV-04-C - init
AICS/SR/SGGIT/CHA/BV-04-C - setup
AICS/SR/SGGIT/CHA/BV-04-C - setup complete
AICS/SR/SGGIT/CHA/BV-04-C - run
AICS/SR/SGGIT/CHA/BV-04-C - test passed
AICS/SR/SGGIT/CHA/BV-04-C - teardown
AICS/SR/SGGIT/CHA/BV-04-C - teardown complete
AICS/SR/SGGIT/CHA/BV-04-C - done

AICS/SR/SGGIT/CHA/BV-05-C - init
AICS/SR/SGGIT/CHA/BV-05-C - setup
AICS/SR/SGGIT/CHA/BV-05-C - setup complete
AICS/SR/SGGIT/CHA/BV-05-C - run
AICS/SR/SGGIT/CHA/BV-05-C - test passed
AICS/SR/SGGIT/CHA/BV-05-C - teardown
AICS/SR/SGGIT/CHA/BV-05-C - teardown complete
AICS/SR/SGGIT/CHA/BV-05-C - done

AICS/SR/SGGIT/CHA/BV-06-C - init
AICS/SR/SGGIT/CHA/BV-06-C - setup
AICS/SR/SGGIT/CHA/BV-06-C - setup complete
AICS/SR/SGGIT/CHA/BV-06-C - run
AICS/SR/SGGIT/CHA/BV-06-C - test passed
AICS/SR/SGGIT/CHA/BV-06-C - teardown
AICS/SR/SGGIT/CHA/BV-06-C - teardown complete
AICS/SR/SGGIT/CHA/BV-06-C - done

AICS/SR/SGGIT/CP/BI-01-C - init
AICS/SR/SGGIT/CP/BI-01-C - setup
AICS/SR/SGGIT/CP/BI-01-C - setup complete
AICS/SR/SGGIT/CP/BI-01-C - run
AICS/SR/SGGIT/CP/BI-01-C - test passed
AICS/SR/SGGIT/CP/BI-01-C - teardown
AICS/SR/SGGIT/CP/BI-01-C - teardown complete
AICS/SR/SGGIT/CP/BI-01-C - done

AICS/SR/SGGIT/CP/BI-02-C - init
AICS/SR/SGGIT/CP/BI-02-C - setup
AICS/SR/SGGIT/CP/BI-02-C - setup complete
AICS/SR/SGGIT/CP/BI-02-C - run
AICS/SR/SGGIT/CP/BI-02-C - test passed
AICS/SR/SGGIT/CP/BI-02-C - teardown
AICS/SR/SGGIT/CP/BI-02-C - teardown complete
AICS/SR/SGGIT/CP/BI-02-C - done

AICS/SR/CP/BV-01-C - init
AICS/SR/CP/BV-01-C - setup
AICS/SR/CP/BV-01-C - setup complete
AICS/SR/CP/BV-01-C - run
[   69.131825] traps: test-vcp[364] general protection fault ip:557087fb0b3f sp:7ffff9a38310 error:0 in test-vcp[1eb3f,557087f95000+29000]
Segmentation fault
```

Sambura, 2025-06-28:
Recently added patch series https://github.com/bluez/bluez/commit/71b5ea9bf3738755bf644d0a21958e54816c54c4 .. https://github.com/bluez/bluez/commit/a209d69767db0f62d6fdc86a7f73ee4ec749c37f (4 commits total) fixes a similar issue and should hopefully also resolve this one. Would appreciate someone testing and verifying this on their end.

gudnimg, 2025-06-28:
> Recently added patch series [71b5ea9](https://github.com/bluez/bluez/commit/71b5ea9bf3738755bf644d0a21958e54816c54c4) .. [a209d69](https://github.com/bluez/bluez/commit/a209d69767db0f62d6fdc86a7f73ee4ec749c37f) (4 commits total) fixes a similar issue and should hopefully also resolve this one. Would appreciate someone testing and verifying this on their end.

Running the test on master branch (8e2425dc2b2553803157d4bee52261e2f4d1ae73) works on my end! Below are the test results for running `unit/test-vcp` which has always failed on my end. All pass now :)

The tests also run more than 10 seconds faster than my previous runs in this thread.

```
Test Summary
------------
VOCS/SR/SGGIT/SER/BV-01-C                            Passed       0.077 seconds
VOCS/SR/SGGIT/CHA/BV-01-C                            Passed       0.070 seconds
VOCS/SR/SGGIT/CHA/BV-02-C                            Passed       0.073 seconds
VOCS/SR/SGGIT/CHA/BV-03-C                            Passed       0.065 seconds
VOCS/SR/SGGIT/CHA/BV-04-C                            Passed       0.068 seconds
VOCS/SR/SGGIT/CP/BI-01-C                             Passed       0.071 seconds
VOCS/SR/SGGIT/CP/BI-02-C                             Passed       0.069 seconds
VOCS/SR/SGGIT/CP/BI-03-C                             Passed       0.076 seconds
VOCS/SR/SPE/BI-01-C                                  Passed       0.073 seconds
VOCS/SR/CP/BV-01-C                                   Passed       0.436 seconds
AICS/SR/SGGIT/CHA/BV-01-C                            Passed       0.077 seconds
AICS/SR/SGGIT/CHA/BV-02-C                            Passed       0.073 seconds
AICS/SR/SGGIT/CHA/BV-03-C                            Passed       0.074 seconds
AICS/SR/SGGIT/CHA/BV-04-C                            Passed       0.072 seconds
AICS/SR/SGGIT/CHA/BV-05-C                            Passed       0.070 seconds
AICS/SR/SGGIT/CHA/BV-06-C                            Passed       0.067 seconds
AICS/SR/SGGIT/CP/BI-01-C                             Passed       0.073 seconds
AICS/SR/SGGIT/CP/BI-02-C                             Passed       0.079 seconds
AICS/SR/CP/BV-01-C                                   Passed       0.440 seconds
AICS/SR/SPE/BI-01-C                                  Passed       0.073 seconds
Total: 20, Passed: 20 (100.0%), Failed: 0, Not Run: 0
Overall execution time: 2.22 seconds
```

macmpi, 2025-09-20:
confirmed fixed with 5.84, thanks.

closed 2025-09-20 by macmpi