bluez/bluez #728: A2DP Headphones Disconnect When Idle/Suspended

GPSolo, 2024-01-21:
I've seen some weird behavior recently from my bluetooth headphones, whenever they connect, they work fine for the first time I play a video or any sort of audio, but once the audio stops going through the headphones bluez seems to try to suspend the device but then it disconnects the headphones. I was messing with my Desktop earlier, trying to install Hyprland on Debian, but only broke other parts of my system so i manually removed it and ever since I've had this problem, but, I also am on unstable Debian "sid" so I'm not sure if its a issue I created or a update changed the behavior of my bluetooth.
Here is my Hardware/OS Info:
```
graphics card:
                       nVidia TU116 [GeForce GTX 1660 SUPER]
cpu:                                                            
                       Intel(R) Core(TM) i5-9400F CPU @ 2.90GHz
network:
  wlp5s0               Intel Wi-Fi 6 AX200NGW
  enp4s0               Gigabyte Onboard Ethernet
network interface:
  wlp5s0               Ethernet network interface
  enp4s0               Ethernet network interface
  lo                   Loopback network interface
bridge:
                       Intel Cannon Lake PCH PCI Express Root Port #1
                       Intel Z390 Chipset LPC/eSPI Controller
                       Intel 6th-10th Gen Core Processor PCIe Controller (x16)
                       Intel Cannon Lake PCH PCI Express Root Port #17
                       Intel Cannon Lake PCH PCI Express Root Port #4
                       Intel 8th Gen Core Processor Host Bridge/DRAM Registers
                       Intel Cannon Lake PCH PCI Express Root Port #9
                       Intel Cannon Lake PCH PCI Express Root Port #3
bluetooth:
                       Intel AX200 Bluetooth
unknown:
                       FPU
                       DMA controller
                       PIC
                       Keyboard controller
                       PS/2 Controller
                       Intel Cannon Lake PCH HECI Controller
                       Intel Cannon Lake PCH SPI Controller
                       Intel Cannon Lake PCH Thermal Controller
                       nVidia TU116 USB Type-C UCSI Controller
                       Intel Cannon Lake PCH Shared SRAM

                       Intel Cannon Lake PCH SMBus Controller

```
and this is the Journalctl Output from my bluetooth.service running in debug mode:
```
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/profile.c:ext_device_probe() Phone Book Access probed with UUID 0000112f-0000-1000-8000-00805f9b34fb
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/service.c:change_state() 0x5557e94045f0: device 04:52:C7:C7:18:2E profile Phone Book Access state changed: unavailable -> disconnected (0)
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/profile.c:register_profile() sender :1.83
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/profile.c:create_ext() Created "Synchronization"
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/profile.c:ext_adapter_probe() "Synchronization" probed
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/profile.c:ext_start_servers() Synchronization listening on chan 14
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/adapter.c:adapter_service_add() /org/bluez/hci0
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/sdpd-service.c:add_record_to_server() Adding record with handle 0x10006
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00000003-0000-1000-8000-00805f9
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00000008-0000-1000-8000-00805f9
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00000100-0000-1000-8000-00805f9
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00001002-0000-1000-8000-00805f9
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00001104-0000-1000-8000-00805f9
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/adapter.c:adapter_service_insert() /org/bluez/hci0
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/adapter.c:add_uuid() sending add uuid command for index 0
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/profile.c:register_profile() sender :1.83
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/profile.c:create_ext() Created "File Transfer"
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/profile.c:ext_adapter_probe() "File Transfer" probed
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/profile.c:ext_start_servers() File Transfer listening on PSM 4103
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/profile.c:ext_start_servers() File Transfer listening on chan 10
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/adapter.c:adapter_service_add() /org/bluez/hci0
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/sdpd-service.c:add_record_to_server() Adding record with handle 0x10007
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00000003-0000-1000-8000-00805f9
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00000008-0000-1000-8000-00805f9
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00000100-0000-1000-8000-00805f9
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00001002-0000-1000-8000-00805f9
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00001106-0000-1000-8000-00805f9
Jan 20 23:54:21 gp-deb bluetoothd[7605]: src/adapter.c:adapter_service_insert() /org/bluez/hci0
[... 350 lines cut ...]
Jan 20 23:54:42 gp-deb bluetoothd[7605]: profiles/audio/avdtp.c:avdtp_ref() 0x5557e9422240: ref=3
Jan 20 23:54:42 gp-deb bluetoothd[7605]: profiles/audio/a2dp.c:setup_ref() 0x5557e9432620: ref=1
Jan 20 23:54:42 gp-deb bluetoothd[7605]: profiles/audio/transport.c:media_request_create() Request created: method=Release id=5
Jan 20 23:54:42 gp-deb bluetoothd[7605]: profiles/audio/transport.c:media_owner_add() Owner :1.52 Request Release
Jan 20 23:54:42 gp-deb bluetoothd[7605]: profiles/audio/transport.c:media_request_reply() Request Release Reply Success
Jan 20 23:54:42 gp-deb bluetoothd[7605]: profiles/audio/transport.c:media_owner_remove() Owner :1.52 Request Release
Jan 20 23:54:42 gp-deb bluetoothd[7605]: profiles/audio/transport.c:transport_set_state() State changed /org/bluez/hci0/dev_04_52_C7_C7_18_2E/sep1/fd0: TRANSPORT_STATE_SUSPENDING -> TRANSPORT_STATE_IDLE
Jan 20 23:54:42 gp-deb bluetoothd[7605]: profiles/audio/transport.c:media_transport_remove_owner() Transport /org/bluez/hci0/dev_04_52_C7_C7_18_2E/sep1/fd0 Owner :1.52
Jan 20 23:54:42 gp-deb bluetoothd[7605]: profiles/audio/transport.c:media_owner_free() Owner :1.52
Jan 20 23:54:42 gp-deb bluetoothd[7605]: profiles/audio/avdtp.c:session_cb()
Jan 20 23:54:42 gp-deb bluetoothd[7605]: profiles/audio/avdtp.c:avdtp_parse_resp() SUSPEND request succeeded
Jan 20 23:54:42 gp-deb bluetoothd[7605]: profiles/audio/avdtp.c:avdtp_sep_set_state() stream state changed: STREAMING -> OPEN
Jan 20 23:54:42 gp-deb bluetoothd[7605]: profiles/audio/sink.c:sink_set_state() State changed /org/bluez/hci0/dev_04_52_C7_C7_18_2E: SINK_STATE_PLAYING -> SINK_STATE_CONNECTED
Jan 20 23:54:42 gp-deb bluetoothd[7605]: profiles/audio/transport.c:transport_update_playing() /org/bluez/hci0/dev_04_52_C7_C7_18_2E/sep1/fd0 State=TRANSPORT_STATE_IDLE Playing=0
Jan 20 23:54:42 gp-deb bluetoothd[7605]: profiles/audio/a2dp.c:suspend_cfm() Source 0x5557e94177f0: Suspend_Cfm
Jan 20 23:54:42 gp-deb systemd[1]: bluetooth.service: Main process exited, code=dumped, status=11/SEGV
Jan 20 23:54:42 gp-deb systemd[1]: bluetooth.service: Failed with result 'core-dump'.
Jan 20 23:54:42 gp-deb systemd[1]: Starting bluetooth.service - Bluetooth service...
Jan 20 23:54:42 gp-deb (uetoothd)[7690]: bluetooth.service: ConfigurationDirectory 'bluetooth' already exists but the mode is different. (File system: 755 ConfigurationDirectoryMode: 555)
Jan 20 23:54:42 gp-deb bluetoothd[7690]: Bluetooth daemon 5.71
Jan 20 23:54:42 gp-deb bluetoothd[7690]: src/main.c:parse_config() parsing /etc/bluetooth/main.conf
Jan 20 23:54:42 gp-deb bluetoothd[7690]: src/main.c:parse_config_string() General.ControllerMode = bredr

```
I'm not sure if this is the right place to come to or not but I figured I needed some help because it has been going for days and I have 0 clue why its happening at all. (forgive me for a very long post, I was trying to copy and paste only the parts I felt we're needed to solve the problem, but I may have gotten a bit too much)

GPSolo, 2024-01-21:
Just did a semi-full reinstall of my debian installation and can say with confidence this is an issue with some sort of update between debian stable and debian unstable, something along the way of these packages I believe is causing this issue. I did maintain the home directory but I do not figure that has anything to do with bluetooth.

vtpoet, 2024-01-21:
I'm also having this issue with my Bose Revolve+. Began with both my Tumbleweed installations, one on Gnome (MS Surface GO) and one on Plasma (Lenovo t470s). On both systems, the Bose would disconnect on Idle/Suspend. Then reconnect, then disconnect, then do a bad reconnect. Because of this bug, and because of a shutdown bug in TW that prevented my MS Surface GO from properly shutting down, I installed Spiral Linux (Debian 12) on the GO. The Bose Rovolve+ operates flawlessly under Debian 12. This would correspond with GPSolo's experience.

gabriele2000, 2024-02-01:
I'm having the same problem.
I'm using an Intel AX210NGW card, but it's not the issue.

I can 100% reproduce the issue by:
- Connecting my Headphones (SONY WH-1000XM5)
- They idle, everything's fine

BUT
- Connecting the headphones
- Connecting the Dualsense (PS5 Controller)
- Headphones goes to idle
- This happens
```
[WH-1000XM5]# [CHG] Device A0:AB:51:A2:0F:FC Connected: yes
[WH-1000XM5]# [CHG] Transport /org/bluez/hci0/dev_AC_80_0A_AE_65_3D/sep3/fd1 State: idle
[WH-1000XM5]# [DEL] Transport /org/bluez/hci0/dev_AC_80_0A_AE_65_3D/sep3/fd1
[WH-1000XM5]# [DEL] Endpoint /org/bluez/hci0/dev_AC_80_0A_AE_65_3D/sep1
[WH-1000XM5]# [DEL] Endpoint /org/bluez/hci0/dev_AC_80_0A_AE_65_3D/sep2
[WH-1000XM5]# [DEL] Endpoint /org/bluez/hci0/dev_AC_80_0A_AE_65_3D/sep3
```

gabriele2000, 2024-03-10:
It seems that the issue is now solved! 
YES!

GPSolo, 2024-04-05:
@gabriele2000 Are you on debian unstable? I have held my bluez packages since I ran into this problem with a version from the bookworm repo _5.66-1+deb12u1._ Did you do anything to get it to work as expected?

gabriele2000, 2024-04-05:
> @gabriele2000 Are you on debian unstable? I have held my bluez packages since I ran into this problem with a version from the bookworm repo _5.66-1+deb12u1._ Did you do anything to get it to work as expected?

False alarm though, even if the problem is less present and a bit different.

I'm on Pop!_Os, I compile and install the stable releases taken from this project's release page.
Here's when it works:
- You connect the controller after connecting the headphones (most of the time it works, the other times it will have the standby issue)
- You connect two bluetooth headphones together using something like Helvum (so that the monitor of one will connect to the sink of the other).
That causes the new node to not go on standby, fixing the issue, but you could have another issue: lack of bandwidth, so the controller will have intermittent lag

Here's what it doesn't work:
- Connecting the controller before the headphones
- When you disconnect the controller and reconnect it, it'll no longer work, you'll have to restart your PC

closed 2024-07-21 by GPSolo