bluez/bluez #730: bluetoothctl v5.72 gatt.select-attribute chooses MAC from disconnected device with same uuid

hkskoglund, 2024-01-24:
I have trouble reading attributes that have the same uuid on two gopros. I was expecing gatt.select-attribute to operate on the currently connected device C6:6B:22:Ed:93:E1, but its using the last connected device? 

Last connected          /org/bluez/hci0/dev_D9_D7_C3_03_2A_3D/service0021/char0022
Currently connected /org/bluez/hci0/dev_C6_6B_22_ED_93_E1/service0021/char0022

bluetoothctl:

```
[GoPro 6288:/service0021/char0022]# info
Device C6:6B:22:ED:93:E1 (random)
        Name: GoPro 6288
        Alias: GoPro 6288
        Appearance: 0x0080 (128)
        Icon: computer
        Paired: yes
        Bonded: yes
        Trusted: yes
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Tx Power                  (00001804-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
        UUID: GoPro, Inc.               (0000fea6-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (b5f90001-aa8d-11e3-9046-0002a5d5c51b)
        UUID: Vendor specific           (b5f90090-aa8d-11e3-9046-0002a5d5c51b)
        ManufacturerData.Key: 0x02f2 (754)
        ManufacturerData.Value:
  01 01 18 3f 00 e1 93 ed 22 6b c6 0a              ...?...."k..    
        Battery Percentage: 0x00 (0)
info88:/service0021/char0022]# info
[GoPro 6288:/service0021/char0022]# gatt.select-attribute b5f90002-aa8d-11e3-9046-0002a5d5c51b
gatt.read
Attempting to read /org/bluez/hci0/dev_D9_D7_C3_03_2A_3D/service0021/char0022
[GoPro 6288:/service0021/char0022]# Failed to read: org.bluez.Error.Failed
[GoPro 6288:/service0021/char0022]# info D9:D7:C3:03:2A:3D
Device D9:D7:C3:03:2A:3D (random)
        Name: GoPro 4298
        Alias: GoPro 4298
        Appearance: 0x0080 (128)
        Icon: computer
        Paired: yes
        Bonded: yes
        Trusted: yes
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Tx Power                  (00001804-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
        UUID: GoPro, Inc.               (0000fea6-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (b5f90001-aa8d-11e3-9046-0002a5d5c51b)
        UUID: Vendor specific           (b5f90080-aa8d-11e3-9046-0002a5d5c51b)
        UUID: Vendor specific           (b5f90090-aa8d-11e3-9046-0002a5d5c51b)
        ManufacturerData.Key: 0x02f2 (754)
        ManufacturerData.Value:
  02 01 39 3f 00 3d 2a 03 c3 d7 d9 0f              ..9?.=*.....    
[GoPro 6288:/service0021/char0022]# 
```

Vudentz, 2024-01-24:
@hkskoglund are you sure there is any service with the UUID on the devices paired with the adapter currently selected? Or are you saying we should restrict the attribute search to only the currently selected adapter, either way it is possible to select by object path in case you want a specific adapter and there are multiple instances of the same UUID accross different adapters.

hkskoglund, 2024-01-25:
@Vudentz I am using an init-script to bluetoothctl for reading values 

    `printf "connect %s\ngatt.select-attribute %s\ngatt.read\nexit\n" "$1" "$2" > "$ble_init_script_filename"`

My understanding is that connect MAC also chooses which device to operate on in the bluetoothctl shell, so that gatt.select-attribute is only searching attributes for this particular connected device

I am confused that gatt.select-attribute chooses an object path from a disconnected device, not the one in the connect command and shell prompt. Runing attribute-info; first match is a disconnected device UUID, next match after select-attribute object-path is correct.

```
[GoPro 6288:/service0021/char0022]# attribute-info  b5f90002-aa8d-11e3-9046-0002a5d5c51b
Characteristic - Vendor specific
        UUID: b5f90002-aa8d-11e3-9046-0002a5d5c51b
        Service: /org/bluez/hci0/dev_D9_D7_C3_03_2A_3D/service0021
        Value:
  47 50 32 36 33 34 34 32 39 38                    GP26344298      
        Flags: read

[GoPro 6288:/service0021/char0022]# select-attribute /org/bluez/hci0/dev_C6_6B_22_ED_93_E1/service0021/char0022

[GoPro 6288:/service0021/char0022]# attribute-info b5f90002-aa8d-11e3-9046-0002a5d5c51b
Characteristic - Vendor specific
        UUID: b5f90002-aa8d-11e3-9046-0002a5d5c51b
        Service: /org/bluez/hci0/dev_C6_6B_22_ED_93_E1/service0021
        Value:
  47 50 32 35 32 31 36 32 38 38                    GP25216288      
        Flags: read
        Flags: write
        MTU: 0x0205 (517)

```

As a workaround I will try to use object path `gatt.select-attribute /org/bluez/hci0/dev_MAC/service0021/char0022`

closed 2025-07-31 by github-actions[bot]