bluez/bluez #732: Pairing fail due to missing key distribution

jamsliu2009, 2024-01-26:
I am developing pairing feature in BlueZ application and encounter a problem which is related to [this commit](https://github.com/torvalds/linux/commit/7aca0ac4792e6cb0f35ef97bfcb39b1663a92fb7#diff-60c49160266f1956b30da7d15585bfd935672e6cb9cda0b080b5b646c925e0a8.).

It seems to have timing issue between SMP distribution and command complete of HCI_OP_WRITE_AUTH_PAYLOAD_TO.

The symptom is, when pairing is proceeding, after remote peer (Peripheral ) sending the Encryption information and Identity information to local Host (BlueZ based Central).
The local host doesn’t move forward to send out the Encryption information and Identity information and pairing procedure is not completed.

From btmon log, there is no “MGMT Event: New Long Term Key (0x000a)” event shows and no LTK created in /var/lib/Bluetooth/
From syslog, there are some SMP commands from ACL are drop as following log snippet.
“Bluetooth: hci0: unexpected SMP command 0x06 from 34:81:f4:ae:0e:b1
Bluetooth: hci0: unexpected SMP command 0x07 from 34:81:f4:ae:0e:b1
Bluetooth: hci0: unexpected SMP command 0x08 from 34:81:f4:ae:0e:b1”

I dig into the SMP part in Linux kernel and found SMP protocol procedure are guarded by allow_cmd flag and SMP_ALLOW_CMD marco to make sure the SMP PDU are received in correct order.
In Our observation, SMP_ALLOW_CMD(SMP_CMD_ENCRYPT_INFO) will be called after command complete of HCI_OP_WRITE_AUTH_PAYLOAD_TO received.
I think BlueZ expect SMP_CMD_ENCRYPT_INFO, SMP_CMD_INITIATOR_IDENT SMP_CMD_IDENT_INFO PDU comes after HCI_OP_WRITE_AUTH_PAYLOAD_TO were complete.

When this command complete event comes later then SMP_CMD_ENCRYPT_INFO, SMP_CMD_INITIATOR_IDENT and SMP_CMD_IDENT_INFO.
smp_sig_channel() rejects the handle of PDU SMP_CMD_ENCRYPT_INFO (0x06), and so the following PDU 0x07, 0x08.
However, these SMP packets and HCI command/event should be independent and can’t guarantee the sequence, it might cause problems.

Please reference following air sniffer log and HCI UART traffic capture. Btmon and Linux syslog related are attached as well.

[image]
[image]

[btmon-pairing-fail.txt](https://github.com/bluez/bluez/files/14061377/btmon-pairing-fail.txt)
[syslog-pairing-fail.txt](https://github.com/bluez/bluez/files/14061379/syslog-pairing-fail.txt)

Vudentz, 2024-01-26:
@jamsliu2009 this shall have been fixed by the following commit:

```
commit f67eabffb57d0bee379994a18ec5f462b2cbdf86
Author: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
Date:   Mon Oct 23 16:26:23 2023 -0700

    Bluetooth: Fix not notifying when connection encryption changes
    
    Some layers such as SMP depend on getting notified about encryption
    changes immediately as they only allow certain PDU to be transmitted
    over an encrypted link which may cause SMP implementation to reject
    valid PDUs received thus causing pairing to fail when it shouldn't.
    
    Fixes: 7aca0ac4792e ("Bluetooth: Wait for HCI_OP_WRITE_AUTH_PAYLOAD_TO to complete")
    Signed-off-by: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
```

jamsliu2009, 2024-02-02:
@Vudentz 
Thanks for your reply. May I know why the original modify was tag on v6.2-rc1 to v6.8-rc1, but this patch was tag on only v6.7-rc7 to v6.8-rc1?

closed 2025-07-30 by github-actions[bot]