bluez/bluez #733: Downgrade to Bluez-5.70.2 solved my problem

berbigou, 2024-01-26:
Hello,
on Archlinux / Plasma / 6.7.0-arch3-1
my Logitech mouse M650 L kept disconnecting (BT icon flickering, not usable) on Bluez-5.72.2, like wrong keys after a new association.
downgrade on Bluez-5.71.3 didn't help (but I didn't test long)

downgrade on Bluez-5.70.2 solved my problem...

So, I post this and hope it will help.

Thanks for your work
Greetings from France

Berbigou

Vudentz, 2024-01-26:
@berbigou any chance to gather some logs, btmon, bluetoothd, etc?

berbigou, 2024-01-27:
(I know I should)
Sorry, as I found a fix, I had no time to dig in...

I will try to investigate. what logs would help (and where are they, please ) ?
Thanks for answering

berbigou, 2024-01-28:
Sorry. 
I came back to bluez-latest (5.72.2) and everything is OK...
Must be stupid.
Excuse me.

Vudentz, 2024-01-29:
Maybe something with the key type is quite right when switching versions or something, or the storage is corrupted and you need to repair to have it fixed, anyway if it is fixed please close it.

closed 2024-01-30 by berbigou