bluez/bluez #734: Bluetooth Key Association Issue with Logitech MX Keys

red-gecko27, 2024-01-29:
**Issue Description:**
I am experiencing an issue with my Logitech MX Keys where after each reboot, I have to re-enter the key association every time.

**Environment:**
- Bluez version: `bluez-5.72-2`
- 6.7.1-arch1-1

**Logs (Not Working - bluez-5.72-2):**
```js
Jan 29 12:17:02 Desktop27 systemd[1]: Starting Bluetooth service...
Jan 29 12:17:02 Desktop27 bluetoothd[1462]: Bluetooth daemon 5.72
Jan 29 12:17:02 Desktop27 bluetoothd[1462]: Starting SDP server
Jan 29 12:17:02 Desktop27 bluetoothd[1462]: src/plugin.c:plugin_init() System does not support csip plugin
Jan 29 12:17:02 Desktop27 bluetoothd[1462]: profiles/audio/micp.c:micp_init() D-Bus experimental not enabled
Jan 29 12:17:02 Desktop27 bluetoothd[1462]: src/plugin.c:plugin_init() System does not support micp plugin
Jan 29 12:17:02 Desktop27 bluetoothd[1462]: src/plugin.c:plugin_init() System does not support vcp plugin
Jan 29 12:17:02 Desktop27 bluetoothd[1462]: src/plugin.c:plugin_init() System does not support mcp plugin
Jan 29 12:17:02 Desktop27 bluetoothd[1462]: src/plugin.c:plugin_init() System does not support bass plugin
Jan 29 12:17:02 Desktop27 bluetoothd[1462]: src/plugin.c:plugin_init() System does not support bap plugin
Jan 29 12:17:02 Desktop27 bluetoothd[1462]: Bluetooth management interface 1.22 initialized
Jan 29 12:17:02 Desktop27 systemd[1]: Started Bluetooth service.
Jan 29 12:17:03 Desktop27 bluetoothd[1462]: Battery Provider Manager created
Jan 29 12:17:03 Desktop27 bluetoothd[1462]: src/device.c:device_set_wake_support() Unable to set wake_support without RPA resolution
Jan 29 12:17:04 Desktop27 bluetoothd[1462]: Failed to load LTKs for hci0: Invalid Parameters (0x0d)
Jan 29 12:17:04 Desktop27 bluetoothd[1462]: Failed to load IRKs for hci0: Invalid Parameters (0x0d)
Jan 29 12:17:21 Desktop27 bluetoothd[1462]: Endpoint registered: sender=:1.58 path=/MediaEndpoint/A2DPSink/sbc
Jan 29 12:17:21 Desktop27 bluetoothd[1462]: Endpoint registered: sender=:1.58 path=/MediaEndpoint/A2DPSource/sbc
Jan 29 12:17:21 Desktop27 bluetoothd[1462]: Endpoint registered: sender=:1.58 path=/MediaEndpoint/A2DPSink/sbc_xq_453
Jan 29 12:17:21 Desktop27 bluetoothd[1462]: Endpoint registered: sender=:1.58 path=/MediaEndpoint/A2DPSource/sbc_xq_453
Jan 29 12:17:21 Desktop27 bluetoothd[1462]: Endpoint registered: sender=:1.58 path=/MediaEndpoint/A2DPSink/sbc_xq_512
Jan 29 12:17:21 Desktop27 bluetoothd[1462]: Endpoint registered: sender=:1.58 path=/MediaEndpoint/A2DPSource/sbc_xq_512
Jan 29 12:17:21 Desktop27 bluetoothd[1462]: Endpoint registered: sender=:1.58 path=/MediaEndpoint/A2DPSink/sbc_xq_552
Jan 29 12:17:21 Desktop27 bluetoothd[1462]: Endpoint registered: sender=:1.58 path=/MediaEndpoint/A2DPSource/sbc_xq_552
Jan 29 12:17:21 Desktop27 bluetoothd[1462]: Endpoint registered: sender=:1.58 path=/MediaEndpoint/A2DPSink/faststream
Jan 29 12:17:21 Desktop27 bluetoothd[1462]: Endpoint registered: sender=:1.58 path=/MediaEndpoint/A2DPSource/faststream

```

**Working Logs (bluez-5.70-1):**
```js
Jan 29 13:28:21 Desktop27 systemd[1]: Starting Bluetooth service...
Jan 29 13:28:21 Desktop27 bluetoothd[1097]: Bluetooth daemon 5.70
Jan 29 13:28:21 Desktop27 bluetoothd[1097]: Starting SDP server
Jan 29 13:28:21 Desktop27 bluetoothd[1097]: src/plugin.c:plugin_init() System does not support csip plugin
Jan 29 13:28:21 Desktop27 bluetoothd[1097]: profiles/audio/micp.c:micp_init() D-Bus experimental not enabled
Jan 29 13:28:21 Desktop27 bluetoothd[1097]: src/plugin.c:plugin_init() System does not support micp plugin
Jan 29 13:28:21 Desktop27 bluetoothd[1097]: src/plugin.c:plugin_init() System does not support vcp plugin
Jan 29 13:28:21 Desktop27 bluetoothd[1097]: src/plugin.c:plugin_init() System does not support mcp plugin
Jan 29 13:28:21 Desktop27 bluetoothd[1097]: src/plugin.c:plugin_init() System does not support bass plugin
Jan 29 13:28:21 Desktop27 bluetoothd[1097]: src/plugin.c:plugin_init() System does not support bap plugin
Jan 29 13:28:21 Desktop27 bluetoothd[1097]: Bluetooth management interface 1.22 initialized
Jan 29 13:28:21 Desktop27 bluetoothd[1097]: src/device.c:device_set_wake_support() Unable to set wake_support without RPA resolution
Jan 29 13:28:21 Desktop27 systemd[1]: Started Bluetooth service.
Jan 29 13:28:21 Desktop27 bluetoothd[1097]: src/device.c:set_wake_allowed_complete() Set device flags return status: Invalid Parameters
Jan 29 13:28:46 Desktop27 bluetoothd[1097]: Endpoint registered: sender=:1.58 path=/MediaEndpoint/A2DPSink/sbc
Jan 29 13:28:46 Desktop27 bluetoothd[1097]: Endpoint registered: sender=:1.58 path=/MediaEndpoint/A2DPSource/sbc
Jan 29 13:28:46 Desktop27 bluetoothd[1097]: Endpoint registered: sender=:1.58 path=/MediaEndpoint/A2DPSink/sbc_xq_453
Jan 29 13:28:46 Desktop27 bluetoothd[1097]: Endpoint registered: sender=:1.58 path=/MediaEndpoint/A2DPSource/sbc_xq_453
Jan 29 13:28:46 Desktop27 bluetoothd[1097]: Endpoint registered: sender=:1.58 path=/MediaEndpoint/A2DPSink/sbc_xq_512
Jan 29 13:28:46 Desktop27 bluetoothd[1097]: Endpoint registered: sender=:1.58 path=/MediaEndpoint/A2DPSource/sbc_xq_512
Jan 29 13:28:46 Desktop27 bluetoothd[1097]: Endpoint registered: sender=:1.58 path=/MediaEndpoint/A2DPSink/sbc_xq_552
Jan 29 13:28:46 Desktop27 bluetoothd[1097]: Endpoint registered: sender=:1.58 path=/MediaEndpoint/A2DPSource/sbc_xq_552
Jan 29 13:28:46 Desktop27 bluetoothd[1097]: Endpoint registered: sender=:1.58 path=/MediaEndpoint/A2DPSink/faststream
Jan 29 13:28:46 Desktop27 bluetoothd[1097]: Endpoint registered: sender=:1.58 path=/MediaEndpoint/A2DPSource/faststream

```

**Additional Information:**
- The issue is observed with `bluez-5.72-2` version.
- The key association does not persist after each reboot.
- Logs indicate failures to load LTKs and IRKs for hci0 with error code Invalid Parameters (0x0d).
- The Bluetooth daemon starts successfully, but the key association problem arises.

**Resolution:**
- downgrading to `bluez-5.70-1` as a temporary workaround until the issue with `bluez-5.72-2` is resolved.

Vudentz, 2024-01-29:
@red-gecko27 try collecting the HCI traces with btmon, I wonder if the key type is invalid for some reason.

red-gecko27, 2024-01-29:
@Vudentz

After downgrading and subsequently upgrading with a reboot, the error appears to have been resolved. It's plausible that my package was corrupted, or there may have been an update to the installation script under Arch.

Upon reviewing the logs, it seems that the issue originated from Bluetooth daemon 5.72. Before the manipulation, the Endpoint registration entries were inconsistent (`sender=:1.58`). However, after the re-installation, the entries became consistent (`sender=:1.61`).

closed 2024-01-29 by red-gecko27