bluez/bluez #736: bluez 1:5.72.r51.g0de32f67f `Failed to start bluetooth.service: Unit bluetooth.service has a bad unit file setting.`

ipaqmaster, 2024-01-31:
I cannot seem to start bluez built from git via the AUR package "bluez-git" which can be found here: https://aur.archlinux.org/packages/bluez-git. The package name hints at the commit built upon: `bluez-git-1:5.72.r51.g0de32f67f-1-x86_64.pkg.tar.zst`

If I downgrade to my most recent working version: `bluez-git-1:5.72.r8.g2ef2f122e-1-x86_64.pkg.tar.zst` I am able to start the `bluetooth` service again.

This appears to be caused by this line in `bluetooth.service`:
    ExecStart=${prefix}/bluetooth/bluetoothd

Replacing the ${prefix} variable with `/usr/lib` allows the service to start once more.

joakim-tjernlund, 2024-02-02:
Just found the exact same error in git

joakim-tjernlund, 2024-02-05:
I suspect this broke the install: https://github.com/bluez/bluez/commit/116524c01de45989d3e17b3a8b29226458a57d1e

@evelikov-work , any ideas?

joakim-tjernlund, 2024-02-07:
Are we the only ones having this problem?

joakim-tjernlund, 2024-02-07:
configure log for Gentoo, same as before:
```
./configure --prefix=/usr --build=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --mandir=/usr/share/man --infodir=/usr/share/info --datadir=/usr/share --sysconfdir=/etc --localstatedir=/var/lib --datarootdir=/usr/share --disable-dependency-tracking --disable-silent-rules --disable-static --docdir=/usr/share/doc/bluez-9999 --htmldir=/usr/share/doc/bluez-9999/html --with-sysroot=/ --libdir=/usr/lib64 --localstatedir=/var --disable-android --enable-datafiles --enable-optimization --disable-debug --enable-pie --enable-threads --enable-library --enable-tools --enable-manpages --enable-monitor --with-systemdsystemunitdir=/lib/systemd/system --with-systemduserunitdir=/usr/lib/systemd/user --disable-btpclient --disable-external-ell --enable-cups --enable-deprecated --enable-experimental --enable-mesh --enable-external-ell --disable-midi --enable-obex --enable-client --enable-systemd --disable-test --enable-udev --enable-hid2hci --enable-sixaxis
configure: loading site script /usr/share/config.site
configure: loading site script /usr/share/config.site.d/80crossdev.conf
checking for a BSD-compatible install... /usr/lib/portage/python3.10/ebuild-helpers/xattr/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to create a pax tar archive... gnutar
checking whether make supports nested variables... (cached) yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for x86_64-pc-linux-gnu-pkg-config... /usr/bin/x86_64-pc-linux-gnu-pkg-config
checking pkg-config is at least version 0.9.0... yes
checking whether make supports the include directive... yes (GNU style)
checking for x86_64-pc-linux-gnu-gcc... x86_64-pc-linux-gnu-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether x86_64-pc-linux-gnu-gcc accepts -g... yes
checking for x86_64-pc-linux-gnu-gcc option to enable C11 features... none needed
checking whether x86_64-pc-linux-gnu-gcc understands -c and -o together... yes
checking dependency style of x86_64-pc-linux-gnu-gcc... none
checking for C/C++ restrict keyword... __restrict__
checking for x86_64-pc-linux-gnu-gcc... (cached) x86_64-pc-linux-gnu-gcc
checking whether the compiler supports GNU C... (cached) yes
checking whether x86_64-pc-linux-gnu-gcc accepts -g... (cached) yes
checking for x86_64-pc-linux-gnu-gcc option to enable C11 features... (cached) none needed
checking whether x86_64-pc-linux-gnu-gcc understands -c and -o together... (cached) yes
checking dependency style of x86_64-pc-linux-gnu-gcc... (cached) none
checking whether x86_64-pc-linux-gnu-gcc accepts -fPIE... yes
checking whether x86_64-pc-linux-gnu-gcc accepts -fsanitize=address... yes
checking whether x86_64-pc-linux-gnu-gcc accepts -fsanitize=leak... yes
checking whether x86_64-pc-linux-gnu-gcc accepts -fsanitize=undefined... yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by x86_64-pc-linux-gnu-gcc... /usr/x86_64-pc-linux-gnu/bin/ld
checking if the linker (/usr/x86_64-pc-linux-gnu/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/x86_64-pc-linux-gnu-nm -B
checking the name lister (/usr/bin/x86_64-pc-linux-gnu-nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/x86_64-pc-linux-gnu/bin/ld option to reload object files... -r
checking for x86_64-pc-linux-gnu-file... no
checking for file... file
checking for x86_64-pc-linux-gnu-objdump... x86_64-pc-linux-gnu-objdump
checking how to recognize dependent libraries... pass_all
checking for x86_64-pc-linux-gnu-dlltool... x86_64-pc-linux-gnu-dlltool
checking how to associate runtime and link libraries... printf %s\n
checking for x86_64-pc-linux-gnu-ar... x86_64-pc-linux-gnu-ar
checking for archiver @FILE support... @
checking for x86_64-pc-linux-gnu-strip... x86_64-pc-linux-gnu-strip
checking for x86_64-pc-linux-gnu-ranlib... x86_64-pc-linux-gnu-ranlib
checking command to parse /usr/bin/x86_64-pc-linux-gnu-nm -B output from x86_64-pc-linux-gnu-gcc object... ok
checking for sysroot... /
checking for a working dd... /bin/dd
checking how to truncate binary pipes... /bin/dd bs=4096 count=1
checking for x86_64-pc-linux-gnu-mt... no
[... 43 lines cut ...]
checking D-Bus system bus services dir... /usr/share/dbus-1/system-services
checking D-Bus session bus services dir... /usr/share/dbus-1/services
/usr/share/zsh/site-functions
checking for libudev >= 196... yes
checking udev directory... /lib/udev
checking for json-c >= 0.13... yes
[... cut ...]
configure: creating ./config.status
config.status: creating lib/bluez.pc
config.status: creating Makefile
config.status: creating mesh/bluetooth-meshd.rst
config.status: creating mesh/bluetooth-mesh.service
config.status: creating obexd/src/obex.service
config.status: creating src/bluetoothd.rst
config.status: creating src/bluetooth.service
config.status: creating tools/bluetooth-logger.service
config.status: creating config.h
config.status: executing depfiles commands
config.status: executing libtool commands
```

joakim-tjernlund, 2024-02-07:
After compile there is ./src/bluetooth.service with:
```
[Unit]
Description=Bluetooth service
Documentation=man:bluetoothd(8)
ConditionPathIsDirectory=/sys/class/bluetooth

[Service]
Type=dbus
BusName=org.bluez
ExecStart=${prefix}/bluetooth/bluetoothd
NotifyAccess=main
#WatchdogSec=10
#Restart=on-failure
CapabilityBoundingSet=CAP_NET_ADMIN CAP_NET_BIND_SERVICE
LimitNPROC=1

# Filesystem lockdown
ProtectHome=true
ProtectSystem=strict
PrivateTmp=true
ProtectKernelTunables=true
ProtectControlGroups=true
StateDirectory=bluetooth
StateDirectoryMode=0700
ConfigurationDirectory=bluetooth
ConfigurationDirectoryMode=0555

# Execute Mappings
MemoryDenyWriteExecute=true

# Privilege escalation
NoNewPrivileges=true

# Real-time
RestrictRealtime=true

[Install]
WantedBy=bluetooth.target
Alias=dbus-org.bluez.service
```

gabriele2000, 2024-02-08:
I'm using Pop_Os 22.04, which is based on Ubuntu 22.04 with patches, improvements and whatnot.

In my case, commit https://github.com/bluez/bluez/commit/d994c9d45dad68bc060add77642fdc37834cd7c1 is the last one that I can build and use.
If I build the next one, since it changed build stuff, the bluetooth adapter doesn't even show up.

Since I'd like to retain some latest fixes, I'm not gonna use the latest release version until another one drops, since 5.72 doesn't have important stuff: there's a lot of latency on 5.72, somehow a newer series of commits fixed that.

My network card is the "Intel AX210" (I have the Intel Killer version of it, but since it uses that chipset, take it as the same thing).

AtelierMizumi, 2024-02-09:
I can confirm I'm facing the same issue. [This commit](https://github.com/bluez/bluez/commit/116524c01de45989d3e17b3a8b29226458a57d1e)  changed the systemd config for bluetooth.service and it's been broken since then

gabriele2000, 2024-02-09:
> if I add --exec_prefix=/usr/libexec to configure I get a working bluetooth.service file

Confirmed.
My complete ./configure options are: `./configure --prefix=/usr --mandir=/usr/share/man --sysconfdir=/etc --localstatedir=/var --exec_prefix=/usr/libexec`

joakim-tjernlund, 2024-02-10:
Adding to configure:
```
--exec-prefix=/usr  --libexecdir=/usr/libexec
```
and using libexecdir in service files:
```
	#fix systemd service files to match intall path
	sed -i 's/@exec_prefix@/@libexecdir@/g' "${S}"/mesh/bluetooth-mesh.service.in
	sed -i 's/@exec_prefix@/@libexecdir@/g' "${S}"/obexd/src/obex.service.in
	sed -i 's/@exec_prefix@/@libexecdir@/g' "${S}"/src/bluetooth.service.in
	sed -i 's/@exec_prefix@/@libexecdir@/g' "${S}"/tools/bluetooth-logger.service.in
```
Fixed the build for me on Gentoo

One could argue autoconf should have done that for you already. Perhaps one can add something to configure.ac
to make that happen.

joakim-tjernlund, 2024-02-10:
Perhaps libexecdir=${prefix}/libexec should be added to lib/bluez.pc as well ?

joakim-tjernlund, 2024-02-10:
autoconf manual, https://www.gnu.org/savannah-checkouts/gnu/autoconf/manual/autoconf-2.72/autoconf.html#Installation-Directory-Variables
has this to say:
```
This includes the note:

   "... you should not use these variables except in makefiles."

and

   "... you should not rely on AC_CONFIG_FILES to replace bindir and
    friends in your shell scripts and other files; instead, let make
    manage their replacement.

along with examples of make rules to do it properly.
```

joakim-tjernlund, 2024-02-15:
These 3 patches is what I think is needed to fix the install in current git.
Note that one needs to specify --exec-prefix and --libexecdir to configure in order to get
working .service files.

```
From 34c67f1ba68c489addcc756ac5f632eaffdc9c19 Mon Sep 17 00:00:00 2001
From: Joakim Tjernlund <Joakim.Tjernlund@infinera.com>
Date: Sat, 10 Feb 2024 15:43:15 +0100
Subject: [PATCH 1/3] Replace @exec_prefix@ with @libexecdir@ in .service
 files.

The programs are installed with libexecdir prefix, make .service files
match that.

---
 mesh/bluetooth-mesh.service.in      | 2 +-
 obexd/src/obex.service.in           | 2 +-
 obexd/src/org.bluez.obex.service.in | 2 +-
 src/bluetooth.service.in            | 2 +-
 tools/bluetooth-logger.service.in   | 2 +-
 5 files changed, 5 insertions(+), 5 deletions(-)

diff --git a/mesh/bluetooth-mesh.service.in b/mesh/bluetooth-mesh.service.in
index 9c3ff01a3..d82043020 100644
--- a/mesh/bluetooth-mesh.service.in
+++ b/mesh/bluetooth-mesh.service.in
@@ -5,7 +5,7 @@ ConditionPathIsDirectory=/sys/class/bluetooth
 [Service]
 Type=dbus
 BusName=org.bluez.mesh
-ExecStart=@exec_prefix@/bluetooth/bluetooth-meshd
+ExecStart=@libexecdir@/bluetooth/bluetooth-meshd
 NotifyAccess=main
 LimitNPROC=1
 ProtectHome=true
diff --git a/obexd/src/obex.service.in b/obexd/src/obex.service.in
index 03d09a679..614ac090e 100644
--- a/obexd/src/obex.service.in
+++ b/obexd/src/obex.service.in
@@ -4,7 +4,7 @@ Description=Bluetooth OBEX service
 [Service]
 Type=dbus
 BusName=org.bluez.obex
-ExecStart=@exec_prefix@/bluetooth/obexd
+ExecStart=@libexecdir@/bluetooth/obexd
 
 [Install]
 Alias=dbus-org.bluez.obex.service
diff --git a/obexd/src/org.bluez.obex.service.in b/obexd/src/org.bluez.obex.service.in
index 14c16d3e3..e53e3f7fa 100644
--- a/obexd/src/org.bluez.obex.service.in
+++ b/obexd/src/org.bluez.obex.service.in
@@ -1,4 +1,4 @@
 [D-BUS Service]
 Name=org.bluez.obex
-Exec=@exec_prefix@/bluetooth/obexd
+Exec=@libexecdir@/bluetooth/obexd
 SystemdService=dbus-org.bluez.obex.service
diff --git a/src/bluetooth.service.in b/src/bluetooth.service.in
index 63e157587..356b85056 100644
--- a/src/bluetooth.service.in
+++ b/src/bluetooth.service.in
@@ -6,7 +6,7 @@ ConditionPathIsDirectory=/sys/class/bluetooth
 [Service]
 Type=dbus
 BusName=org.bluez
-ExecStart=@exec_prefix@/bluetooth/bluetoothd
+ExecStart=@libexecdir@/bluetooth/bluetoothd
 NotifyAccess=main
 #WatchdogSec=10
 #Restart=on-failure
diff --git a/tools/bluetooth-logger.service.in b/tools/bluetooth-logger.service.in
index 5657c8d84..31fabe8d3 100644
--- a/tools/bluetooth-logger.service.in
+++ b/tools/bluetooth-logger.service.in
@@ -4,7 +4,7 @@ ConditionPathIsDirectory=/sys/class/bluetooth
 
 [Service]
 Type=simple
-ExecStart=@exec_prefix@/bluetooth/btmon-logger -p -b /var/log/bluetooth/hci.log
+ExecStart=@libexecdir@/bluetooth/btmon-logger -p -b /var/log/bluetooth/hci.log
 NotifyAccess=main
 CapabilityBoundingSet=CAP_NET_RAW
 LimitNPROC=1
-- 
2.43.0
```

```
From f91dd4f2dd5c7f9c48135d09196be6d6e9960463 Mon Sep 17 00:00:00 2001
From: Joakim Tjernlund <Joakim.Tjernlund@infinera.com>
Date: Thu, 15 Feb 2024 09:41:46 +0100
Subject: [PATCH 2/3] bluez.pc.in: Add definition of libexecdir

---
 lib/bluez.pc.in | 1 +
 1 file changed, 1 insertion(+)

diff --git a/lib/bluez.pc.in b/lib/bluez.pc.in
index 3d6e59616..1c2f30b33 100644
--- a/lib/bluez.pc.in
+++ b/lib/bluez.pc.in
@@ -1,6 +1,7 @@
 prefix=@prefix@
 exec_prefix=@exec_prefix@
 libdir=@libdir@
+libexecdir=@libexecdir@
 includedir=@includedir@
  
 Name: BlueZ
-- 
2.43.0
```

```
From 443565970e8133b2e0df54f038a805e334eabe4f Mon Sep 17 00:00:00 2001
From: Joakim Tjernlund <Joakim.Tjernlund@infinera.com>
Date: Thu, 15 Feb 2024 10:03:27 +0100
Subject: [PATCH 3/3] bootstrap-configure: specfiy exec-prefix and libexecdir

---
 bootstrap-configure | 2 ++
 1 file changed, 2 insertions(+)

diff --git a/bootstrap-configure b/bootstrap-configure
index 1c26909b9..d927ff9c6 100755
--- a/bootstrap-configure
+++ b/bootstrap-configure
@@ -8,6 +8,8 @@ fi
     ./configure --enable-maintainer-mode \
 		--enable-debug \
 		--prefix=/usr \
+		--exec-prefix=/usr \
+		--libexecdir=/usr/libexec \
 		--mandir=/usr/share/man \
 		--sysconfdir=/etc \
 		--localstatedir=/var \
-- 
2.43.0
```

evelikov-work, 2024-02-15:
Hey, sorry for the delay. This account is purely for corporate purposes - please use @evelikov going forward.

I am quite busy today/tomorrow although I will try to get back to this by Monday latest.

evelikov-work, 2024-02-15:
Or I can simply delete this account - both actions will result in equally unhappy corporate overlords :smiling_imp: 

Joking aside, thanks for the report and sorry for the breakage. I've tracked down why the commit worked on my end and will send a fix shortly.

joakim-tjernlund, 2024-02-15:
> Joking aside, thanks for the report and sorry for the breakage. I've tracked down why the commit worked on my end and will send a fix shortly.

Please look at the patches I have provided here. I think they are good

joakim-tjernlund, 2024-02-16:
I sent my patches to the list too

joakim-tjernlund, 2024-02-20:
Did a quick build test WITHOUT mesh and see I see:
```
org.bluez.service
[D-BUS Service]
Name=org.bluez
Exec=/bin/false
User=root
SystemdService=dbus-org.bluez.service
```

joakim-tjernlund, 2024-02-20:
> Is there anything wrong with the `org.bluez.service` file? If so, please document your observations and expectations in a separate issue.
> 
> If you see anything wrong within the following files, let me know. I'll be happy to help. Otherwise I think we can close this.
> 
> ```
>  mesh/bluetooth-mesh.service.in
>  obexd/src/obex.service.in
>  obexd/src/org.bluez.obex.service.in
>  src/bluetooth.service.in
>  tools/bluetooth-logger.service.in
> ```

I noticed the Exec=/bin/false in some .service files after install:
```
grep -r /bin/false -r
usr/share/dbus-1/system-services/org.bluez.service:Exec=/bin/false
usr/share/dbus-1/system-services/org.bluez.mesh.service:Exec=/bin/false
```
Just odd and I don't know if it should be so or what caused it.

joakim-tjernlund, 2024-02-21:
> > I noticed the Exec=/bin/false in some .service files after install
> 
> Those are preexisting and expected.
> 
> Tl:Dr: is that majority of dbus services have moved from being explicitly managed* by dbus (hence the `Exec=/bin/false`) to forwarding the service management to systemd (aka SystemdService=...).
> 
> For more details you can see the dbus spec and also cross reference with the existing dbus services on your system*.
> 
> This is orthogonal to the original issue and deserves it's own issue/discussion.

OK, so I guess building with --disable-systemd should replace these /bin/false with the rigth thing.

> 
> @ipaqmaster @joakim-tjernlund - can you please confirm if git works for you? Just tried the AUR `bluez-git` package and it just works over here.

Yes, it works for me too. I still add libexec to configure though.

joakim-tjernlund, 2024-02-25:
> @joakim-tjernlund as mentioned previously - please open a separate issue with the `Exec=/bin/false` concerns.

https://github.com/bluez/bluez/issues/753

closed 2024-03-07 by BluezTestBot

gabriele2000, 2025-12-19:
> worst package ever. crashed my whole system

While complaining is free, writing a complaint takes time.
Could you please dedicate a bit more of your time to describe what issue occurred, so that a useful comment can trigger my inbox notification next time?