bluez/bluez #737: the first BT keyboard keypress to wake up PC is eaten

bam80, 2024-02-06:
This long standing issue is driving me nuts.

On the BT-paired keyboard, the fist keypress is always gets "eaten" if PC is asleep.

With the same hardware, the behavior doesn't reproduce on Windows and Android - the host wakes up upon the key tap _and_ the very first keystroke is handled correctly there.

This bug influences usability of any BT keyboard on Linux very badly.

Vudentz, 2024-02-06:
> This long standing issue is driving me nuts.
> 
> On the BT-paired keyboard, the fist keypress is always gets "eaten" if PC is asleep.
> 
> With the same hardware, the behavior doesn't reproduce on Windows and Android - the host wakes up upon the key tap _and_ the very first keystroke is handled correctly there.
> 
> This bug influences usability of any BT keyboard on Linux very badly.
> 
> PS: IIRC, the problem is also reproduces even when the PC is not asleep, but the BT keyboard itself is falling asleep. Then, e.g. in terminal, you can't be assure your first keypress will appear there if there was pause between the keystrokes. This problem is even worse and again, it doesn't reproduce on Windows/Android.

Ok, does it gets "eaten" by what layer is the question, perhaps have a btmon trace so we can see it is really received, also it could be perhaps that is discarded because of the now infamous race between USB endpoints where the connection event is processed after the data packet so it is discarded because its handle cannot be found.

bam80, 2024-02-08:
> perhaps have a btmon trace so we can see it is really received

It does, here is the trace.

The issue happens after my keyboard disconnects on timeout 10 min. This is powersaving feature present on all BT keyboards I have (the delay may vary).
The Disconnection Request:
```
> ACL Data RX: Handle 256 flags 0x02 dlen 12                                                                                                                                                                                          #254 [hci0] 2241.727478
      L2CAP: Disconnection Request (0x06) ident 23 len 4
        Destination CID: 65
        Source CID: 72
> ACL Data RX: Handle 256 flags 0x02 dlen 12                                                                                                                                                                                          #255 [hci0] 2241.727481
      L2CAP: Disconnection Request (0x06) ident 24 len 4
        Destination CID: 64
        Source CID: 71
< ACL Data TX: Handle 256 flags 0x00 dlen 12                                                                                                                                                                                          #256 [hci0] 2241.727564
      L2CAP: Disconnection Response (0x07) ident 23 len 4
        Destination CID: 65
        Source CID: 72
< ACL Data TX: Handle 256 flags 0x00 dlen 12                                                                                                                                                                                          #257 [hci0] 2241.727576
      L2CAP: Disconnection Response (0x07) ident 24 len 4
        Destination CID: 64
        Source CID: 71
< HCI Command: Exit Sniff Mode (0x02|0x0004) plen 2                                                                                                                                                                                   #258 [hci0] 2241.727582
        Handle: 256 Address: 7C:1E:52:A8:89:77 (Microsoft)
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                                                                             #259 [hci0] 2241.729043
      Exit Sniff Mode (0x02|0x0004) ncmd 1
        Status: Success (0x00)
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                                                                                #260 [hci0] 2241.732089
        Num handles: 1
        Handle: 256 Address: 7C:1E:52:A8:89:77 (Microsoft)
        Count: 1
        Latency: 4 msec (3-6 msec ~4 msec)
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                                                                                #261 [hci0] 2241.734091
        Num handles: 1
        Handle: 256 Address: 7C:1E:52:A8:89:77 (Microsoft)
        Count: 1
        Latency: 6 msec (3-6 msec ~5 msec)
> HCI Event: Mode Change (0x14) plen 6                                                                                                                                                                                                #262 [hci0] 2242.137203
        Status: Success (0x00)
        Handle: 256 Address: 7C:1E:52:A8:89:77 (Microsoft)
        Mode: Active (0x00)
        Interval: 0.000 msec (0x0000)
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                                                                                        #263 [hci0] 2242.267164
        Status: Success (0x00)
        Handle: 256 Address: 7C:1E:52:A8:89:77 (Microsoft)
        Reason: Remote User Terminated Connection (0x13)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                                                                                                 {0x0001} [hci0] 2242.267227
        BR/EDR Address: 7C:1E:52:A8:89:77 (Microsoft)
        Reason: Connection terminated by remote host (0x03)
```
Note "Reason: Remote User Terminated Connection" above.

So the keyboard disconnected itself on timeout
and now I press and hold "a" key:
```
> HCI Event: Connect Request (0x04) plen 10                                                                                                                                                                                           #264 [hci0] 2275.993162
        Address: 7C:1E:52:A8:89:77 (Microsoft)
        Class: 0x002540
          Major class: Peripheral (mouse, joystick, keyboards)
          Minor class: 0x10
          Limited Discoverable Mode
        Link type: ACL (0x01)
< HCI Command: Accept Connection Request (0x01|0x0009) plen 7                                                                                                                                                                         #265 [hci0] 2275.993313
        Address: 7C:1E:52:A8:89:77 (Microsoft)
        Role: Central (0x00)
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                                                  [... cut ...]
aaaaaaaaaaaaaaaaaaaaaaaaaaaa> ACL Data RX: Handle 256 flags 0x02 dlen 14                                                                                                                                                                                          
#316 [hci0] 2315.479398
      Channel: 65 len 10 [PSM 19 mode Basic (0x00)] {chan 1}
        a1 01 00 00 00 00 00 00 00 00                    ..........      
```
We see the "a" key press event and this time many "a" chars in terminal were received, normally!
After all, I released the key and the corresponding event followed.

bam80, 2024-02-09:
> @bam80 try if evtest is able to capture the first key

It seems evtest doesn't allow me to do that,
as soon as the keyboard disconnects, get an error:
```
evtest: error reading: No such device
```

> Btw, what kernel version and BlueZ are you testing with?

```sh
$ uname -a
Linux host 6.5.0-15-generic #15-Ubuntu SMP PREEMPT_DYNAMIC Tue Jan  9 17:03:36 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

$ bluetoothd -v
5.68

bam80, 2024-02-09:
@Vudentz would you be able to reproduce the issue on your side maybe?

Vudentz, 2024-02-09:
> @Vudentz would you be able to reproduce the issue on your side maybe?

```
sudo evtest
Select the device event number [0-18]: 18
Input driver version is 1.0.1
Input device ID: bus 0x5 vendor 0x46d product 0xb025 version 0x14
Input device name: "Logitech MX Anywhere 3"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 272 (BTN_LEFT)
    Event code 273 (BTN_RIGHT)
    Event code 274 (BTN_MIDDLE)
    Event code 275 (BTN_SIDE)
    Event code 276 (BTN_EXTRA)
    Event code 277 (BTN_FORWARD)
    Event code 278 (BTN_BACK)
    Event code 279 (BTN_TASK)
    Event code 280 (?)
    Event code 281 (?)
    Event code 282 (?)
    Event code 283 (?)
    Event code 284 (?)
    Event code 285 (?)
    Event code 286 (?)
    Event code 287 (?)
  Event type 2 (EV_REL)
    Event code 0 (REL_X)
    Event code 1 (REL_Y)
    Event code 6 (REL_HWHEEL)
    Event code 8 (REL_WHEEL)
    Event code 11 (REL_WHEEL_HI_RES)
    Event code 12 (REL_HWHEEL_HI_RES)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
Event: time 1707426255.722758, type 2 (EV_REL), code 1 (REL_Y), value -1
Event: time 1707426255.722758, -------------- SYN_REPORT ------------
Event: time 1707426255.728893, type 2 (EV_REL), code 0 (REL_X), value 1
Event: time 1707426255.728893, type 2 (EV_REL), code 1 (REL_Y), value -1
```

bam80, 2024-02-09:
@Vudentz right, but when your mouse goes asleep (does it sleep?), you should get the same error I mentioned, and `evtest` quits as there is no more event device it was tracking.

Vudentz, 2024-02-09:
> @Vudentz right, but when your mouse goes asleep (does it sleep?), you should get the same error I mentioned, and `evtest` quits as there is no more event device it was tracking.

Right, you need to capture it from the very start so it would need to work even before it is connected. I guess we could add a hack to keep the device node around for some time before the device reconnects, that way you could capture it, but it will probably requires changes and chances are that if we keep the device node in place it perhaps wouldnt manifest the problem anymore, that said perhaps that would be a workaround until we figure out where the report is being dropped.

bam80, 2024-02-09:
> I guess we could add a hack to keep the device node around for some time before the device reconnects

Oh, that would be awesome to investigate! I'm in situation when any workaround would help, as I just can't use my PC normally.

Luiz, I would be very appreciated for any help towards the solving. It's just not much I can do myself..

Vudentz, 2024-02-09:
```
diff --git a/profiles/input/hog-lib.c b/profiles/input/hog-lib.c
index 7ff1ede3db35..67492a63eca3 100644
--- a/profiles/input/hog-lib.c
+++ b/profiles/input/hog-lib.c
@@ -1309,11 +1309,35 @@ static bool cancel_gatt_req(const void *data, const void *user_data)
        return g_attrib_cancel(hog->attrib, req->id);
 }
 
+static void uhid_destroy(struct bt_hog *hog)
+{
+       int err;
+       struct uhid_event ev;
+
+       if (!hog->uhid_created)
+               return;
+
+       bt_uhid_unregister_all(hog->uhid);
+
+       memset(&ev, 0, sizeof(ev));
+       ev.type = UHID_DESTROY;
+
+       err = bt_uhid_send(hog->uhid, &ev);
+
+       if (err < 0) {
+               error("bt_uhid_send: %s", strerror(-err));
+               return;
+       }
+
+       hog->uhid_created = false;
+}
+
 static void hog_free(void *data)
 {
        struct bt_hog *hog = data;
 
        bt_hog_detach(hog);
+       uhid_destroy(hog);
 
        queue_destroy(hog->input, free);
        queue_destroy(hog->bas, (void *) bt_bas_unref);
@@ -1823,29 +1847,6 @@ bool bt_hog_attach(struct bt_hog *hog, void *gatt)
        return true;
 }
 
-static void uhid_destroy(struct bt_hog *hog)
-{
-       int err;
-       struct uhid_event ev;
-
-       if (!hog->uhid_created)
-               return;
-
-       bt_uhid_unregister_all(hog->uhid);
-
-       memset(&ev, 0, sizeof(ev));
-       ev.type = UHID_DESTROY;
-
-       err = bt_uhid_send(hog->uhid, &ev);
-
-       if (err < 0) {
-               error("bt_uhid_send: %s", strerror(-err));
-               return;
-       }
-
-       hog->uhid_created = false;
-}
-
 void bt_hog_detach(struct bt_hog *hog)
 {
        GSList *l;
@@ -1879,7 +1880,6 @@ void bt_hog_detach(struct bt_hog *hog)
        queue_remove_all(hog->gatt_op, cancel_gatt_req, hog, destroy_gatt_req);
        g_attrib_unref(hog->attrib);
        hog->attrib = NULL;
-       uhid_destroy(hog);
 }
 
 int bt_hog_set_control_point(struct bt_hog *hog, bool suspend)
```

bam80, 2024-02-09:
Hmm, it seems I still have the event device disappearing on the disconnect..

What I did:
- Build the PR, stop system BT daemon and run the new one:
```sh
   git clone https://github.com/tedd-an/bluez-upstream-test.git
   cd bluez-upstream-test/
   gh pr checkout 567
   ./bootstrap
   ./configure
   make
   sudo systemctl disable bluetooth.service --now
   sudo src/bluetoothd -d
# connect the keyboard..
```
- In the other terminal, run `evtest` and wait for the timeout..
```
$ sudo evtest 
[sudo] password for bam: 
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Power Button
/dev/input/event1:	Logitech Wireless Keyboard PID:4023
/dev/input/event2:	Logitech Wireless Mouse PID:4022
/dev/input/event3:	cros_ec_buttons
/dev/input/event4:	Microsoft Bluetooth Mobile Keyboard 6000 Keyboard
/dev/input/event5:	C-Media Electronics Inc. USB Audio Device
Select the device event number [0-5]: 4
Input driver version is 1.0.1
Input device ID: bus 0x5 vendor 0x45e product 0x762 version 0x13
Input device name: "Microsoft Bluetooth Mobile Keyboard 6000 Keyboard"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 1 (KEY_ESC)
...
    Event code 613 (KEY_KBDINPUTASSIST_CANCEL)
  Event type 2 (EV_REL)
    Event code 6 (REL_HWHEEL)
    Event code 12 (REL_HWHEEL_HI_RES)
  Event type 3 (EV_ABS)
    Event code 32 (ABS_VOLUME)
      Value      0
      Min        0
      Max     1023
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
  Event type 17 (EV_LED)
    Event code 0 (LED_NUML) state 0
    Event code 1 (LED_CAPSL) state 0
    Event code 2 (LED_SCROLLL) state 0
Key repeat handling:
  Repeat type 20 (EV_REP)
    Repeat code 0 (REP_DELAY)
      Value    250
    Repeat code 1 (REP_PERIOD)
      Value     33
Properties:
Testing ... (interrupt to exit)
Event: time 1707496217.803341, type 17 (EV_LED), code 0 (LED_NUML), value 1
Event: time 1707496217.803341, -------------- SYN_REPORT ------------
Event: time 1707496227.243530, type 17 (EV_LED), code 0 (LED_NUML), value 0
Event: time 1707496227.243530, -------------- SYN_REPORT ------------
Event: time 1707496238.131889, type 17 (EV_LED), code 0 (LED_NUML), value 1
Event: time 1707496238.131889, -------------- SYN_REPORT ------------
expected 24 bytes, got -1

evtest: error reading: No such device
```

bam80, 2024-02-09:
System log on the disconnection, if it's useful:
```
...
фев 09 20:17:01 host CRON[428937]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
фев 09 20:17:01 host CRON[428938]: (root) CMD (cd / && run-parts --report /etc/cron.hourly)
фев 09 20:17:01 host CRON[428937]: pam_unix(cron:session): session closed for user root
фев 09 20:17:20 host bluetoothd[428475]: profiles/input/device.c:ctrl_watch_cb() Device 7C:1E:52:A8:89:77 disconnected
фев 09 20:17:20 host bluetoothd[428475]: profiles/input/device.c:intr_watch_cb() Device 7C:1E:52:A8:89:77 disconnected
фев 09 20:17:20 host bluetoothd[428475]: src/service.c:change_state() 0x5612e2eb9a40: device 7C:1E:52:A8:89:77 profile input-hid state changed: connected -> disconnected (0)
фев 09 20:17:20 host bluetoothd[428475]: profiles/input/device.c:input_device_enter_reconnect_mode() path=/org/bluez/hci0/dev_7C_1E_52_A8_89_77 reconnect_mode=device
фев 09 20:17:20 host sudo[428619]: pam_unix(sudo:session): session closed for user root
фев 09 20:17:21 host bluetoothd[428475]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x000c
фев 09 20:17:21 host bluetoothd[428475]: src/adapter.c:dev_disconnected() Device 7C:1E:52:A8:89:77 disconnected, reason 3
фев 09 20:17:21 host bluetoothd[428475]: src/adapter.c:adapter_remove_connection()
фев 09 20:17:21 host bluetoothd[428475]: plugins/policy.c:disconnect_cb() reason 3
фев 09 20:17:21 host bluetoothd[428475]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 7C:1E:52:A8:89:77 type 0 status 0xe
фев 09 20:17:21 host bluetoothd[428475]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
фев 09 20:17:21 host bluetoothd[428475]: src/device.c:device_bonding_failed() status 14
фев 09 20:17:21 host bluetoothd[428475]: src/adapter.c:resume_discovery()

Vudentz, 2024-02-09:
> System log on the disconnection, if it's useful:
>
> ```
> ...
> фев 09 20:17:01 host CRON[428937]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
> фев 09 20:17:01 host CRON[428938]: [...]

Oh, so it is Classic, not LE peripheral, yeah that is completely different as it does use hidp rather than hog, well I can probably do something similar to Classic but you will need to set UserspaceHID=true in input.conf, we have been considering that for a while since hidp being in the kernel doesn't buy us much and is a lot more complicated, anyway try changing that first to see if that changes anything in the behavior.

bam80, 2024-02-09:
> well I can probably do something similar to Classic but you will need to set UserspaceHID=true in input.conf, we have been considering that for a while since hidp being in the kernel doesn't buy us [...]

I tried to restart the bluetooth systemd service with `UserspaceHID=true` but the behavior is the same.
So as I understand more changes are needed to cover the Classic BT?

Vudentz, 2024-02-09:
> I tried to restart the bluetooth systemd service with `UserspaceHID=true` but the behavior is the same. So as I understand more changes are needed to cover the Classic BT?

Ive uploaded a new set (v2) try with that instead.

bam80, 2024-02-09:
So I retested with the new patchset but the result is the same:
```
фев 10 01:51:10 Feedme bluetoothd[634232]: profiles/input/device.c:intr_watch_cb() Device 7C:1E:52:A8:89:77 disconnected
фев 10 01:51:10 Feedme bluetoothd[634232]: src/service.c:change_state() 0x564c4e359180: device 7C:1E:52:A8:89:77 profile input-hid state changed: connected -> disconnected (0)
фев 10 01:51:10 Feedme bluetoothd[634232]: profiles/input/device.c:input_device_enter_reconnect_mode() path=/org/bluez/hci0/dev_7C_1E_52_A8_89_77 reconnect_mode=device
фев 10 01:51:10 Feedme rtkit-daemon[2093]: Supervising 13 threads of 10 processes of 1 users.
фев 10 01:51:10 Feedme rtkit-daemon[2093]: Supervising 13 threads of 10 processes of 1 users.
фев 10 01:51:10 Feedme acpid[1647]: input device has been disconnected, fd 20
фев 10 01:51:10 Feedme sudo[634423]: pam_unix(sudo:session): session closed for user root
фев 10 01:51:10 Feedme bluetoothd[634232]: profiles/input/device.c:ctrl_watch_cb() Device 7C:1E:52:A8:89:77 disconnected
фев 10 01:51:10 Feedme bluetoothd[634232]: src/shared/att.c:can_read_data() (chan 0x564c4e365880) ATT PDU received: 0x1b
фев 10 01:51:11 Feedme bluetoothd[634232]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x000c
фев 10 01:51:11 Feedme bluetoothd[634232]: src/adapter.c:dev_disconnected() Device 7C:1E:52:A8:89:77 disconnected, reason 3
фев 10 01:51:11 Feedme bluetoothd[634232]: src/adapter.c:adapter_remove_connection()
фев 10 01:51:11 Feedme bluetoothd[634232]: plugins/policy.c:disconnect_cb() reason 3
фев 10 01:51:11 Feedme bluetoothd[634232]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 7C:1E:52:A8:89:77 type 0 status 0xe
фев 10 01:51:11 Feedme bluetoothd[634232]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
фев 10 01:51:11 Feedme bluetoothd[634232]: src/device.c:device_bonding_failed() status 14
фев 10 01:51:11 Feedme bluetoothd[634232]: src/adapter.c:resume_discovery()
```
`evtest`:
```
...
Testing ... (interrupt to exit)
expected 24 bytes, got -1

evtest: error reading: No such device
```

bam80, 2024-02-12:
Hi Luiz, any idea what's wrong?

Vudentz, 2024-02-12:
> Hi Luiz, any idea what's wrong?

Try with v3, looks like v2 didn't really default to use uhid because I missed the following change:

`static bool uhid_enabled = true;`

bam80, 2024-02-13:
> Try with v3, looks like v2 didn't really default to use uhid because I missed the following change:
> 
> `static bool uhid_enabled = true;`

@Vudentz I retested but still having the same result.

Could you try to reproduce on your side?
If not, I'm ready to test more patches.

bam80, 2024-02-14:
Hi Luiz, any thoughts?

Vudentz, 2024-02-14:
> Hi Luiz, any thoughts?

Ive sent a v5, the condition for not destroying the device node was wrong, this time it shall not destroy the input device on disconnection.

bam80, 2024-02-14:
> Ive sent a v5, the condition for not destroying the device node was wrong, this time it shall not destroy the input device on disconnection.

Yep that helped, thanks a lot!
Would you apply the patches upstream then?

Unfortunately, I don't have BT5 keyboard on hands but maybe we could apply the LE part blindly? Or find someone else to test it.
I have BT5 mouse, though. Not sure if it can replace the keyboard in the test.

Vudentz, 2024-02-14:
> > Ive sent a v5, the condition for not destroying the device node was wrong, this time it shall not destroy the input device on disconnection.
> 
> Yep that helped, thanks a lot! Would you apply the patches upstream then?

So this fixes the missing of the first key press?

> Unfortunately, I don't have BT5 keyboard on hands but maybe we could apply the LE part blindly? Or find someone else to test it. I have BT5 mouse, though. Not sure if it can replace the keyboard in the test.

I have LE HID (aka HoG) that I tested the first patch, so it should be alright to apply these changes.

bam80, 2024-02-14:
> So this fixes the missing of the first key press?

Yes, now the behavior is exactly the same as on Windows/Android.
The first press comes with a little delay, probably because the keyboard re-connects and such things, but the most importantly the press doesn't get lost anymore!

Actually, as I recall now the problem was not only with the first press, but for all the presses happen within the re-connection period (which could be up to several seconds) . Now it's all fixed, which just adds value to the fix.

bam80, 2024-02-14:
The only thing leaving me clueless how this could happen it wasn't fixed for a decade or more?

Vudentz, 2024-02-15:
> The only thing leaving me clueless how this could happen it wasn't fixed for a decade or more?

Most likely is a combination of people being used to it, and therefore not caring to file an issue, and BT keyboards not being that popular like mices for example, anyway it is good to finally have it fixed.

bam80, 2024-02-15:
The next step is to merge upstream and close it, right?

closed 2024-02-19 by Vudentz

bam80, 2024-02-19:
I'm not sure if the patches were upstreamed, were they?

bam80, 2024-11-13:
That has to be clarified yet, but seems things worked flawlessly up to Bluez 5.78 version, and then regressed again in 5.79.
Any idea what could go wrong?
I can assist in debugging, thanks.
@Vudentz

bam80, 2024-11-13:
> That has to be clarified yet, but seems things worked flawlessly up to Bluez 5.78 version, and then regressed again in 5.79.

I've tested and can confirm the regression is in Bluez.
Please reopen.

Vudentz, 2024-11-14:
There was too many bugs related to this behavior so we need to revert it and introduce an option, see 1782bfd791f76317a3f3ded2884606c1b5b164eb

bam80, 2024-11-14:
Thank you for informing me Luiz.

Still, I'm not sure the solution chosen is satisfyable.
I looked at the related issues, they all seem to be related to particular device models, not even a class.
Whereas the issue here reveals itself for all Bluetooth keyboards without exceptions, and I can assure you this ruins user experience with these keyboards on Linux.

Also please note Android/Windows don't have this issue and don't require any options - the keyboards just work out of the box there.

In this situation, I would consider making the `UserspaceHID=persist` the default option value, until proper solution is found.
What do you think?

Vudentz, 2024-11-14:
> Thank you for informing me Luiz.
>
> Still, I'm not sure the solution chosen is satisfyable. I looked at the related issues, they all seem to be related to particular device models, not even a class. [...]

Yeah, the fact that Android works sort of points out to userspace not picking up events for some reason, since it also uses uHID and I don't think it persists either.

> In this situation, I would consider making the `UserspaceHID=persist` the default option value, until proper solution is found. What do you think?

Problem is that many hid drivers are shared with USB which does actually disconnects, so we probably need to investigate what the heck is going on that events are lost.

bam80, 2024-11-14:
Can we reopen the issue then,
as most users are unaware of the option (and it's relation to the issue) and for them, the issue is effectively unsolved?
Thanks.

bam80, 2024-11-18:
> Does the following change makes any difference:

Apparently not

bam80, 2024-12-01:
> Oh, so it is Classic, not LE peripheral, yeah that is completely different as it does use hidp rather than hog, well I can probably do something similar to Classic but you will need to set UserspaceHID=true in input.conf

Hi Luiz,
I bout LE keyboard (BT 5.0 actually: [Gembird KBW-7](http://gembird.ru/catalog/keyboards/kits/kbw-7/)),
but apparently `UserspaceHID=persist` option doesn't work for it:
whenever the keyboard disconnects, the first press(es?) is lost.
The option does work for Classic BT keyboard, though (yes I have both keyboards now so can compare).

Any ideas?

Vudentz, 2024-12-02:
> > I bout LE keyboard (BT 5.0 actually: [Gembird KBW-7](http://gembird.ru/catalog/keyboards/kits/kbw-7/)),
> > but apparently `UserspaceHID=persist` option doesn't work for it
> 
> Is the fix intended to work for BLE (BT 5.0 particularly) also, or it just wasn't implemented for it?

It should work for LE, yes, so you are loosing the first key even with persistent? Does evtest show it?

bam80, 2024-12-03:
Neither `evtest` nor `btmon` show the first key press for the new keyboard, even in BT-3.0 mode. That's with "persistent".
So I'm kinda.. pissed off.
I didn't expect the situation with even new BT hardware is so painfully bad.
Now I have to sell it out,  but that's only fraction of the problem.

I wonder if keyboards with free firmware (QMK, ZMK, etc.) do it properly, at least?
I fed up with the surprises and will probably be targeting those keyboards, despite the price.