bluez/bluez #738: Logitech MX Anywhere 3 is not detected

mtbrobotanist, 2024-02-06:
My MX Anywhere 3 is not detected at all by bluez. Scan never shows it even though I have put it in pairing mode by holding the connect button until the light blinks rapidly. I have tried all three profiles on the mouse.

Here is all the relevant info I can think of:

Ubuntu 20.04.6 LTS   (dual boot with windows 11, works fine there)

```
$ uname -a
Linux yolo 5.15.0-91-generic #101~20.04.1-Ubuntu SMP Thu Nov 16 14:22:28 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

$ bluetoothctl --version
bluetoothctl: 5.53

$ btmgmt info
Index list with 1 item
hci0:	Primary controller
	addr 30:89:4A:28:CE:67 version 11 manufacturer 2 class 0x3c0000
	supported settings: powered connectable fast-connectable discoverable bondable link-security ssp br/edr hs le advertising secure-conn debug-keys privacy configuration static-addr 
	current settings: powered connectable discoverable bondable ssp br/edr secure-conn 
	name yolo
	short name 
hci0:	Configuration options
	supported options: public-address 
	missing options: 

$ hciconfig -a
hci0:	Type: Primary  Bus: USB
	BD Address: 30:89:4A:28:CE:67  ACL MTU: 1021:4  SCO MTU: 96:6
	UP RUNNING PSCAN ISCAN 
	RX bytes:37433855 acl:190 sco:731679 events:6952 errors:0
	TX bytes:39140311 acl:1431 sco:731557 commands:4924 errors:0
	Features: 0xbf 0xfe 0x0f 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'yolo'
	Class: 0x3c0000
	Service Classes: Rendering, Capturing, Object Transfer, Audio
	Device Class: Miscellaneous, 
	HCI Version:  (0xb)  Revision: 0x3362
	LMP Version:  (0xb)  Subversion: 0x3362
	Manufacturer: Intel Corp. (2)
```

Might be worth mentioning it doesn't seems to be detected as a usb device either. but it receives charge fine.

mtbrobotanist, 2024-02-07:
So I updated to Ubuntu 22.04.3 and the kernel to 6.4.0-060400-generic. Same problem.

mtbrobotanist, 2024-02-08:
Update: i tried a second MX Anywhere 3 and it is the same as the first: undetected by direct USB connection and undetected by bluetooth. However both mouses work when using the Logitech Unifying Receiver. I want to investigate further, but I don't know where to start. I suppose I can try the latest kernel. Any other suggestions?

Vudentz, 2024-02-08:
@mtbrobotanist Well it works just fine on fedora:

```
Select the device event number [0-18]: 18
Input driver version is 1.0.1
Input device ID: bus 0x5 vendor 0x46d product 0xb025 version 0x14
Input device name: "Logitech MX Anywhere 3"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 272 (BTN_LEFT)
    Event code 273 (BTN_RIGHT)
    Event code 274 (BTN_MIDDLE)
    Event code 275 (BTN_SIDE)
    Event code 276 (BTN_EXTRA)
    Event code 277 (BTN_FORWARD)
    Event code 278 (BTN_BACK)
    Event code 279 (BTN_TASK)
    Event code 280 (?)
    Event code 281 (?)
    Event code 282 (?)
    Event code 283 (?)
    Event code 284 (?)
    Event code 285 (?)
    Event code 286 (?)
    Event code 287 (?)
  Event type 2 (EV_REL)
    Event code 0 (REL_X)
    Event code 1 (REL_Y)
    Event code 6 (REL_HWHEEL)
    Event code 8 (REL_WHEEL)
    Event code 11 (REL_WHEEL_HI_RES)
    Event code 12 (REL_HWHEEL_HI_RES)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
Event: time 1707426255.722758, type 2 (EV_REL), code 1 (REL_Y), value -1
Event: time 1707426255.722758, -------------- SYN_REPORT ------------
Event: time 1707426255.728893, type 2 (EV_REL), code 0 (REL_X), value 1
Event: time 1707426255.728893, type 2 (EV_REL), code 1 (REL_Y), value -1
```

mtbrobotanist, 2024-02-10:
Update: The mouse specs are here: https://support.logi.com/hc/en-us/articles/360053565493-MX-Anywhere-3-Technical-Specifications
It lists "- Bluetooth Low Energy technology".

Meanwhile, I had Low Energy mode (le) disabled in my config. `/etc/bluetooth/main.conf` had `ControllerMode=bredr` instead of `dual`. Switching to dual and restarting bluetooth service fixed it.

Sorry for the trouble.

closed 2024-02-10 by mtbrobotanist

Flashwalker, 2024-09-07:
Works for me again, devices are detectable via cli again.  

I'm using logi bolt adapter now, so it works.  

Can't really say that is was helpfull, adding my user to the input group, or just a logi bolt, or some update or something else.

walmer26, 2024-10-24:
> input

I am having the same exact issue and it is very annoying having forgeting the device and re-pair because it simply wont work after boot. If trying to switch ON the connection at setting it truns back OFF automatically. 

❯ hciconfig -a
hci0:	Type: Primary  Bus: USB
	BD Address: 9C:DA:3E:B2:91:6A  ACL MTU: 1021:4  SCO MTU: 96:6
	UP RUNNING PSCAN 
	RX bytes:357789 acl:18197 sco:0 events:3458 errors:0
	TX bytes:613389 acl:460 sco:0 commands:2703 errors:0
	Features: 0xff 0xfe 0x0f 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF 
	Link mode: PERIPHERAL ACCEPT 
	Name: 'HP-Spectre'
	Class: 0x7c0000
	Service Classes: Rendering, Capturing, Object Transfer, Audio, Telephony
	Device Class: Miscellaneous, 
	HCI Version: 4.2 (0x8)  Revision: 0x100
	LMP Version: 4.2 (0x8)  Subversion: 0x100
	Manufacturer: Intel Corp. (2)

❯ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 064e:3401 Suyin Corp. HP TrueVision FHD RGB-IR
Bus 001 Device 003: ID 8087:0a2b Intel Corp. Bluetooth wireless interface
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

❯ bluetoothd -v
5.72

❯ uname -a
Linux HP-Spectre 6.8.0-47-generic #47-Ubuntu SMP PREEMPT_DYNAMIC Fri Sep 27 21:40:26 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

❯ bluetoothctl --version
bluetoothctl: 5.72