bluez/bluez #744: configuring bluez for ios/macos

Ben1980, 2024-02-12:
I currently work on a wrapper of libdivecomputer for ios/macos. This library is using bluez for the Bluetooth connection and therefore i need to configure/build bluez for ios/macs. Unfortunately i get the error `configure: error: signalfd support is required` from the config tool. From googleing i don't see any support for ios/macos for signalfd.
What would be the propper configuration setting for bluez in my case? I know that somehow it should work because there is a ios/macos build of subsurface which is using the libdivecomputer library.

Vudentz, 2024-02-12:
I assume that use qt framework and that has it internals hooked to the OS Bluetooth APIs, so in sort it doesnt use BlueZ to access the Bluetooth hardware on ios/macos.

Ben1980, 2024-02-13:
Your statement about Qt could be right. As far as I know they use it for the mobile version.

closed 2025-05-18 by Vudentz