bluez/bluez #745: unable to pair my BT remote with bluez 5.66

ANANTHMARIMUTHU, 2024-02-14:
Recently I have upgraded my bluez version from 5.48 to 5.66 but BT remote pairing is not stable with the 5.66. I am facing below error in bluez while problem happened.

= bluetoothd: ../bluez-5.66/profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Reque..   138.223963
= bluetoothd: ../bluez-5.66/profiles/deviceinfo/dis.c:read_pnpid_cb() Error reading PNP_ID value: Request attr..   138.223994
= bluetoothd: ../bluez-5.66/profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Request attri..   138.224016
= bluetoothd: ../bluez-5.66/profiles/input/hog-lib.c:external_report_reference_cb() Read External Report Refer..   138.224037
= bluetoothd: ../bluez-5.66/profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attr..   138.224058

following commands used for pairing in bluetoothctl
#pairable on
# pair <MAC>

after this pairing was successful and immediately entered into reconnect loop.
[CHG] Device ServicesResolved: no
[CHG] Device Connected: no 
[CHG] Device 20:E7:B6:34:5F:13 Connected: yes
[CHG] Device Connected: no
[CHG] Device 20:E7:B6:34:5F:13 Connected: yes

kernel version is : 5.4

To check this issue I have downgraded my bluez version as 5.48. my BT remote properly working with the bluez 5.48 without any issue. What needs to be configured in the new bluez 5.66 to avoid the PNP ID failure?

Vudentz, 2024-02-15:
> Recently I have upgraded my bluez version from 5.48 to 5.66 but BT remote pairing is not stable with the 5.66. I am facing below error in bluez while problem happened.
> 
> = bluetoothd: ../bluez-5.66/profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Reque.. 138.223963 = bluetoothd: ../bluez-5.66/profiles/deviceinfo/dis.c:read_pnpid_cb() Error reading PNP_ID value: Request attr.. 138.223994 = bluetoothd: ../bluez-5.66/profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Request attri.. 138.224016 = bluetoothd: ../bluez-5.66/profiles/input/hog-lib.c:external_report_reference_cb() Read External Report Refer.. 138.224037 = bluetoothd: ../bluez-5.66/profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attr.. 138.224058
> 
> following commands used for pairing in bluetoothctl #pairable on #pair MAC ADDRESS
> 
> after this pairing was successful and immediately entered into reconnect loop. [CHG] Device ServicesResolved: no [CHG] Device Connected: no [CHG] Device Connected: yes [CHG] Device Connected: no [CHG] Device Connected: yes
> 
> kernel version is : 5.4
> 
> To check this issue I have downgraded my bluez version as 5.48. my BT remote properly working with the bluez 5.48 without any issue. What needs to be configured in the new bluez 5.66 to avoid the PNP ID failure?

Any reason you couldn't update to something more recent e.g. 5.72? We may need the HCI traces which you can collect with use of btmon, btw perhaps it is a good idea to use connect command instead of pair since we auto update the security in case an attribute requires it.

ANANTHMARIMUTHU, 2024-02-15:
Thanks for the update.
There is no reason. already bluez 5.66 properly ported So, I am using it. I will collect and share the btmon logs. I have tried with "connect" instead of pair and facing the same error. What could be possibility for this error?

aleritty, 2024-02-16:
I'm affected too on Linux 6.7.0-0-MANJARO x86_64.

```
bluetoothd[618]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
bluetoothd[618]: profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Request attribute has encountered an unlikely error
bluetoothd[618]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
bluetoothd[618]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
bluetoothd[618]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
bluetoothd[618]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
bluetoothd[618]: profiles/input/hog-lib.c:proto_mode_read_cb() Protocol Mode characteristic read failed: Request attribute has encountered an unlikely error
```

ANANTHMARIMUTHU, 2024-02-20:
[btmon.txt](https://github.com/bluez/bluez/files/14339911/btmon.txt)

ANANTHMARIMUTHU, 2024-02-23:
I have tested with the bluez 5.72 and facing the same error. mentioned the bluez log below,
```
src/shared/att.c:disconnect_cb() (chan 0x28e7ca8) Failed to obtain disconnect error: Bad file descriptor
src/shared/att.c:disconnect_cb() Channel 0x28e7ca8 disconnected: Success
/bluez-5.72/src/shared/gatt-client.c:service_changed_register_cb() Failed to register handler for "Service Changed"
profiles/gap/gas.c:read_device_name_cb() Reading device name failed with ATT errror: 14
profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely err
profiles/battery/battery.c:read_initial_battery_level_cb() Reading battery level failed with ATT errror: 14
profiles/deviceinfo/dis.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
attrib/gattrib.c:g_attrib_unref() 0x28e7e38: g_attrib_unref=11
profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Request attribute has encountered an unlikely error
attrib/gattrib.c:g_attrib_unref() 0x28e7e38: g_attrib_unref=10
profiles/input/hog-lib.c:external_report_reference_cb() Read External Report Reference descriptor failed: Request attribute ha
attrib/gattrib.c:g_attrib_unref() 0x28e7e38: g_attrib_unref=9
profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attribute has encountered an unlikely error
attrib/gattrib.c:g_attrib_unref() 0x28e7e38: g_attrib_unref=8
profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an u
attrib/gattrib.c:g_attrib_unref() 0x28e7e38: g_attrib_unref=7
profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attribute has encountered an unlikely error
attrib/gattrib.c:g_attrib_unref() 0x28e7e38: g_attrib_unref=6
profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an u
attrib/gattrib.c:g_attrib_unref() 0x28e7e38: g_attrib_unref=5
profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attribute has encountered an unlikely error
attrib/gattrib.c:g_attrib_unref() 0x28e7e38: g_attrib_unref=4
profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an u
attrib/gattrib.c:g_attrib_unref() 0x28e7e38: g_attrib_unref=3
src/device.c:att_disconnected_cb()
src/device.c:att_disconnected_cb() Success (0)
```

ANANTHMARIMUTHU, 2024-02-23:
Do we need to enable any configuration in the kernel 5.4 for the bluez 5.66 and 5.72?

ANANTHMARIMUTHU, 2024-03-05:
anybody has input for this issue?

ANANTHMARIMUTHU, 2024-03-18:
@Vudentz Could you please help me on this?

Vudentz, 2024-03-18:
@ANANTHMARIMUTHU check with the latest tree, there has been some changes on the way we handle input devices, if there is still pairing problem then we need the btmon logs to check what is going on.

aleritty, 2024-03-20:
I'm on bluez 5.73.r21, on Linux 6.7.7 and the problem still occurs for me.

Vudentz, 2024-03-20:
> I'm on bluez 5.73.r21, on Linux 6.7.7 and the problem still occurs for me.

Logs?

ANANTHMARIMUTHU, 2024-03-20:
@Vudentz tested with the bluez 5.73 and facing the same error. Attached the btmon logs 
[btmon_logs.txt](https://github.com/bluez/bluez/files/14669899/btmon_logs.txt)

Vudentz, 2024-03-20:
```
> HCI Event: Authenticated Payload Timeout Expired (0x57) plen 2                                        #215 [hci0] 71.539769
        Handle: 64 Address: E4:A6:34:CB:61:9F (Universal Electronics, Inc.)
```

Link layer issue, not really pairing, you either have to contact the manufactorer to know what is going on or try with another controller.

ANANTHMARIMUTHU, 2024-03-20:
[btmon_scan_pair.txt](https://github.com/bluez/bluez/files/14671107/btmon_scan_pair.txt)
In this I have not seen the "authentication Payload "error.

Vudentz, 2024-03-21:
```
luetoothd[652]: @ MGMT Command: Us.. (0x001d) plen 7  {0x0001} [hci0] 6.762065
        LE Address: D0:EB:20:46:DD:EA (Static)
@ MGMT Event: Authentication Failed (0x0011) plen 8    {0x0001} [hci0] 6.762073
        LE Address: D0:EB:20:46:DD:EA (Static)
        Status: Authentication Failed (0x05)
< ACL Data TX: Handle 512 flags 0x00 dlen 6                 #27 [hci0] 6.762076
      SMP: Pairing Failed (0x05) len 1
        Reason: Passkey entry failed (0x01)
@ MGMT Event: Command Complete (0x0001) plen 10        {0x0001} [hci0] 6.762078
      User Confirmation Negative Reply (0x001d) plen 7
        Status: Success (0x00)
        LE Address: D0:EB:20:46:DD:EA (Static)
```

User Confirmation Negative Reply, so if you reject the pairing it obviously won't pair... Or perhaps you don't even have a pairing agent active on your system? If you are using bluetoothctl you can pass -a auto that way it responds automatically to this sort of requests.

ANANTHMARIMUTHU, 2024-03-21:
> ```
>         Handle: 64 Address: E4:A6:34:CB:61:9F (Universal Electronics, Inc.)
> ```
>
> Link layer issue, not really pairing, you either have to contact the manufactorer to know what is going on or try [...]

Same controller properly working with the bluez 5.48. [btmon_scan_pair.txt](https://github.com/bluez/bluez/files/14671107/btmon_scan_pair.txt) -> In this log I can not see the "Authenticated Payload Timeout Expired".

aleritty, 2024-03-22:
> ```
> luetoothd[652]: @ MGMT Command: Us.. (0x001d) plen 7  {0x0001} [hci0] 6.762065
>         LE Address: D0:EB:20:46:DD:EA (Static)
> @ MGMT Event: Authentication Failed (0x0011) plen 8    {0x0001} [...]

Of course I'm not manually refusing the pairing, I'll try with bluetoothctl but I'm quite sure the problem will reappear at restart.

ANANTHMARIMUTHU, 2024-03-26:
@Vudentz 
[btmon_bluez5.48_working.txt](https://github.com/bluez/bluez/files/14762373/btmon_bluez5.48_working.txt)
[btmon_bluez5.66_not_working.txt](https://github.com/bluez/bluez/files/14762392/btmon_bluez5.66_not_working.txt)
attached the working and not working btmon files.

aleritty, 2024-03-26:
> For me it is not working after setup the "AlwaysPairable=true". Still facing the similar disconnect issue. @aleritty Can you please share your bluetooth.main.conf file to verify my configuration.

Yes, this is my /etc/bluetooth/main.conf

```
[General]

AlwaysPairable = true
FastConnectable = true
Privacy = off
JustWorksRepairing = always
```

ANANTHMARIMUTHU, 2024-04-04:
@aleritty Thanks for the information. But, still I am facing the same error with the above configuration?

ANANTHMARIMUTHU, 2024-04-04:
> @Vudentz [btmon_bluez5.48_working.txt](https://github.com/bluez/bluez/files/14762373/btmon_bluez5.48_working.txt) [btmon_bluez5.66_not_working.txt](https://github.com/bluez/bluez/files/14762392/btmon_bluez5.66_not_working.txt) bluez 5.66 not working log: [bluez.log](https://github.com/bluez/bluez/files/14763372/bluez.log)
> 
> attached the working and not working btmon files.

@Vudentz got any chance to check these logs?

dkarnikis, 2024-04-05:
I'm running bluez 5.65 and Linux Kernel 6.4.7. Both have custom patches applied. I'm facing a similar issue, initial pairing with the ble device works fine and I can communicate normally. 
As soon as i disconnect the device with `bluetoothctl remove BLE_MAC` (or another way to make the ble device disappear) and then try to reconnect, I get into the exactly same sequence; connect/disconnect and the unlikely error as described before.
More specifically, my connection fails with `PINS OR KEY MISSING 0x06` even though ltk are present

The previous solutions from @aleritty don't solve my problem.

I can modify the kernel, userspace and of course bluez. I can provide pcap logs, traces, etc.

Firestar-Reimu, 2024-04-15:
Archlinux + bluez 5.73 + Logitech MX Anywhere 3S

Same issue as [this](https://bbs.archlinux.org/viewtopic.php?id=292866)

> Basically when I turn on my laptop the mouse starts connecting and immediately disconnecting in an infinite loop.

Every time I open my computer I need to delete and reconnect it.

I tried https://wiki.archlinux.org/title/Bluetooth_mouse#Problems_with_the_Logitech_BLE_mouse_(M557,_M590,_M720,_anywhere_mouse_2,_etc)

the first steps are fine, but when I do:

> 10. Power the bluetooth off and on.

it will not automatically connect the mouse, I need to manually `connect XX:XX:XX:XX:XX:XX`

ANANTHMARIMUTHU, 2024-04-16:
@Vudentz 
I have compared the bluez 5.48 (working) and bluez 5.66(not working) btmon logs. seems some of the GATT characteristics (0x0003 - device name, 0040 - PNP ID, etc) read operation not initiated from the bluez 5.66.

**bluez 5.48 btmon - read operation for device name and PNP ID**
< ACL Data TX: Handle 64 flags 0x00 dlen 7                                                                #53 [hci0] 7.089350
      ATT: Read Request (0x0a) len 2
        Handle: 0x0003
< ACL Data TX: Handle 64 flags 0x00 dlen 13                                                               #54 [hci0] 7.089645
      ATT: Read By Type Response (0x09) len 8
        Attribute data length: 7
        Attribute data list: 1 entry
        Handle: 0x0007
        Value: 200800052a
> HCI Event: Number of Completed Packets (0x13) plen 5                                                    #55 [hci0] 7.138102
        Num handles: 1
        Handle: 64
        Count: 2
> ACL Data RX: Handle 64 flags 0x02 dlen 15                                                               #56 [hci0] 7.234958
      ATT: Read Response (0x0b) len 10
        Value: 506c6174636f20505231
< ACL Data TX: Handle 64 flags 0x00 dlen 7                                                                #63 [hci0] 7.430501
      ATT: Read Request (0x0a) len 2
        Handle: 0x0040
< ACL Data TX: Handle 64 flags 0x00 dlen 10                                                               #64 [hci0] 7.430799
      ATT: Find Information Response (0x05) len 5
        Format: UUID-16 (0x01)
        Handle: 0x0009
        UUID: Client Characteristic Configuration (0x2902)
> HCI Event: Number of Completed Packets (0x13) plen 5                                                    #65 [hci0] 7.479298
        Num handles: 1
        Handle: 64
        Count: 2
        
Like this some other characteristics read also not initiated in the Bluez 5.66. I have used the same Bluetooth device with the Bluez 5.48 and 5.66. my device is properly working with the bluez 5.48 but not working with the bluez 5.66.

Attached the btmon logs: 
[bluez5.48_btmon.log](https://github.com/bluez/bluez/files/15001701/bluez5.48_btmon.log)
[bluez5.66_btmon.log](https://github.com/bluez/bluez/files/15001702/bluez5.66_btmon.log)

Firestar-Reimu, 2024-04-17:
Update to 5.74, no fixed.

g0ne150, 2024-04-29:
Update to 5.75, no fixed.

Vudentz, 2024-04-30:
```
< HCI Command: LE Remote Connection Parameter Request Reply (0x08|0x0020) plen 14                       #168 [hci0] 18.986704
        Handle: 64 Address: E4:A6:34:CB:61:9F (Universal Electronics, Inc.)
        Min connection interval: 15.00 msec (0x000c)
        Max connection interval: 15.00 msec (0x000c)
        Connection latency: 331 (0x014b)
        Supervision timeout: 15000 msec (0x05dc)
        Min connection length: 0.000 msec (0x0000)
        Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Complete (0x0e) plen 6                                                             #169 [hci0] 18.988094
      LE Remote Connection Parameter Request Reply (0x08|0x0020) ncmd 1
        Status: Success (0x00)
        Handle: 64 Address: E4:A6:34:CB:61:9F (Universal Electronics, Inc.)
> HCI Event: LE Meta Event (0x3e) plen 10                                                               #170 [hci0] 19.328457
      LE Connection Update Complete (0x03)
        Status: Success (0x00)
        Handle: 64 Address: E4:A6:34:CB:61:9F (Universal Electronics, Inc.)
        Connection interval: 15.00 msec (0x000c)
        Connection latency: 331 (0x014b)
        Supervision timeout: 15000 msec (0x05dc)
```

These parameters seems to be exact the same in 5.48 vs 5.66, what kernel version is being used?

ANANTHMARIMUTHU, 2024-04-30:
@Vudentz
Yes, this is remote device to respond. I have tried to read the 'databash hash' characteristic value via gatttool. I got response from the remote device without any issue. But, did not respond while read from bluez.

After commented the below line of code from the 'src/shared/gatt-client.c' file, disconnect not happened.
-	if (success && read_db_hash(op))
-		return;

TimVosch, 2024-05-01:
I'm having a similar issue. My Corsair Harpoon mouse stopped connecting while performing a system update a few weeks ago. 

I've used BTMon to get some logs. The device in question is "ED:9F:68:86:63:CD". 
I've tried using bluez-git, changing the configs as recommended above, removing `/var/bluetooth/<ctrl_id>`  and resetting the mouse (many times).

The mouse does work on my Android phone. I'll see if I can try a different PC.

```
bluetoothd[37400]: @ MGMT Command: Stop Discovery (0x0024) plen 1                                                                                                                                                                    {0x0001} [hci0] 59.778461
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
< HCI Command: Inquiry Cancel (0x01|0x0002) plen 0                                                                                                                                                                                        #18 [hci0] 59.778658
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                                                               #19 [hci0] 59.780079
      Inquiry Cancel (0x01|0x0002) ncmd 2
        Status: Success (0x00)
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                                                                                                                                                                           #20 [hci0] 59.780243
        Extended scan: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                                                               #21 [hci0] 59.781112
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 2
        Status: Success (0x00)
@ MGMT Event: Discovering (0x0013) plen 2                                                                                                                                                                                            {0x0001} [hci0] 59.781202
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
        Discovery: Disabled (0x00)
@ MGMT Event: Command Complete (0x0001) plen 4                                                                                                                                                                                       {0x0001} [hci0] 59.781280
      Stop Discovery (0x0024) plen 1
        Status: Success (0x00)
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
bluetoothd[37400]: @ MGMT Command: Stop Discovery (0x0024) plen 1                                                                                                                                                                    {0x0001} [hci0] 59.781497
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
@ MGMT Event: Command Complete (0x0001) plen 4                                                                                                                                                                                       {0x0001} [hci0] 59.781508
      Stop Discovery (0x0024) plen 1
        Status: Rejected (0x0b)
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
bluetoothd[37400]: @ MGMT Command: Pair Device (0x0019) plen 8                                                                                                                                                                       {0x0001} [hci0] 59.781556
        LE Address: ED:9F:68:86:63:CD (Static)
        Capability: DisplayYesNo (0x01)
< HCI Command: LE Set Extended Scan Parameters (0x08|0x0041) plen 13                                                                                                                                                                      #22 [hci0] 59.781705
        Own address type: Public (0x00)
        Filter policy: Ignore not in accept list (0x01)
        PHYs: 0x05
        Entry 0: LE 1M
          Type: Passive (0x00)
          Interval: 60.000 msec (0x0060)
          Window: 60.000 msec (0x0060)
        Entry 1: LE Coded
          Type: Passive (0x00)
          Interval: 180.000 msec (0x0120)
          Window: 180.000 msec (0x0120)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                                                               #23 [hci0] 59.782015
      LE Set Extended Scan Parameters (0x08|0x0041) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                                                                                                                                                                           #24 [hci0] 59.782132
        Extended scan: Enabled (0x01)
[... 224 lines cut ...]
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                                                                                            [... cut ...]
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                                                              #53 [hci0] 236.639394
      Write Scan Enable (0x03|0x001a) ncmd 2
        Status: Success (0x00)
pai
bluetoothctl[44226]: @ MGMT Open: bluetoothctl (privileged) version 1.22                                                                                                                                                                   {0x0002} 313.433624
bluetoothctl[44226]: @ MGMT Close: bluetoothctl
```

Vudentz, 2024-05-01:
```
bluetoothd[37400]: @ MGMT Command: Cancel Pair Device (0x001a) plen 7                                                                                                                                                                {0x0001} [hci0] 84.891522
        LE Address: ED:9F:68:86:63:CD (Static)
```

So it seems the pairing is being cancelled, so not the same issue it seems.

duguyipiao, 2024-05-14:
> Yes, this is my /etc/bluetooth/main.conf
>
> ```
> [General]
>
> AlwaysPairable = true
> FastConnectable = true
> Privacy = off
> JustWorksRepairing = always
> ```

For my Logitech MX Vertical, enabling `JustWorksRepairing = always` makes the mouse work.  
(Previously, I have to delete and readd the device after every reboot.)

thibaultmol, 2024-06-18:
Same problem here, also on Nixos also using a logitech mouse (mx master). also added that to my config and that worked (after a reboot).

```
$ uname -a
Linux Framework-THIB 6.6.33 #1-NixOS SMP PREEMPT_DYNAMIC Wed Jun 12 09:13:03 UTC 2024 x86_64 GNU/Linux

$ bluetoothctl --version
bluetoothctl: 5.75
```

johnlinahan, 2024-07-14:
Had the same issue with MX Anywhere 3. Added `JustWorksRepairing = always` and it worked after `sudo systemctl restart bluetooth.service`.

joaodefelipe, 2024-08-15:
Same here, with a Logitech M720 Triathlon. With `JustWorksRepairing = always`, it started *just working* ~(who would've thought?)~ as soon as I restarted the daemon. Of course, `confirm`, while somewhat annoying, works as well.

Vudentz, 2024-08-15:
> Same here, with a Logitech M720 Triathlon. With `JustWorksRepairing = always`, it started _just working_ ~(who would've thought?)~ as soon as I restarted the daemon. Of course, `confirm`, while somewhat annoying, works as well.

Authentication vs Authorization, JustWorks is suppose to bypass authentication not authorization, if there are implementation doing both by default they are probably vulnerable to impersonation attacks.

noleti, 2025-01-15:
One more data point for whoever else has this issue (as I am on the common Ubuntu 24.04 LTS): I believe the fix in https://github.com/bluez/bluez/issues/745#issuecomment-2160670429 also fixes my issues I had in Ubuntu after upgrading to 24.04 LTS. Before applying the fix, previously paired devices would not be reconnected to automatically, and I would be unable to connect again afterwards. I could also not change audio profiles (headphones) or similar. Applying the change in `/etc/bluetooth/main.conf` fixed this.

```
 ➜ uname -a
Linux not-ThinkPad-X1-Carbon-6th 6.8.0-51-generic #52-Ubuntu SMP PREEMPT_DYNAMIC Thu Dec  5 13:09:44 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

bluetoothctl --version
bluetoothctl: 5.72

 ➜ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04.1 LTS
Release:	24.04
Codename:	noble
```

closed 2025-07-30 by github-actions[bot]

Firestar-Reimu, 2025-07-30:
My MX Anywhere 3S will sometimes lost connection when I switched to other device and switch back, and I should delete that device and re-pair it