bluez/bluez #747: "bluetoothctl" documentation / source nowhere to be found

dennisorlando, 2024-02-17:
Hi,
`bluetoothctl` appears to be the only linux command able to read GATT values of BLE services, yet there is absolutely no documentation regarding this tool, which leads to a lot of frustration and the need to switch to plain C or Python bluetooth bindings. 
Am I missing something? Is this the right place for this issue?

dennisorlando, 2024-02-17:
> If you look at the git repo you'll see a handful of `bluetoothctlXXX.rst` files, which the build generates into manpages assuming `./configure --disable-manpages ...` is not used.
> 
> ~Although looking at my distribution of choice - Arch - it looks like the manual pages are build but not part of any package.~ They're available, I was looking at the wrong place 🤦 Eg [bluetoothctl-advertise.1](https://man.archlinux.org/man/extra/bluez-utils/bluetoothctl-advertise.1.en)

Hmm most of the submenu manpages don't really show any useful information apart from the syntax and a really short description which is already available when using the command.
'bluetoothctl-advertise' seems to be the only exception

dennisorlando, 2024-02-17:
btw do you happen to know where the source is? In Arch, `bluetoothctl` gets installed under `bluez-utils` which doesn't have any upstream url. I grepped this repo but also couldn't find anything

dennisorlando, 2024-02-21:
> @dennisorlando if you ever feel like writing some documentation, I doubt the maintainers will object ;-)

hehe 😂
I actually thought about that but I'm not really that knowledgeable about this stuff, I think I would have to dive deep into the sources and specifications
Might do it in the future 🤷

potto216, 2024-02-21:
Hi @dennisorlando I helped with the bluetoothctl page updates a few months ago and am working on further updates. It's helpful to know use cases to decide what and how much to document. For the gatt menu were you primarily interested in setting up a gatt server? Like for example setting up a service with read only and read/write characteristics to test a client. For example:

```
[bluetooth]# menu gatt
[bluetooth]# register-service e2d36f99-8909-4136-9a49-d825508b297b
[bluetooth]# register-characteristic 0x1234 read
[bluetooth]# register-characteristic 0x4567 read,write
[bluetooth]# register-application
[bluetooth]# back
[bluetooth]# discoverable on
[bluetooth]# advertise on
```

dennisorlando, 2024-02-22:
Hmm the information I was specifically looking for was:
- when you select a GATT Characteristics and then run "read", the actual values of given characteristics are printed with some sort of "delay". I wanted to make a bash script to automatically fetch a value (e.g. the battery level of a service), but I wasn't able to because those "values" didn't appear to be in "stdin" and I couldn't find any documentation (without looking at the source) that tells me how that value is being displayed.

In general, I wanted to use bluetoothctl as an instrument to use GATT, but given the lack of documentation I would have to first read the GATT and maybe the BLE specifications to understand how to use bluetoothctl.

Here are some examples, all referring to the gatt submenu which I've used:

<img src="https://github.com/bluez/bluez/assets/47061464/539a6053-29fd-42f9-b178-bbd9df3bd887)
what's "offset"? What are you reading? You can figure it out by playing with it, but it's not written anywhere. 

[image]  
(Ignoring the "select attribute" string which I think is just a bug) what information are we reading? I suppose there are some specific fields of a GATT service which this command reads, but they are not listed anywhere; I'm not sure if I can use this tool to read some custom information that I've added to a GATT service because I'm not sure what info it actually prints.
As an example, attribute-info on a Battery Service prints if it's a primary battery or not. I searched through the GATT specs and found no mention of any "Primary" field. I suppose that attribute-info checks whether a battery service has the "main" identifier, but it's not stated anywhere. 

[image]  

I actually have no idea what this does. Is this a lock? Does it mean that the value can't be changed while you are reading? Is it server-specific? 

[image]  

I suppose the correct documentation for this is something like "Turn on/off notifications for when the selected attribute's value changes". "Notify attribute value" doesn't make that much sense

[image]  

I'm using bluetoothctl on my laptop, connected via BLE to a bluetooth keyboard.
A clone command doesn't make sense; does it mean that I can somehow modify the attributes of the 'remote' BLE device? 
What does it mean by "cloning a device"? Wouldn't a duplicated attribute cause trouble? (i.e. duplicated addresses?)
I suppose you can clone attributes only if you are using bluetoothctl to create a gatt server.

[image]  

Register for what? Why would you have to register something? 
There are 3 flags being mentioned (read, write, notify...); what's after those three dots? I would have to read the GATT manual to find out where the "read, write, notify" flags are defined and see if there's anything else.

[image]  

Included in what? 

I know that there are some things that the user is expected to know before using this tool (there's no need to tell the user what's an attribute), but I think these descriptions are just too barebone

potto216, 2024-02-23:
Hi @dennisorlando thanks for the comments I'll write a more detailed response this weekend, but in general read, write, or read/write specifies the access of that characteristic in the GATT database. So, if you are designing a database for your Bluetooth device that has values to share with the world such as the devices battery level, and sensor readings the Bluetooth device would make those "read", so they can only be read and not overwritten by others. However, if that same device wanted to allow others to change a value such as the color of a light, then the device would make the characteristic "write" so the outside world can change it.  If that Bluetooth device wanted to announce to the world when a value had changed then setting the notify on the characteristic would announce it when it changes.  Oh and all of this happens between two connected devices.

I was curious as to your use case of the battery level. Did you want to share the battery information from the device you are running `bluetoothctl` on, or did you want to see another device's battery information by connecting to it with `bluetoothctl`?  Also it sounds like you want to automate this process. You can autoamate `bluetoothctl` with the `expect ` program or you can use another Bluez program `gatttool` which is made for the command line. For example, if you wanted to read the battery level on another device you could have (I haven't tested this to see if it actually works)

```
#!/bin/bash

# BLE device MAC address
DEVICE_MAC="XX:XX:XX:XX:XX:XX"
# Battery Level GATT Characteristic UUID
CHAR_UUID="00002a19-0000-1000-8000-00805f9b34fb"

# Reading the battery level GATT characteristic
read_value=$(gatttool -b $DEVICE_MAC --char-read --uuid=$CHAR_UUID 2>&1)

# Check for read error
if echo "$read_value" | grep -q "Error"; then
  echo "Failed to read battery level."
  exit 1
fi

# Extracting the value
# The output format of gatttool is "Characteristic value/descriptor: XX", we need the XX part
# The battery level is a single byte, so we convert from hexadecimal to decimal
value_hex=$(echo "$read_value" | awk '{print $NF}') # Assuming the value is the last word
value=$((16#$value_hex)) # Convert hex to decimal

# Display the battery level
echo "Battery Level: $value%"

# Example logic condition using the battery level
if [ $value -lt 20 ]; then
  echo "Battery low: Consider recharging."
else
  echo "Battery level is sufficient."
fi
```

dennisorlando, 2024-02-23:
Thanks for the answer!
I actually ended up using [pydbus](https://pypi.org/project/pydbus/) to read the values, for which I now have a currently working program. In my use case, I have a [ZMK](https://zmk.dev/) split keyboard (left and right) which has two separated batteries. A recent pull request was merged into ZMK that allows monitoring of the battery levels of both sides, via means of multiple "Battery Service"s on a single BLE device. 
There isn't any host application that is able to display both battery levels, thus I wanted to make my own, which I'm currently able to using a mix of pydbus and bluetoothctl. 
I haven't tried using `expect`

The underlying issue seems to be recursive: I've just checked `gatttool` but it lacks proper documentation just as `bluetoothctl`. It's also deprecated, which kind of pushes me away from giving it a try.

potto216, 2024-02-24:
@dennisorlando Glad you got a solution working (I didn't realize that `gatttool` was deprecated) and thanks for the info on ZMK. If you run into limitations with pydbus, another option is [BlueR](https://github.com/bluez/bluer) which is the Rust user level interface to BlueZ. The project has command line tools that can be used for GATT. There is a [discussion board](https://github.com/bluez/bluer) for help. 
If you want to leave this issue open I'll post a draft of the `bluetoothctl` GATT documentation for feedback.

potto216, 2024-02-27:
@dennisorlando let me know your thoughts on the overview https://github.com/potto216/bluez/blob/master/client/bluetoothctl-gatt.rst#generic-attribute-commands.

dennisorlando, 2024-02-27:
> @dennisorlando let me know your thoughts on the overview https://github.com/potto216/bluez/blob/master/client/bluetoothctl-gatt.rst#generic-attribute-commands.

I've given it a quick look and I like it, it seems like it's really clear. I'll read it again in the next days

potto216, 2024-03-10:
@Vudentz what are your thoughts on how to organize the "Generic Attribute Commands" section of https://github.com/potto216/bluez/blob/master/client/bluetoothctl-gatt.rst? I realize it is too long to have at the beginning, but I think the information is important since people using the program may not be familiar with GATT concepts. One option is I could move the bulk of the GATT description to the end of the page after the command descriptions.

Egyember, 2025-01-03:
I think a more detailed explanation would be great on the details of commands. For example what the type argument do for write command and what are the possible values?

potto216, 2025-01-05:
@Egyember I think the type arguments for the GATT write are "write", "reliable-write", and "write-without-response" based on this section of `src/gatt-client.c`.
```
	/*
	 * Decide which write to use based on characteristic properties. For now
	 * we don't perform signed writes since gatt-client doesn't support them
	 * and the user can always encrypt the through pairing. The procedure to
	 * use is determined based on the following priority:
	 *
	 *   * "reliable-write" property set -> reliable long-write.
	 *   * "write" property set -> write request.
	 *     - If value is larger than MTU - 3: long-write
	 *   * "write-without-response" property set -> write command.
	 */
	if ((!type && (chrc->ext_props & BT_GATT_CHRC_EXT_PROP_RELIABLE_WRITE))
			|| (type && !strcasecmp(type, "reliable"))) {
		supported = true;
		chrc->write_op = start_long_write(msg, chrc->value_handle, gatt,
						true, value, value_len, offset,
						chrc, chrc_write_complete);
		if (chrc->write_op)
			return NULL;
	}

	if ((!type && chrc->props & BT_GATT_CHRC_PROP_WRITE) ||
			(type && !strcasecmp(type, "request"))) {
		uint16_t mtu;
		
};
```
Have you tried the [BlueR command line tool `gattcat`](https://github.com/bluez/bluer/blob/master/bluer-tools/src/gattcat.rs)? They have a responsive [discussion board](https://github.com/bluez/bluer/discussions) for questions which is part of the BlueZ family.

Vudentz, 2025-01-06:
> I think a more detailed explanation would be great on the details of commands. For example what the type argument do for write command and what are the possible values?

Are you talking about https://github.com/bluez/bluez/blob/master/client/bluetoothctl-gatt.rst#write?

That is documented in https://github.com/bluez/bluez/blob/master/doc/org.bluez.GattCharacteristic.rst#void-writevaluearraybyte-value-dict-options, that said we could probably do a better job document it in the bluetoothctl documentation so user don't have to go to the D-Bus documentation in order to figure it out, lately Ive been adding examples alongside the command description to make it even more clear.

dennisorlando, 2025-01-06:
> ...we could probably do a better job document it in the bluetoothctl documentation so user don't have to go to the D-Bus documentation in order to figure it out..

Perhaps redirecting to the d-bus documentation might be a cheap compromise?

Vudentz, 2025-01-16:
> I know this discussion has mostly been about GATT stuff, but given the title of this issue, I'm assuming it covers documentation gaps of any kind. Else I can open a new issue.
> 
> For me, there were many gaps that made it difficult to use. For example just the command line...
> 
> My (Ubuntu) man page says the command invocation is: `bluetoothctl [options]` and the doc [here](https://github.com/bluez/bluez/blob/master/client/bluetoothctl.rst) says: `bluetoothctl [-a capability] [-e] [-m] [-t seconds] [-v] [-h]` But `bluetoothctl --help` says: `bluetoothctl [--options] [commands]`

Yeah it seems the synopsis is sort of outdated in this respect, but it is not like we don't have each command documented, in fact each menu now has it own documentation.

> I looked at these regarding whether a command like "scan on" can be given on the command line. Two of the sources say it cannot and one says it can. It seems like it can, but then for the particular invocation of: `bluetoothctl scan on` It does not appear to actually conduct a scan (in version 5.72) when run in this non-interactive way (and exits instantly). I'm guessing this is because I need to provide "--timeout" for a non-synchronous/long-running command like "scan on"
> 
> And then I've found that "scan on" seems to only continue scanning for about 4 minutes. Is this hardware-dependent, or is it just built into bluez?

Not sure, check with btmon if the controller fails to scan after that.

> Is there any difference between running commands interactively vs non-interactively? Can the stdout of bluetoothctl be set to line-buffered via flags? Doesn't seem to be documented that it is not line-buffered. Is the syntax of the output for "scan on" documented anywhere? The --help says "[commands]" with an "s". What is the syntax for providing multiple commands there?

I guess I will need to check what do you mean by line-buffered, it uses libreadline so Im not really sure we can change it to be line-buffered. Anyway perhaps you want to check how scripts (-s) works since you seem to be interested in entering multiple commands.
 
> As far as I know, bluetoothctl is the single (and official) program used for command-line Bluetooth control on linux, so I would expect it to have comprehensive documentation akin to "man bash" or the like, since Bluetooth is a pretty fundamental and universal user-facing system.

Well if you provide resources to hire someone to work on it things may change faster, otherwise it probably will take me some time to figure out how to document it in a user friendly manner.

> So although I am interested in the answers to the above questions, my point is that they should be documented for all in the official documentation, and their presence here is more rhetorical.
> 
> And there is hcitool... `hcitool scan --help` `Usage: scan [--length=N] [--numrsp=N] [--iac=lap] [--flush] [--class] [--info] [--oui] [--refresh]` As far as I can see, there is no documentation anywhere of what those options mean.
> 
> > > [@dennisorlando](https://github.com/dennisorlando) if you ever feel like writing some documentation, I doubt the maintainers will object ;-)
> > 
> > 
> > hehe 😂 I actually thought about that but I'm not really that knowledgeable about this stuff, I think I would have to dive deep into the sources and specifications Might do it in the future 🤷
> 
> I would love to contribute too, but updating documentation for a large set of commands like this requires a ton of knowledge of the implementation which I and most users of the tool do not have, so it's impractical to do except in slow motion over years in tiny bits by studying source and asking knowledgeable contributors. Perhaps that is the only way.

Well you have to start somewhere if you really want to contribute.

Vudentz, 2025-01-16:
> I guess I will need to check what do you mean by line-buffered, it uses libreadline so Im not really sure we can change it to be line-buffered. Anyway perhaps you want to check how scripts (-s) [...]

We do actually use setlinebuf:

https://github.com/bluez/bluez/blob/master/src/shared/shell.c#L1271

But that doesn't seems to be set for non-interactive mode, which perhaps is what you are referring to? In that it is probably very trivial to be addressed, although I would expect application to parse the exit value not its output, otherwise we had to keep some sort of pattern for errors, etc, but bluetoothctl command output is not intended as a stable API.

Vudentz, 2025-01-17:
@surlyhacker can you check if the above changes address your comments regarding bluetoothctl not setting stdout to be line-buffered?

ferrieux, 2025-04-19:
I'm also struggling to understand how to use `bluetoothctl` to register an endpoint.

The existence of an "interactive" mode is encouraging... except it is not really interactive: no completion, no way to list valid values expected nor even valid syntax, and even ^C is *disabled* in the middle of the interaction. 

But the most frustrating part is that "examples" from `man bluetoothctl-endpoint` are ... just plain wrong.

As proof: if I invoke `endpoint.register` with just UUID and codec, I get prompted with "capabilities", which are supposed to be optional. But I'm stubbornly requested to enter them anyway. What is the expected syntax here ?

```shell
[bluetooth]# endpoint.register 00001131-0000-1000-8000-00805f9b34fb 0x00
[/local/endpoint/ep2] Enter capabilities:
```

Then, if I enter some random string, say ".", I get "Invalid value at index 0", and then the dialog proceeds to the next question anyway .... what's the point then ?

```shell
[/local/endpoint/ep2] Enter capabilities: .
Invalid value at index 0
```

Then, in the next questions there is a slight amount of help, so  I'm guided to enter values that seem valid... to no avail , as the dialog ends up with a mysterious "Failed register endpoint" and a glorious segfault:

```shell
[/local/endpoint/ep2] Enter Metadata (value/no): no
[/local/endpoint/ep2] Auto Accept (yes/no): y
[/local/endpoint/ep2] Max Transports (auto/value): a
[/local/endpoint/ep2] Locations: 0
[/local/endpoint/ep2] Supported Context (value): 0
[/local/endpoint/ep2] Context (value): 0
[/local/endpoint/ep2] CIG (auto/value): 0
[/local/endpoint/ep2] CIS (auto/value): 0
Failed register endpoint
[bluetooth]# Segmentation fault
```

So, I agree a hundred percent with the OP's feeling that "surely there must be some non-obvious documentation somewhere", given the central status of `bluetoothctl`. Or is it just some unmaintained "toy sample program" ?

Vudentz, 2025-04-21:
> I'm also struggling to understand how to use `bluetoothctl` to register an endpoint.
> 
> The existence of an "interactive" mode is encouraging... except it is not really interactive: no completion, no way to list valid values expected nor even valid syntax, and even ^C is _disabled_ in the middle of the interaction.
> 
> But the most frustrating part is that "examples" from `man bluetoothctl-endpoint` are ... just plain wrong.
> 
> As proof: if I invoke `endpoint.register` with just UUID and codec, I get prompted with "capabilities", which are supposed to be optional. But I'm stubbornly requested to enter them anyway. What is the expected syntax here ?
> 
> [bluetooth]# endpoint.register 00001131-0000-1000-8000-00805f9b34fb 0x00
> [/local/endpoint/ep2] Enter capabilities:

Were in the documentation it is stated that 0x1131 is valid though? That is for HSP, endpoint.register only works with A2DP and BAP UUIDs, anyway I can probably make it check that instead of trying to ask for codec capabilities.

> Then, if I enter some random string, say ".", I get "Invalid value at index 0", and then the dialog proceeds to the next question anyway .... what's the point then ?
> 
> [/local/endpoint/ep2] Enter capabilities: .
> Invalid value at index 0
> 
> Then, in the next questions there is a slight amount of help, so I'm guided to enter values that seem valid... to no avail , as the dialog ends up with a mysterious "Failed register endpoint" and a glorious segfault:
> 
> [/local/endpoint/ep2] Enter Metadata (value/no): no
> [/local/endpoint/ep2] Auto Accept (yes/no): y
> [/local/endpoint/ep2] Max Transports (auto/value): a
> [/local/endpoint/ep2] Locations: 0
> [/local/endpoint/ep2] Supported Context (value): 0
> [/local/endpoint/ep2] Context (value): 0
> [/local/endpoint/ep2] CIG (auto/value): 0
> [/local/endpoint/ep2] CIS (auto/value): 0
> Failed register endpoint
> [bluetooth]# Segmentation fault

Well you entered something invalid expect something invalid, garbage in == garbage out, can we make it bluetoothctl ultra strict, absolutely, will it help you to figure out HSP don't work with MediaEndpoint, I gues not...
 
> So, I agree a hundred percent with the OP's feeling that "surely there must be some non-obvious documentation somewhere", given the central status of `bluetoothctl`. But reading this thread, I have the feeling that only GATT is decently covered. So, how are people supposed to learn to use the "endpoint" submenu for example ?

Wait, why do you think endpoint submenu is intended for regular users? bluetoothctl is just a reference client so upper layer have examples of how the D-Bus API works, in fact if you had succeeded in registering an endpoint for HSP you would probably break the integration with your sound system which is where the likes of HSP is implemented.

ferrieux, 2025-04-21:
Ah, understood. My apologies for expecting too much from "just a reference client" :}
One reason is that the "old generation" (pre-bluez5) tools somehow gave the impression to be Swiss Army knives, able to do everything from a Shell script. As I keep reading that they are  deprecated, and replaced by bluetoothd, I assumed a similar coverage. My bad.

Vudentz, 2025-04-21:
> Ah, understood. My apologies for expecting too much from "just a reference client" :} One reason is that the "old generation" (pre-bluez5) tools somehow gave the impression to be Swiss Army knives, [...]

What are you talking about, BlueZ 4 never had anything like bluetoothctl... and for the most part all the tools that worked with BlueZ 4 works with BlueZ 5 at lower layers, which is basically just HCI RAW socket, that said there is some 12+ years that we released BlueZ 5.

seamusdemora, 2025-05-05:
Hello -  Pardon the interruption, but I've been searching for some help on a problem when I found this thread. 

The problem is trying to connect a second speaker using `bluetoothctl`. I've just "upgraded" my system - bought a couple of Bose Flex BT speakers. I can connect either one of the speakers, but cannot connect both. I wondered if this had something to do with specifying the A2DP profile. I found something yesterday that stated the `bluetoothctl` command to accomplish this was: 

```
bluetoothctl XX:YY:ZZ:AA:BB:CC a2dp
```

I tried this, but it did not work. 

Running a "headless" Debian bookworm distro w/ pipewire ver `Compiled with libpipewire 1.2.7` if that makes any difference. I thought that `a2dp` was "automatic", but it's been anything but that since I've "upgraded" to 2 speakers.

seamusdemora, 2025-05-05:
> Probably due to: [#1037](https://github.com/bluez/bluez/issues/1037)

Maybe...  but as I read the posts in that thread, their problem seems to be with 2 different device sets - e.g. headphones and speakers. My problem is simpler (at least I think it is :) as it concerns only one pair of speakers.

Vudentz, 2025-05-05:
> > Probably due to: [#1037](https://github.com/bluez/bluez/issues/1037)
> 
> Maybe... but as I read the posts in that thread, their problem seems to be with 2 different device sets - e.g. headphones and speakers. My problem is simpler (at least I think it is :) as it concerns only one pair of speakers.

Are there more than one device involved? It it is then is probably due to pipewire having a single endpoint, so it can only connect to a single device at time.

seamusdemora, 2025-05-05:
> Are there more than one device involved? It it is then is probably due to pipewire having a single endpoint, so it can only connect to a single device at time.

Well - there are two separate speakers, if that's what you mean. 

Your second point *raised my eyebrows*  :)   I was under the impression that `pipewire` could support "high end" audio systems/surround sound/etc... How does it do that if it can only connect to a single endpoint?

closed 2026-09-24 by github-actions[bot]