bluez/bluez #750: Audio Connection Failures with Qualcomm QCNCM865

GreyXor, 2024-02-20:
Hello,
I am encountering an issue with my Qualcomm QCNCM865 (WCN7850/WCN7851 chipset) device. Whenever I attempt to connect to any audio device such as headphones or speakers, the connection fails inexplicably. Despite troubleshooting with bluetoothctl, I am unable to pinpoint the exact cause as it doesn't provide any specific error details.

- I can still successfully connect my mouse via Bluetooth without any issues.
- Before that, I had an RZ616 bluetooth card, and all worked normally.

# System
Arch Linux with all packages up-to-date (linux-firmware, Linux 6.7.5, bluez 5.72, pipewire 1.0.3)

# Dump
`sudo btmon -w hcidump.log `
[hcidump.log](https://github.com/bluez/bluez/files/14343212/hcidump.log)

Vudentz, 2024-02-20:
@GreyXor HCI trace looks clean, do you have the syslog/journalctl of bluetoothd when the problem occurs?

GreyXor, 2024-02-21:
Another one with less noise 
[bluetoothd.txt](https://github.com/bluez/bluez/files/14357863/bluetoothd.txt)

GreyXor, 2024-03-06:
Hello @Vudentz, do you need something else to investigate ? Thank you

Vudentz, 2024-03-06:
```
Feb 20 17:04:39 gregoire-laptop bluetoothd[201843]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 70:5A:6F:62:B1:35: Host is down (112)
Feb 20 17:04:39 gregoire-laptop bluetoothd[201843]: profiles/audio/avdtp.c:avdtp_ref() 0x559999ff4fc0: ref=3
Feb 20 17:04:39 gregoire-laptop bluetoothd[201843]: profiles/audio/avdtp.c:connection_lost() Disconnected from 70:5A:6F:62:B1:35
Feb 20 17:04:39 gregoire-laptop bluetoothd[201843]: profiles/audio/a2dp.c:discover_cb() version 0x0000 err 0x7fffea01d9e0
Feb 20 17:04:39 gregoire-laptop bluetoothd[201843]: profiles/audio/avdtp.c:avdtp_unref() 0x559999ff4fc0: ref=2
Feb 20 17:04:39 gregoire-laptop bluetoothd[201843]: src/service.c:change_state() 0x559999fe9b10: device 70:5A:6F:62:B1:35 profile a2dp-sink state changed: connecting -> disconnected (-112)
Feb 20 17:04:39 gregoire-laptop bluetoothd[201843]: src/device.c:device_profile_connected() a2dp-sink Host is down (112)
Feb 20 17:04:39 gregoire-laptop bluetoothd[201843]: profiles/audio/a2dp.c:setup_unref() 0x559999fe7df0: ref=0
Feb 20 17:04:39 gregoire-laptop bluetoothd[201843]: profiles/audio/a2dp.c:setup_free() 0x559999fe7df0
Feb 20 17:04:39 gregoire-laptop bluetoothd[201843]: profiles/audio/avdtp.c:avdtp_unref() 0x559999ff4fc0: ref=1
Feb 20 17:04:39 gregoire-laptop bluetoothd[201843]: profiles/audio/sink.c:sink_set_state() State changed /org/bluez/hci0/dev_70_5A_6F_62_B1_35: SINK_STATE_CONNECTING -> SINK_STATE_DISCONNECTED
```

Seems like this is the result of page timeout which result in Host is down (112) error, so it doesn't seem your controller is able to find the device to connect.

GreyXor, 2024-03-07:
@Vudentz thank you, that would mean that my controller is at fault? It's only the audio peripherals that are causing the problem.

Vudentz, 2024-03-08:
@GreyXor have you tried with another controller, perhaps there is something special to audio that causes this connection timeout.

GreyXor, 2024-03-08:
> @GreyXor have you tried with another controller, perhaps there is something special to audio that causes this connection timeout.

Yes, with AMD RZ616 everything is working as expected. I only have this issue with QCNCM865

Vudentz, 2024-03-08:
> > @GreyXor have you tried with another controller, perhaps there is something special to audio that causes this connection timeout.
> 
> Yes, i've tested with AMD RZ616 everything is working as expected. Only QCNCM865 won't connect to audio devices.

Well then I guess it is time to consult with the manufactorer to know if there is something else they are expecting.

GreyXor, 2024-04-03:
> Well then I guess it is time to consult with the manufactorer to know if there is something else they are expecting.

Thanks, can you confirm that it's not an bluez issue ?

GreyXor, 2024-05-20:
@Vudentz hello, gentle ping

GreyXor, 2024-05-27:
> Well then I guess it is time to consult with the manufactorer to know if there is something else they are expecting.

@Vudentz Thank you, so you're confirming that it's not a bluez issue?

Idesmi, 2024-08-27:
@GreyXor Do you have updates on this? I am a new user of QCNCM865 and experiencing a similar problem.

GreyXor, 2024-08-27:
> @GreyXor Do you have updates on this? I am a new user of QCNCM865 and experiencing a similar problem.

Sadly no, I don't know what to do to debug more this issue. HCI look clean :shrug: 
Do you have AMD CPU ? I maybe found something that it's related to AMD Sound Engine (or something like that)

fbruffaert, 2024-09-01:
Having the same issue with latest kernel.

GreyXor, 2024-09-20:
> It pairs, trusts and connects fine, but when the audio device comes up it crashes wireplumber and disconnects from the device:
> 
> ```
> kernel: ath12k_pci 0000:01:00.0: failed to enqueue rx buf: -28
> bluetoothd[9308]: profiles/audio/avdtp.c:cancel_request() Start: Connection timed out (110)
> wireplumber[3952]: spa.bluez5: Acquire /org/bluez/hci0/dev_AC_80_0A_7B_AC_0D/sep3/fd0 returned error: org.bluez.Error.Failed
> wireplumber[3952]: pw.node: (bluez_output.AC_80_0A_7B_AC_0D.1-159) running -> error (Received error event)
> wireplumber[3952]: spa.bluez5: Failure in Bluetooth audio transport /org/bluez/hci0/dev_AC_80_0A_7B_AC_0D/sep3/fd0
> pipewire[3951]: pw.node: (bluez_output.AC_80_0A_7B_AC_0D.1-159) running -> error (Received error event)
> bluetoothd[9308]: profiles/audio/avdtp.c:cancel_request() Abort: Connection timed out (110)
> kded6[4174]: org.kde.pulseaudio: No object for name "bluez_output.AC_80_0A_7B_AC_0D.1"
> kdeconnectd[4512]: 2024-09-20T11:04:20 org.kde.pulseaudio: No object for name "bluez_output.AC_80_0A_7B_AC_0D.1"
> plasmashell[4223]: org.kde.pulseaudio: No object for name "bluez_output.AC_80_0A_7B_AC_0D.1"
> wireplumber[3952]: wp-event-dispatcher: <WpAsyncEventHook:0x55c61b0eb400> failed: failed to configure si-standard-link <WpSiStandardLink:0x55c61b2dba50>
> plasmashell[4223]: org.kde.pulseaudio: context kaput
> ```

Thanks, so there's a chance that the issue is from pipewire/wireplumber? Should I report it there ?

clementpoiret, 2024-10-08:
Same issue here...
However, audio doesn't cause disconnection on my side, it is only very noisy / loose a lot of packets. It's quite impossible to ear something clearly.

GreyXor, 2024-10-08:
FYI I also report it here : https://gitlab.freedesktop.org/pipewire/wireplumber/-/issues/723

closed 2024-11-21 by GreyXor

ormandj, 2024-11-21:
> I’ve identified the issue causing Bluetooth audio problems and that's not a bluez issue. I will push a patch to Linux to resolve it.

Could you update this ticket with a link when you do? I'd like to follow along. We greatly appreciate your work on this!

GreyXor, 2024-11-21:
> > I’ve identified the issue causing Bluetooth audio problems and that's not a bluez issue. I will push a patch to Linux to resolve it.
> 
> Could you update this ticket with a link when you do? I'd like to follow along. We greatly appreciate your work on this!

Here it is, thanks!

https://patchwork.kernel.org/project/bluetooth/patch/20241121163259.180589-1-greyxor@protonmail.com/

jdelic, 2024-11-22:
@GreyXor 

Thank you for finding the missing vendor id! I just applied your patch to stock kernel 6.11.9 and tested it on my system (Gigabyte Aorus Master X870e) which has a WCN785x

```
$ lspci -v
0f:00.0 Network controller: Qualcomm Technologies, Inc WCN785x Wi-Fi 7(802.11be) 320MHz 2x2 [FastConnect 7800] (rev 01
)
        Subsystem: Foxconn International, Inc. WCN785x Wi-Fi 7(802.11be) 320MHz 2x2 [FastConnect 7800]
        Flags: bus master, fast devsel, latency 0, IRQ 232, IOMMU group 24
        Memory at a0200000 (64-bit, non-prefetchable) [size=2M]
        Capabilities: [40] Power Management version 3
        Capabilities: [50] MSI: Enable+ Count=16/32 Maskable+ 64bit-
        Capabilities: [70] Express Endpoint, MSI 00
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [148] Secondary PCI Express
        Capabilities: [158] Transaction Processing Hints
        Capabilities: [1e4] Latency Tolerance Reporting
        Capabilities: [1ec] L1 PM Substates
        Kernel driver in use: ath12k_pci
        Kernel modules: ath12k

$ cat /sys/kernel/debug/devices
T:  Bus=01 Lev=01 Prnt=01 Port=09 Cnt=03 Dev#=  4 Spd=12   MxCh= 0
D:  Ver= 1.10 Cls=e0(wlcon) Sub=01 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=0489 ProdID=e10d Rev= 0.01
C:* #Ifs= 2 Cfg#= 1 Atr=e0 MxPwr=100mA
I:* If#= 0 Alt= 0 #EPs= 3 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=81(I) Atr=03(Int.) MxPS=  16 Ivl=1ms
E:  Ad=82(I) Atr=02(Bulk) MxPS=  64 Ivl=0ms
E:  Ad=02(O) Atr=02(Bulk) MxPS=  64 Ivl=0ms
I:* If#= 1 Alt= 0 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=83(I) Atr=01(Isoc) MxPS=   0 Ivl=1ms
E:  Ad=03(O) Atr=01(Isoc) MxPS=   0 Ivl=1ms
I:  If#= 1 Alt= 1 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=83(I) Atr=01(Isoc) MxPS=   9 Ivl=1ms
E:  Ad=03(O) Atr=01(Isoc) MxPS=   9 Ivl=1ms
I:  If#= 1 Alt= 2 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=83(I) Atr=01(Isoc) MxPS=  17 Ivl=1ms
E:  Ad=03(O) Atr=01(Isoc) MxPS=  17 Ivl=1ms
I:  If#= 1 Alt= 3 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=83(I) Atr=01(Isoc) MxPS=  25 Ivl=1ms
E:  Ad=03(O) Atr=01(Isoc) MxPS=  25 Ivl=1ms
I:  If#= 1 Alt= 4 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=83(I) Atr=01(Isoc) MxPS=  33 Ivl=1ms
E:  Ad=03(O) Atr=01(Isoc) MxPS=  33 Ivl=1ms
I:  If#= 1 Alt= 5 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=83(I) Atr=01(Isoc) MxPS=  49 Ivl=1ms
E:  Ad=03(O) Atr=01(Isoc) MxPS=  49 Ivl=1ms
I:  If#= 1 Alt= 6 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=83(I) Atr=01(Isoc) MxPS=  63 Ivl=1ms
E:  Ad=03(O) Atr=01(Isoc) MxPS=  63 Ivl=1ms
I:  If#= 1 Alt= 7 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=83(I) Atr=01(Isoc) MxPS=  65 Ivl=1ms
E:  Ad=03(O) Atr=01(Isoc) MxPS=  65 Ivl=1ms
```

Unfortunately I still get only timeouts when trying to connect to bluetooth headphones. Any ideas what to try next?

jdelic, 2024-11-22:
@GreyXor 
> Hi, I see that your ProdID is e10d. My patch only add e10a. I don't know what Qualcomm's point is in adding 50 different ProdIDs. 

oh, of course :man_facepalming: 

While I try this... should that ProdID also have `BTUSB_VALID_LE_STATES` like the other entry for WCN785x? I couldn't so far find out what that actually does, but I definitely do see LE devices around me in bluetoothctl, so maybe _you_ know?

GreyXor, 2024-11-22:
True that audio work but not LE Audio. So maybe that something to check, thank you

GreyXor, 2024-11-22:
I don't see any others ID that use BTUSB_INVALID_LE_STATES. And that's not make the LC3 working

yoqto, 2024-12-22:
> https://patchwork.kernel.org/project/bluetooth/patch/20241121180742.156230-1-greyxor@protonmail.com/

I manually added the VID and PID as per your patch and this fixes bluetooth on my MSI MPG X870E Carbon Wifi - thank you so much!

garciadias, 2025-01-04:
Hi @yoqto,
Sorry, I am having the same problem, but I am not sure how to apply the patch.
Which file have you modified? I can't find the `drivers/bluetooth/btusb.c` on my system.

I can only find binary files with similar name:
```
/usr/lib/modules/6.11.4-301.fc41.x86_64/kernel/drivers/bluetooth/btusb.ko.xz
/usr/lib/modules/6.12.6-200.fc41.x86_64/kernel/drivers/bluetooth/btusb.ko.xz
/usr/lib/modules/6.12.7-200.fc41.x86_64/kernel/drivers/bluetooth/btusb.ko.xz
/.snapshots/1/snapshot/usr/lib/modules/6.11.4-301.fc41.x86_64/kernel/drivers/bluetooth/btusb.ko.xz
/.snapshots/1/snapshot/usr/lib/modules/6.12.6-200.fc41.x86_64/kernel/drivers/bluetooth/btusb.ko.xz
/.snapshots/1/snapshot/usr/lib/modules/6.12.7-200.fc41.x86_64/kernel/drivers/bluetooth/btusb.ko.xz
```

Here are all the logs and specifications for my system:

```
         .';:cccccccccccc:;,.             ----------
      .;cccccccccccccccccccccc;.          OS: Fedora Linux 41 (Workstation Edition) x86_64
    .:cccccccccccccccccccccccccc:.        Host: Blade 14 - RZ09-0508 (10.04)
  .;ccccccccccccc;.:dddl:.;ccccccc;.      Kernel: Linux 6.12.7-200.fc41.x86_64
 .:ccccccccccccc;OWMKOOXMWd;ccccccc:.     Uptime: 2 mins
.:ccccccccccccc;KMMc;cc;xMMc;ccccccc:.    Packages: 2424 (rpm), 22 (flatpak), 104 (brew)
,cccccccccccccc;MMM.;cc;;WW:;cccccccc,    Shell: fish 3.7.1
:cccccccccccccc;MMM.;cccccccccccccccc:    Display (BOE0B94): 2560x1600 @ 60 Hz (as 1704x1065) in 14" [Built-in]
:ccccccc;oxOOOo;MMM000k.;cccccccccccc:    DE: GNOME 47.2
cccccc;0MMKxdd:;MMMkddc.;cccccccccccc;    WM: Mutter (Wayland)
ccccc;XMO';cccc;MMM.;cccccccccccccccc'    WM Theme: Adwaita
ccccc;MMo;ccccc;MMW.;ccccccccccccccc;     Theme: Adwaita [GTK2/3/4]
ccccc;0MNc.ccc.xMMd;ccccccccccccccc;      Icons: Adwaita [GTK2/3/4]
cccccc;dNMWXXXWM0:;cccccccccccccc:,       Font: Cantarell (11pt) [GTK2/3/4]
cccccccc;.:odl:.;cccccccccccccc:,.        Cursor: Adwaita (24px)
ccccccccccccccccccccccccccccc:'.          Terminal: tmux 3.5a
:ccccccccccccccccccccccc:;,..             CPU: AMD Ryzen 9 8945HS w/ Radeon 780M Graphics (16) @ 5.26 GHz
 ':cccccccccccccccc::;,.                  GPU 1: NVIDIA GeForce RTX 4070 Max-Q / Mobile [Discrete]
                                          GPU 2: AMD Phoenix3 [Integrated]
                                          Memory: 6.61 GiB / 30.62 GiB (22%)
                                          Swap: 0 B / 8.00 GiB (0%)
                                          Disk (/): 192.36 GiB / 952.27 GiB (20%) - btrfs
                                          Local IP (wlp3s0): 192.168.1.36/24
                                          Battery (Blade): 72% (3 hours, 10 mins remaining) [Discharging]
                                          Locale: en_GB.UTF-8

                                                                  
                                                                  
=== lspci -knn | grep -i net ===
pcilib: Error reading /sys/bus/pci/devices/0000:00:08.3/label: Operation not permitted
03:00.0 Network controller [0280]: Qualcomm Technologies, Inc WCN785x Wi-Fi 7(802.11be) 320MHz 2x2 [FastConnect 7800] [17cb:1107] (rev 01)

=== lsmod | grep ath12 ===
ath12k                589824  0
qmi_helpers            40960  1 ath12k
mac80211             1777664  1 ath12k
cfg80211             1404928  2 ath12k,mac80211
mhi                   155648  2 ath12k,qrtr_mhi

=== lsmod | grep bt ===
btusb                  77824  0
btrtl                  36864  1 btusb
btintel                69632  1 btusb
btbcm                  24576  1 btusb
btmtk                  32768  1 btusb
bluetooth            1052672  44 btrtl,btmtk,btintel,btbcm,bnep,btusb,rfcomm

=== systemctl status bluetooth ===
● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset: enabled)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf, 50-keep-warm.conf
     Active: active (running) since Sat 2025-01-04 11:51:53 GMT; 2min 42s ago
[... 158 lines cut ...]
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=07 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 6.12
S:  Manufacturer=Linux 6.12.7-200.fc41.x86_64 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=0000:67:00.4
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=08 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=10000 MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.10 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 6.12
S:  Manufacturer=Linux 6.12.7-200.fc41.x86_64 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=0000:67:00.4
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

```

yoqto, 2025-01-04:
> Hi @yoqto, Sorry, I am having the same problem, but I am not sure how to apply the patch. Which file have you modified? I can't find the `drivers/bluetooth/btusb.c` on my system.

Hi @garciadias,

I build my kernel from the upstream vanilla sources and run another distro. For Fedora in particular, I believe that [this guide for testing kernel patches](https://docs.fedoraproject.org/en-US/quick-docs/kernel-testing-patches/) would be the easiest way to get you started on testing a patch.

Looking at your posted specs, however, this may or may not work: the line with `Vendor=xxx ProdId=yyy` was cut off from your `/sys/kernel/debug/usb/devices` output. If it shows `Vendor=0489 ProdID=e10a` for you, here's the patch I applied on 6.12.6:

```patch
diff --git a/drivers/bluetooth/btusb.c b/drivers/bluetooth/btusb.c
index 11755cb1eb16..96d33a6c909f 100644
--- a/drivers/bluetooth/btusb.c
+++ b/drivers/bluetooth/btusb.c
@@ -371,6 +371,8 @@ static const struct usb_device_id quirks_table[] = {
        /* QCA WCN785x chipset */
        { USB_DEVICE(0x0cf3, 0xe700), .driver_info = BTUSB_QCA_WCN6855 |
                                                     BTUSB_WIDEBAND_SPEECH },
+       { USB_DEVICE(0x0489, 0xe10a), .driver_info = BTUSB_QCA_WCN6855 |
+                                                    BTUSB_WIDEBAND_SPEECH },
 
        /* Broadcom BCM2035 */
        { USB_DEVICE(0x0a5c, 0x2009), .driver_info = BTUSB_BCM92035 },
```
It seems that a bunch of different BT chipsets are grouped under vendor `0489`. According to these two pages for your hardware ([1](https://linux-hardware.org/?probe=cc1f5421e7), [2](https://linux-hardware.org/?probe=bf83f11224)), you should have vendor `0489`, product id `e11b`, not product id `e10a`. Going by [neighboring product IDs (`e118`, `e11e`)](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/bluetooth/btusb.c?h=v6.12.7#n645), you might have to use `BTUSB_MEDIATEK` instead of `BTUSB_QCA_WCN6855` with your product id.

With that patch applied, I can see the module loading its firmware at boot on my system:
```
# dmesg | grep -i blue
[   21.086045] [   T2366] Bluetooth: hci0: using rampatch file: qca/rampatch_usb_00190200.bin
[   21.086145] [   T2366] Bluetooth: hci0: QCA: patch rome 0x190200 build 0x5656, firmware rome 0x190200 build 0x43fb
[   21.622267] [   T2366] Bluetooth: hci0: using NVM file: qca/nvm_usb_00190200.bin
```
File names `qca/*.bin` are relative to `/usr/lib/firmware/` as installed via the `linux-firmware` package and should exist on your system (Fedora 41 ships version 20241017 as per [this page](https://packages.fedoraproject.org/pkgs/linux-firmware/linux-firmware/) -> [upstream list of files in `qca/`](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/qca?h=20241017)).

Best of luck!

garciadias, 2025-01-04:
Hi @yoqto,

I have updated my comment to include the vendor and product id. They are: Vendor=1d6b ProdID=0002, which certainly is not in the patch.  

You gave me a lot to investigate, so I will follow the links and try to learn more about the issue.

Thank you very much! I really appreciate your help.

yoqto, 2025-01-04:
No problem!

Vendor `1d6b` is [Linux Foundation](https://devicehunt.com/view/type/usb/vendor/1D6B) - that output doesn't make sense :raised_eyebrow:
`Vendor=1532 ProdID=02b6` is your [keyboard](https://github.com/openrazer/openrazer/tree/master). I believe you missed a few lines when copying the output of the command.

garciadias, 2025-01-04:
Sorry, you are right. Something strange was happening in my terminal. Now I see two lines with other vendors:
P:  Vendor=1532 ProdID=02b6 Rev= 2.00
S:  Manufacturer=Razer
S:  Product=Razer Blade
and
P:  Vendor=0489 ProdID=e11b Rev= 0.01

Z-Y00, 2025-01-31:
For those of you need help about compiling Linux kernel and apply the patch mentioned above, you can refer to my note on how to do it.
I build the kernel module and applied the patch, it now works very well for my bluetooth:

https://github.com/Z-Y00/scripts/blob/master/linux_bluetooth_recompile.md#english

tarka, 2025-03-26:
@GreyXor I've encountered this on an MSI MAG B850 Tomahawk Max WIFI. Manually applying your `0x0489, 0xe10a` patch to 6.14 fixed it. I looks like your patch didn't make it into mainline?

GreyXor, 2025-03-27:
> [@GreyXor](https://github.com/GreyXor) I've encountered this on an MSI MAG B850 Tomahawk Max WIFI. Manually applying your `0x0489, 0xe10a` patch to 6.14 fixed it. I looks like your patch didn't make it into mainline?
> 
> Update: It looks like this device along with a bunch of others was added to bluetooth-next overnight; specifically [this commit](https://web.git.kernel.org/pub/scm/linux/kernel/git/bluetooth/bluetooth-next.git/commit/?id=2dd1c1eee3e496fcc16971be4db5bb792a36025c).

Hello, exactly. Linux 6.15 will have the fix