bluez/bluez #752: Logitech MX Master 3S unable to connect after hibernation

Thulium-Drake, 2024-02-21:
Hi there!

My Framework 13 laptop has some issues reconnecting to my mouse after it's been hibernated. When resuming the laptop, I see a lot of notifications mentioning connect and disconnect events for the device. In the bluetoothd logs I see the following messages:

```
Feb 08 12:29:48 archon bluetoothd[5234]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
Feb 08 12:29:48 archon bluetoothd[5234]: profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Request attribute has encountered an unlikely error
Feb 08 12:29:48 archon bluetoothd[5234]: profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attribute has encountered an unlikely error
Feb 08 12:29:48 archon bluetoothd[5234]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Feb 08 12:29:48 archon bluetoothd[5234]: profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attribute has encountered an unlikely error
Feb 08 12:29:48 archon bluetoothd[5234]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Feb 08 12:29:48 archon bluetoothd[5234]: profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attribute has encountered an unlikely error
Feb 08 12:29:48 archon bluetoothd[5234]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Feb 08 12:29:48 archon bluetoothd[5234]: profiles/input/hog-lib.c:proto_mode_read_cb() Protocol Mode characteristic read failed: Request attribute has encountered an unlikely error
Feb 08 12:29:49 archon bluetoothd[5234]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
Feb 08 12:29:49 archon bluetoothd[5234]: profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Request attribute has encountered an unlikely error
Feb 08 12:29:49 archon bluetoothd[5234]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Feb 08 12:29:49 archon bluetoothd[5234]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Feb 08 12:29:49 archon bluetoothd[5234]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Feb 08 12:29:49 archon bluetoothd[5234]: profiles/input/hog-lib.c:proto_mode_read_cb() Protocol Mode characteristic read failed: Request attribute has encountered an unlikely error
Feb 08 12:29:49 archon bluetoothd[5234]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
Feb 08 12:29:49 archon bluetoothd[5234]: profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Request attribute has encountered an unlikely error
Feb 08 12:29:49 archon bluetoothd[5234]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Feb 08 12:29:49 archon bluetoothd[5234]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Feb 08 12:29:49 archon bluetoothd[5234]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Feb 08 12:29:49 archon bluetoothd[5234]: profiles/input/hog-lib.c:proto_mode_read_cb() Protocol Mode characteristic read failed: Request attribute has encountered an unlikely error
```
I'm running Fedora 39 with ```Linux archon 6.7.3-200.fc39.x86_64 #1 SMP PREEMPT_DYNAMIC Thu Feb  1 03:29:52 UTC 2024 x86_64 GNU/Linux```

> rpm -qa | grep blue
```
$ rpm -qa | grep blue
bluez-5.72-1.fc39.x86_64
bluez-libs-5.72-1.fc39.x86_64
bluez-obexd-5.72-1.fc39.x86_64
blueman-2.3.5-8.fc39.x86_64
```

> bluetoothctl show
```
Controller 14:AC:60:D8:3B:C8 (public)
	Manufacturer: 0x0046 (70)
	Version: 0x0b (11)
	Name: archon
	Alias: archon
	Class: 0x007c010c (8126732)
	Powered: yes
	PowerState: on
	Discoverable: no
	DiscoverableTimeout: 0x000000b4 (180)
	Pairable: yes
	UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
	UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00005005-0000-1000-8000-0002ee000001)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
	UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v1D6Bp0246d0548
	Discovering: no
	Roles: central
	Roles: peripheral
Advertising Features:
	ActiveInstances: 0x00 (0)
	SupportedInstances: 0x14 (20)
	SupportedIncludes: tx-power
	SupportedIncludes: appearance
	SupportedIncludes: local-name
	SupportedSecondaryChannels: 1M
	SupportedSecondaryChannels: 2M
	SupportedSecondaryChannels: Coded
```

> bluetoothctl info
```
Device D0:AE:C4:43:A7:4B (random)
	Name: MX Master 3S
	Alias: MX Master 3S
	Appearance: 0x03c2 (962)
	Icon: input-mouse
	Paired: yes
	Bonded: yes
	Trusted: no
	Blocked: no
	Connected: yes
	WakeAllowed: yes
	LegacyPairing: no
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
	UUID: Human Interface Device    (00001812-0000-1000-8000-00805f9b34fb)
	UUID: Logitech International SA (0000fd72-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00010000-0000-1000-8000-011f2000046d)
	Modalias: usb:v046DpB034d0006
	ManufacturerData.Key: 0x0006 (6)
	ManufacturerData.Value:
  03 00 80                                         ...             
	ServiceData.0000fd72-0000-1000-8000-00805f9b34fb:
  10 01 01 34 b0 02 02 00                          ...4....        
	Battery Percentage: 0x64 (100)
```
Please let me know if you need more information :-)

Thulium-Drake, 2024-02-22:
So I have more information on _when_ this occurs, every time the laptop has been powered off (hibernation, reboots and poweroffs).
So the good news is, I can reproduce it really easily, bad news, it's still really annoying :clown_face:

AriaMoradi, 2024-03-17:
Facing the same issue on Arch with my mouse and keyboard, after a reboot the devices connect automatically but no input is sent. The only way to fix it is to remove and re-pair the devices on every reboot.

erkkimon, 2024-04-11:
Exactly. I can confirm this happens on Arch Linux, super consistently. :grinning:

jiriks74, 2024-05-16:
I'm facing this issue on NixOS unstable using `Bluetooth daemon 5.72` and Logitech Lift. The laptop is Lenovo Ideapad 5 Pro 14iAH7 with Intel Bluetooth hardware.

<details>
<summary>Output of <code>hciconfig -a</code></summary>

```
hci0:   Type: Primary  Bus: USB
        BD Address: 14:75:5B:C5:07:B9  ACL MTU: 1021:4  SCO MTU: 96:6
        UP RUNNING PSCAN 
        RX bytes:26142 acl:108 sco:0 events:3408 errors:0
        TX bytes:759834 acl:72 sco:0 commands:3244 errors:0
        Features: 0xbf 0xfe 0x0f 0xfe 0xdb 0xff 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
        Link policy: RSWITCH SNIFF 
        Link mode: PERIPHERAL ACCEPT 
        Name: 'IdeaPad5Pro'
        Class: 0x7c010c
        Service Classes: Rendering, Capturing, Object Transfer, Audio, Telephony
        Device Class: Computer, Laptop
        HCI Version: 5.2 (0xb)  Revision: 0x367b
        LMP Version: 5.2 (0xb)  Subversion: 0x367b
        Manufacturer: Intel Corp. (2)
```

</details>

jiriks74, 2024-05-16:
The timeframe points to this commit 247ae8524888268bb075ff90a3d77a9b4449d08f

> [!Note]
> To be fair I do not understand Bluetooth and the project, I'm just looking for changes made to the files that report issues during the time frame this issue appeared.

jiriks74, 2024-05-16:
I can confirm that 247ae8524888268bb075ff90a3d77a9b4449d08f is not the culprit as reverting it didn't do anything.

jiriks74, 2024-05-16:
Rolling back to `5.70` works for me. I'll investigate which version introduced this bug.

jiriks74, 2024-05-16:
`5.71` is unaffected. This means that this bug was introduced in `5.72`.

jiriks74, 2024-05-16:
@Thulium-Drake were there any gatt db errors in your logs?

AriaMoradi, 2024-05-21:
> Rolling back to 5.70 works for me. I'll investigate which version introduced this bug.

I downgraded `bluez` to `5.70-2` on ArchLinux an can confirm that the issue is resolved.

jiriks74, 2024-05-21:
> I downgraded `bluez` to `5.70-2`

You can go up to `5.71` if that's what you want. Weird that this had no acknowledgement yet...

Thulium-Drake, 2024-05-21:
Sorry for the late reply, I was enjoying a week off :)

With regards to downgrading, I already recently upgraded my laptop to Fedora 40, which runs a newer version of bluez:

> rpm -qa | grep blue

```
bluez-libs-5.75-1.fc40.x86_64
bluez-5.75-1.fc40.x86_64
bluez-obexd-5.75-1.fc40.x86_64
blueman-2.4.1-1.fc40.x86_64
```

The good news is that the effect didn't change :-) the bad news is that I don't know if I can still downgrade to that version (it's not in the repos)

@jiriks74 no, that term doesn't ring a bell, I added all the logs I could find that triggered when I tried to recreate the problem. Is there any location I should look for them? So far I checked journal and dmesg.

jiriks74, 2024-05-21:
> Sorry for the late reply, I was enjoying a week off :)

I hope you enjoyed that well then!

> it's not in the repos

I guess that the other option is to uninstall it and compile it yourself?

> I added all the logs I could find that triggered when I tried to recreate the problem. Is there any location I should look for them? So far I checked journal and dmesg.

Not in particular. I had it in the journal so I was just curious whether you had that as well. Did it appear now that you have a newer version?

Iraijus, 2024-06-23:
I wanted to add to this issue, since it's still open and Thulium-Drake mentioned that the problem happened on a Framework laptop. I'm using Arch Linux on a Framework 16 and had the same issue with a Logitech MX Vertical and an Iqunix F96 keyboard. Both are now able to reconnect again after hibernation, because I downgraded bluez to 5.70-2.

I didn't want to just add a 'same problem' message, but it did seem like the extra info could be helpful in some way. In any case, I'm glad that I don't need to repair my devices after every reboot anymore.

AriaMoradi, 2024-06-24:
> I downgraded bluez to 5.70-2

The last good release on Arch is 5.71-3

To add the the issue my keyboard is Rapoo model K8100M and the Mouse is a Xiaomi model WXSMSBMW02.

The issue shouldn't be isolated to Logitech, unless both companies are using the same tech stack as Logitech.

Thulium-Drake, 2024-06-24:
Thanks for the updates folks! I haven't had the time to downgrade the bluez package on my laptop yet (I'll have to recompile it as it's not present in F40..)

Meanwhile, as it's been multiple devices, on both reboots and hibernation, I've updated the title to better reflect this!

AriaMoradi, 2024-06-24:
#875 might be a duplicate or related issue.

TrainDoctor, 2024-06-24:
For anyone with a Framework 16, I downgraded to bluez 5.71-3.fc40 in an attempt to resolve a similair issue on that hardware and it had no effect in resolving the issue. I have reported the bug to fedora and continue to try to report the bug to Framework in the hopes that it will be resolved. Hopefully ya'll will be able to get your issue's resolved as well.

Thulium-Drake, 2024-08-04:
So, the plot thickens... And not in a good way, so @kevinquick and I both have the same Framework model (13) with the same ```MEDIATEK Corp. MT7922 802.11ax PCI Express Wireless Network Adapter``` and he tested the Logitech mouse on his laptop (which also runs Fedora 40!) without issues. So we decided to switch adapters:

For clarity:

- My laptop (OG):
  - Mouse connects fine
  - Reboots/Hibernation (suspend to disk) leads to the behaviour descrived in the OP
  - Side issue, not bluetooth, but related to the chip, after hibernation the chip had to be "reset" by unloading the kernel module before WiFi connectivity is available again
- @kevinquick's laptop (OG):
  - Mouse connects fine
  -  Reboots had no impact, hibernation is not set up on his laptop, so we didn't test that

Then after the switch, _BOTH_ our laptops have no issues whatsoever with bluetooth connectivity with the same mouse. Both laptops still have a connection available with the Logitech mouse after a reboot.

So, this rules out everything in the chain from the hardware in the mouse to the hardware in the laptop, but I have no f-ing idea _why_. "Fun" additional note, when we switch the adapters back, the problems return to my laptop only.

Thanks mediatek, I guess? :dizzy_face:

jiriks74, 2024-08-05:
So, for me the problems went away. I have a Lenovo IdeaPad 5 Pro with a 12th gen Core i7. I think the wireless card is from Intel and I had the same issues with my Logitech Lift.

I reinstalled the OS since my last comment, NixOS unstable, and the problems are now gone. I think some bluez update fized it for me?

MightyPork, 2024-09-11:
Another confirmation from Arch Linux, after this change my MX Master 3 works again. The system is fully updated otherwise.

```
downgraded bluez (5.78-1 -> 5.70-2)
```

AriaMoradi, 2024-09-19:
While trying to troubleshoot my PS3 controller I found #673 and applied the fix from https://github.com/bluez/bluez/issues/673#issuecomment-1858431729 and that fixed my keyboard and mouse issue completley. Related issue?

SouthPenguinJay, 2024-10-04:
Could very well be related, some of seems to have it solved while others do not, I for example have tried about everything in the other thread to no success.

> While trying to troubleshoot my PS3 controller I found #673 and applied the fix from https://github.com/bluez/bluez/issues/673#issuecomment-1858431729 and that fixed my keyboard and mouse issue completley. Related issue?

jiriks74, 2024-10-04:
@SouthPenguinJay 
> I for example have tried about everything in the other thread to no success.

Does an older version work for you?

gguman, 2024-12-29:
Archlinux. Bluez 5.79-1. The problem still persists

partofthething, 2025-05-18:
FWIW I upgraded my framework 13 to ubuntu 25.04 and have not experienced this issue for some time. I currently am on bluez version `5.79-2ubuntu1`

CRiS-SVRS, 2025-07-30:
I have this exact issue with a bt mouse on arch and latest bluez 5.83. 
Downgraded bluez, bluez-libs, bluez-utils and bluez-hid2hci to 5.82 and the issue is gone.

MightyPork, 2025-07-30:
> 5.82

what is your mouse? if a patch downgrade really works, that should be bisectable. strange that it'd oscillate between broken and working like this though

CRiS-SVRS, 2025-07-30:
sorry, but as soon as i woke up my pc the issue is there again... 
i downgraded all versions until 5.78 but no way, after stand by the mouse won't reconnect and the journal shows those errors.
anyway mine is a budget hp bt mouse.

CRiS-SVRS, 2025-08-01:
downgraded to 5.75 and then to 5.70. nothing seems to work... 
every time i start pc or wake it from standby, pc and mouse try to connect for a long time. if i shutdown the mouse, the errors pop up in journal. after they appear, i power on the mouse again and they connect instantly and the connection never drops.

CRiS-SVRS, 2025-08-14:
I would like to add that this behavior only occurs at the first bluetooth connection, which does not necessarily have to be a mouse. I tested a pair of headphones and a controller and the exact same thing happens.

closed 2025-09-29 by github-actions[bot]