bluez/bluez #755: bluez: gatt_db_service_destroy(): bluetoothd killed by SIGSEGV

kparal, 2024-02-26:
I've seen this bluez crash when I removed my bluetooth mouse connection from gnome-control-center. It was removed, but bluetoothd crashed at the same time.

Traceback:
```
Thread 1 (Thread 0x7f4151b7d800 (LWP 1472)):
#0  malloc_consolidate (av=0x7f4152029ac0 <main_arena>) at malloc.c:4842
        idx = <optimized out>
        fb = 0x7f4152029ad0 <main_arena+16>
        maxfb = 0x7f4152029b18 <main_arena+88>
        p = 0x5b15496b0
        nextp = <optimized out>
        unsorted_bin = 0x7f4152029b20 <main_arena+96>
        first_unsorted = <optimized out>
        nextchunk = <optimized out>
        size = <optimized out>
        nextsize = <optimized out>
        prevsize = <optimized out>
        nextinuse = <optimized out>
#1  0x00007f4151eed4d5 in _int_free_maybe_consolidate (av=0x7f4152029ac0 <main_arena>, size=<optimized out>) at malloc.c:4772
        __PRETTY_FUNCTION__ = "_int_free_maybe_consolidate"
#2  0x00007f4151eed7fe in _int_free_maybe_consolidate (size=<optimized out>, av=<optimized out>) at malloc.c:4695
        __PRETTY_FUNCTION__ = "_int_free_maybe_consolidate"
        heap = <optimized out>
#3  0x00007f4151eed9ea in _int_free (av=<optimized out>, p=p@entry=0x55e6b14c4020, have_lock=<optimized out>, have_lock@entry=0) at malloc.c:4639
        size = <optimized out>
        fb = <optimized out>
#4  0x00007f4151ef03de in __GI___libc_free (mem=0x55e6b14c4030) at malloc.c:3391
        ar_ptr = <optimized out>
        p = 0x55e6b14c4020
        err = 0
#5  0x000055e6b0a1bfbd in gatt_db_service_destroy (data=0x55e6b14b9620) at src/shared/gatt-db.c:438
        service = 0x55e6b14b9620
        i = <optimized out>
#6  0x000055e6b0a24461 in queue_remove_all (destroy=<optimized out>, user_data=0x0, function=0x0, queue=0x55e6b14a3390) at src/shared/queue.c:341
        tmp = 0x55e6b149d200
        entry = 0x0
        count = <optimized out>
        data = <optimized out>
        entries = <optimized out>
#7  queue_destroy (destroy=<optimized out>, queue=0x55e6b14a3390) at src/shared/queue.c:60
No locals.
#8  queue_destroy (destroy=<optimized out>, queue=0x55e6b14a3390) at src/shared/queue.c:55
No locals.
#9  gatt_db_destroy (db=0x55e6b1485480) at src/shared/gatt-db.c:459
No locals.
#10 gatt_db_unref (db=0x55e6b1485480) at src/shared/gatt-db.c:472
No locals.
#11 0x000055e6b09f2cd9 in gatt_db_unref (db=<optimized out>) at src/shared/gatt-db.c:466
No locals.
#12 device_free (user_data=0x55e6b14aeba0) at src/device.c:818
        device = 0x55e6b14aeba0
        __btd_debug_desc = {file = 0x55e6b0a5b841 "src/device.c", flags = 0}
        __func__ = "device_free"
#13 0x000055e6b0a0ef0c in remove_interface (data=0x55e6b149aa40, name=name@entry=0x55e6b0a5b82f "org.bluez.Device1") at gdbus/object.c:660
        iface = 0x55e6b14b1a30
#14 0x000055e6b0a0f24f in g_dbus_unregister_interface (connection=0x55e6b146f1e0, path=<optimized out>, name=0x55e6b0a5b82f "org.bluez.Device1") at gdbus/object.c:1394
        data = 0x55e6b149aa40
#15 0x000055e6b09e4f7d in dev_disconnected (adapter=0x55e6b14b7090, addr=0x55e6b1471cb9, reason=2 '\002') at src/adapter.c:8475
        device = 0x55e6b14aeba0
        dst = "C3:0C:C9:46:1B:63"
        __btd_debug_desc = {file = 0x55e6b0a47f41 "src/adapter.c", flags = 0}
        __func__ = "dev_disconnected"
#16 0x000055e6b0a13522 in request_complete (mgmt=mgmt@entry=0x55e6b1471c30, status=0 '\000', opcode=opcode@entry=20, index=index@entry=0, length=length@entry=7, param=0x55e6b1471cb9) at src/shared/mgmt.c:306
        match = {opcode = 20, index = 0}
        request = <optimized out>
        __func__ = "request_complete"
#17 0x000055e6b0a13f0e in can_read_data (io=<optimized out>, user_data=0x55e6b1471c30) at src/shared/mgmt.c:394
        mgmt = 0x55e6b1471c30
        hdr = <optimized out>
        cc = 0x55e6b1471cb6
        cs = <optimized out>
        bytes_read = <optimized out>
        opcode = 20
        event = <optimized out>
        index = <optimized out>
        length = 10
        __func__ = "can_read_data"
#18 0x000055e6b0a35a19 in watch_callback (channel=<optimized out>, cond=<optimized out>, user_data=<optimized out>) at src/shared/io-glib.c:157
        watch = <optimized out>
        result = <optimized out>
        destroy = <optimized out>
#19 0x00007f41520e3e5c in g_main_dispatch (context=0x55e6b146c7d0) at ../glib/gmain.c:3476
        dispatch = 0x7f4152141830 <g_io_unix_dispatch>
        prev_source = 0x0
        begin_time_nsec = 46385347380262
        was_in_call = 0
        user_data = 0x55e6b1471fb0
        callback = 0x55e6b0a359f0 <watch_callback>
        cb_funcs = 0x7f41521d0380 <g_source_callback_funcs>
        cb_data = 0x55e6b1472060
        need_destroy = <optimized out>
        source = 0x55e6b1471fe0
        current = 0x55e6b1476e90
        i = 0
        __func__ = <optimized out>
#20 g_main_context_dispatch_unlocked (context=0x55e6b146c7d0) at ../glib/gmain.c:4284
No locals.
#21 0x00007f415213ef18 in g_main_context_iterate_unlocked.isra.0 (context=0x55e6b146c7d0, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../glib/gmain.c:4349
        max_priority = 2147483647
        timeout = 999
        some_ready = 1
        nfds = 31
        allocated_nfds = <optimized out>
        fds = 0x55e6b14a01e0
        begin_time_nsec = 46385001396669
#22 0x00007f41520e5447 in g_main_loop_run (loop=0x55e6b146c9b0) at ../glib/gmain.c:4551
        self = <optimized out>
        __func__ = "g_main_loop_run"
#23 0x000055e6b096ac4d in mainloop_run () at src/shared/mainloop-glib.c:66
No locals.
#24 mainloop_run_with_signal (func=0x55e6b09be520 <signal_callback>, user_data=0x0) at src/shared/mainloop-notify.c:188
        data = 0x55e6b1476cf0
        io = 0x55e6b1476d10
        ret = <optimized out>
        data = <optimized out>
        io = <optimized out>
        ret = <optimized out>
        __n = <optimized out>
        __s = <optimized out>
        __p = <optimized out>
#25 main (argc=<optimized out>, argv=<optimized out>) at src/main.c:1452
        context = <optimized out>
        err = 0x0
        sdp_mtu = 0
        sdp_flags = <optimized out>
        gdbus_flags = <optimized out>
        __func__ = "main"
        __btd_debug_desc = {file = 0x55e6b0a5398f "src/main.c", flags = 0}
```

Full traceback details:
https://bugzilla-attachments.redhat.com/attachment.cgi?id=2018658

Downstream Fedora bug with more details:
https://bugzilla.redhat.com/show_bug.cgi?id=2265883

Version:
```
bluez-5.72-1.fc39
Fedora 39
```

Vudentz, 2024-02-27:
@kparal could you please try to reproduce while running bluetoothd under valgrind, that should produce a more readable backtrace.

kparal, 2024-02-28:
Hi, I tried to remove my mouse a few more times. It worked about 10 times, but crashed once more, with a different traceback:
```
Thread 1 (Thread 0x7fc6adf8f800 (LWP 1402)):
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
        tid = <optimized out>
        ret = 0
        pd = <optimized out>
        old_mask = {__val = {94660775391184}}
        ret = <optimized out>
#1  0x00007fc6ae2f38a3 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
No locals.
#2  0x00007fc6ae2a18ee in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
        ret = <optimized out>
#3  0x00007fc6ae2898ff in __GI_abort () at abort.c:79
        save_stage = 1
        act = {__sigaction_handler = {sa_handler = 0x20, sa_sigaction = 0x20}, sa_mask = {__val = {140491304171164, 140724144264000, 140491304171882, 7, 6220111616, 128, 10182242192194494208, 2, 94660775389200, 140724144264016, 140491304184387, 94660775389200, 140724144264032, 140724144264064, 140491304054377, 0}}, sa_flags = 1925144320, sa_restorer = 0x5617ede431e0}
#4  0x00007fc6ae28a7d0 in __libc_message (fmt=fmt@entry=0x7fc6ae40756a "%s\n") at ../sysdeps/posix/libc_fatal.c:150
        ap = {{gp_offset = 16, fp_offset = 32764, overflow_arg_area = 0x7ffce4a154a0, reg_save_area = 0x7ffce4a15430}}
        fd = 2
        list = <optimized out>
        nlist = <optimized out>
        cp = <optimized out>
#5  0x00007fc6ae2fd7a5 in malloc_printerr (str=str@entry=0x7fc6ae40a820 "malloc(): unsorted double linked list corrupted") at malloc.c:5765
No locals.
#6  0x00007fc6ae300cac in _int_malloc (av=av@entry=0x7fc6ae43bac0 <main_arena>, bytes=bytes@entry=104) at malloc.c:4079
        next = <optimized out>
        iters = <optimized out>
        nb = <optimized out>
        idx = 7
        bin = <optimized out>
        victim = <optimized out>
        size = <optimized out>
        victim_index = <optimized out>
        remainder = <optimized out>
        remainder_size = <optimized out>
        block = <optimized out>
        bit = <optimized out>
        map = <optimized out>
        fwd = <optimized out>
        bck = <optimized out>
        tcache_unsorted_count = 14
        tcache_nb = <optimized out>
        tc_idx = 5
        return_cached = 0
        __PRETTY_FUNCTION__ = "_int_malloc"
#7  0x00007fc6ae302e61 in __libc_calloc (n=n@entry=1, elem_size=elem_size@entry=104) at malloc.c:3747
        av = <optimized out>
        oldtop = 0x5617edf0d310
        sz = 104
        oldtopsize = 1178864
        mem = <optimized out>
        clearsize = <optimized out>
        nclears = <optimized out>
        d = <optimized out>
        bytes = 104
        __PRETTY_FUNCTION__ = "__libc_calloc"
        p = <optimized out>
        csz = <optimized out>
#8  0x00007fc6ae4fc2e2 in g_malloc0 (n_bytes=104) at ../glib/gmem.c:163
        mem = <optimized out>
#9  0x00007fc6ae4f2bbc in g_source_new (source_funcs=source_funcs@entry=0x7fc6ae5e22a0 <g_idle_funcs>, struct_size=struct_size@entry=104) at ../glib/gmain.c:1047
        source = <optimized out>
        __func__ = "g_source_new"
#10 0x00007fc6ae4f4ffb in idle_source_new (one_shot=0) at ../glib/gmain.c:6296
        source = <optimized out>
        idle_source = <optimized out>
        source = <optimized out>
        idle_source = <optimized out>
#11 idle_add_full (priority=priority@entry=200, one_shot=one_shot@entry=0, function=function@entry=0x5617ed9f9670 <process_changes>, data=data@entry=0x5617ede5d1b0, notify=notify@entry=0x0) at ../glib/gmain.c:6340
        source = <optimized out>
[... 67 lines cut ...]
        data = 0x5617ede4acf0
        io = 0x5617ede4ad10
        ret = <optimized out>
        data = <optimized out>
        io = <optimized out>
        ret = <optimized out>
        __n = <optimized out>
        __s = <optimized out>
        __p = <optimized out>
#27 main (argc=<optimized out>, argv=<optimized out>) at src/main.c:1452
        context = <optimized out>
        err = 0x0
        sdp_mtu = 0
        sdp_flags = <optimized out>
        gdbus_flags = <optimized out>
        __func__ = "main"
        __btd_debug_desc = {file = 0x5617eda3f98f "src/main.c", flags = 0}
```

More details is in:
https://bugzilla.redhat.com/show_bug.cgi?id=2266554

So it seem to be a race condition, which is not completely easy to hit.

I'm not familiar with valgrind. How can I use it to provide useful output? On my Fedora distro, is the easiest way to override `/usr/lib/systemd/system/bluetooth.service` and adjust `ExecStart=/usr/libexec/bluetooth/bluetoothd` ? How should I modify the command line?

nullr0ute, 2024-02-28:
> I'm not familiar with valgrind. How can I use it to provide useful output? On my Fedora distro, is the easiest way to override `/usr/lib/systemd/system/bluetooth.service` and adjust `ExecStart=/usr/libexec/bluetooth/bluetoothd` ? How should I modify the command line?

The abrt service should be able to process the output of a crash to produce something useful on Fedora, I tend to find it's more straight forward to get it to run it locally as it tends to actually produce something where as the hosted service doesn't always work.

kparal, 2024-02-28:
Both bugs that I linked are processed by ABRT, so they should contain full debugging information, IIUIC.

> I tend to find it's more straight forward to get it to run it locally as it tends to actually produce something where as the hosted service doesn't always work.

It's always run locally nowadays (at least in my experience), using debuginfod.

closed 2025-09-22 by github-actions[bot]