bluez/bluez #763: Disabling The Scan Filter After Connection

onurcoskun14, 2024-03-06:
Hello,

I have a project that scans ble sensor around and listen the advertisement changes and on top of that sometimes connect to sensor and get data. I set a scan filter so i can only get adv data from my sensors. But after connectting a sensor filter get disabled. I tried to stop scan before connectting but it did not change anything.

Features:
Rpi4- Linux Cblt-Gw 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
Bluez 5.66

onurcoskun14, 2024-03-06:
When i try the same thing in ubuntu 20.04 with bluez 5.53 it does not happen.

Vudentz, 2024-03-06:
@onurcoskun14 by disabling I suppose you mean it stop scanning? We do suspend scan while connecting but it shall resume after the connection is established.

onurcoskun14, 2024-03-07:
No not stoping. Scan is on but filter is disabled. It acts like there is no filter.

onurcoskun14, 2024-03-11:
@Vudentz No not stoping. Scan is on but filter is disabled. It acts like there is no filter.