bluez/bluez #765: Unable to pair and connect Wiimote on Debian 12

vanfanel, 2024-03-06:
Hi there,

I'm trying to pair an original Wiimote on Debian 12 using bluetoothctl.
bluetoothctl version on Debian is **5.66**, which I believe is recent enough to have the Wiimote support pluging included.
Kernel is 6.1.0-17-amd64, which is also recent enough.

I have the `hid-wiimote` kernel module loaded.

It's successfully detected:
```
Agent registered
[CHG] Controller F0:20:FF:0D:C1:8A Pairable: yes
[bluetooth]# scan on
Discovery started
[CHG] Controller F0:20:FF:0D:C1:8A Discovering: yes
[NEW] Device 00:22:D7:E7:5B:D3 Nintendo RVL-CNT-01
```

Then, if I try to pair it with my computer, I get this:
```

[bluetooth]# pair 00:22:D7:E7:5B:D3 
Attempting to pair with 00:22:D7:E7:5B:D3
[CHG] Device 00:22:D7:E7:5B:D3 Connected: yes
[CHG] Device 00:22:D7:E7:5B:D3 Bonded: yes
[CHG] Device 00:22:D7:E7:5B:D3 Modalias: usb:v057Ep0306d0600
[CHG] Device 00:22:D7:E7:5B:D3 UUIDs: 00001000-0000-1000-8000-00805f9b34fb
[CHG] Device 00:22:D7:E7:5B:D3 UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[CHG] Device 00:22:D7:E7:5B:D3 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device 00:22:D7:E7:5B:D3 ServicesResolved: yes
[CHG] Device 00:22:D7:E7:5B:D3 Paired: yes
Pairing successful
[CHG] Device 00:22:D7:E7:5B:D3 WakeAllowed: yes
[CHG] Device 00:22:D7:E7:5B:D3 ServicesResolved: no
[CHG] Device 00:22:D7:E7:5B:D3 Connected: no
```

Seems to be paired, but not connected, right?

Then, if I try to connect after that:

```
[bluetooth]# connect 00:22:D7:E7:5B:D3 
Attempting to connect to 00:22:D7:E7:5B:D3
[CHG] Device 00:22:D7:E7:5B:D3 Connected: yes
Request PIN code
[agent] Enter PIN code: Failed to connect: org.bluez.Error.Failed br-connection-create-socket
[agent] Enter PIN code: Request canceled
[CHG] Device 00:22:D7:E7:5B:D3 Connected: no
```
I am not entering any PIN, as it's not needed.
If I am not mistaken, Bluez should manage that internally using it's wiimote plugin.

So, maybe the Debian 12 build doesn't have the wiimote plugin support included? That would be very strange indeed...
HOW can I verify that, please?

vanfanel, 2024-03-07:
I verified that bluez on Debian 12 has wiimote plugin enabled:
```

root@debian:~# grep wiimote /usr/libexec/bluetooth/bluetoothd 
grep: /usr/libexec/bluetooth/bluetoothd: binary file matches
```

So it should work. But it does not.

Zipdox2, 2024-03-08:
I also have this problem

queenkjuul, 2024-04-23:
Having the same problem with Ubuntu 23.10, trying the 24.04 beta as we speak to see if it's any different. Can connect and then pair with `bluetoothctl` but nothing else on the system sees the remote. dmesg does not indicate a driver is loaded, though it confirms the connection. 

Passing through the same bluetooth adapter directly to Dolphin works with this remote, so I know it isn't the adapter or the controller, it's the software

Leviticoh, 2024-04-27:
not sure if it's the same, but on arch linux i found the same behaviour since bluez 5.73, by bisecting i found that the first bad commit was "f174724c76c6c8b91276f444142fed9c2537e5c1 bluetoothd: change plugin loading alike obexd"
i hope this is useful

schorschii, 2024-07-10:
I'm on Ubuntu 22.04 with bluetoothctl 5.64 and had the same issue. I solved it by putting `ClassicBondedOnly=false` into `/etc/bluetooth/input.conf` and restarted the bluetooth service.

ali-layken, 2024-11-09:
This still exists

moiri, 2025-07-31:
I ran into the same problem and here is how I finally got the Wiimote to pair and connect without a PIN prompt.

Changes (I am not sure if any of these are strictly necessary but this is what ended up working for me):
- `/etc/bluetooth/main.conf`:
    - `AlwaysPairable = true
    - `FastConnectable = true`
    - `AutoEnable = true`
- `/etc/bluetooth/input.conf`:
    - `ClassicBondedOnly = false`
- add user to input group (`sudo usermod -a -G input localtest`)
- add udev rules in `/etc/udev/rules/99-wiimote.rules`:
  ```
  KERNELS=="0005:057E:0306.*", MODE="0666"
  KERNELS=="0005:057E:0330.*", MODE="0666"
  ```

Process of successfully pairing:
1. `bluetoothctl`
2. `agent off`
3. `agent NoInputNoOutput`
4. `default-agent`
5. `scan on`
6. put device in pairing mode with red button in battery compartment
    important:
    1. do not use any other button combination
    2. make sure to do the following while the device remains in pairing mode
8. `trust <MAC>`
9. `pair <MAC>`
10. `connect <MAC>`

After this process, when pressing a button on the Wiimote it connects automatically. This does not always work though. It is sometimes necessary to press a button multiple times for the Wiimote to connect.