bluez/bluez #767: bluetoothd crash

kirankrishnappa-intel, 2024-03-07:
@Vudentz  - Here is the valgrind report. Crash was observed when I keep restarting PipeWire multiple times.

dbus[62219]: arguments to dbus_message_iter_append_basic() were incorrect, assertion "_dbus_check_is_valid_path (*string_p)" failed in file ../../../dbus/dbus-message.c line 2765.
This is normally a bug in some application using the D-Bus library.

  D-Bus not built with -rdynamic so unable to print a backtrace
==62219== 
==62219== Process terminating with default action of signal 6 (SIGABRT)
==62219==    at 0x4A8D9FC: __pthread_kill_implementation (pthread_kill.c:44)
==62219==    by 0x4A8D9FC: __pthread_kill_internal (pthread_kill.c:78)
==62219==    by 0x4A8D9FC: pthread_kill@@GLIBC_2.34 (pthread_kill.c:89)
==62219==    by 0x4A39475: raise (raise.c:26)
==62219==    by 0x4A1F7F2: abort (abort.c:79)
==62219==    by 0x49B4ED5: ??? (in /usr/lib/x86_64-linux-gnu/libdbus-1.so.3.19.13)
==62219==    by 0x49DB68F: _dbus_warn_check_failed (in /usr/lib/x86_64-linux-gnu/libdbus-1.so.3.19.13)
==62219==    by 0x49CAA46: dbus_message_iter_append_basic (in /usr/lib/x86_64-linux-gnu/libdbus-1.so.3.19.13)
==62219==    by 0x163E0C: get_device (bap.c:302)
==62219==    by 0x1CE855: append_property (object.c:498)
==62219==    by 0x1CF021: append_properties (object.c:527)
==62219==    by 0x1CF0B2: append_interface (object.c:542)
==62219==    by 0x48E0B1F: g_slist_foreach (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4)
==62219==    by 0x1CEA9E: append_interfaces (object.c:1104)
==62219==    by 0x1CEA9E: append_object (object.c:1119)
==62219== 
==62219== FILE DESCRIPTORS: 24 open (3 std) at exit.
==62219== Open pf-31 socket 23:
==62219==    at 0x4B1EB3B: socket (syscall-template.S:120)
==62219==    by 0x171624: create_io (btio.c:1962)
==62219==    by 0x174BA4: bt_io_listen (btio.c:2107)
==62219==    by 0x1A2BD8: ext_start_servers (profile.c:1416)
==62219==    by 0x1A2BD8: ext_adapter_probe (profile.c:1476)
==62219==    by 0x18BD68: probe_profile (adapter.c:5135)
==62219==    by 0x19B467: adapter_add_profile (adapter.c:5152)
==62219==    by 0x48E0B1F: g_slist_foreach (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4)
==62219==    by 0x1A1D7F: create_ext (profile.c:2423)
==62219==    by 0x1A1D7F: register_profile (profile.c:2496)
==62219==    by 0x1D09E8: process_message (object.c:246)
==62219==    by 0x49BF868: dbus_connection_dispatch (in /usr/lib/x86_64-linux-gnu/libdbus-1.so.3.19.13)
==62219==    by 0x1CD077: message_dispatch (mainloop.c:59)
==62219==    by 0x48C4C43: g_main_context_dispatch (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4)
==62219== 
==62219== Open pf-31 socket 22:
==62219==    at 0x4B1EB3B: socket (syscall-template.S:120)
==62219==    by 0x1717D8: create_io (btio.c:1947)
==62219==    by 0x174BA4: bt_io_listen (btio.c:2107)
==62219==    by 0x159830: server_start (server.c:292)
==62219==    by 0x18BD68: probe_profile (adapter.c:5135)
==62219==    by 0x1A32EB: btd_profile_foreach (profile.c:740)
==62219==    by 0x19E495: adapter_register (adapter.c:9382)
==62219==    by 0x19E495: read_info_complete (adapter.c:10314)
==62219==    by 0x1D5FF1: request_complete (mgmt.c:306)
==62219==    by 0x1D7295: can_read_data (mgmt.c:394)
==62219==    by 0x1FFAA8: watch_callback (io-glib.c:157)
==62219==    by 0x48C4C43: g_main_context_dispatch (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4)
==62219==    by 0x491A257: ??? (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4)
==62219== 
==62219== Open pf-31 socket 21:
==62219==    at 0x4B1EB3B: socket (syscall-template.S:120)
==62219==    by 0x1717D8: create_io (btio.c:1947)
==62219==    by 0x174BA4: bt_io_listen (btio.c:2107)
==62219==    by 0x1597FB: server_start (server.c:279)
==62219==    by 0x18BD68: probe_profile (adapter.c:5135)
==62219==    by 0x1A32EB: btd_profile_foreach (profile.c:740)
==62219==    by 0x19E495: adapter_register (adapter.c:9382)
==62219==    by 0x19E495: read_info_complete (adapter.c:10314)
==62219==    by 0x1D5FF1: request_complete (mgmt.c:306)
==62219==    by 0x1D7295: can_read_data (mgmt.c:394)
==62219==    by 0x1FFAA8: watch_callback (io-glib.c:157)
==62219==    by 0x48C4C43: g_main_context_dispatch (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4)
==62219==    by 0x491A257: ??? (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4)
[... 195 lines cut ...]
==62219==    at 0x4B1CCDB: eventfd (syscall-template.S:120)
==62219==    by 0x49154BC: ??? (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4)
==62219==    by 0x48C153B: g_main_context_new_with_flags (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4)
==62219==    by 0x48C1644: g_main_context_default (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4)
==62219==    by 0x48C16AC: g_main_loop_new (in /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4)
==62219==    by 0x2000F0: mainloop_init (mainloop-glib.c:35)
==62219==    by 0x12CF81: main (main.c:1387)
==62219== 
==62219== 
==62219== HEAP SUMMARY:
==62219==     in use at exit: 127,484 bytes in 2,188 blocks
==62219==   total heap usage: 79,051 allocs, 76,863 frees, 30,076,576 bytes allocated
==62219== 
==62219== LEAK SUMMARY:
==62219==    definitely lost: 0 bytes in 0 blocks
==62219==    indirectly lost: 0 bytes in 0 blocks
==62219==      possibly lost: 0 bytes in 0 blocks
==62219==    still reachable: 127,484 bytes in 2,188 blocks
==62219==         suppressed: 0 bytes in 0 blocks
==62219== Reachable blocks (those to which a pointer was found) are not shown.
==62219== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==62219== 
==62219== For lists of detected and suppressed errors, rerun with: -s
==62219== ERROR SUMMARY: 370 errors from 3 contexts (suppressed: 0 from 0)
Aborted

m0n5t3r, 2025-05-20:
I am seeing this too, specifically when a chromium application starts (Microsoft Teams in this case), on Ubuntu 24.04:

Bluez version: 5.72-0ubuntu5.1
Chromium version:   136.0.7103.92, snap rev. 3125

the logs are very similar to the original report:
```
bluetoothd[1248778]: dbus[1248778]: arguments to dbus_message_iter_append_basic() were incorrect, assertion "_dbus_check_is_valid_path (*string_p)" failed in file ../../../dbus/dbus-message.c line 2780.
bluetoothd[1248778]: This is normally a bug in some application using the D-Bus library.
bluetoothd[1248778]:   D-Bus not built with -rdynamic so unable to print a backtrace
[...]
systemd[1]: bluetooth.service: Main process exited, code=dumped, status=6/ABRT
systemd[1]: bluetooth.service: Failed with result 'core-dump'.
```

if there's anything I can do to provide further info I'm willing to try.

eliasp, 2025-09-13:
Just happened here a few times on NixOS with BlueZ 5.83 and Pipewire 1.4.6 as well, after restarting a user's `pipewire.service` a few times to test some module config changes. Weirdly, it didn't happen immediately when restarting pipewire, but a few minutes later while just using the device regularly, while not doing any BT or sound-related actions.

```
Module /nix/store/9s8r7f3ijyhmc5v379iv8qcjjqwhxsgx-bluez-5.83/libexec/bluetooth/bluetoothd without build-id.
Module libcap.so.2 without build-id.
Module libpcre2-8.so.0 without build-id.
Module libasound.so.2 without build-id.
Stack trace of thread 1681:
#0  0x000000000042afbc release_stream (/nix/store/9s8r7f3ijyhmc5v379iv8qcjjqwhxsgx-bluez-5.83/libexec/bluetooth/bluetoothd + 0x2afbc)
#1  0x00007f0b0e409f30 g_slist_foreach (libglib-2.0.so.0 + 0x89f30)
#2  0x0000000000428c79 connection_lost (/nix/store/9s8r7f3ijyhmc5v379iv8qcjjqwhxsgx-bluez-5.83/libexec/bluetooth/bluetoothd + 0x28c79)
#3  0x000000000042beac session_cb (/nix/store/9s8r7f3ijyhmc5v379iv8qcjjqwhxsgx-bluez-5.83/libexec/bluetooth/bluetoothd + 0x2beac)
#4  0x00007f0b0e3e2f6d g_main_context_dispatch_unlocked (libglib-2.0.so.0 + 0x62f6d)
#5  0x00007f0b0e3e52b8 g_main_context_iterate_unlocked.isra.0 (libglib-2.0.so.0 + 0x652b8)
#6  0x00007f0b0e3e5da7 g_main_loop_run (libglib-2.0.so.0 + 0x65da7)
#7  0x000000000050c675 mainloop_run (/nix/store/9s8r7f3ijyhmc5v379iv8qcjjqwhxsgx-bluez-5.83/libexec/bluetooth/bluetoothd + 0x10c675)
#8  0x000000000050cb21 mainloop_run_with_signal (/nix/store/9s8r7f3ijyhmc5v379iv8qcjjqwhxsgx-bluez-5.83/libexec/bluetooth/bluetoothd + 0x10cb21)
#9  0x000000000040b953 main (/nix/store/9s8r7f3ijyhmc5v379iv8qcjjqwhxsgx-bluez-5.83/libexec/bluetooth/bluetoothd + 0xb953)
#10 0x00007f0b0de2a4d8 __libc_start_call_main (libc.so.6 + 0x2a4d8)
#11 0x00007f0b0de2a59b __libc_start_main@@GLIBC_2.34 (libc.so.6 + 0x2a59b)
#12 0x000000000040c1d5 _start (/nix/store/9s8r7f3ijyhmc5v379iv8qcjjqwhxsgx-bluez-5.83/libexec/bluetooth/bluetoothd + 0xc1d5)
ELF object binary architecture: AMD x86-64
```