bluez/bluez #771: DualShock 3 not connecting with bluez 5.73

kattjevfel, 2024-03-08:
After updating bluez 5.72 --> 5.73, I can still successfully pair my DS3, but connecting always fails. It will be listed as connected but the lights keep flashing and doesn't show up a gamepad.

journal output with 5.72:
```
Mar 08 18:24:05 main kernel: sony 0005:054C:0268.0008: unknown main item tag 0x0
Mar 08 18:24:06 main kernel: input: Sony PLAYSTATION(R)3 Controller Motion Sensors as /devices/pci0000:00/0000:00:08.1/0000:0b:00.3/usb3/3-4/3-4.4/3-4.4:1.0/bluetooth/hci0/hci0:11/0005:054C:0268.0008/input/input22
Mar 08 18:24:06 main kernel: input: Sony PLAYSTATION(R)3 Controller as /devices/pci0000:00/0000:00:08.1/0000:0b:00.3/usb3/3-4/3-4.4/3-4.4:1.0/bluetooth/hci0/hci0:11/0005:054C:0268.0008/input/input21
Mar 08 18:24:06 main kernel: sony 0005:054C:0268.0008: input,hidraw6: BLUETOOTH HID v80.00 Joystick [Sony PLAYSTATION(R)3 Controller] on 5c:f3:70:8d:ab:4a
```

journal output with 5.73:
```
Mar 08 18:18:41 main kernel: sony 0005:054C:0268.0007: unknown main item tag 0x0
Mar 08 18:18:46 main kernel: sony 0005:054C:0268.0007: Failed to set controller into operational mode
Mar 08 18:18:46 main kernel: sony 0005:054C:0268.0007: hidraw6: BLUETOOTH HID v80.00 Joystick [Sony PLAYSTATION(R)3 Controller] on 5c:f3:70:8d:ab:4a
Mar 08 18:18:46 main kernel: sony 0005:054C:0268.0007: failed to claim input
```

I use Arch Linux with the testing repos enabled, bluez 5.73-3, linux-zen 6.7.9.zen1-1
Downstream issue: https://gitlab.archlinux.org/archlinux/packaging/packages/bluez/-/issues/6

Vudentz, 2024-03-08:
@kattjevfel same kernel? That would be really weird since we haven't change much, well except that we changed to use uhid by default:

```
commit 9698870015b0972d3d93f30a6d9de1ed029ba87c
Author: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
Date:   Fri Feb 9 13:47:22 2024 -0500

    input.conf: Make UserspaceHID defaults to true
    
    This makes UserspaceHID defaults to true so the plugin has more control
    over the input device lifetime.

diff --git a/profiles/input/device.c b/profiles/input/device.c
index 6c64ff1c1c52..ff7e3482d0eb 100644
--- a/profiles/input/device.c
+++ b/profiles/input/device.c
@@ -81,7 +81,7 @@ struct input_device {
 };
 
 static int idle_timeout = 0;
-static bool uhid_enabled = false;
+static bool uhid_enabled = true;
 static bool classic_bonded_only = true;
 
 void input_set_idle_timeout(int timeout)
diff --git a/profiles/input/input.conf b/profiles/input/input.conf
index d8645f3dd664..00a34eb63de1 100644
--- a/profiles/input/input.conf
+++ b/profiles/input/input.conf
@@ -9,7 +9,7 @@
 #IdleTimeout=30
 
 # Enable HID protocol handling in userspace input profile
-# Defaults to false (HIDP handled in HIDP kernel module)
+# Defaults to true (Use UHID instead of kernel HIDP)
 #UserspaceHID=true
 
 # Limit HID connections to bonded devices
```

Try setting main.conf:UserspaceHID=false to see if that helps.

kattjevfel, 2024-03-08:
I assume you meant input.conf? Either way I set `UserspaceHID=false` in input.conf and it does indeed fix it. but it also still says "Defaults to false" in my copy.

JockeTF, 2024-03-10:
I had the same issue! Setting `UserspaceHID` to `false` in `/etc/bluetooth/input.conf` fixed the issue for me as well.

Vudentz, 2024-03-11:
Can you guys try applying the following set to check if that fixes the problem, I tried with LE HID but unfortunately I don't a Classic one to test with:

https://patchwork.kernel.org/project/bluetooth/patch/20240308221007.250681-1-luiz.dentz@gmail.com/

kattjevfel, 2024-03-11:
I tried the patch out, but all it appears to do is get rid of the dmesg output that was there previously, but still doesn't work without `UserspaceHID=false`.

kattjevfel, 2024-03-11:
The v2 brings back the kernel messages (exact same as first post), but nothing else. bluetoothd is also completely silent while connecting.

JockeTF, 2024-03-11:
Same for me. Tried the V2 patch, but it still requires setting `UserspaceHID` to `false`.

Vudentz, 2024-03-11:
v3: https://patchwork.kernel.org/project/bluetooth/list/?series=834439

Also make sure you have been passing -d to bluetootd otherwise it won't be printing much anyway.

kattjevfel, 2024-03-11:
I had not been passing -d, I have done so now with the v3:
```
Mar 11 21:23:47 main bluetoothd[453878]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x000b
Mar 11 21:23:47 main bluetoothd[453878]: src/adapter.c:connected_callback() hci0 device 00:1E:3D:29:FE:0E connected eir_len 5
Mar 11 21:23:48 main bluetoothd[453878]: profiles/input/server.c:connect_event_cb() Incoming connection from 00:1E:3D:29:FE:0E on PSM 17
Mar 11 21:23:48 main bluetoothd[453878]: profiles/input/device.c:input_device_set_channel() idev 0x62655936a080 psm 17
Mar 11 21:23:48 main bluetoothd[453878]: profiles/input/server.c:confirm_event_cb()
Mar 11 21:23:48 main bluetoothd[453878]: profiles/input/server.c:connect_event_cb() Incoming connection from 00:1E:3D:29:FE:0E on PSM 19
Mar 11 21:23:48 main bluetoothd[453878]: profiles/input/device.c:input_device_set_channel() idev 0x62655936a080 psm 19
Mar 11 21:23:48 main bluetoothd[453878]: src/service.c:change_state() 0x62655936a040: device 00:1E:3D:29:FE:0E profile input-hid state changed: disconnected -> connected (0)
Mar 11 21:23:48 main bluetoothd[453878]: src/service.c:btd_service_ref() 0x62655936a040: ref=3
Mar 11 21:23:48 main bluetoothd[453878]: plugins/policy.c:service_cb() Added input-hid reconnect 0
Mar 11 21:23:48 main bluetoothd[453878]: profiles/input/device.c:hidp_send_set_report()
Mar 11 21:23:48 main bluetoothd[453878]: profiles/input/device.c:hidp_recv_ctrl_handshake()
Mar 11 21:23:48 main bluetoothd[453878]: profiles/input/device.c:hidp_recv_ctrl_handshake() SET_REPORT successful
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)
```

The last line keeps repeating many times a second forever. Still the same behaviour.

JockeTF, 2024-03-11:
Looks about the same to me!

Vudentz, 2024-03-11:
So this tells me the SET_REPORT, etc, does seems to work:
```

Mar 11 21:23:48 main bluetoothd[453878]: profiles/input/device.c:hidp_send_set_report()
Mar 11 21:23:48 main bluetoothd[453878]: profiles/input/device.c:hidp_recv_ctrl_handshake()
Mar 11 21:23:48 main bluetoothd[453878]: profiles/input/device.c:hidp_recv_ctrl_handshake() SET_REPORT successful
```

And that we are receiving input events:

`Mar 11 21:23:49 main bluetoothd[453878]: profiles/input/device.c:uhid_send_input_report() HID report (49 bytes)`

So something else might be causing the problem then, or perhaps it is a timing issue where the initial handshake doesn't complete in time, etc.

Can you guys collect the HCI trace using btmon using both methods?

JockeTF, 2024-03-11:
For completeness, here's the log with `UserspaceHID` to `false`.

```
bluetoothd[163860]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x000b
bluetoothd[163860]: src/adapter.c:connected_callback() hci0 device 00:06:F5:XX:XX:XX connected eir_len 5
bluetoothd[163860]: profiles/input/server.c:connect_event_cb() Incoming connection from 00:06:F5:XX:XX:XX on PSM 17
bluetoothd[163860]: profiles/input/device.c:input_device_set_channel() idev 0x5decxxxxxxc0 psm 17
bluetoothd[163860]: profiles/input/server.c:confirm_event_cb()
bluetoothd[163860]: profiles/input/server.c:connect_event_cb() Incoming connection from 00:06:F5:XX:XX:XX on PSM 19
bluetoothd[163860]: profiles/input/device.c:input_device_set_channel() idev 0x5decxxxxxxc0 psm 19
bluetoothd[163860]: src/service.c:change_state() 0x5decxxxxxx80: device 00:06:F5:XX:XX:XX profile input-hid state changed: disconnected -> connected (0)
bluetoothd[163860]: src/service.c:btd_service_ref() 0x5decxxxxxx80: ref=3
bluetoothd[163860]: plugins/policy.c:service_cb() Added input-hid reconnect 0
kernel: sony 0005:xxxx:xxxx.xxxx: unknown main item tag 0x0
kernel: input: Sony PLAYSTATION(R)3 Controller Motion Sensors as /devices/.../bluetooth/.../input/input50
kernel: input: Sony PLAYSTATION(R)3 Controller as /devices/.../bluetooth/hci0/.../input/input49
kernel: sony 0005:xxxx:xxxx.xxxx: input,hidraw8: BLUETOOTH HID v80.00 Joystick [Sony PLAYSTATION(R)3 Controller] on d4:d2:52:xx:xx:xx
```

... Need to get some sleep now!

Vudentz, 2024-03-11:
This might be the problem:

```
diff --git a/profiles/input/device.c b/profiles/input/device.c
index 0d32b705bd00..298fca7c2195 100644
--- a/profiles/input/device.c
+++ b/profiles/input/device.c
@@ -445,6 +445,7 @@ static void hidp_recv_ctrl_handshake(struct input_device *idev, uint8_t param)
                        timeout_remove(idev->report_req_timer);
                        idev->report_req_timer = 0;
                }
+               uhid_send_set_report_reply(idev, idev->report_rsp_id, 0);
                idev->report_rsp_id = 0;
        }
 }

```

That said I don't think the HIDP_TRANS_HANDSHAKE were supposed to be used as response, so this just works by a coincidence but given that there are no calls to hidp_recv_ctrl_data with HIDP_TRANS_SET_REPORT perhaps that is intentional, anyway I preparing a v4 to test this theory.

Vudentz, 2024-03-12:
v4: https://patchwork.kernel.org/project/bluetooth/list/?series=834468

jerekulmala, 2024-03-12:
Tested with v4. The controller connects now and seems to work.

kattjevfel, 2024-03-12:
I too can confirm the v4 works for me without setting `UserspaceHID=false`.

closed 2024-03-13 by BluezTestBot

LizziAS, 2024-03-17:
I do not believe this issue is entirely resolved.  using an arch based system i have had the errors above and this fix seems to have resolved it mostly but if i disconnect my ps4 controller or it times out to save battery then when i reconnect it the led is white and not red and doesnt get picked up by steam even though bluez reports it as connected.  please advise.  having to delete it from bluez and pair it over again and then retrust and reconnect it once disconnecting it or timing out to save battery is not ideal.  only other option seems to be a reboot.  ***It used to be i could just press the home/ps button to reconnect it***  sure miss those days of wild abandon.  ;)

aiudirog, 2024-03-17:
@LizziAS there is a separate issue for the DS4/5: #777

Also, you can just disable `UserspaceHID` in `inputs.conf` to have it working like before until the bug is resolved.

LizziAS, 2024-03-17:
TY!  edited and rebooting now

Krauzer94, 2024-04-13:
I still can't connect my controller, I followed [these instructions from Arch Linux Wiki](https://wiki.archlinux.org/title/Gamepad#Pairing_via_Bluetooth), and also [these other ones from Gentoo Wiki](https://wiki.gentoo.org/wiki/Sony_DualShock#DualShock_3), my controller worked perfectly until this new update dropped.

[Here is my log](https://wiki.gentoo.org/wiki/Sony_DualShock#DualShock_3) of trying to connect it via the mothods above mentioned.

Spike1377, 2024-11-17:
can someone add a onscreen message saying to set UserspaceHID to false to get it to work

lctrr, 2025-04-28:
The provided solution doesn't work for me unfortunately, but the controller connects if setting

`ClassicBondedOnly=false`

Not sure if something else changed

Holbormon, 2026-10-03:
For what it's worth, and even though I'm not sure how many of you are still following that old issue and despairing over needing to compromise security for usability with PS3 controllers, i'll mention that the instructions from the Arch wiki's Gamepad page at section [`3.9 Playstation 3/4 Controllers`](https://wiki.archlinux.org/title/Gamepad#PlayStation_3%2F4_controller) (NOT to be confused with `3.8 Playstation 3 controllers`) worked flawlessly for wireless pairing, WITHOUT requiring me to touch anything in /etc/bluetooth/input.conf (in particular, ClassicBondedOnly does not need changing from its default `true` value)