bluez/bluez #773: Build error with 5.73 when —disable-cups is set: conditional "CUPS_SERVERBIN" was never defined

heitbaum, 2024-03-09:
Build error with 5.73 when —disable-cups is set.

```
configure: error: conditional "CUPS_SERVERBIN" was never defined.
Usually this means the macro was only invoked conditionally.
```

reverting d5a9195a6269e703c6e3a47c40982ec94d8cc838 allows the build to complete.

gudnimg, 2024-06-29:
I believe I'm seeing the same issue when attempting to upgrade BlueZ 5.72 to 5.76 with Yocto, by default cups is disabled.
[image]

gudnimg, 2024-07-01:
Sorry to ping you @Vudentz 🙏 I noticed there is a patch for this issue in the mailing list https://lore.kernel.org/linux-bluetooth/20240416052603.4373-2-polynomial-c@gmx.de/ Am wondering if the reason the patch is not merged is due to some formatting problems? I can submit a new patch if needed.

Please let me know if I can help somehow get this fixed upstream.

Vudentz, 2024-07-01:
That was failing our CI tests for some reason.

gudnimg, 2024-07-05:
Damn, I forgot to check for too long lines. I'll resubmit the patch later 🤦  https://patchwork.kernel.org/project/bluetooth/patch/20240705084216.552838-1-gudni.m.g@gmail.com/

closed 2024-07-08 by BluezTestBot