bluez/bluez #774: bluetooth does not work after resuming from hibernation

arenevier, 2024-03-09:
I am using archlinux. When I resume from hibernation, bluetooth does not work. The bluetooth service seems "stuck". bluetoothctl timeouts with:

```
bluetoothctl
Waiting to connect to bluetoothd...[bluetooth]#
```

Bluetooth service is show as running

```

● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset: disabled)
     Active: active (running) since Fri 2024-03-08 18:24:44 PST; 7min ago
       Docs: man:bluetoothd(8)
   Main PID: 1187 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 47732)
     Memory: 2.8M (peak: 3.5M)
        CPU: 1min 32.389s
     CGroup: /system.slice/bluetooth.service
             └─1187 /usr/lib/bluetooth/bluetoothd

Mar 08 18:24:44 archlinux bluetoothd[1187]: src/plugin.c:init_plugin() System does not support vcp plugin
Mar 08 18:24:44 archlinux bluetoothd[1187]: profiles/audio/micp.c:micp_init() D-Bus experimental not enabled
Mar 08 18:24:44 archlinux bluetoothd[1187]: src/plugin.c:init_plugin() System does not support micp plugin
Mar 08 18:24:44 archlinux bluetoothd[1187]: src/plugin.c:init_plugin() System does not support ccp plugin
Mar 08 18:24:44 archlinux bluetoothd[1187]: src/plugin.c:init_plugin() System does not support csip plugin
Mar 08 18:24:44 archlinux bluetoothd[1187]: Bluetooth management interface 1.22 initialized
Mar 08 18:24:44 archlinux systemd[1]: Started Bluetooth service.
Mar 08 18:24:45 archlinux bluetoothd[1187]: Battery Provider Manager created
Mar 08 18:24:45 archlinux bluetoothd[1187]: src/device.c:device_set_wake_support() Unable to set wake_support without RPA resolution
Mar 08 18:24:45 archlinux bluetoothd[1187]: src/device.c:set_wake_allowed_complete() Set device flags return status: Invalid Parameters
```

If I restart it, bluetooth service takes a very long time to stop (like 1 or 2 minutes). But eventually, after a restart, bluetooth works correctly.

It seems like a regression. It was working  fine with 5.72, but does not with 5.73

Cuttergott, 2024-03-14:
Same for me.
However, I can launch bluetoothctl, but it says no controller is available

`
[bluetooth]# list
[bluetooth]# show
No default controller available
[bluetooth]# devices
No default controller available
`

When I stop bluetooth.service with systemctl stop bluetooth.service, my bt devices suddenly connect. It seems like 

`[bluetooth]# list
Agent registered
[CHG] Controller 34:2E:B7:DF:08:80 Name: Maximegalon
[CHG] Controller 34:2E:B7:DF:08:80 Alias: Maximegalon
hci0 class of device changed: 0x00010c
[CHG] Controller 34:2E:B7:DF:08:80 Class: 0x0000010c (268)
hci0 new_settings: powered ssp br/edr le secure-conn 
[CHG] Controller 34:2E:B7:DF:08:80 PowerState: on
[CHG] Controller 34:2E:B7:DF:08:80 Powered: yes
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 0000111f-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 00001108-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 0000180a-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 0000111f-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 00001108-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 0000180a-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 0000111f-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 00001108-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 0000180a-0000-1000-8000-00805f9b34fb
hci0 name changed: BlueZ 5.73
[CHG] Controller 34:2E:B7:DF:08:80 Alias: BlueZ 5.73
[CHG] Device DE:CA:15:9D:F7:71 Connected: yes
hci0 new_settings: powered connectable ssp br/edr le secure-conn 
hci0 new_settings: powered connectable discoverable ssp br/edr le secure-conn 
hci0 new_settings: powered connectable discoverable bondable ssp br/edr le secure-conn 
[CHG] Controller 34:2E:B7:DF:08:80 Discoverable: yes
[CHG] Controller 34:2E:B7:DF:08:80 Pairable: yes
hci0 class of device changed: 0x20010c
[CHG] Controller 34:2E:B7:DF:08:80 Class: 0x0020010c (2097420)
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 0000111f-0000-1000-8000-00805f9b34fb
[CHG] Controller 34:2E:B7:DF:08:80 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
`

0rtz, 2024-03-17:
Same for me

arenevier, 2024-03-18:
After resuming from hibernation, top shows  that bluetoothd is using 100% of the cpu. It seems stucks in an infinite loop.

I was able to bisect the problem, and it seems to have been introduced by https://github.com/bluez/bluez/commit/44d3f67277f83983e1e9697eda7b9aeb40ca231d

arenevier, 2024-03-18:
@Vudentz

arenevier, 2024-03-18:
It would make it possibly a duplicate of #784

arenevier, 2024-03-18:
maybe #783  too??

nullr0ute, 2024-03-18:
Also #778 too

Vudentz, 2024-03-18:
@nullr0ute I think it has something todo with:

```
commit 44d3f67277f83983e1e9697eda7b9aeb40ca231d
Author: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
Date:   Tue Feb 27 11:40:57 2024 -0500

    device: Consider service state on device_is_connected
    
    This checks if there is any service connected on device_is_connected
    since some profiles maybe probed using advertising data which doesn't
    require a connection.

```

closed 2024-03-18 by BluezTestBot

AdamWill, 2024-03-19:
This is looking good to fix #778 for me, since updating to [this build with a backport of the fix](https://bodhi.fedoraproject.org/updates/FEDORA-2024-f219b6c25c), my wheel scrolling has not slowed down all day 🥳

gguman, 2025-01-03:
> I am using archlinux. When I resume from hibernation, bluetooth does not work. The bluetooth service seems "stuck". bluetoothctl timeouts with:
>
> ```
> bluetoothctl
> Waiting to connect to [...]

Exactly the same issue still persists with 5.79. Thanks @0rtz for script. But how the problem should be solve completely without handmade code?