bluez/bluez #777: DualShock 4 and DualSense error on reconnect after upgrading to 5.73

aiudirog, 2024-03-11:
This may be related to #771 but after upgrading to 5.73, I can no longer reconnect my PS4 & PS5 controllers. They connect and work fine the first time, but if I disconnect and reconnect they will say they're connected but no inputs work and journalctl spams this >700x/second until I force disconnect the controller:

```
Mar 10 23:08:27 Ryzen-Arch kernel: playstation 0005:054C:09CC.0009: Unhandled reportID=1
```

My first fix attempt was to swap to the LTS kernel, but that didn't work. Then I noticed Bluez had also updated since this last worked, so I rolled all bluez-* packages back to 5.72 and that seems to resolve the problem.

<details>
<summary>System Info</summary>

```
System:
  Host: Ryzen-Arch Kernel: 6.7.9-zen1-1-zen arch: x86_64 bits: 64
  Desktop: KDE Plasma v: 6.0.1 Distro: Arch Linux
Machine:
  Type: Desktop Mobo: Micro-Star model: MPG X570 GAMING PLUS (MS-7C37) v: 2.0
    serial: <superuser required> UEFI: American Megatrends LLC. v: A.D1
    date: 03/03/2021
CPU:
  Info: 12-core AMD Ryzen 9 5900X [MT MCP] speed (MHz): avg: 2536
    min/max: 2200/4950
Graphics:
  Device-1: AMD Navi 21 [Radeon RX 6950 XT] driver: amdgpu v: kernel
  Display: wayland server: X.org v: 1.21.1.11 with: Xwayland v: 23.2.4
    compositor: kwin_wayland driver: X: loaded: amdgpu dri: radeonsi gpu: amdgpu
    resolution: 1: 1920x1080 2: 1920x1080
  API: OpenGL v: 4.6 compat-v: 4.5 vendor: amd mesa v: 24.0.2-arch1.2
    renderer: AMD Radeon RX 6950 XT (radeonsi navi21 LLVM 17.0.6 DRM 3.57
    6.7.9-zen1-1-zen)
Network:
  Device-1: Realtek RTL8111/8168/8211/8411 PCI Express Gigabit Ethernet
    driver: r8169
Drives:
  Local Storage: total: 28.22 TiB used: 10.03 TiB (35.5%)
Info:
  Memory: total: 32 GiB available: 31.26 GiB used: 4.67 GiB (14.9%)
  Processes: 536 Uptime: 5m Shell: Zsh inxi: 3.3.33
```

</details>

aiudirog, 2024-03-11:
The suggested fix from the other issue of setting `UserspaceHID=false` in `input.conf` seems to resolve this issue as well.

It also fixes a minor annoyance I have with some automation scripts since updating. I have a UDEV rule which checks the MAC address of connected Bluetooth controllers and changes my display output to the appropriate device, based on the controller that was connected. I was previously doing this using the DEVPATH provided by UDEV (example: `/sys/devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:06.0/0000:28:00.0/usb1/1-4/1-4:1.0/bluetooth/hci0/hci0:71/`) and then looking for `*:*/uevent` (which maps to something like `0005:054C:09CC.000D/uevent`) which contained a field for `HID_UNIQ=<MAC>`. When using the userspace HID driver, this file apparently no longer gets created and I have been at a loss for another way to check which device drove the UDEV event when there are multiple connected.

Vudentz, 2024-03-11:
https://github.com/bluez/bluez/issues/771#issuecomment-1989350849

Vudentz, 2024-03-11:
Might need to find one of these since Im lacking a peripheral for testing classic HID, maybe something like the following would work:

https://www.amazon.com/DualShock-Wireless-Controller-PlayStation-Sony-4/dp/B0125JFATY/ref=sr_1_2?crid=RISM0CG0SVEK&dib=eyJ2IjoiMSJ9.iao30N-tJ5v5pXiYNTS_M7D32JrSzKidVGRah7Q8cQ3HUk54CkCnyK4mkq1WqI-1sYW3-A1Ikinckdp0SPnVlNO70TmcTeSVZ-8VNWDmXEUaZb7YNPXJ4WG7W5mOYAkHr6sRrKYhefU1L4pCzXQQ4IYYJIAuu2WwzRnT6TvwyGo3-029k_4X_tx2LrnORd_In-SoTNEIUP-PCX97L5v9qssNNqEIeF9CwaFkHWLXezY.YAhKD4R8UDW3yvpVHKkWffnStyFhDjrSXBU6TVQI51k&dib_tag=se&keywords=dualshock+4&qid=1710188049&s=videogames&sprefix=dual%2Cvideogames%2C4834&sr=1-2

aiudirog, 2024-03-11:
> [#771 (comment)](https://github.com/bluez/bluez/issues/771#issuecomment-1989350849)

I'm totally down to compile and test things, but I've not compiled Bluez before or with patches. Do you have a quick guide to get setup?

> Might need to find one of these since Im lacking a peripheral for testing classic HID, maybe something like the following would work: [...]

Might be a good idea, just watch out for knockoffs as I've found they're very hit or miss on working at all. The DualSense also regularly goes on sale for ~$45 new and is (IMO) much nicer.

aiudirog, 2024-03-12:
@Vudentz I was able to figure out how to patch the PKGBUILD and test the v4 patch series, but unfortunately it doesn't fix the problem. :(

<details>

<summary>In case you want to check my work and make sure I didn't miss anything...</summary>

Repo Status:

```
$ git remote -v
origin  https://gitlab.archlinux.org/archlinux/packaging/packages/bluez.git (fetch)
origin  https://gitlab.archlinux.org/archlinux/packaging/packages/bluez.git (push)

$ git --no-pager show HEAD --no-patch
commit d9bacb6b1a00f17e7c71c84fa92ec629175581f8 (HEAD -> main, tag: 5.73-3, origin/main, origin/HEAD)
Author: Andreas Radke <andyrtr@archlinux.org>
Date:   Fri Mar 8 17:36:20 2024 +0100

    upgpkg: 5.73-3; bring back proper config dir permission 0555; sort fakeinstall commands by path/file (parts of MR2)
```

PKGBUILD Changes:

```diff
$ git --no-pager diff PKGBUILD
diff --git a/PKGBUILD b/PKGBUILD
index 3cde825..37c05c3 100644
--- a/PKGBUILD
+++ b/PKGBUILD
@@ -20,6 +20,12 @@ sha256sums=('257e9075ce05c70d48c5defd254e78c418416f7584b45f9dddc884ff88e3fc53'
             '46c021be659c9a1c4e55afd04df0c059af1f3d98a96338236412e449bf7477b4')
 validpgpkeys=('E932D120BC2AEC444E558F0106CA9F5D1DCF2659') # Marcel Holtmann <marcel@holtmann.org>
 
+prepare() {
+    cd "${pkgname}"-${pkgver}
+    patch --forward --strip=1 --input=../../BlueZ-v4-1-4-shared-uhid-Add-dedicated-functions-for-each-UHID-opcode.patch 
+}
+
+
 build() {
   cd "${pkgname}"-${pkgver}
   ./configure \
```

Series patch downloaded from [here](https://patchwork.kernel.org/series/834468/mbox/)

Confirm patch applied by PKGBUILD:

```
$ patch --directory=./src/bluez-5.73/ --forward --strip=1 --input=../../BlueZ-v4-1-4-shared-uhid-Add-dedicated-functions-for-each-UHID-opcode.patch
patching file src/shared/uhid.c
Reversed (or previously applied) patch detected!  Skipping patch.
4 out of 4 hunks ignored -- saving rejects to file src/shared/uhid.c.rej
patching file src/shared/uhid.h
Reversed (or previously applied) patch detected!  Skipping patch.
2 out of 2 hunks ignored -- saving rejects to file src/shared/uhid.h.rej
patching file profiles/input/hog-lib.c
Reversed (or previously applied) patch detected!  Skipping patch.
14 out of 14 hunks ignored -- saving rejects to file profiles/input/hog-lib.c.rej
patching file profiles/input/device.c
Reversed (or previously applied) patch detected!  Skipping patch.
10 out of 10 hunks ignored -- saving rejects to file profiles/input/device.c.rej
patching file unit/test-uhid.c
Reversed (or previously applied) patch detected!  Skipping patch.
3 out of 3 hunks ignored -- saving rejects to file unit/test-uhid.c.rej
```

Installed with `makepkg -si` and following all prompts

</details>

kode54, 2024-03-12:
v4 of the patch series fixed the problem for us, and also fixed a problem I was having with an Xbox Series X|S Wireless Controller.

aiudirog, 2024-03-12:
> v4 of the patch series fixed the problem for us, and also fixed a problem I was having with an Xbox Series X|S Wireless Controller.

It didn't make my problem worse, just seems to be unrelated unfortunately. Luckily disabling UserspaceHID still works for now.

Let me know if there's a new set of patches I should test.

Kimikz, 2024-03-13:
if i connect and disconnect my dualshock 4 i get a insane log spam of 

`20:59:11 kernel: playstation 0005:054C:09CC.0007: Unhandled reportID=1`

and

`20:06:19 bluetoothd: profiles/input/device.c:hidp_send_message() BT socket not connected`

like almost 1000 entries per second.

kode54, 2024-03-13:
Same. I was too quick to report it fixed things for me. Restarting Bluetooth services stops the spam, until the next time it decides to do it.

aiudirog, 2024-03-13:
> if i connect and disconnect my dualshock 4 i get a insane log spam of
> 
> `20:59:11 kernel: playstation 0005:054C:09CC.0007: Unhandled reportID=1`
> 
> and
> 
> `20:06:19 bluetoothd: profiles/input/device.c:hidp_send_message() BT socket not connected`
> 
> like almost 1000 entries per second.

Glad to know it's not just me haha

> Same. I was too quick to report it fixed things for me. Restarting Bluetooth services stops the spam, until the next time it decides to do it.

I believe that's because it also disconnects the controller. Once the the device has connected and disconnected once, it will spam that for the duration of any subsequent connections until the Bluetooth service is restarted. It's honestly super reproducible, so it should be obvious to tell if it's fixed.

Vudentz, 2024-03-13:
Hmm, interesting, so looks like keeping the device node alive while disconnected perhaps is not the best idea if the driver keeps sending messages like that, now what kind of message is that? We could perhaps pretend it is alive by auto responding while disconnected, since Id like to keep the current logic so we don't loose events when reconnecting.

aiudirog, 2024-03-13:
> Hmm, interesting, so looks like keeping the device node alive while disconnected perhaps is not the best idea if the driver keeps sending messages like that, now what kind of message is that?

This is the relevant code in the playstation driver: [DualSense](https://github.com/torvalds/linux/blob/9187210eee7d87eea37b45ea93454a88681894a4/drivers/hid/hid-playstation.c#L1321-L1343),  [DualShock 4](https://github.com/torvalds/linux/blob/9187210eee7d87eea37b45ea93454a88681894a4/drivers/hid/hid-playstation.c#L2186-L2215)
 
It looks like it's trying to handle initialization for the device, but the state is incorrect. Maybe when in the userspace HID mode, that node being kept alive leads to a state mismatch with the kernel driver that may have cleared it? 

> We could perhaps pretend it is alive by auto responding while disconnected, since Id like to keep the current logic so we don't loose events when reconnecting.

To be clear, the messages spam only while the device is connected but only after the first time it disconnected and has been reconnected, while the bluetooth daemon hasn't been restarted. So if the device is disconnected, there is no spam. Also while the spam is happening, the device does not work. It's lit up, but no inputs are registered.

Vudentz, 2024-03-13:
Just sent a patch for fixing if the driver start pocking the device while offline the code will just destroy the input device:

https://patchwork.kernel.org/project/bluetooth/patch/20240313211944.1440225-1-luiz.dentz@gmail.com/

Although not ideal it is probably best to avoid causing drivers to misbehave since they normally expect the same behavior as to USB which is always connected.

aiudirog, 2024-03-13:
No dice unfortunately. Just checking, I did have to apply this patch with the v4 patches, right? It wouldn't apply cleanly otherwise.

Vudentz, 2024-03-14:
Any logs? It should have stopped after the first message from the driver.

aiudirog, 2024-03-14:
No new logs, just the spam of:
```
Mar 13 18:44:08 Ryzen-Arch kernel: playstation 0005:054C:09CC.0009: Unhandled reportID=1
```

Vudentz, 2024-03-14:
v2: https://patchwork.kernel.org/project/bluetooth/list/?series=835319

aiudirog, 2024-03-14:
Same spam of message unfortunately

aiudirog, 2024-03-14:
I understand, but unfortunately that doesn't seem to be the case:
[image]

Vudentz, 2024-03-14:
> I understand, but unfortunately that doesn't seem to be the case: [...]

What is the output of bluetoothd? Are you sure you applied the v2? Or perhaps it is busy looping or something, that is really weird to see so many messages at that short of interval.

aiudirog, 2024-03-14:
> What is the output of bluetoothd?

Nothing showed in the logs after the service start on either the first connetion or the re-connection. I then enabled debug logging with the `-d` flag and here are the logs:

<details>

<summary>Service startup</summary>

```
Mar 14 19:15:07 Ryzen-Arch systemd[1]: Starting Bluetooth service...
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: Bluetooth daemon 5.73
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/main.c:parse_config() parsing /etc/bluetooth/main.conf
Mar 14 19:15:07 Ryzen-Arch systemd[1]: Started Bluetooth service.
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/adapter.c:adapter_init() sending read version command
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: Starting SDP server
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/sdpd-service.c:register_device_id() Adding device id record for 0002:1d6b:0246:0549
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:plugin_init() Loading builtin plugins
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Loading hostname plugin
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: plugins/hostname.c:read_dmi_fallback() chassis: desktop
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: plugins/hostname.c:read_dmi_fallback() major: 0x01 minor: 0x01
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: plugins/hostname.c:read_transient_hostname() read transient hostname: 'Ryzen-Arch'
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Plugin hostname loaded
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Loading wiimote plugin
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Plugin wiimote loaded
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Loading autopair plugin
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Plugin autopair loaded
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Loading policy plugin
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Plugin policy loaded
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Loading a2dp plugin
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Plugin a2dp loaded
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Loading avrcp plugin
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Plugin avrcp loaded
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Loading network plugin
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: profiles/network/manager.c:read_config() /etc/bluetooth/network.conf: Key file does not have key “DisableSecurity” in group “General”
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: profiles/network/manager.c:read_config() Config options: Security=true
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Plugin network loaded
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Loading input plugin
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: profiles/input/manager.c:input_init() input.conf: UserspaceHID=true
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Plugin input loaded
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Loading hog plugin
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: profiles/input/suspend-none.c:suspend_init()
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Plugin hog loaded
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Loading gap plugin
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Plugin gap loaded
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Loading scanparam plugin
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Plugin scanparam loaded
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Loading deviceinfo plugin
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Plugin deviceinfo loaded
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Loading midi plugin
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Plugin midi loaded
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Loading battery plugin
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Plugin battery loaded
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Loading sixaxis plugin
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: plugins/sixaxis.c:sixaxis_init()
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Plugin sixaxis loaded
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Loading bap plugin
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/profile.c:btd_profile_register() D-Bus experimental not enabled
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:init_plugin() System does not support bap plugin
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Loading bass plugin
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/profile.c:btd_profile_register() D-Bus experimental not enabled
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:init_plugin() System does not support bass plugin
Mar 14 19:15:07 Ryzen-Arch bluetoothd[127878]: src/plugin.c:add_plugin() Loading mcp plugin
[... cut ...]
Mar 14 19:27:10 Ryzen-Arch bluetoothd[128494]: src/device.c:device_bonding_failed() status 14
Mar 14 19:27:10 Ryzen-Arch bluetoothd[128494]: src/adapter.c:resume_discovery()
```

</details>

> Are you sure you applied the v2?

Yes:

```
$ cd src/bluez-5.73
$ patch --forward --strip=1 --input=../../BlueZ-v2-input-device-Force-UHID_DESTROY-on-error.patch
patching file profiles/input/device.c
Reversed (or previously applied) patch detected!  Skipping patch.
8 out of 8 hunks ignored -- saving rejects to file profiles/input/device.c.rej

And I just force rebuilt everything to confirm.

Vudentz, 2024-03-15:
The following sequence only appears at first connection:

```
Mar 14 19:19:45 Ryzen-Arch bluetoothd[128494]: profiles/input/device.c:hidp_send_get_report()
Mar 14 19:19:45 Ryzen-Arch bluetoothd[128494]: profiles/input/device.c:hidp_recv_ctrl_data()
Mar 14 19:19:45 Ryzen-Arch bluetoothd[128494]: profiles/input/device.c:uhid_send_get_report_reply() HID report (49 bytes)
Mar 14 19:19:45 Ryzen-Arch bluetoothd[128494]: profiles/input/device.c:hidp_send_get_report()
Mar 14 19:19:45 Ryzen-Arch bluetoothd[128494]: profiles/input/device.c:hidp_recv_ctrl_data()
Mar 14 19:19:45 Ryzen-Arch bluetoothd[128494]: profiles/input/device.c:uhid_send_get_report_reply() HID report (41 bytes)
Mar 14 19:19:45 Ryzen-Arch bluetoothd[128494]: profiles/input/device.c:hidp_send_output()
```

So I assume it needs to be performed at very reconnection, or the driver needs to be changed in order to perform the init sequence of or something, since the report size appears to be different.

aiudirog, 2024-03-15:
That makes sense to me. This also only occurs when selecting the userspace HID mode. I don't know much about how the two modes are implemented, but would comparing them be possible to see if the kernel level one performs this initialization?

Vudentz, 2024-03-20:
> That makes sense to me. This also only occurs when selecting the userspace HID mode. I don't know much about how the two modes are implemented, but would comparing them be possible to see if the kernel level one performs this initialization?

Can you try with the following change, it shall actually honor the IdleTimeout set on input.conf so if you keep it disconnected long enough it should take of the problem, that said perhaps 30 seconds is a little too long and if you connect in between it would still fail.

aiudirog, 2024-03-20:
@Vudentz Is there a specific patchset for me to test or should I try that linked PR?

Vudentz, 2024-03-21:
@aiudirog try with following: https://patchwork.kernel.org/project/bluetooth/patch/20240320130119.854959-1-luiz.dentz@gmail.com/ if you set the IdleTimeout=30 that should destroy the input device node after being idle for that long.

aiudirog, 2024-03-21:
@Vudentz It says `Hunk #7 FAILED at 370` when trying to apply that patch on top of 5.73 with v4 patch from the other issue. Should I be building off master or a different tag for this one?

Vudentz, 2024-03-21:
> @Vudentz It says `Hunk #7 FAILED at 370` when trying to apply that patch on top of 5.73 with v4 patch from the other issue. Should I be building off master or a different tag for this one?

Hmm, any chance to compile it from source directly?

aiudirog, 2024-03-22:
It required a bit of tweaking on the PKGBUILD, but I did get it to compile & install [the branch backing the linked PR.](https://github.com/BluezTestBot/bluez/tree/836747)

With `UserspaceHID=true` and `IdleTimeout=1`, the issue seems to go away if at least 1 minute passes between disconnecting and reconnecting. Shorter than that or without the `IdleTimeout` configured and it continues to appear. Also, the comments in `input.conf` say that `IdleTimeout` is actually in minutes, so that lines up with my experience. I tried setting `IdleTimeout=0.001` to go shorter, but that didn't seem to work.

So yeah, seems like whatever destruction `IdleTimeout` is driving should probably always be promptly happening - at least for these controllers.

Vudentz, 2024-03-22:
> It required a bit of tweaking on the PKGBUILD, but I did get it to compile & install [the branch backing the linked PR.](https://github.com/BluezTestBot/bluez/tree/836747)
>
> With `UserspaceHID=true` [...]

Then we need a device specific configuration which is something I was trying to avoid, or have it as VirtualUnplugOnDisconnect but that sort of defeat the purpose of using uhid to keep the devices 'virtually' plugged while disconnected to prevent loosing key presses, etc, when reconnecting.

aiudirog, 2024-03-22:
I feel like that concept applies more to keyboards/mice, which turn off when not in use to save power and need to quickly reconnect, whereas game controllers tend to shut completely down and need to be explicitly reconnected by pushing their logo button. Is there a way to cleanly distinguish between the two without going down to the level of device specific configuration?

kode54, 2024-03-22:
The problem happening with the gamepads is/was that something would flake out and they'd disconnect and reconnect periodically in the middle of gaming sessions. Worse still, sometimes they'd get stuck reporting axes and/or buttons pressed out of idle state, even across reconnecting. So I'd end up with a game controller with its stick permanently stuck off center, making it impossible to temporarily switch to mouse+keyboard to finish the match when a controller refused to reconnect. And switching the controller to cabled left it with the Bluetooth instance still jamming axes and/or buttons simultaneously.

aiudirog, 2024-03-22:
@kode54 I haven't run into your issue of stuck state, but I do occasionally get disconnects where I have to hit the logo button to turn it back on reconnect. I haven't tested what happens there in UserspaceHID mode though.

kode54, 2024-03-22:
@aiudirog Ah, indeed, I only experienced the stuck input state with UserspaceHID enabled. The configuration file here seems to indicate that it's enabled by default, so I had to turn it off to avoid the problems. The Arch wiki claims this setting does not need to be changed under GNOME, but does not mention that it should already be enabled by default when freshly installed and not otherwise changed.

hasanjones, 2024-04-01:
> @aiudirog Ah, indeed, I only experienced the stuck input state with UserspaceHID enabled. The configuration file here seems to indicate that it's enabled by default, so I had to turn it off to avoid [...]

I can confirm the same behavior with my Dualsense. Setting UserspaceHID to false allows it to reconnect properly.

Vudentz, 2024-04-01:
Send a new version: https://patchwork.kernel.org/project/bluetooth/list/?series=840335

This shall, not tested, replay the initial control messages which hopefully would be enough for thing to work when the connection is restored without destroying the input node.

kode54, 2024-04-01:
@Vudentz does not apply cleanly to 5.73.

Vudentz, 2024-04-02:
@kode54 here is the diff against 5.73:

[5.73-input.diff.txt](https://github.com/bluez/bluez/files/14841631/5.73-input.diff.txt)

kode54, 2024-04-02:
Seems to be working with default timeout disabled, I'll have to test with a timeout and see if it fixes that case.

Vudentz, 2024-04-03:
> Seems to be working with default timeout disabled, I'll have to test with a timeout and see if it fixes that case.

@kode54 No need to add timeout, the replay logic would resend the control PDU when reconnecting so that should bring back the same behavior as destroying the UHID device. Could you please attach the output of bluetoothd when reconnecting?

kode54, 2024-04-05:
Here's `journalctl -b <the boot where it worked> -u bluetooth`:
[bluetooth.txt](https://github.com/bluez/bluez/files/14881469/bluetooth.txt)

Nothing about HID there.

kode54, 2024-04-05:
Here's the binary file that btmon generated.
[btmon.bin.gz](https://github.com/bluez/bluez/files/14890479/btmon.bin.gz)

Vudentz, 2024-04-05:
v6: https://patchwork.kernel.org/project/bluetooth/list/?series=841948

kode54, 2024-04-05:
@Vudentz What about 5.73? Should I really be running -git bluez?

Vudentz, 2024-04-08:
> @Vudentz What about 5.73? Should I really be running -git bluez?
> 
> Edit: Never mind, I've installed bluez-git with the patch applied. Will test.
> 
> Edit 2: Here's a log and a bin: [bluetooth.txt.gz](https://github.com/bluez/bluez/files/14891258/bluetooth.txt.gz) [btmon.bin.gz](https://github.com/bluez/bluez/files/14891259/btmon.bin.gz)
> 
> Also, it's a legitimate Sony V2 controller. Manufacturer code of "G" matches. Though much of the bottom label has worn off, since the clear coat sticker eventually peeled off.

Looks like it is working, at least the report length looks proper and we do record the control messages at the start:

```
bluetoothd[1985837]: profiles/input/device.c:hidp_send_get_report() 
bluetoothd[1985837]: profiles/input/device.c:hidp_record_message() output[0]: hdr 0x43 size 1
bluetoothd[1985837]: profiles/input/device.c:uhid_send_input_report() HID report (10 bytes)
bluetoothd[1985837]: profiles/input/device.c:hidp_recv_ctrl_data() 
bluetoothd[1985837]: profiles/input/device.c:uhid_send_get_report_reply() HID report (49 bytes)
bluetoothd[1985837]: profiles/input/device.c:hidp_record_message() input[0]: hdr 0xa3 size 49
bluetoothd[1985837]: profiles/input/device.c:uhid_send_input_report() HID report (10 bytes)
bluetoothd[1985837]: profiles/input/device.c:hidp_send_get_report() 
bluetoothd[1985837]: profiles/input/device.c:hidp_record_message() output[1]: hdr 0x43 size 1
bluetoothd[1985837]: profiles/input/device.c:uhid_send_input_report() HID report (78 bytes)
bluetoothd[1985837]: profiles/input/device.c:uhid_send_input_report() HID report (78 bytes)
bluetoothd[1985837]: profiles/input/device.c:hidp_recv_ctrl_data() 
bluetoothd[1985837]: profiles/input/device.c:uhid_send_get_report_reply() HID report (41 bytes)
bluetoothd[1985837]: profiles/input/device.c:hidp_record_message() input[1]: hdr 0xa3 size 41
```

closed 2024-04-08 by BluezTestBot

urbenlegend, 2024-04-16:
I am on Kernel 6.8.5 and BlueZ 5.75. When I connect, disconnect, and reconnect, I no longer get the error log spam but instead `bluetoothd` crashes. I have to restart the `bluetooth` service and then it works fine until the next disconnect / reconnect cycle.

```
#0  g_main_context_add_poll_unlocked (context=0x5c9c88622800, priority=0, fd=0x5c9c88633cd0) at ../glib/glib/gmain.c:4628
#1  0x00007cb4e24a4b10 in unblock_source (source=0x5c9c88633c70) at ../glib/glib/gmain.c:3266
#2  0x00007cb4e24a8368 in g_main_dispatch (context=0x5c9c88622800) at ../glib/glib/gmain.c:3366
#3  0x00007cb4e25073bf in g_main_context_dispatch_unlocked (context=0x5c9c88622800) at ../glib/glib/gmain.c:4152
#4  g_main_context_iterate_unlocked.isra.0 (context=0x5c9c88622800, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../glib/glib/gmain.c:4217
#5  0x00007cb4e24a8dc7 in g_main_loop_run (loop=0x5c9c886229f0) at ../glib/glib/gmain.c:4419
#6  0x00005c9c87652ef3 in mainloop_run () at src/shared/mainloop-glib.c:66
#7  mainloop_run_with_signal (func=0x5c9c876a64e0 <signal_callback>, user_data=0x0) at src/shared/mainloop-notify.c:188
#8  main (argc=<optimized out>, argv=<optimized out>) at src/main.c:1456
```

kernle32dll, 2024-04-16:
> I am on Kernel 6.8.5 and BlueZ 5.75. When I connect, disconnect, and reconnect, I no longer get the error log spam but instead `bluetoothd` crashes. I have to restart the `bluetooth` service and [...]

Can confirm - same happening here

Vudentz, 2024-04-16:
Probably related to https://github.com/bluez/bluez/issues/815 which has been fixed but it is not part of 5.75.

jonian, 2024-04-21:
@Vudentz, I have tried v5.75 with the patch that fixes #815 in LibreELEC. The controller works on reconnect and the log spamming stops but the controller light stays white and is not turning blue.

Vudentz, 2024-04-22:
> @Vudentz, I have tried v5.75 with the patch that fixes #815 in LibreELEC. The controller works on reconnect and the log spamming stops but the controller light stays white and is not turning blue.

What turning blue means though? Does it work or does it not?

jonian, 2024-04-22:
Thanks for the quick reply! The controller works but I haven't done any extensive testing. 

I'm not sure what the white light means, but a Dual Shock 4 controller that has been connected successfully is supposed to turn blue. This was the behavior with bluez 5.66 and older versions that LibreELEC was using in the previous version. 

When I connect the controller the first time the light turns blue, If I disconnect the controller and reconnect then the light remains white. If I restart bluetooth service and reconnect, the light turns blue.

The same behavior also on Arch Linux with version 5.75 but if I set `UserspaceHID=false` in `/etc/bluetooth/input.conf` the controller has blue light on first connection and blue light on reconnect.

So it seems there is some connection issue. Any advice on how I can debug it and any logs I can provide to help fix this issue?

kode54, 2024-04-22:
The color would be the player number indicator, replicating the behavior of the game console by setting the color on connect. Though nothing stops something from changing the color manually later, either.

kode54, 2024-04-29:
FYI, this doesn't fix the reconnect handshake messages. As mentioned above, DualShock 4 doesn't replay the LED player indicator setting message. And adding one that wasn't mentioned above: Xbox Series X|S Wireless Controller, when paired with xpad neo driver, doesn't send the connect time "hello" vibration messages on reconnects after booting. It only sends it the first time the controller connects.

Vudentz, 2024-04-29:
> FYI, this doesn't fix the reconnect handshake messages. As mentioned above, DualShock 4 doesn't replay the LED player indicator setting message. And adding one that wasn't mentioned above: Xbox [...]

Ok, we maybe need to do the destroy of uhid logic based on the device type then, is there any property or anything we cause to determine if the HID device is actually a controller or something similar?