bluez/bluez #782: Bluetoothctl-mesh request name failed

sidelux, 2024-03-15:
Hello, i'm with bluez 5.73 on a Raspberry P4B but when i run the following command:
```
cd /home/user/bluez-5.73/mesh/
./bluetooth-meshd
```
I get this output:
```
D-Bus ready
Request name failed
```
What can i do to fix it?
Thanks a lot.

bgix, 2024-03-16:
It needs to be run with root privileges and it needs to have the dbus
settings for it properly installed

sidelux, 2024-03-18:
With sudo seems to works but what can i do from this output?
```
sudo ./bluetooth-meshd
D-Bus ready
Loading node configuration from /var/lib/bluetooth/mesh
index 0 up:1 pwr: 1 mesh: 1
Controller hci 0 already in use (ad1)
mesh_ready_callback
Added Network Interface on /org/bluez/mesh
```
Seems stuck on last line... There is some documentations to know how it work?

skalazion, 2024-12-19:
@Sidelux bluetooth-meshd is a daemon (background process) that sets up and manages a Bluetooth Mesh network. When you run it with sudo ./bluetooth-meshd, the process initializes and runs in the background, as indicated by the output you see.

Run the daemon in debug mode using: 
```
sudo ./bluetooth-meshd -nd
``` 
This will run it in non-daemon mode and display verbose debug logs in your terminal.

closed 2025-07-28 by github-actions[bot]