bluez/bluez #783: Bluetooth keyboard connecting and disconnecting in infinite loop

wreckdump, 2024-03-16:
I am using a bluetooth keyboard, and I am having trouble connecting to my computer using bluez 5.73-4. Here is what I have done. In `bluetoothctl` environment, I turned on the scan, and I found the keyboard MAC address and when I pair it I get

```
# [NEW] Device E1:EB:D6:7B:94:2E BT60
# pair E1:EB:D6:7B:94:2E
# [CHG] Device E1:EB:D6:7B:94:2E Connected: yes
# Failed to pair: org.bluez.Error.AuthenticationFailed
# [CHG] Device E1:EB:D6:7B:94:2E Connected: no
```

So, when I `trust` that MAC address and try to connect it, then I get

```
# [CHG] Device E1:EB:D6:7B:94:2E Trusted: yes
# Changing E1:EB:D6:7B:94:2E trust succeeded
# connect E1:EB:D6:7B:94:2E
[CHG] Device E1:EB:D6:7B:94:2E Connected: yes
[BT60]# Connection successful
[CHG] Device E1:EB:D6:7B:94:2E Connected: no
[CHG] Device E1:EB:D6:7B:94:2E Connected: yes
[CHG] Device E1:EB:D6:7B:94:2E Connected: no
[CHG] Device E1:EB:D6:7B:94:2E Connected: yes
[CHG] Device E1:EB:D6:7B:94:2E Connected: no
[CHG] Device E1:EB:D6:7B:94:2E Connected: yes
[CHG] Device E1:EB:D6:7B:94:2E Connected: no
[CHG] Device E1:EB:D6:7B:94:2E Connected: yes
[CHG] Device E1:EB:D6:7B:94:2E Connected: no
[CHG] Device E1:EB:D6:7B:94:2E Connected: yes
[CHG] Device E1:EB:D6:7B:94:2E Connected: no
[CHG] Device E1:EB:D6:7B:94:2E Connected: yes
[CHG] Device E1:EB:D6:7B:94:2E Connected: no
[CHG] Device E1:EB:D6:7B:94:2E Connected: yes
[CHG] Device E1:EB:D6:7B:94:2E Connected: no
[CHG] Device E1:EB:D6:7B:94:2E Connected: yes
[CHG] Device E1:EB:D6:7B:94:2E Connected: no
[CHG] Device E1:EB:D6:7B:94:2E Con...
```

This goes on forever. Looking at the `dmesg` output, some of the relevant message seems to be

```
Bluetooth: hci0: unexpected SMP command 0x0b from e1:eb:d6:7b:94:2e
```

I've been googling this issue for hours, but I can't find any solution to this. Weird thing is the keyboard used to work fine, but when I removed the device from my computer and try to pair/connect it again, it is acting up. I don't know what to do. Any help would be grateful.

Thank you.

Vudentz, 2024-03-18:
https://github.com/bluez/bluez/issues/774#issuecomment-2004303720

closed 2024-03-18 by BluezTestBot

wreckdump, 2024-03-18:
It did not fix the issue. Same problem persists.

Vudentz, 2024-03-18:
Perhaps I had the wrong impression then, since we didn't change anything specially to pairing, not sure how it went this way with just userspace changes since SMP happen to be in the kernel, or did you update your kernel as well?

wreckdump, 2024-03-18:
I found the cause of the issue. It was not the issue with the bluez, it was caused by the keyboard. I forgot to clear the pairing info in the keyboard and it was rejecting any new paring or connection. Sorry for the confusion. Thank you.